geerlingguy/raspberry-pi-pcie-devices #1: Test SATA adapter(s)

geerlingguy, 2020-10-22:
6by9 on the Pi Forums [mentioned](https://www.raspberrypi.org/forums/viewtopic.php?p=1744542#p1744542):

> For those wanting to know about PCI-e compatibility, I have one here with a Pericom PI7C9X 1 to 3 way PCI-e bridge, and Marvell 9215 4 port SATA card connected to that. (My VL805 USB3 card is still to be delivered). With a couple of extra kernel modules enabled (mainly CONFIG_ATA, and CONFIG_SATA_AHCI) it's the basis of my next NAS.
>
> <24W with a pair of 8TB SATA drives spinning and a 240GB SSD. <10W with the spinning rust in standby.

geerlingguy, 2020-10-22:
I'm going to try out the [IO Crest 4-port SATA adapter](https://amzn.to/3knaCqM).

geerlingguy, 2020-10-24:
It has arrived!

geerlingguy, 2020-10-24:
And... I just realized I have no SATA power supply cable, just the data cable. So I'll have to wait for one of those to come in before I can actually test one of my SATA drives.

geerlingguy, 2020-10-24:
First light is good:

```
$ lspci
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 SATA controller: Marvell Technology Group Ltd. Device 9215 (rev 11)
```

geerlingguy, 2020-10-24:
Though `dmesg` shows that it's hitting BAR default address space limits again:

```
[    0.925795] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.925818] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    0.925884] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x0603ffffff -> 0x00f8000000
[    0.925948] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0100000000
[    0.953526] brcm-pcie fd500000.pcie: link up, 5 GT/s x1 (SSC)
[    0.953827] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    0.953844] pci_bus 0000:00: root bus resource [bus 00-ff]
[    0.953866] pci_bus 0000:00: root bus resource [mem 0x600000000-0x603ffffff] (bus address [0xf8000000-0xfbffffff])
[    0.953933] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    0.954172] pci 0000:00:00.0: PME# supported from D0 D3hot
[    0.957560] PCI: bus0: Fast back to back transfers disabled
[    0.957582] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    0.957802] pci 0000:01:00.0: [1b4b:9215] type 00 class 0x010601
[    0.957874] pci 0000:01:00.0: reg 0x10: [io  0x8000-0x8007]
[    0.957911] pci 0000:01:00.0: reg 0x14: [io  0x8040-0x8043]
[    0.957947] pci 0000:01:00.0: reg 0x18: [io  0x8100-0x8107]
[    0.957984] pci 0000:01:00.0: reg 0x1c: [io  0x8140-0x8143]
[    0.958021] pci 0000:01:00.0: reg 0x20: [io  0x800000-0x80001f]
[    0.958058] pci 0000:01:00.0: reg 0x24: [mem 0x00900000-0x009007ff]
[    0.958095] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0003ffff pref]
[    0.958262] pci 0000:01:00.0: PME# supported from D3hot
[    0.961586] PCI: bus1: Fast back to back transfers disabled
[    0.961605] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    0.961674] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    0.961698] pci 0000:01:00.0: BAR 6: assigned [mem 0x600000000-0x60003ffff pref]
[    0.961722] pci 0000:01:00.0: BAR 5: assigned [mem 0x600040000-0x6000407ff]
[    0.961744] pci 0000:01:00.0: BAR 4: no space for [io  size 0x0020]
[    0.961759] pci 0000:01:00.0: BAR 4: failed to assign [io  size 0x0020]
[    0.961774] pci 0000:01:00.0: BAR 0: no space for [io  size 0x0008]
[    0.961788] pci 0000:01:00.0: BAR 0: failed to assign [io  size 0x0008]
[    0.961803] pci 0000:01:00.0: BAR 2: no space for [io  size 0x0008]
[    0.961817] pci 0000:01:00.0: BAR 2: failed to assign [io  size 0x0008]
[    0.961831] pci 0000:01:00.0: BAR 1: no space for [io  size 0x0004]
[    0.961845] pci 0000:01:00.0: BAR 1: failed to assign [io  size 0x0004]
[    0.961860] pci 0000:01:00.0: BAR 3: no space for [io  size 0x0004]
[    0.961873] pci 0000:01:00.0: BAR 3: failed to assign [io  size 0x0004]
[    0.961891] pci 0000:00:00.0: PCI bridge to [bus 01]
[    0.961914] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    0.962217] pcieport 0000:00:00.0: enabling device (0140 -> 0142)
[    0.962439] pcieport 0000:00:00.0: PME: Signaling with IRQ 55
[    0.962813] pcieport 0000:00:00.0: AER: enabled with IRQ 55
```

geerlingguy, 2020-10-24:
I just increased the BAR allocation following [the directions in this Gist](https://gist.github.com/geerlingguy/9d78ea34cab8e18d71ee5954417429df), but when I rebooted (without the card in), I got:

```
[    0.926161] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.926184] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    0.926247] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    0.926312] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0100000000
[    1.521386] brcm-pcie fd500000.pcie: link down
```

geerlingguy, 2020-10-24:
Ah... looking closer, those 'failed to assign' errors are for IO BARs, which are unsupported on the Pi.

So... I posted [in the BAR space thread on Pi Forums](https://www.raspberrypi.org/forums/viewtopic.php?p=1747597#p1747597) asking 6by9 if that user has had the same logs and if they can be safely ignored. Still waiting on a way to power my drive so I can do an end-to-end test :)

kitlith, 2020-10-27:
something else that may be interesting is if you can get a sas adapter/raid card working. I know I was looking into SBCs w/ pcie awhile back for the purpose of building a low power/low heat host for some sas drives I have. (ended up just throwing it in a computer and not running 24/7)

geerlingguy, 2020-10-27:
That would be an interesting thing to test, though it'll have to wait a bit as I'm trying to get through some other cards and might also test 2.5 Gbps or 5 Gbps networking if I am able to!

geerlingguy, 2020-10-27:
Without the kernel modules enabled, `lsblk` shows no device:

```
$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 29.8G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 29.6G  0 part /
```

Going to try adding those modules and see what happens!

geerlingguy, 2020-10-27:
```
# Install dependencies
sudo apt install -y git bc bison flex libssl-dev make libncurses5-dev

# Clone source
git clone --depth=1 https://github.com/raspberrypi/linux

# Apply default configuration
cd linux
export KERNEL=kernel7l # use kernel8 for 64-bit, or kernel7l for 32-bit
make bcm2711_defconfig

# Customize the .config further with menuconfig
make menuconfig
# (search for `/ata` and `/ahci`, enable in the proper section, save, then exit)
nano .config
# (edit CONFIG_LOCALVERSION and add a suffix that helps you identify your build)

# Build the kernel and copy everything into place
make -j4 zImage modules dtbs # 'Image' on 64-bit
sudo make modules_install
sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/$KERNEL.img
```

geerlingguy, 2020-10-28:
Yahoo, it worked!

```
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    1 223.6G  0 disk 
├─sda1        8:1    1   256M  0 part /media/pi/boot
└─sda2        8:2    1 223.3G  0 part /media/pi/rootfs
mmcblk0     179:0    0  29.8G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  29.6G  0 part /
```

geerlingguy, 2020-10-28:
Repartitioning the drive:

```
sudo fdisk /dev/sda
d 1    # delete partition 1
d 2    # delete partition 2
n    # create new partition
p    # primary (default)
1    # partition 1 (default)
2048    # First sector (default)
468862127    # Last sector (default)
w    # write new partition table
```

Got the following:

```
The partition table has been altered.
Failed to remove partition 1 from system: Device or resource busy
Failed to remove partition 2 from system: Device or resource busy
Failed to add partition 1 to system: Device or resource busy

The kernel still uses the old partitions. The new table will be used at the next reboot. 
Syncing disks.
```

Rebooted the Pi, then:

```
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    1 223.6G  0 disk 
└─sda1        8:1    1 223.6G  0 part 
mmcblk0     179:0    0  29.8G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  29.6G  0 part /
```

To format the device, use `mkfs`:

```
$ sudo mkfs.ext4 /dev/sda1
mke2fs 1.44.5 (15-Dec-2018)
Discarding device blocks: done                            
Creating filesystem with 58607510 4k blocks and 14655488 inodes
Filesystem UUID: dd4fa95d-edbf-4696-a9e1-ddf1f17da580
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624, 11239424, 20480000, 23887872

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (262144 blocks): done
Writing superblocks and filesystem accounting information: done 
```

Then mount it somewhere:

```
$ sudo mkdir /mnt/sata-sda
$ sudo mount /dev/sda1 /mnt/sata-sda
$ mount
...
/dev/sda1 on /mnt/sata-sda type ext4 (rw,relatime)
```

geerlingguy, 2020-10-28:
Performance testing of the [Kingston SA400S37/240G drive](https://amzn.to/2HBLs9k):

| Test | Result |
| --- | --- |
| hdparm | 314.79 MB/s |
| dd | 189.00 MB/s |
| random 4K read | 22.98 MB/s |
| random 4K write | 55.02 MB/s |

Compare that to the same drive over USB 3.0 using a USB to SATA adapter:

| Test | Result |
| --- | --- |
| hdparm | TODO MB/s |
| dd | TODO MB/s |
| random 4K read | TODO MB/s |
| random 4K write | TODO MB/s |

mo-g, 2020-10-30:
I'm curious about other OS's. Obviously, Raspbian is a good basis - but as I recall, Fedora Pi 64-bit uses their own custom kernel. I'd be interested in seeing what they've "left in" from the standard kernel config.

I'm looking forward to picking one of these up in a month or so when they become available to the public, then I'll give it a try!

Side note for your list page - could you include PCI ID's as well as just the brand names of the cards? It'll help avoid confusion where cards have multiple revisions, as well as help non-US users identify comparable cards in their own markets.

Great work in the meantime! :+1:

mi-hol, 2020-11-02:
> And someone just mentioned they have some RAID cards they'd be willing to send me. Might have to pony up for a bunch of hard drives and have my desk turn into some sort of frankemonster NAS-of-many-drives soon!

It would be great to test a RAID card based on Marvell 88SE9128 chipset, because it is used by many suppliers

geerlingguy, 2020-11-04:
Trying again today (but [cross-compiling](https://www.jeffgeerling.com/blog/2020/cross-compiling-raspberry-pi-os-linux-kernel-on-macos) this time since it's oh-so-much faster) now that I have two drives and the appropriate power adapters. I'm planning on just testing a file copy between the drives for now, I'll get into other tests later.

geerlingguy, 2020-11-04:
Hmm... putting this on pause. My cross compilation is not dropping in the AHCI module for some reason, probably a bad `.config` :/

geerlingguy, 2020-11-04:
Also, the adapter gets _hot_ after prolonged use.

mi-hol, 2020-11-06:
> Hmm... putting this on pause. My cross compilation is not dropping in the AHCI module for some reason, probably a bad `.config` :/

Sorry to hear about your issue,
I is a quite interesting topic to build a next generation NAS using [OpenMediaVault](github.com/openmediavault/openmediavault/)

geerlingguy, 2020-11-09:
(For anyone interested in testing on an LSI/IBM SAS card, check out https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/18)

geerlingguy, 2020-11-09:
My desk is becoming a war zone:

[image]

Plan is to set up a RAID (probably either 0 if I feel more YOLO-y or 1/10 if I'm more stable-minded) with either 2 or 4 drives, using [mdadm](https://raid.wiki.kernel.org/index.php/A_guide_to_mdadm).

I was having trouble with the SAS card, not sure if the cards are bad or they just don't work at all with the Pi :(

geerlingguy, 2020-11-10:
Testing also with an NVMe using the IO Crest PCIe switch:

```
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    1 223.6G  0 disk 
sdb           8:16   1 223.6G  0 disk 
└─sdb1        8:17   1 223.6G  0 part 
mmcblk0     179:0    0  29.8G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  29.6G  0 part /
nvme0n1     259:0    0 232.9G  0 disk
```

I'll post some benchmarks copying files between one of the SSDs and the NVMe; will be interesting to see how many MB/sec they can pump through the switch.

geerlingguy, 2020-11-10:
For a direct file copy from one drive to another:

```
# fallocate -l 10G /mnt/nvme/test.img
# pv /mnt/nvme/test.img > /mnt/sata-sda/test.img
```

I got an average of 190 MiB/sec, or about 1.52 Gbps. So two-way, that's 3.04 Gbps (under the 3.2 Gbps I was hoping for, but that's maybe down to PCIe switching?

It looks like CPU goes to 99% as SDA takes more than 50% of the CPU—see atop results during a copy:

[image]

geerlingguy, 2020-11-10:
Also comparing raw disk speeds through the PCIe switch:

### Kingston SSD

| Test | Result |
| --- | --- |
| hdparm | 364.23 MB/s |
| dd | 148.00 MB/s |
| random 4K read | 28.89 MB/s |
| random 4K write | 58.01 MB/s |

### Samsung EVO 970 NVMe

| Test | Result |
| --- | --- |
| hdparm | 363.81 MB/s |
| dd | 166.00 MB/s |
| random 4K read | 46.50 MB/s |
| random 4K write | 75.41 MB/s |

geerlingguy, 2020-11-10:
```
# Install mdadm.
sudo apt install -y mdadm

# Create a RAID0 array using sda1 and sdb1.
sudo mdadm --create --verbose /dev/md0 --level=0 --raid-devices=2 /dev/sda1 /dev/sdb1

# Create a mount point for the new RAID device.
sudo mkdir /mnt/raid0

# Format the RAID device.
sudo mkfs.ext4 /dev/md0

# Mount the RAID device.
sudo mount /dev/md0 /mnt/raid0
```

Benchmarking the device:

TODO

geerlingguy, 2020-11-10:
Software RAID1 (mirrored) testing using `mdadm`:

```
# Install mdadm.
sudo apt install -y mdadm

# Create a RAID1 array using sda1 and sdb1.
sudo mdadm --create --verbose /dev/md0 --level=1 --raid-devices=2 /dev/sda1 /dev/sdb1

# Create a mount point for the new RAID device.
sudo mkdir /mnt/raid1

# Format the RAID device.
sudo mkfs.ext4 /dev/md1

# Mount the RAID device.
sudo mount /dev/md0 /mnt/raid1
```

Benchmarking the device:

| Test | Result |
| --- | --- |
| hdparm | TODO MB/s |
| dd | TODO MB/s |
| random 4K read | TODO MB/s |
| random 4K write | TODO MB/s |

TODO.

geerlingguy, 2020-11-10:
Hmm... I'm seeing `md0_resync` continue to run for a long while _after_ the test. So how are they getting out of sync in the first place? Maybe it is trying to sync data that was already on the drive? I thought I had reformatted them though...

geerlingguy, 2020-11-10:
It looks like the resync was the major issue—now that it's complete, numbers are looking much better:

| Test | Result |
| --- | --- |
| hdparm | 352.38 MB/s |
| dd | 112.00 MB/s |
| random 4K read | 27.83 MB/s |
| random 4K write | 42.12 MB/s |

geerlingguy, 2020-11-10:
What I'd like to test with my 4 spinning disks once I get the rest of my SATA cables in the mail today:

  - Set them up in RAID 10 (mirrored stripe) so I get 1 TB of space for 4 500 GB drives.
  - Configure Samba and create a share on the drive.
  - Connect from my Mac and see if I can saturate the 1 Gbps onboard ethernet connection (~100 MB/sec).
  - (Maybe) Connect to 10 Gbps network via PCIe switch slot #2 and see if that gives any _more) than 100 MB/sec copy performance.

geerlingguy, 2020-11-10:
For the spinning disks (500GB WD5000AVDS), I [partitioned, formatted, and mounted them](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/1#issuecomment-717663894), then I ran my benchmarking tests against them:

| Test | Result |
| --- | --- |
| hdparm | 72.43 MB/s |
| dd | 67.30 MB/s |
| random 4K read | 0.48 MB/s |
| random 4K write | 0.60 MB/s |

Sometimes you forget just how good we have it with flash memory nowadays. These drives are _not_ a great option as boot volumes for the Pi :P

I then put two of them in a stripe with `mdadm`, and ran the same test:

| Test | Result |
| --- | --- |
| hdparm | TWO MB/s |
| dd | TODO MB/s |
| random 4K read | TODO MB/s |
| random 4K write | TODO MB/s |

geerlingguy, 2020-11-10:
I also set up SMB, and averaged 75 MB/sec copy performance over the Pi's built-in Gigabit interface for a single large file.

geerlingguy, 2020-11-10:
Ouch, the initial resync is even slower on these spinny disk drives than it was on the SSDs (which, of course, are half the size in the first place, in addition to being twice as fast). 1% per minute on the sync.

Apparently [you could completely skip this option with `--assume-clean`](https://raid.wiki.kernel.org/index.php/Initial_Array_Creation)... but there are many caveats and that's not really intended to happen unless you're in a disaster recovery scenario and you don't want anything to touch the drives when you initialize the RAID device.

So good to know that you should probably plan on letting your array sync up the first time you get it running.

geerlingguy, 2020-11-10:
Hmm... now trying all four drives:

```
$ sudo mdadm --create --verbose /dev/md0 --level=0 --raid-devices=4 /dev/sda1 /dev/sdb1 /dev/sdc1 /dev/sdd1
mdadm: chunk size defaults to 512K
mdadm: Defaulting to version 1.2 metadata
mdadm: RUN_ARRAY failed: Unknown error 524
```

I then zeroed out the superblock:

```
sudo mdadm --zero-superblock /dev/sda1 /dev/sdb1 /dev/sdc1 /dev/sdd1
```

But then when I tried to create again, I got:

```
mdadm: super1.x cannot open /dev/sdd1: Device or resource busy
mdadm: /dev/sdd1 is not suitable for this array.
mdadm: create aborted
```

So I'm going to reboot and try again. Maybe I have a bad drive 😢

geerlingguy, 2020-11-10:
No matter what I try, I keep getting `mdadm: RUN_ARRAY failed: Unknown error 524` in the end.

geerlingguy, 2020-11-10:
Weird. After finding [this question on Stack Exchange](), I tried:

```
# echo 1 > /sys/module/raid0/parameters/default_layout
```

And this time, it works:

```
$ sudo mdadm --create --verbose /dev/md0 --level=0 --raid-devices=4 /dev/sda1 /dev/sdb1 /dev/sdc1 /dev/sdd1
mdadm: chunk size defaults to 512K
mdadm: Defaulting to version 1.2 metadata
mdadm: array /dev/md0 started.
```

We'll see how much further I can go.

geerlingguy, 2020-11-10:
All four drives in RAID0:

| Test | Result |
| --- | --- |
| hdparm | 327.32 MB/s |
| dd | 155.00 MB/s |
| random 4K read | 4.46 MB/s |
| random 4K write | 4.71 MB/s |

Note: The card is getting HOT:

[image]

geerlingguy, 2020-11-10:
Another fun thing I just noticed—`ext4lazyinit` is still running and making it so I can't unmount the volume without forcing it. If I'm going to repartition and reformat anyways, what's the point of letting it finish?

geerlingguy, 2020-11-10:
Resetting the array:

```
sudo umount /mnt/raid0
sudo mdadm --stop /dev/md0
sudo mdadm --zero-superblock /dev/sda1 /dev/sdb1 /dev/sdc1 /dev/sdd1
```

Then set it to RAID 10:

```
# Install mdadm.
sudo apt install -y mdadm

# Create a RAID10 array using four drives.
sudo mdadm --create --verbose /dev/md0 --level=10 --raid-devices=4 /dev/sda1 /dev/sdb1 /dev/sdc1 /dev/sdd1

# Create a mount point for the new RAID device.
sudo mkdir -p /mnt/raid10

# Format the RAID device.
sudo mkfs.ext4 /dev/md0

# Mount the RAID device.
sudo mount /dev/md0 /mnt/raid1
```

Confirm the RAID 10 drive gives me 1 TB of mirrored/striped storage:

```
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/md0        915G   77M  869G   1% /mnt/raid1

$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE   MOUNTPOINT
sda           8:0    1 465.3G  0 disk   
└─sda1        8:1    1 465.3G  0 part   
  └─md0       9:0    0 930.3G  0 raid10 /mnt/raid1
sdb           8:16   1 465.3G  0 disk   
└─sdb1        8:17   1 465.3G  0 part   
  └─md0       9:0    0 930.3G  0 raid10 /mnt/raid1
sdc           8:32   1 465.3G  0 disk   
└─sdc1        8:33   1 465.3G  0 part   
  └─md0       9:0    0 930.3G  0 raid10 /mnt/raid1
sdd           8:48   1 465.8G  0 disk   
└─sdd1        8:49   1 465.8G  0 part   
  └─md0       9:0    0 930.3G  0 raid10 /mnt/raid1
mmcblk0     179:0    0  29.8G  0 disk   
├─mmcblk0p1 179:1    0   256M  0 part   /boot
└─mmcblk0p2 179:2    0  29.6G  0 part   /
```

geerlingguy, 2020-11-10:
It bears repeating:

[image]

PixlRainbow, 2020-11-16:
Have you tested if you can boot from a drive attached through PCIE?

**EDIT**: It appears that as of now the Raspberry Pi firmware only supports SD card, USB and Network boot. However, you could _potentially_ boot a U-Boot shell from the SD card, load a efi driver for NVMe drives, then load the OS efi bootloader from the drive. But this appears to be completely untested on raspberry pi although it has been found to work on the Rock Pi (Rockchip ARM, not Broadcom).
TianoCore has a more "finished" UEFI implementation on the Raspberry Pi. Unfortunately, the project's NVMe efi driver cannot be built for arm, though it may be able to load a driver from another project.

markbirss, 2020-11-19:
@geerlingguy 

You should seriously look at using ZFS raidz over mdadm raid

"calculator" 
https://calomel.org/zfs_raid_speed_capacity.html

Official OpenZFS guide now including installation on Raspberry Pi

https://openzfs.github.io/openzfs-docs/Getting%20Started/Ubuntu/Ubuntu%2020.04%20Root%20on%20ZFS%20for%20Raspberry%20Pi.html

geerlingguy, 2020-11-20:
Another quick note, just to make sure I point it out: to reset the drives, you can run `sudo wipefs -a /dev/sd[a-d]`. Don't, uh... do that when you're not certain you want to wipe all the drives though :D

geerlingguy, 2020-11-20:
Now this is weird... I kept trying to create an array with 4 SSDs, but kept getting results like:

```
$ sudo mdadm --create --verbose /dev/md0 --level=0 --raid-devices=4 /dev/sd[a-d]1
mdadm: super1.x cannot open /dev/sda1: Device or resource busy
mdadm: ddf: Cannot use /dev/sda1: Device or resource busy
mdadm: Cannot use /dev/sda1: It is busy
mdadm: cannot open /dev/sda1: Device or resource busy
```

But sometimes (after doing a reset where I stopped md0, zeroed the drives, and removed md0), it would be sdb. Sometimes sdc. Sometimes sdd. Sometimes more than one, but never the same.

So it looked like a race condition, and lo and behold, searching around, I found this post from 2012: [mdadm: device or resource busy](http://dev.bizo.com/2012/07/mdadm-device-or-resource-busy.html), and in it, it is suggested disabling udev events during creation:

```
$ sudo udevadm control --stop-exec-queue
$ sudo mdadm --create ...
$ sudo udevadm control --start-exec-queue
```

Lo and behold... that worked!

geerlingguy, 2020-11-20:
Some benchmarks for 4 Kingston SSDs (2x 120 GB, 2x 240 GB) below:

### RAID 0

| Test | Result |
| --- | --- |
| hdparm | 296.21 MB/s |
| dd | 169.67 MB/s |
| random 4K read | 28.33 MB/s |
| random 4K write | 61.85 MB/s |

### RAID 10

| Test | Result |
| --- | --- |
| hdparm | ()/3 MB/s |
| dd | ()/3 MB/s |
| random 4K read | ()/3 MB/s |
| random 4K write | ()/3 MB/s |

geerlingguy, 2020-11-29:
This video will (hopefully be epic), and still, sadly, won't cover probably more than 50% of what I've learned testing this card. Working on the final script now, hopefully I'll be able to start recording either late tomorrow or early in the week, once I get my notes finished for my Kubernetes 101 series episode!

geerlingguy, 2020-11-29:
`iperf3` measured 942 Mbps between the Pi's 1 Gbps port and my MacBook Pro through a CalDigit TB3 hub, so the maximum possible transfer rate I could achieve is 118 MB/sec on this connection:

| Scenario | Result |
| --- | --- |
| SMB RAID 10 Kingston SSD x4 | TODO MB/sec |
| NFS RAID 10 Kingston SSD x4 | TODO MB/sec |

### SMB Setup

TODO.

### NFS Setup

TODO.

geerlingguy, 2020-11-30:
One last thing I want to test—overall energy consumption between the 4 drives and the IO board + CM4 + IO Crest card. I'll plug it all in through a Kill-A-Watt and observe.

geerlingguy, 2020-11-30:
For NFS threads:

```
# Change RPCNFSDCOUNT from 8 to 1.
sudo nano /etc/default/nfs-kernel-server

# Restart nfsd.
sudo systemctl restart nfs-kernel-server

# Confirm there's now one thread.
ps aux | grep nfsd
```

And the result? Even with only one thread, I was able to hit 900+ Mbps and sustain 105+ MB/sec with NFS (though the single thread was hitting 75-100% CPU usage on one core now).

So something about the NFS protocol seems to be slightly more efficient than Samba—at least on Linux—in general, regardless of the threading model.

geerlingguy, 2020-11-30:
Energy consumption (4x Kingston SSD via dedicated AC adapter + IO Board, CM4, IOCrest card via AC adapter):

  - Idle: 6W
  - Avg during file copy: 11W
  - Peak consumption: 12.2W

geerlingguy, 2020-12-02:
One more thing I was wondering—is there a technical reason to partition the drives before adding them to the array (vs. just using sda/sdb/etc.)? [This SO answer about creating an array using partitions vs. the whole disk](https://unix.stackexchange.com/a/323425/16194) seemed to have a few good arguments in favor of pre-partitioning.

elFarto, 2020-12-02:
The only technical reason I can think of is to get the correct block alignment for drives that have a 4k native sector size and a 512b logical block size (and SSDs where you want to align to the erase block size). But this isn't a requirement, it'll just slow the drives down a little bit if they're not aligned.

Because the wasted space due to alignment is usually greater than that of partitioning, and partitioning has other advantages, you might as well partition them.

geerlingguy, 2020-12-02:
@elFarto - It seemed like that was the most compelling universal reason—though the note about some motherboards (and chipsets, I presume) having trouble with unpartitioned block devices made me nervous enough to recommend always partitioning before RAIDing.

paulwratt, 2020-12-03:
disk block i/o sizes have been verified to play a part in slowdowns, especially (but not only) with SSD's. A proper prepartition just means the start of you partitions is on whatever the block size boundry. (Apparently windows can _not_ properly align the first partition, other partitions are fine).

okket, 2020-12-05:
FYI: Here are my configuration options for Samba to make it faster interacting with a macOS client. It has been a while since I researched this topic, but I remember that plain Samba performance was abysmal, esp. with large directories. The `xattr` stuff seems to help a lot.

```
[global]
        min protocol = SMB2 
        ea support = yes
        vfs objects = catia fruit streams_xattr acl_xattr shadow_copy2 
        fruit:metadata = stream
        fruit:model = MacPro
        fruit:posix_rename = yes 
        fruit:veto_appledouble = no
        fruit:wipe_intentionally_left_blank_rfork = yes 
        fruit:delete_empty_adfiles = yes

[TimeMachine]
        path = <path>
        browseable = yes
        writeable = yes
        read only = no
        create mask = 0600
        directory mask = 0700
        spotlight = yes
        vfs objects = catia fruit streams_xattr
        fruit:time machine = yes
        fruit:time machine max size = <size in TB> T
        valid users = <usernames>
```

geerlingguy, 2020-12-22:
Video is here: [I built the fastest Raspberry Pi SATA RAID NAS!](https://www.youtube.com/watch?v=oWev1THtA04).

geerlingguy, 2021-01-09:
Closing issues where testing is at least mostly complete, to keep the issue queue tidy.

closed 2021-01-09 by geerlingguy

jamesy0ung, 2021-04-15:
Hey Jeff this is not working for me.
I compiled the latest kernel, copied it over and booted it fine without the card.
When I decompiled the device tree blob to do the bar patch, I get these errors

root@buster:/mnt/pi-fat32# dtc -I dtb -O dts bcm2711-rpi-cm4.dtb -o test.dts
test.dts: Warning (unit_address_vs_reg): /soc: node has a reg or ranges property, but no unit name
test.dts: Warning (unit_address_vs_reg): /soc/axiperf: node has a reg or ranges property, but no unit name
test.dts: Warning (unit_address_vs_reg): /soc/gpiomem: node has a reg or ranges property, but no unit name
test.dts: Warning (unit_address_vs_reg): /emmc2bus: node has a reg or ranges property, but no unit name
test.dts: Warning (unit_address_vs_reg): /scb: node has a reg or ranges property, but no unit name
test.dts: Warning (unit_address_vs_reg): /v3dbus: node has a reg or ranges property, but no unit name
test.dts: Warning (pci_device_reg): /scb/pcie@7d500000/pci@1,0: PCI unit address format error, expected "0,0"
test.dts: Warning (avoid_unnecessary_addr_size): /soc/firmware: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
test.dts: Warning (unique_unit_address): /soc/mmc@7e300000: duplicate unit-address (also used in node /soc/mmcnr@7e300000)
test.dts: Warning (unique_unit_address): /soc/firmwarekms@7e600000: duplicate unit-address (also used in node /soc/smi@7e600000)
test.dts: Warning (clocks_property): /__symbols__:clocks: property size (21) is invalid, expected multiple of 4
test.dts: Warning (gpios_property): /aliases:gpio: property size (19) is invalid, expected multiple of 4
test.dts: Warning (gpios_property): /__symbols__:gpio: property size (19) is invalid, expected multiple of 4

root@buster:/mnt/pi-fat32# dtc -I dts -O dtb test.dts -o test.dtb
test.dtb: Warning (unit_address_vs_reg): /soc: node has a reg or ranges property, but no unit name
test.dtb: Warning (unit_address_vs_reg): /soc/axiperf: node has a reg or ranges property, but no unit name
test.dtb: Warning (unit_address_vs_reg): /soc/gpiomem: node has a reg or ranges property, but no unit name
test.dtb: Warning (unit_address_vs_reg): /emmc2bus: node has a reg or ranges property, but no unit name
test.dtb: Warning (unit_address_vs_reg): /scb: node has a reg or ranges property, but no unit name
test.dtb: Warning (unit_address_vs_reg): /v3dbus: node has a reg or ranges property, but no unit name
test.dtb: Warning (pci_device_reg): /scb/pcie@7d500000/pci@1,0: PCI unit address format error, expected "0,0"
test.dtb: Warning (avoid_unnecessary_addr_size): /soc/firmware: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
test.dtb: Warning (unique_unit_address): /soc/mmc@7e300000: duplicate unit-address (also used in node /soc/mmcnr@7e300000)
test.dtb: Warning (unique_unit_address): /soc/firmwarekms@7e600000: duplicate unit-address (also used in node /soc/smi@7e600000)
test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 1 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 3 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 5 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 7 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 9 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 11 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/serial@7e201000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/serial@7e201000:clocks: cell 2 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/mmc@7e202000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/i2s@7e203000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/spi@7e204000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/i2c@7e205000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/dpi@7e208000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/dpi@7e208000:clocks: cell 2 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/dsi@7e209000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/dsi@7e209000:clocks: cell 2 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/dsi@7e209000:clocks: cell 4 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/aux@7e215000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/serial@7e215040:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/spi@7e215080:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/spi@7e2150c0:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/pwm@7e20c000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/hvs@7e400000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/dsi@7e700000:clocks: cell 0 is not a phandle reference
test.dtb: Warning (clocks_property): /soc/dsi@7e700000:clocks: cell 2 is not a phandle reference
[... cut ...]
With the original dtb, it boots fine, but I get the asynchronous serror interrupt if no sata drives are plugged in. WIth drive plugged in, the card does not work either presumably because of bar space. Here is the entries in dmesg

[    1.247086] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.249747] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.252416] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.255082] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.577589] brcm-pcie fd500000.pcie: link down

What should I do?

stamaali4, 2021-06-09:
@geerlingguy Thanks for a detailed writeup...

I am trying to recompile to the kernel to enable SATA Support on IO Crest 4 Port MARVELL 9215 to use it with Raspberry Pi CM4 using UBUNTU 20.04 .

I tried the exact same steps mentioned; however the disks are not detected.... I am giving the try again now as I missed the steps mentioned for increasing the BAR Address(i did see the BAR Address spaces errors in dmesg) space for PCIe.... wanted to check with you(I am a novice in linux) if the steps that you shared applicable for Ubuntu 20.04 or are they specific to Raspbian OS (Buster)... if they are specific to BUSTER can you please share the changes in the above steps for Ubuntu OS 20.4 .... additional details are mentioned below.

1. output of uname -a: Linux cvbkpgtw 5.4.0-1035-raspi #38-Ubuntu SMP PREEMPT Tue Apr 20 21:37:03 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

2. output of lspci -vv
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2711 (rev 20) (prog-if 00 [Normal decode])
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 43
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: 00000000-00000fff [size=4K]
        Memory behind bridge: f8000000-f80fffff [size=1M]
        Prefetchable memory behind bridge: [disabled]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: <access denied>
        Kernel driver in use: pcieport

01:00.0 SATA controller: Marvell Technology Group Ltd. Device 9215 (rev 11) (prog-if 01 [AHCI 1.0])
        Subsystem: Marvell Technology Group Ltd. Device 9215
        Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 0
        Region 0: I/O ports at 0000
        Region 1: I/O ports at 0000
        Region 2: I/O ports at 0000
        Region 3: I/O ports at 0000
        Region 4: I/O ports at 0000
        Region 5: Memory at 600040000 (32-bit, non-prefetchable) [size=2K]
        Expansion ROM at 600000000 [size=256K]
        Capabilities: <access denied>

3. output of os-release: cvadmin@cvbkpgtw:~$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.2 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

4. Raspberry Pi CM4 - 4GB RAM and 32GB eMMC with IO Board

5. Connected 4 SATA Drives - 1 SSD 240GB(Seagate) & 3 HDDx2TB(Seagate)

6. Booting from eMMC

Please let me know if further details are need, thanks for your help in advance....

App-Teck, 2021-07-30:
Hey , i didnt get this par while config the SATA: 
"nano .config
# (edit CONFIG_LOCALVERSION and add a suffix that helps you identify your build)
# Build the kernel and copy everything into place"

I got in nano .config at line : "CONFIG_LOCALVERSION="-v7l""  
but where to add the commands: below? Thanks for your he;lp
"make -j4 zImage modules dtbs # 'Image' on 64-bit
sudo make modules_install
sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/$KERNEL.img"

Appreciate your help.

[16 more replies]