geerlingguy/raspberry-pi-pcie-devices #2: Test GPU (Zotac Nvidia GeForce GT 710)

geerlingguy, 2020-10-22:
I'm currently testing this GPU ([Zotac Nvidia GeForce GT 710](https://amzn.to/3dSLCW2)), so until that's done, I wanted an issue open to track where that work is going on.

Currently I have the driver installed, and everything sorta works, but I've had to do a lot to get it to that state. Right now I'm installing the CUDA tools to see if I can get some programming done on the GPU. The X window system did not seem to like the driver for some reason.

Anyways, some relevant links to follow:

  - Gist: [Setting up the Nvidia GeForce GT 710 on Raspberry Pi Compute Module 4](https://gist.github.com/geerlingguy/9f1510ab028e68b712381520308db2af)
  - Nvidia bug report: [Drivers fail to load with NVRM: GPU 0004:01:00.0: RmInitAdapter failed! (0x25:0x54:1211)](https://forums.developer.nvidia.com/t/gtx-1080-drivers-fail-to-load-with-nvrm-gpu-000400-0-rminitadapter-failed-0x251211/156902)

geerlingguy, 2020-10-22:
Can't get simple CUDA examples to work either, as they trigger dmesg failure notices—see https://gist.github.com/geerlingguy/9f1510ab028e68b712381520308db2af#gistcomment-3500278

geerlingguy, 2020-10-25:
Just posting as a point of reference, 96Boards mentioned getting an Nvidia 710 working with one of their own boards: https://www.96boards.org/blog/oxalis-gpu/

geerlingguy, 2020-10-25:
One other thing I'd like to try at some point—I just received a 1x to 16x adapter with external power, so I can supply power to the board without leeching off the IO Board's 12v 2A power supply. Might work better, probably won't :P

geerlingguy, 2020-10-27:
It looks like some users of the ROCKPRO64 (using the RK3399, which has x4 PCIe) also attempted and failed to use a GPU on that flavor of the ARM SoC: [Considering buying and question on the PCIe slot](https://forum.pine64.org/showthread.php?tid=6205&page=3)

geerlingguy, 2020-10-27:
Some good reading here: [PCI BARs and other means of accessing the GPU](https://envytools.readthedocs.io/en/latest/hw/bus/bars.html)

NostalgiaRunner, 2020-10-28:
Greetings Jeff,
I recently saw your GPU on CM4 video and blog post and had an idea for a way to get either dGPU working. I don't know if this would be worth the rabbit trail, but If you're willing to try something a little less than on-metal, my idea is to use QEMU. QEMU utilizes the Linux kernel's KVM hypervisor and is able to emulate X86 and X86_64 operating systems on ARM64 hardware. In my previous ventures with ProxMox (VMWare ESXI's open-source QEMU based competition), I was able to passthrough a cheap GT 430 GPU to my very old BIOS age Xeon server. If you wish to try it, I think it'd be possible. Atleast on paper it is. Run QEMU on a headless minimal Pi image, create an x86 VM with your choice of x86 image, and configure the QEMU VM to passthrough the GPU device. Here's a link to the guide I used for ProxMox. Though I have my doubts that it'll be 100% the same process or success level. https://pve.proxmox.com/wiki/Pci_passthrough#GPU_Passthrough While this wouldn't be running the GPU on-metal, I think it would still be an incredible feat if the VM is able to utilize the card, as this would make the idea of custom ARM based desktop motherboards for x86 work a little more appealing. Some final notes are... there's a potential software dilemma. QEMU is and has been able to run x86 VMs on the RPi. However, it is not able to address all 8GB of RAM for the 8GB model, for use across all VMs. ESXI ARM Fling on the other hand is a free??? beta testing product at this time, and is able to address all 8GB on the host. However, it is my recollection that ESXI ARM Fling is unable to run x86 VMs, and potentially unable to passthrough PCIe devices. These are my thoughts and findings, and I hope they may be of use to you. Cheers!

geerlingguy, 2020-10-28:
Over in the raspberrypi/linux project, it looks like this commit (https://github.com/raspberrypi/linux/commit/54db4b2fa4d17251c2f6e639f849b27c3b553939) has increased the default BAR allocation to 1GB by default—nice!

geerlingguy, 2020-11-04:
Before I throw in the towel on this thing, I'm going to try to use the external powered PCIe switch to see if it makes any difference. I'll first try with Nouveau, by [cross-compiling the kernel from my Mac](https://www.jeffgeerling.com/blog/2020/cross-compiling-raspberry-pi-os-linux-kernel-on-macos).

geerlingguy, 2020-11-04:
I'm not even seeing the card when it's plugged into the switch:

```
$ lspci
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
02:01.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
02:02.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
```

No errors in dmesg.

geerlingguy, 2020-11-04:
And if I plug it straight in, I get `pcie link down` each time.

geerlingguy, 2020-11-04:
And using my other powered PCIe adapter I get `pcie link down` as well. Darn. No dice with this card currently.

geerlingguy, 2020-11-18:
All right, spinning this card up again once more using the expanded BAR space suggested in this comment: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/6#issuecomment-728206583

geerlingguy, 2020-11-18:
No matter what I do, `pcie link down` is always the result. Maybe I borked the card?

geerlingguy, 2020-11-18:
Well, now it's finally booting at least, if I don't use external power.

geerlingguy, 2020-11-18:
```
[   50.212050] [drm] radeon kernel modesetting enabled.
[   50.212392] radeon 0000:01:00.0: remove_conflicting_pci_framebuffers: bar 0: 0x700000000 -> 0x70fffffff
[   50.212403] radeon 0000:01:00.0: remove_conflicting_pci_framebuffers: bar 2: 0x600000000 -> 0x60001ffff
[   50.212511] pci 0000:00:00.0: enabling device (0000 -> 0002)
[   50.212536] radeon 0000:01:00.0: enabling device (0000 -> 0002)
[   50.213507] [drm] initializing kernel modesetting (CEDAR 0x1002:0x68F9 0x1545:0x5450 0x00).
[   50.213710] [drm:radeon_device_init [radeon]] *ERROR* Unable to find PCI I/O BAR
[   50.329687] radeon 0000:01:00.0: Expecting atombios for evergreen GPU
[   50.329697] radeon 0000:01:00.0: Fatal error during GPU init
[   50.329703] [drm] radeon: finishing device.
[   50.329709] [TTM] Memory type 2 has not been initialized
[   50.337212] radeon: probe of 0000:01:00.0 failed with error -22
```

arrowmaster, 2020-12-16:
ASUS has a similar board to this with twice the memory and four HDMI ports in exchange for only using PCIe Gen2. That may have alternative use cases for those that need the PCIe 1x physical size.

https://www.asus.com/Motherboards-Components/Graphics-Cards/ASUS/GT710-4H-SL-2GD5/

6by9, 2020-12-16:
@arrowmaster I have an Asus GT710-4H-SL-2GD5 in front of me. The nouveau driver blows up with it during initialisation.
More details in https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=288902

I had some advice from the dri-nouveau mailing list, but time isn't allowing me to play with it at present.

geerlingguy, 2021-01-09:
Closing issues where testing is at least mostly complete, to keep the issue queue tidy.

(If someone else can get the GT 7210 to light up, I'd be happy to have a new thread on it!)

closed 2021-01-09 by geerlingguy

geerlingguy, 2024-10-30:
Just wanted to confirm on Pi 5 I'm still getting kernel errors with this card and the nouveau drivers:

```
[    5.916841] nouveau 0000:01:00.0: DRM: VRAM: 1024 MiB
[    5.916850] nouveau 0000:01:00.0: DRM: GART: 1048576 MiB
[    5.916856] nouveau 0000:01:00.0: DRM: TMDS table version 2.0
[    5.916859] nouveau 0000:01:00.0: DRM: DCB version 4.0
[    5.916863] nouveau 0000:01:00.0: DRM: DCB outp 00: 01000f02 00020030
[    5.916870] nouveau 0000:01:00.0: DRM: DCB outp 01: 02011f62 00020010
[    5.916873] nouveau 0000:01:00.0: DRM: DCB outp 02: 02022f10 00000000
[    5.916877] nouveau 0000:01:00.0: DRM: DCB conn 00: 00001031
[    5.916880] nouveau 0000:01:00.0: DRM: DCB conn 01: 00002161
[    5.916883] nouveau 0000:01:00.0: DRM: DCB conn 02: 00000200
[    5.919269] nouveau 0000:01:00.0: DRM: failed to initialise sync subsystem, -22
[    5.919943] Unable to handle kernel NULL pointer dereference at virtual address 0000000000003fc0
[    5.928802] Mem abort info:
[    5.931628]   ESR = 0x0000000096000145
[    5.935392]   EC = 0x25: DABT (current EL), IL = 32 bits
[    5.940727]   SET = 0, FnV = 0
[    5.943784]   EA = 0, S1PTW = 0
[    5.946947]   FSC = 0x05: level 1 translation fault
[    5.951844] Data abort info:
[    5.954729]   ISV = 0, ISS = 0x00000145, ISS2 = 0x00000000
[    5.960234]   CM = 1, WnR = 1, TnD = 0, TagAccess = 0
[    5.965311]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    5.970660] user pgtable: 16k pages, 47-bit VAs, pgdp=0000000107a44000
[    5.977217] [0000000000003fc0] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[    5.985967] Internal error: Oops: 0000000096000145 [#1] PREEMPT SMP
[    5.992256] Modules linked in: algif_hash algif_skcipher af_alg bnep spidev hci_uart btbcm bluetooth aes_ce_blk aes_ce_cipher ghash_ce gf128mul sha2_ce ecdh_generic sha256_arm64 ecc sha1_ce libaes raspberrypi_hwmon brcmfmac_wcc hid_apple nouveau(+) rpivid_hevc(C) pisp_be v4l2_mem2mem videobuf2_dma_contig i2c_brcmstb brcmfmac videobuf2_memops vc4 drm_exec spi_bcm2835 videobuf2_v4l2 brcmutil snd_soc_hdmi_codec cec videodev drm_dma_helper i2c_algo_bit cfg80211 drm_display_helper snd_soc_core gpio_keys snd_compress drm_ttm_helper snd_pcm_dmaengine ttm snd_pcm drm_kms_helper videobuf2_common mc rfkill snd_timer pwm_fan v3d snd gpu_sched drm_shmem_helper hid_multitouch joydev rp1_adc raspberrypi_gpiomem binfmt_misc nvmem_rmem uio_pdrv_genirq uio drm i2c_dev drm_panel_orientation_quirks backlight fuse dm_mod ip_tables x_tables ipv6
[    6.065732] CPU: 2 PID: 331 Comm: (udev-worker) Tainted: G        WC         6.6.58-v8-16k+ #12
[    6.074467] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    6.080318] pstate: 00400009 (nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    6.087305] pc : dcache_inval_poc+0x28/0x58
[    6.091505] lr : arch_sync_dma_for_cpu+0x34/0x50
[    6.096136] sp : ffffc000808eb720
[    6.099456] x29: ffffc000808eb720 x28: ffffd00044c99a78 x27: 0000000000000001
[    6.106618] x26: ffffc000808ebce0 x25: ffffc000808ebc90 x24: ffffffffffffffff
[    6.113781] x23: 0000000000000000 x22: ffff8001009998c0 x21: 0000000000004000
[    6.120943] x20: 0000000000000000 x19: ffffffffffffffff x18: ffffffffffffffff
[    6.128105] x17: 202c6d6574737973 x16: ffffd0008413de98 x15: 0000000000000100
[    6.135268] x14: ffff80010a3ab000 x13: 02dc8759a673a8ce x12: 0000000000000001
[    6.142430] x11: ffffd00044d04a28 x10: 0000000000000001 x9 : ffffd0008413e04c
[    6.149592] x8 : 0000000000000000 x7 : ffff800105526aa0 x6 : ffffffefffffffff
[    6.156754] x5 : ffff800105526aa0 x4 : 0000000000000000 x3 : 000000000000003f
[    6.163919] x2 : 0000000000000040 x1 : 0000000000003fc0 x0 : ffffffffffffffff
[    6.171090] Call trace:
[    6.173551]  dcache_inval_poc+0x28/0x58
[    6.177413]  dma_unmap_page_attrs+0x1b4/0x1d0
[    6.181789]  nvkm_fb_dtor+0xc0/0x108 [nouveau]
[    6.186380]  nvkm_subdev_del+0x78/0x128 [nouveau]
[    6.191209]  nvkm_device_del+0x80/0x130 [nouveau]
[    6.196037]  nouveau_drm_probe+0x154/0x230 [nouveau]
[    6.201127]  local_pci_probe+0x48/0xb8
[    6.204886]  pci_device_probe+0xac/0x1c8
[    6.208818]  really_probe+0x150/0x2c0
[    6.212489]  __driver_probe_device+0x80/0x140
[    6.216857]  driver_probe_device+0x44/0x170
[    6.221050]  __driver_attach+0x9c/0x1b0
[    6.224894]  bus_for_each_dev+0x80/0xe8
[    6.228740]  driver_attach+0x2c/0x40
[    6.232323]  bus_add_driver+0xec/0x218
[    6.236079]  driver_register+0x68/0x138
[    6.239924]  __pci_register_driver+0x54/0x68
[    6.244205]  nouveau_drm_init+0x214/0x3ff8 [nouveau]
[    6.249300]  do_one_initcall+0x60/0x2c0
[    6.253146]  do_init_module+0x60/0x218
[    6.256905]  load_module+0x1de0/0x2090
[    6.260663]  __do_sys_init_module+0x19c/0x1e0
[    6.265031]  __arm64_sys_init_module+0x24/0x38
[    6.269486]  invoke_syscall+0x50/0x128
[    6.273246]  el0_svc_common.constprop.0+0xc8/0xf0
[    6.277965]  do_el0_svc+0x24/0x38
[    6.281287]  el0_svc+0x40/0xe8
[    6.284347]  el0t_64_sync_handler+0x100/0x130
[    6.288716]  el0t_64_sync+0x190/0x198
[    6.292387] Code: d1000443 ea03003f 8a230021 54000040 (d50b7e21) 
[    6.298501] ---[ end trace 0000000000000000 ]---
```

geerlingguy, 2025-09-30:
Tested again using yanghaku's DRM patch and nouveau on kernel `rpi-6.15.y`, and I get:

```
[    4.653099] nouveau 0001:01:00.0: enabling device (0000 -> 0002)
[    4.653138] nouveau 0001:01:00.0: NVIDIA GK208B (b060b0b1)
[    4.800766] nouveau 0001:01:00.0: bios: version 80.28.a6.00.6d
[    4.801230] ------------[ cut here ]------------
[    4.801235] nouveau 0001:01:00.0: swiotlb addr 0x00000010fae00000+4096 overflow (mask ffffffff, bus limit ffffffffff).
[    4.801251] WARNING: CPU: 1 PID: 374 at kernel/dma/swiotlb.c:1594 swiotlb_map+0x2c0/0x2f0
[    4.801260] Modules linked in: nouveau(+) spidev brcmfmac_wcc hci_uart btbcm bluetooth drm_gpuvm i2c_algo_bit ecdh_generic ecc aes_ce_blk aes_ce_cipher ghash_ce gf128mul drm_exec gpu_sched sha2_ce sha256_arm64 brcmfmac rpi_hevc_dec sha1_ce drm_display_helper pisp_be v4l2_mem2mem sha1_generic brcmutil videobuf2_dma_contig cfg80211 raspberrypi_hwmon drm_ttm_helper videobuf2_memops rfkill videobuf2_v4l2 spi_bcm2835 videodev videobuf2_common gpio_keys ttm mc drm_client_lib drm_kms_helper rp1_adc rp1_pio raspberrypi_gpiomem rp1_fw rp1_mailbox nvmem_rmem sg joydev hid_multitouch uio_pdrv_genirq uio i2c_dev ledtrig_pattern drm fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[    4.801326] CPU: 1 UID: 0 PID: 374 Comm: (udev-worker) Not tainted 6.15.11-v8-4k-gpu+ #1 PREEMPT 
[    4.801330] Hardware name: Raspberry Pi 500 Rev 1.0 (DT)
[    4.801332] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    4.801335] pc : swiotlb_map+0x2c0/0x2f0
[    4.801337] lr : swiotlb_map+0x2c0/0x2f0
[    4.801339] sp : ffffffc0837cb630
[    4.801340] x29: ffffffc0837cb640 x28: ffffff8102126800 x27: ffffffd08d667000
[    4.801344] x26: 00000013812e8000 x25: fffffffece04ba00 x24: 0000000000001000
[    4.801348] x23: ffffff8080db10c8 x22: 0000000000000000 x21: 0000000000000000
[    4.801352] x20: 00000003812e8000 x19: 00000013812e8000 x18: 00000000ffffffff
[    4.801356] x17: 667265766f203639 x16: 30342b3030303030 x15: 6561663031303030
[    4.801360] x14: 3030307830207264 x13: 2e29666666666666 x12: 666666662074696d
[    4.801363] x11: 00000000000000c0 x10: 0000000000001a70 x9 : ffffffd08c4eb990
[    4.801367] x8 : ffffff83812c3c10 x7 : 0000000000000001 x6 : 0000000002e9558a
[    4.801371] x5 : 000000011e2d2400 x4 : 0000000000001ae0 x3 : 0000000000000008
[    4.801375] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff83812c2140
[    4.801379] Call trace:
[    4.801380]  swiotlb_map+0x2c0/0x2f0 (P)
[    4.801383]  dma_map_page_attrs+0x1d8/0x368
[    4.801389]  nvkm_fb_ctor+0xc4/0x100 [nouveau]
[    4.801545]  gf100_fb_new_+0x68/0x98 [nouveau]
[    4.801675]  gk110_fb_new+0x38/0x50 [nouveau]
[    4.801802]  nvkm_device_ctor+0xaec/0x36d8 [nouveau]
[    4.801931]  nvkm_device_pci_new+0xe0/0x2b8 [nouveau]
[    4.802059]  nouveau_drm_probe+0x4c/0x1e0 [nouveau]
[    4.802189]  local_pci_probe+0x48/0xb8
[    4.802195]  pci_device_probe+0xcc/0x1d8
[    4.802198]  really_probe+0xc4/0x2a8
[    4.802203]  __driver_probe_device+0x80/0x140
[    4.802206]  driver_probe_device+0x44/0x170
[    4.802210]  __driver_attach+0x9c/0x1b0
[    4.802213]  bus_for_each_dev+0x84/0xf0
[    4.802216]  driver_attach+0x2c/0x40
[    4.802219]  bus_add_driver+0xec/0x218
[    4.802223]  driver_register+0x68/0x138
[    4.802225]  __pci_register_driver+0x54/0x68
[    4.802228]  nouveau_drm_init+0x24c/0xff8 [nouveau]
[    4.802357]  do_one_initcall+0x64/0x288
[    4.802360]  do_init_module+0x60/0x230
[    4.802363]  load_module+0x1c70/0x1f08
[    4.802366]  __do_sys_init_module+0x188/0x1f8
[    4.802369]  __arm64_sys_init_module+0x24/0x38
[    4.802372]  invoke_syscall+0x50/0x120
[    4.802376]  el0_svc_common.constprop.0+0xc8/0xf0
[    4.802380]  do_el0_svc+0x24/0x38
[    4.802383]  el0_svc+0x30/0xd0
[    4.802386]  el0t_64_sync_handler+0x10c/0x138
[    4.802389]  el0t_64_sync+0x198/0x1a0
[    4.802391] ---[ end trace 0000000000000000 ]---
```