geerlingguy/raspberry-pi-pcie-devices #4: Test GPU (

geerlingguy, 2020-10-22:
I want to see if an AMD card works out of the box with the drivers built into Linux, as everyone on the Internet seems to say. For X86 Linux, that definitely seems to be the case, but will it work on ARM32? ARM64?

I settled on the [Radeon HD 5450 1GB PCIe 2.1 card](https://amzn.to/3jjWNYD), mostly because it's available at a local retailer for $35.

Phoronix did a pretty extensive article on this board, and while it's no screamer... or even that fast... it is a simple, fanless, low-power board, and that might just be perfect for the Pi CM4. Here's that article: [ATI Radeon HD 5450 On Linux](https://www.phoronix.com/scan.php?page=article&item=ati_radeonhd_5450&num=1).

I don't expect it to be fast, or amazing, but I do expect to get it to work. Maybe.

dtischler, 2020-10-22:
AMD cards do work on Arm64...but...that is on machines with proper UEFI+ACPI and enough BAR.   So, not apples to apples here.  Just saying that the drivers are in fact functional.

geerlingguy, 2020-10-22:
@dtischler - Very good to know! I'm heading out to pick up the card from Micro Center in a little bit... fingers crossed this experience is a little easier :)

geerlingguy, 2020-10-22:
```
$ sudo lspci -v
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20) (prog-if 00 [Normal decode])
	Flags: fast devsel
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 00000000-00000fff
	Memory behind bridge: f8000000-f80fffff
	Capabilities: [48] Power Management version 3
	Capabilities: [ac] Express Root Port (Slot-), MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
	Capabilities: [240] L1 PM Substates

01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Cedar [Radeon HD 5000/6000/7350/8350 Series] (prog-if 00 [VGA controller])
	Subsystem: VISIONTEK Cedar [Radeon HD 5000/6000/7350/8350 Series]
	Flags: fast devsel, IRQ 255
	Memory at <unassigned> (64-bit, prefetchable) [disabled]
	Memory at 600000000 (64-bit, non-prefetchable) [disabled] [size=128K]
	I/O ports at <unassigned> [disabled]
	[virtual] Expansion ROM at 600020000 [disabled] [size=128K]
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Legacy Endpoint, MSI 00
	Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150] Advanced Error Reporting

01:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Cedar HDMI Audio [Radeon HD 5400/6300/7300 Series]
	Subsystem: VISIONTEK Cedar HDMI Audio [Radeon HD 5400/6300/7300 Series]
	Flags: fast devsel, IRQ 255
	Memory at 600040000 (64-bit, non-prefetchable) [disabled] [size=16K]
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Legacy Endpoint, MSI 00
	Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150] Advanced Error Reporting
```

geerlingguy, 2020-10-22:
Same BAR address space issue as the Zotac GeForce GT 710:

```
[    1.047548] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.047591] pci 0000:00:00.0: BAR 9: no space for [mem size 0x10000000 64bit pref]
[    1.047604] pci 0000:00:00.0: BAR 9: failed to assign [mem size 0x10000000 64bit pref]
[    1.047618] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.047638] pci 0000:01:00.0: BAR 0: no space for [mem size 0x10000000 64bit pref]
[    1.047650] pci 0000:01:00.0: BAR 0: failed to assign [mem size 0x10000000 64bit pref]
[    1.047664] pci 0000:01:00.0: BAR 2: assigned [mem 0x600000000-0x60001ffff 64bit]
[    1.047701] pci 0000:01:00.0: BAR 6: assigned [mem 0x600020000-0x60003ffff pref]
[    1.047716] pci 0000:01:00.1: BAR 0: assigned [mem 0x600040000-0x600043fff 64bit]
[    1.047749] pci 0000:01:00.0: BAR 4: no space for [io  size 0x0100]
[    1.047760] pci 0000:01:00.0: BAR 4: failed to assign [io  size 0x0100]
[    1.047774] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.047793] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    1.047897] pci 0000:01:00.1: D0 power state depends on 0000:01:00.0
```

So I'm going to follow the same process for increasing the BAR addressable memory range (see step 1 here: https://gist.github.com/geerlingguy/9f1510ab028e68b712381520308db2af).

geerlingguy, 2020-10-22:
Well that's weird. I extended the address range and am still gettting:

```
[    1.006265] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.006284] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.006343] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x060fffffff -> 0x00e0000000
[    1.006398] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0100000000
[    1.023282] brcm-pcie fd500000.pcie: link up, 2.5 GT/s x1 (SSC)
[    1.023571] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.023586] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.023603] pci_bus 0000:00: root bus resource [mem 0x600000000-0x60fffffff] (bus address [0xe0000000-0xefffffff])
[    1.023655] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.023874] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.027287] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.027475] pci 0000:01:00.0: [1002:68f9] type 00 class 0x030000
[    1.027563] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x0fffffff 64bit pref]
[    1.027603] pci 0000:01:00.0: reg 0x18: [mem 0x00000000-0x0001ffff 64bit]
[    1.027630] pci 0000:01:00.0: reg 0x20: [io  0x0000-0x00ff]
[    1.027673] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0001ffff pref]
[    1.027702] pci 0000:01:00.0: enabling Extended Tags
[    1.027858] pci 0000:01:00.0: supports D1 D2
[    1.027913] pci 0000:01:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x1 link at 0000:00:00.0 (capable of 32.000 Gb/s with 2.5 GT/s x16 link)
[    1.028051] pci 0000:01:00.0: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[    1.028125] pci 0000:01:00.1: [1002:aa68] type 00 class 0x040300
[    1.028209] pci 0000:01:00.1: reg 0x10: [mem 0x00000000-0x00003fff 64bit]
[    1.028315] pci 0000:01:00.1: enabling Extended Tags
[    1.028470] pci 0000:01:00.1: supports D1 D2
[    1.031733] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.031776] pci 0000:00:00.0: BAR 9: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    1.031791] pci 0000:00:00.0: BAR 8: no space for [mem size 0x00100000]
[    1.031803] pci 0000:00:00.0: BAR 8: failed to assign [mem size 0x00100000]
[    1.031823] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    1.031858] pci 0000:01:00.0: BAR 2: no space for [mem size 0x00020000 64bit]
[    1.031870] pci 0000:01:00.0: BAR 2: failed to assign [mem size 0x00020000 64bit]
[    1.031883] pci 0000:01:00.0: BAR 6: no space for [mem size 0x00020000 pref]
[    1.031894] pci 0000:01:00.0: BAR 6: failed to assign [mem size 0x00020000 pref]
[    1.031907] pci 0000:01:00.1: BAR 0: no space for [mem size 0x00004000 64bit]
[    1.031918] pci 0000:01:00.1: BAR 0: failed to assign [mem size 0x00004000 64bit]
[    1.031929] pci 0000:01:00.0: BAR 4: no space for [io  size 0x0100]
[    1.031940] pci 0000:01:00.0: BAR 4: failed to assign [io  size 0x0100]
[    1.031953] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.031978] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x60fffffff 64bit pref]
[    1.032077] pci 0000:01:00.1: D0 power state depends on 0000:01:00.0
```

geerlingguy, 2020-10-22:
I was also looking into RadeonOpenCompute, but it seems Debian 10 is not a supported OS: https://rocmdocs.amd.com/en/latest/Current_Release_Notes/Current-Release-Notes.html#list-of-supported-operating-systems

geerlingguy, 2020-10-22:
I'm going to re-flash 32-bit Pi OS and see if that makes any difference.

dtischler, 2020-10-22:
Before taking a saw to the x1 slot, you could try a SATA, Network, or other PCIe card in the x16 slot, and see if that enumerates and functions properly, just to confirm the adapter works.  If not, then, reach for the saw.

geerlingguy, 2020-10-22:
@dtischler - Good point. Will test another card.

geerlingguy, 2020-10-23:
Same thing under 32-bit Pi OS, unfortunately:

```
[    0.955978] pci 0000:00:00.0: BAR 9: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    0.955997] pci 0000:00:00.0: BAR 8: no space for [mem size 0x00100000]
[    0.956013] pci 0000:00:00.0: BAR 8: failed to assign [mem size 0x00100000]
[    0.956039] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    0.956084] pci 0000:01:00.0: BAR 2: no space for [mem size 0x00020000 64bit]
[    0.956100] pci 0000:01:00.0: BAR 2: failed to assign [mem size 0x00020000 64bit]
[    0.956118] pci 0000:01:00.0: BAR 6: no space for [mem size 0x00020000 pref]
[    0.956134] pci 0000:01:00.0: BAR 6: failed to assign [mem size 0x00020000 pref]
[    0.956151] pci 0000:01:00.1: BAR 0: no space for [mem size 0x00004000 64bit]
[    0.956167] pci 0000:01:00.1: BAR 0: failed to assign [mem size 0x00004000 64bit]
[    0.956183] pci 0000:01:00.0: BAR 4: no space for [io  size 0x0100]
[    0.956200] pci 0000:01:00.0: BAR 4: failed to assign [io  size 0x0100]
```

elFarto, 2020-10-23:
It looks like you're running out of BAR space again. BAR 9 gets allocated all the BAR space available, which is why the rest fail (well, apart from the IO one).

geerlingguy, 2020-10-23:
@elFarto - I just noted over in the Pi Forums issue that expanding the space to 1 GB seems to have fixed that issue:

```
ranges = <0x02000000 0x0 0xc0000000 0x6 0x00000000 0x0 0x40000000>;
```

However, I'm now realizing that just like the nouveau driver, it seems the amd drm drivers are nowhere in sight on Raspberry Pi OS (`find /lib/modules/$(uname -r) -type f -name '*.ko*' | grep amd` reveals nothing).

So I'm going to try to rebuild the kernel with those modules, on 32-bit Pi OS (I tried earlier on 64-bit, but must've missed something because I couldn't get nouveau to work). Fingers crossed!

geerlingguy, 2020-10-23:
Process for [building](https://www.raspberrypi.org/documentation/linux/kernel/building.md) and [configuring](https://www.raspberrypi.org/documentation/linux/kernel/configuring.md) the kernel:

```
# Install dependencies
sudo apt install -y git bc bison flex libssl-dev make

# Clone source
git clone --depth=1 https://github.com/raspberrypi/linux

# Apply default configuration
cd linux
export KERNEL=kernel7l # use kernel8 for 64-bit, or kernel7l for 32-bit
make bcm2711_defconfig

# Customize the .config further with menuconfig
sudo apt install -y libncurses5-dev
make menuconfig
# (search for /amd (and /nouveau if desired), enable in the proper section, save, then exit)

# Build the kernel and copy everything into place
make -j4 zImage modules dtbs # 'Image' on 64-bit
sudo make modules_install
sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/$KERNEL.img

# Redo the changes from this Gist to expand BAR space.
```

geerlingguy, 2020-10-23:
I'm now seeing the amdgpu module:

```
pi@raspberrypi:~ $ sudo modinfo amdgpu
filename:       /lib/modules/5.4.72-v7l-jjggpu+/kernel/drivers/gpu/drm/amd/amdgpu/amdgpu.ko
license:        GPL and additional rights
description:    AMD GPU
author:         AMD linux driver team
...
```

Then:

```
$ sudo modprobe amdgpu
$ dmesg | grep amdgpu
[  131.521279] [drm] amdgpu kernel modesetting enabled.
```

geerlingguy, 2020-10-23:
Drat! I think the amdgpu driver only works with newer Radeons ('Southern Islands' and newer).

Might have to re-recompile the kernel with the radeon driver, which according to this, supports the 'Evergreen' generation, which is the family the 5450 belongs to: https://www.x.org/wiki/RadeonFeature/

Should've probably done that from the start, oops.

dtischler, 2020-10-23:
Doh - I could have warned you of that, I was not paying close enough attention.  You are correct though, AMDGPU would be for newer cards.

geerlingguy, 2020-10-23:
Well, it's no biggie, I was just assuming that "the past 20 years" was modern here, where it's really just the past 5-10 years.

Interestingly, since I had also compiled nouveau (but it's not enabled), I am now finding that with my custom kernel, the whole Pi locks up during boot if I have the Nvidia card plugged in—it gets past the initial boot, the HDMI0 display turns on to a flashing cursor, then the cursor stops flashing and 'poof', it's locked up. No SSH access, no keyboard access. Weird. That doesn't happen with the Radeon plugged in.

geerlingguy, 2020-10-23:
Rebuilding the kernel the 4th time today, this time choosing the `radeon` driver and not `amdgpu`...

geerlingguy, 2020-10-23:
Well this is weird... `lspci` is showing nothing after recompiling, and I see:

```
[    0.865707] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.865731] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    0.865803] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    0.865875] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0100000000
[    1.464096] brcm-pcie fd500000.pcie: link down
...
[    5.292508] vc4-drm gpu: HDMI-A-2: EDID is invalid:
[    5.292530] 	[00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    5.292544] 	[00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    5.292557] 	[00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    5.292570] 	[00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    5.292583] 	[00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    5.292595] 	[00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    5.292608] 	[00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    5.292621] 	[00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Rebooting again. I have not yet enabled the `radeon` kernel module.

geerlingguy, 2020-10-23:
I keep getting that `[    1.464096] brcm-pcie fd500000.pcie: link down` after every reboot... switching back to a known-good microSD to make sure I haven't done something bad to the hardware.

geerlingguy, 2020-10-23:
After re-allocating 1GB of memory to the PCIe bus again, the device is getting recognized again (phew!), so back to recompiling the kernel the 6th time today.

geerlingguy, 2020-10-23:
Sooooo close:

Darn, *so close*:

```
$ dmesg | grep radeon
[    0.000000] Linux version 5.4.72-v7l-radeon+ (pi@raspberrypi) (gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)) #1 SMP Fri Oct 23 22:10:38 BST 2020
[    4.737476] [drm] radeon kernel modesetting enabled.
[    4.737774] radeon 0000:01:00.0: remove_conflicting_pci_framebuffers: bar 0: 0x0 -> 0xfffffff
[    4.737793] radeon 0000:01:00.0: remove_conflicting_pci_framebuffers: bar 2: 0x10000000 -> 0x1001ffff
[    4.737852] radeon 0000:01:00.0: enabling device (0140 -> 0142)
[    4.743219] [drm:radeon_device_init [radeon]] *ERROR* Unable to find PCI I/O BAR
[    4.896672] radeon 0000:01:00.0: Expecting atombios for evergreen GPU
[    4.896692] radeon 0000:01:00.0: Fatal error during GPU init
[    4.896708] [drm] radeon: finishing device.
[    4.991240] radeon: probe of 0000:01:00.0 failed with error -22
```

It looks like the radeon driver [i]requires[/i] the presence of the I/O BAR. Maybe that's also what's happening with the Nvidia proprietary driver, and it's just not giving a helpful error message about that?

geerlingguy, 2020-10-23:
I think we may be out of luck, in this case, as @pelwell [mentions](https://www.raspberrypi.org/forums/viewtopic.php?p=1746705#p1746705) that the spec sheet for the BCM2711 says:

> Supports accessing external PCIe configuration space and memory space (no support for I/O space).

@trejan on the Pi Forum also [said this](https://www.raspberrypi.org/forums/viewtopic.php?p=1746927#p1746927):

> Digging around, it looks like Nvidia have been reusing the same basic design for this for several generations now. That IOBAR is used for the legacy VGA I/O ports and also BIOS configuration. The x86 CPU is still in real mode at that point so can't access the other memory ranges. The Radeon Open Compute documentation also mentions that their cards have an IOBAR and it used for the same reasons but say it isn't needed if the card is operating headless for compute.
> 
> Not sure what the Nvidia closed source driver is doing though. Something not initialised properly or is it trying to access an I/O port? The Tegra PCIe controller apparently does support IOBARs according the source code.

I think we can say conclusively that this Radeon board will not work, though I can't say that conclusively about the Nvidia board. If the IO bar is used only for VGA, it would be nice if a driver could ignore it's absence, and maybe Nvidia's driver could at some point...

I wonder if there are any video cards that are not massively expensive (and likely to need too many resources to work with the Pi) that do not require an I/O BAR?

geerlingguy, 2020-10-27:
Some good reading here: [PCI BARs and other means of accessing the GPU](https://envytools.readthedocs.io/en/latest/hw/bus/bars.html)

scarburato, 2020-10-27:
The PCI I/O BAR error should not be lethal as it doesn't cause the probe to fail (see https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/radeon_device.c#L1427 ) it just print the error and then continues 

The probe fails with error 22 which is `Invalid argument`, which seems to be generated here ( https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/evergreen.c#L5186 you can also see the string "expecting atombios for evergreen gpu" being printed). The last error is then printed here ( https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/radeon_kms.c#L136 )

I'm not exactly sure what is this atombios if it's something on the gpu or on (x86, amd64) cpus or with missing IO BARs... anyway here ( https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/radeon_bios.c#L708 ) is where the driver seems to its thing to check this atom bios presence 

Not exactly sure if this might be helpful but those are my 2 cent from a rapid look to the Kernel code

geerlingguy, 2020-10-27:
Related: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/6

Also, I found out more clues through the grapevine today:

> You need to find out if the platform supports a cache coherent PCIe interface. The PCI spec requires it, but not all platforms (especially non-x86 platforms) support that. ARM has some IP you need to include for that to work correctly. Also, most non-x86 platforms seems to have problems with writecombining. They may need to patch `drm_arch_can_wc_memory()` for their platform; see https://patchwork.kernel.org/project/dri-devel/patch/20181220145657.304-1-alexander.deucher@amd.com/

geerlingguy, 2020-10-27:
Further:

> tl;dr - It's not the board, it's the old card.
> 
> Looks like the driver is unable to access the PCI ROM which is required to fetch the vbios image. The driver needs the vbios image to get board specific details (display topology, clocks, voltages, etc.). You'd probably have better luck on a newer board supported by amdgpu.
>
> The radeon driver requires that the PCI ROM BAR be accessible to fetch the vbios image, but on the amdgpu driver, we can read back the rom via MMIO registers.

So it seems like #6 might be the way to go, and might offer a more fruitful result. Thanks especially to Djhg2000 for the recommendation on YouTube, looks like he was on to something!

geerlingguy, 2020-10-28:
Over in the raspberrypi/linux project, it looks like this commit (https://github.com/raspberrypi/linux/commit/54db4b2fa4d17251c2f6e639f849b27c3b553939) has increased the default BAR allocation to 1GB by default—nice!

geerlingguy, 2020-10-29:
`unrelenting.technology` over on my blog [posted](https://www.jeffgeerling.com/comment/12488#comment-12488):

> Very interesting that the BAR space is configurable on the Broadcrap SoC.
> 
> You can get around the I/O BAR issue on radeon: in your kernel source, go to drivers/gpu/drm/radeon/radeon_device.c, remove these lines: https://github.com/torvalds/linux/blob/598a597636f8618a0520fd3ccefedaed9e4709b0/drivers/gpu/drm/radeon/radeon_device.c#L1419-L1428 (in `radeon_device_init()` the `/* io port mapping */` section).
> 
> This is a silly check left over there, you can see another place in that file says DRM_ERROR("Unable to find PCI I/O BAR; using MMIO for ATOM IIO\n"); so the cards are perfectly fine without the I/O BAR. In the FreeBSD port of the driver, these lines are just ifdef'd out, sooooo yeah :) And of course there is no such mistake in the amdgpu kernel driver so indeed modern Radeons likely would just work.

myfreeweb, 2020-10-30:
^ that would be me :) didn't find the github repo yesterday

Ah I see that this error is already non-fatal as mentioned above.

> which seems to be generated here ( https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/evergreen.c#L5186

oh. So, `Expecting atombios for evergreen GPU`, *without* `Unable to locate a BIOS ROM` or `BIOS signature incorrect`, without the [failing right before that message](https://elixir.bootlin.com/linux/v5.9.2/source/drivers/gpu/drm/radeon/evergreen.c#L5183) (note: `ASIC_IS_AVIVO(rdev) == (rdev->family >= CHIP_RS600)`, [CEDAR is far newer than RS600](https://elixir.bootlin.com/linux/v5.9.2/source/drivers/gpu/drm/radeon/radeon_family.h#L57)), this means that `radeon_get_bios` succeeded but `is_atom_bios == false`.

So either the firmware on the card is corrupted / some weird version, or the PCIe memory reads somehow failed in a way that

- the 0xAA55 magic number is fine
- the x86-bios part's magic is fine
- but the "ATOM" magic string is not fine.

I don't know how PCIe reads could fail in *that* way, maybe it's the former option.

Does the card work on an x86 PC? (Both Windows and Linux tests would be interesting)

ZILtoid1991, 2020-10-30:
There's probably some way to get around the I/O BAR issue by allocating it as normal BAR space in kernel, but I cannot guarantee that it would properly work, especially without glitches. This would also need a hack in the kernel's source code. Or at least it seems to me after reading up for it on osdev wiki.

I/O BARs are for emulating the behavior of memory mapped I/O devices without any loss to legacy functionality.

myfreeweb, 2020-10-31:
@ZILtoid1991 Radeon GPUs do not require I/O BARs, this has already been posted a few times in this thread. The radeon driver will happily use memory BARs, even if after logging a "scary" error message.

pppq, 2020-10-31:
It might legitimately have a "COM" BIOS (whatever that means, that was the other option in the debug message). There are some images uploaded from GPU-Z on TechPowerUp's site, which have the same PCI ID and refer to ATOM in the identifier strings, eg. one from an MSI card: https://www.techpowerup.com/vgabios/153193/msi-hd5450-1024-120919

myfreeweb, 2020-10-31:
@pppq COMBIOS is something really, really ancient that could never be legitimately found on this generation of GPUs. `radeon_get_bios` is used on all generations, it's common code. `evergreen_init` of course expects ATOMBIOS, but e.g. [`r100_init`](https://elixir.bootlin.com/linux/v5.9.2/source/drivers/gpu/drm/radeon/r100.c#L4050) expects COMBIOS. For some perspective, [r100 is from the year 2000](https://en.wikipedia.org/wiki/Radeon_R100_series).

PixlRainbow, 2020-10-31:
Interesting note: someone HAS successfully got hardware accelerated graphics a Radeon card on a non-x86 computer on Linux before, but it wasn't an ARM computer -- it was RISC-V. It was also *some* AMD-era Radeon, not an ATi-era Radeon. [They don't really go into much details though.](https://abopen.com/news/building-a-risc-v-pc/).

ZILtoid1991, 2020-10-31:
They also could've enabled the I/O BARs somehow.

myfreeweb, 2020-10-31:
@PixlRainbow I do have a Radeon RX 480 (also tested a Radeon HD 7950) running on an ARM computer (SolidRun MACCHIATObin) too. On FreeBSD instead of Linux, even :)

> the patches they had to make to the drivers to make them work

This is related to building userspace software, which in our case is already built by the distro. This is irrelevant.

> They also could've enabled the I/O BARs somehow

@ZILtoid1991 no need for "somehow", if your PCIe host controller is decent, I/O BARs Just Work™. They DO NOT require having ISA-level I/O ports like x86 `inb` `outb`. The controller just maps them into memory, of course.

---

Again, **I/O BARs are not required**! Everyone: read https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/4#issuecomment-719649428 carefully.

PixlRainbow, 2020-11-01:
Wait, which IO board has he mounted the CM4 onto?

paulwratt, 2020-11-04:
Just a note:
> It was also some AMD-era Radeon, not an ATi-era Radeon. 

the 5450 (HD5000) series _IS_ ATI-era hardware, even though it is produced by AMD. The HD7000 series _IS_ AMD-era, but (sorry) not sure about the HD6000 series. (dont quote me, but) I believe the HD5000 series to be the last of the ATI-era hardware. I know because I spent time trying to get both HIS HD5450 PCI Silent and an HD7450 PCIe Silent drivers on an old Dell/HP with MacOSX Leopard/Snow Leopard around 2011/12

myfreeweb, 2020-11-04:
> ATI-style Radeons

> The HD7000 series is AMD-era

This marketing mess is confusing, use the architecture names.
The first [TeraScale](https://en.wikipedia.org/wiki/TeraScale_%28microarchitecture%29) GPUs shipped in 2007, next year after AMD bought ATI. The ATI *brand* lasted until 2010, but the modern architecture [GCN](https://en.wikipedia.org/wiki/Graphics_Core_Next) shipped in 2012, so these timelines are not synchronized.

You CAN NOT tell the architecture by the "HD7xxx" first digit because of the massive amount of rebrands. 7670 and lower 7xxx numbers are TeraScale 2, just like the 5450 discussed here: https://en.wikipedia.org/wiki/Template:AMD_Radeon_HD_7xxx
This even happened into the next naming scheme for a bit, see R5 235/230/225/220.

Very helpful bookmark: https://en.wikipedia.org/wiki/Template:AMD_GPU_features

> I further suggest it would also be possible to get a vulkan driver to build with hardware support too

Mesa that comes in the distros already includes Vulkan support, namely the RADV driver. But Vulkan is only supported on GCN-era (amdgpu) GPUs, not on anything this old. TeraScale hardware DOES NOT RUN VULKAN.

> > with the knowledge presented in this thread (disabling some lines as BSD does) and the use of those patched user-space drivers (MESA, XORG)

Again, it turned out that these lines were NOT THE ISSUE (they just logged a message but did not end the loading process), everyone please stop repeating my mistake.

And NO USERSPACE PATCHES SHOULD BE NECESSARY. aarch64 is not riscv64. aarch64 Just Works with mainline Mesa.

geerlingguy, 2020-11-18:
Was going to test this card again today with the expanded BAR space (see https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/2#issuecomment-730025910), but this card is also giving me 'link down' all the time now :-/

geerlingguy, 2020-11-18:
Well, now it's finally booting at least, if I don't use external power.

```
[   50.212050] [drm] radeon kernel modesetting enabled.
[   50.212392] radeon 0000:01:00.0: remove_conflicting_pci_framebuffers: bar 0: 0x700000000 -> 0x70fffffff
[   50.212403] radeon 0000:01:00.0: remove_conflicting_pci_framebuffers: bar 2: 0x600000000 -> 0x60001ffff
[   50.212511] pci 0000:00:00.0: enabling device (0000 -> 0002)
[   50.212536] radeon 0000:01:00.0: enabling device (0000 -> 0002)
[   50.213507] [drm] initializing kernel modesetting (CEDAR 0x1002:0x68F9 0x1545:0x5450 0x00).
[   50.213710] [drm:radeon_device_init [radeon]] *ERROR* Unable to find PCI I/O BAR
[   50.329687] radeon 0000:01:00.0: Expecting atombios for evergreen GPU
[   50.329697] radeon 0000:01:00.0: Fatal error during GPU init
[   50.329703] [drm] radeon: finishing device.
[   50.329709] [TTM] Memory type 2 has not been initialized
[   50.337212] radeon: probe of 0000:01:00.0 failed with error -22
```

geerlingguy, 2020-11-18:
The `[TTM] Memory type 2 has not been initialized` seems different...

PixlRainbow, 2020-11-19:
probably should try a graphics card that has its own external power connector next time? So we aren't relying on pulling power through the PCIe slot.

Coreforge, 2020-12-26:
Got a CM4 now, so I've started some testing with a simillar card. It's a Radeon HD6450 2GB, which shouldn't be too different (should I open a new Issue anyways or continue in this one?).
I flashed the current 64bit version of Raspberry Pi OS, compiled the 5.10.y kernel with the radeon and amdgpu driver, blacklisted the radeon driver and modprobed it. It crashed as expected, but left a trace in dmesg.

`[  193.679976] [drm] radeon kernel modesetting enabled.
[  193.680427] pci 0000:00:00.0: enabling device (0000 -> 0002)
[  193.680449] radeon 0000:01:00.0: enabling device (0000 -> 0002)
[  193.680918] [drm] initializing kernel modesetting (CAICOS 0x1002:0x6779 0x174B:0xA004 0x00).
[  193.681196] [drm:radeon_device_init [radeon]] *ERROR* Unable to find PCI I/O BAR
[  193.807109] radeon 0000:01:00.0: Expecting atombios for evergreen GPU
[  193.807122] radeon 0000:01:00.0: Fatal error during GPU init
[  193.807129] [drm] radeon: finishing device.
[  193.807137] [TTM] Memory type 2 has not been initialized
[  193.814644] radeon: probe of 0000:01:00.0 failed with error -22
[  193.814653] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000168
[  193.825542] Mem abort info:
[  193.828420]   ESR = 0x96000005
[  193.831516]   EC = 0x25: DABT (current EL), IL = 32 bits
[  193.836934]   SET = 0, FnV = 0
[  193.840045]   EA = 0, S1PTW = 0
[  193.843224] Data abort info:
[  193.846276]   ISV = 0, ISS = 0x00000005
[  193.850187]   CM = 0, WnR = 0
[  193.853204] user pgtable: 4k pages, 39-bit VAs, pgdp=00000000423bc000
[  193.859742] [0000000000000168] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  193.868594] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[  193.874243] Modules linked in: radeon i2c_algo_bit ttm fuse sha256_generic cfg80211 rfkill 8021q garp stp llc vc4 cec v3d drm_kms_helper gpu_sched drm bcm2835_v4l2(C) bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_soc_core snd_compress drm_panel_orientation_quirks videodev snd_bcm2835(C) snd_pcm_dmaengine raspberrypi_hwmon snd_pcm snd_timer mc vc_sm_cma(C) snd syscopyarea sysfillrect sysimgblt fb_sys_fops rpivid_mem backlight uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[  193.927068] CPU: 1 PID: 446 Comm: Xorg Tainted: G         C        5.10.2-v8+ #1
[  193.934564] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[  193.941092] pstate: 00000005 (nzcv daif -PAN -UAO -TCO BTYPE=--)
[  193.947249] pc : radeon_driver_open_kms+0x40/0x190 [radeon]
[  193.952929] lr : radeon_driver_open_kms+0x38/0x190 [radeon]
[  193.958573] sp : ffffffc011f7b9e0
[  193.961927] x29: ffffffc011f7b9e0 x28: 0000000000000000 
[  193.967312] x27: 0000000000000000 x26: 0000000000000000 
[  193.972695] x25: ffffff806517b8c8 x24: 000000005cbd5280 
[  193.978079] x23: ffffff806517b800 x22: 0000000000000000 
[  193.983463] x21: ffffff805cb48800 x20: 0000000000000001 
[  193.988847] x19: ffffff806517b800 x18: 0000000000000000 
[  193.994232] x17: 0000000000000000 x16: 0000000000000000 
[  193.999614] x15: 0000000000000000 x14: 00000020632d6c6c 
[  194.004998] x13: 0006001072657474 x12: 0000000000000018 

Message from syslogd@raspberrypi at Dec 26 01:28:51 ...
 kernel:[  193.868594] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[  194.010381] x11: 0101010101010101 x10: 00000000000019e0 
[  194.015766] x9 : ffffffc010aa37b8 x8 : ffffff805cb48a00 
[  194.021149] x7 : 0000000000000000 x6 : 0000000000000000 
[  194.026533] x5 : ffffffc011239000 x4 : ffffffc011f7b9b0 
[  194.031916] x3 : ffffffc011239190 x2 : 0000000000000001 
[  194.037300] x1 : ffffff8042ff1e40 x0 : 0000000000000001 
[  194.042684] Call trace:
[  194.045193]  radeon_driver_open_kms+0x40/0x190 [radeon]
[  194.050562]  drm_file_alloc+0x150/0x238 [drm]
[  194.055012]  drm_open+0xe0/0x278 [drm]
[  194.058844]  drm_stub_open+0xb0/0x168 [drm]
[  194.063087]  chrdev_open+0xac/0x1a8
[  194.066619]  do_dentry_open+0x134/0x3a0
[  194.070501]  vfs_open+0x34/0x40
[  194.073680]  path_openat+0x87c/0xdf8
[  194.077298]  do_filp_open+0x80/0x108
[  194.080917]  do_sys_openat2+0x1f0/0x2a0
[  194.084800]  do_sys_open+0x60/0xb0
[  194.088243]  __arm64_sys_openat+0x2c/0x38
[  194.092305]  el0_svc_common.constprop.0+0x84/0x1e8
[  194.097158]  do_el0_svc+0x2c/0x98
[  194.100515]  el0_svc+0x20/0x30
[  194.103605]  el0_sync_handler+0xb0/0xb8
[  194.107489]  el0_sync+0x174/0x180
[  194.110847] Code: f9400c00 95de6f05 2a0003f4 37f807c0 (b9416ac0) 
[  194.117024] ---[ end trace 47e97aeaf44f2e91 ]---`
The ssh session I probed the module from also got some syslogd messages, so ssh seems to live long enough to get error messages out.
`Message from syslogd@raspberrypi at Dec 26 01:28:51 ...
 kernel:[  193.868594] Internal error: Oops: 96000005 [#1] PREEMPT SMP

Message from syslogd@raspberrypi at Dec 26 01:28:51 ...
 kernel:[  194.110847] Code: f9400c00 95de6f05 2a0003f4 37f807c0 (b9416ac0) `
If I don't blacklist radeon, the pi starts to start the desktop (console gets cleared) and just shows a blinking cursor until I power it down.

Here's also the BAR space allocation if anyone's interested 
`[    1.298057] pci 0000:00:00.0: BAR 9: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    1.301382] pci 0000:00:00.0: BAR 8: assigned [mem 0x610000000-0x6100fffff]
[    1.304690] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[... cut ...]
[    1.314597] pci 0000:01:00.1: BAR 0: assigned [mem 0x610040000-0x610043fff 64bit]
[    1.317882] pci 0000:01:00.0: BAR 4: no space for [io  size 0x0100]
[    1.321130] pci 0000:01:00.0: BAR 4: failed to assign [io  size 0x0100]
[    1.324352] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.327560] pci 0000:00:00.0:   bridge window [mem 0x610000000-0x6100fffff]
[    1.330762] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x60fffffff 64bit pref]
[    1.334156] pci 0000:01:00.1: D0 power state depends on 0000:01:00.0
`

Coreforge, 2020-12-26:
This is interesting. When I connected to the pi over ssh yesterday, it locked up after loading the radeon module. When connecting over serial though, I can still use the shell to some extend, so it doesn't lock up completely.

6by9, 2020-12-26:
```[  193.807109] radeon 0000:01:00.0: Expecting atombios for evergreen GPU```
is the main issue there.
https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/evergreen.c#L5186
```
	if (!rdev->is_atom_bios) {
		dev_err(rdev->dev, "Expecting atombios for evergreen GPU\n");
		return -EINVAL;
	}
```
ie it just gives up if the situation doesn't quite fit with what they expect. The Pi has no BIOS, therefore I don't know what it's finding there, nor whether it actually really matters.

One could try removing that failure path and seeing what happens next, but I'd be surprised if it just worked.
I think the kernel splat is likely to be a error handling path failing and causing a splat.

Wiki does seem to have a good set of articles describing the different Radeon cards. https://en.wikipedia.org/wiki/Radeon_HD_5000_series
Evergreen is the HD5000 series. The "Northern Islands" is meant to be the HD6000 range (Caicos for 6450), so I'm not quite sure why it's trying to load Evergreen for an HD6450.
It's identified correctly as ```CAICOS``` too
```[  193.680918] [drm] initializing kernel modesetting (CAICOS 0x1002:0x6779 0x174B:0xA004 0x00)```,
which comes from https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/radeon_device.c#L1312
https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/radeon_asic.c#L2419 is the case for the Evergreen HD5000's, and https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/radeon_asic.c#L2438 is the case for "Northern Islands" HD6000's.

geerlingguy, 2020-12-26:
Speaking of the `HD6450` in particular (I'm fine with that being merged into this issue; it would be nice to pop it into the database site too!)—I remember speaking to a few people who seem to have more close affiliation with AMD and they suggested that generation and older would not likely be easy (heh, 'easy'... none of these GPUs I would qualify as 'easy'!) to get working on ARM due to some design constraints from the older generations.

But I am far from qualified to know what I'm talking about with old GPUs—as I've said elsewhere, the last time I ever spent much time (until now) worrying about GPUs and drivers was when the ATi Rage Pro 128 was new and amazing.

6by9, 2020-12-26:
Looking at the table on Wiki, I wonder if looking at the newer cards using the drm/amdgpu driver instead of drm/radeon is likely to give better results. Time to try and find a cheap Rx300 series card.

I've not worried about x86 GPUs either for a long while - I've never been a heavy gamer, so integrated graphics has normally worked for me. Although I did get the Acer Revo R3610 with added Nvidia ION GPU so it actually made a reasonable media player!

Coreforge, 2020-12-27:
Since I can't get the debugger to work, I'm using printk to check where it crashes. 
It probably crashes when I'm copying the BIOS buffer into a temporary buffer to print it to then check if it's a read error or if something else is happening. I'm not too great with C, so it's probably a pretty easy error. atombios seems to be a bios in bytecode that gets interpreted by a specific interpreter that is in the PCI ROM for x86 or in the driver for other architectures. It doesn't seem to be exclusive to evergreen, so I don't know why that's in evergreen.c, but it makes sense it's using it. Removing the check would probably result in an error further down the line when it's being interpreted, so that's not really a good option. One way to get around this could be to read out the bios with a PC and copy it over to the pi, and modify the driver so that it uses the image instead of reading it out. The fglrx driver apparently does that when it can't read the bios correctly, but that one isn't available for arm.
The code for copying parts of the bios:
`void *testbios = kmalloc(128,GFP_KERNEL);
	memcpy(testbios,rdev->bios_header_start,128);
	printk("first 128 bytes after BIOS header start: %*ph",testbios);
	memcpy(testbios,rdev->bios,128);
	printk("first 128 bytes from bios: %*ph",testbios);
	kfree(testbios);`

My guess is that either testbios isn't being properly allocated, or one of the addresses is NULL.

Coreforge, 2020-12-27:
Got it working. The first 64 bytes after the start of the header are `00 00 01 01 41 54 01 01 41 54 c3 03 74 01 c3 03 74 01 22 04 00 00 22 04 00 00 04 a0 04 02 04 a0 04 02 a0 00 00 00 a0 00 00 00 02 10 79 67 02 10 79 67 00 00 00 03 00 00 00 03 00 00 00 00 00 00`
The ATOM magic starts at the 5th byte, so the AT are there, but the second half is missing.

The start of the BIOS is `55 aa 80 e9 55 aa 80 e9 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 02 00 00 04 02 00 00 4d 44 00 00 4d 44 00 00 00 00 00 00 00 00 00 00 20 37 36 31 20 37 36 31 32 30 00 00 32 30 00 00`
I'll dump the full BIOS on the PI to see what's going on. I'll also try to remove the check and see what's happening.

Coreforge, 2020-12-27:
Removing the ATOM magic check gets it further (obviously), but it just fails later, as expected.
```

[   75.096398] [drm:radeon_atombios_init [radeon]] *ERROR* Unable to find PCI I/O BAR; using MMIO for ATOM IIO
[   75.096409] Invalid ATI magic
[   75.096419] radeon 0000:01:00.0: Fatal error during GPU init
[   75.096428] [drm] radeon: finishing device.
[   75.096434] [TTM] Memory type 2 has not been initialized
[   75.104133] radeon: probe of 0000:01:00.0 failed with error -12

```
Since the ATI magic is invalid too, I don't think just removing all these checks is going to work well. File IO doesn't seem to easy in kernel modules, so I haven't dumped the BIOS yet. A BIOS image can just be in a header file however, so I'll try that next.

Coreforge, 2020-12-27:
It's getting further with the BIOS image now. It's just a rom from techpowerup since I didn't want to go through the trouble of reading it out right now. The pi crashes before though and locks up shortly after, but it still prints for a bit in dmesg. It doesn't react to anything though (neither serial nor ssh).

```
[   68.275242] ------------[ cut here ]------------
[   68.275244] field width 1152254592 too large
[   68.275245] WARNING: CPU: 1 PID: 724 at lib/vsprintf.c:2506 set_field_width+0x94/0xa0
[   68.275247] Modules linked in: radeon(+) i2c_algo_bit ttm sha256_generic cfg80211 rfkill 8021q garp stp llc vc4 cec v3d drm_kms_helper gpu_sched drm bcm2835_v4l2(C) bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops snd_soc_core videobuf2_v4l2 videobuf2_common videodev drm_panel_orientation_quirks raspberrypi_hwmon snd_compress snd_bcm2835(C) snd_pcm_dmaengine snd_pcm mc snd_timer vc_sm_cma(C) snd syscopyarea sysfillrect sysimgblt rpivid_mem fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[   68.275299] CPU: 1 PID: 724 Comm: insmod Tainted: G         C        5.10.2-v8+ #2
[   68.275300] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   68.275302] pstate: 60000085 (nZCv daIf -PAN -UAO -TCO BTYPE=--)
[   68.275303] pc : set_field_width+0x94/0xa0
[   68.275304] lr : set_field_width+0x94/0xa0
[   68.275305] sp : ffffffc01200b3f0
[   68.275307] x29: ffffffc01200b3f0 x28: ffffffc0090aff5c 
[   68.275310] x27: ffffffc0114a084a x26: 0000000000000020 
[   68.275313] x25: ffffffc0090aff5c x24: 00000000000003d0 
[   68.275316] x23: 00000000ffffffd0 x22: ffffffc010b45070 
[   68.275319] x21: ffffffc01200b780 x20: ffffffc01200b490 
[   68.275323] x19: 0000000044ae0280 x18: 0000000000000001 
[   68.275326] x17: ffffff8041d7f0a0 x16: 0000000000000000 
[   68.275329] x15: ffffff804360e020 x14: ffffffffffffffff 
[   68.275332] x13: ffffff807fb8b8e8 x12: ffffff807fb8993b 
[   68.275335] x11: 0000000005f5e100 x10: abcc77118461cefd 
[   68.275338] x9 : ffffffc0100e95dc x8 : 000000000000000b 
[   68.275341] x7 : 0000000000000064 x6 : ffffff807fb8994f 
[   68.275344] x5 : 00000000fffffff2 x4 : ffffff807fb898f8 
[   68.275347] x3 : 0000000000000000 x2 : 0000000000000023 
[   68.275350] x1 : 0000000000000000 x0 : 0000000000000000 
[   68.275353] Call trace:
[   68.275354]  set_field_width+0x94/0xa0
[   68.275355]  vsnprintf+0x1b8/0x730
[   68.275357]  vscnprintf+0x30/0x68
[   68.275358]  vprintk_store+0x78/0x230
[   68.275359]  vprintk_emit+0xfc/0x328
[   68.275361]  vprintk_default+0x40/0x50
[   68.275362]  vprintk_func+0xfc/0x308
[   68.275363]  printk+0x68/0x90
[   68.275364]  radeon_get_bios+0xdcc/0xed0 [radeon]
[   68.275366]  evergreen_init+0x20/0x358 [radeon]
[   68.275367]  radeon_device_init+0x4c4/0xa68 [radeon]
[   68.275369]  radeon_driver_load_kms+0x74/0x170 [radeon]
[   68.275370]  drm_dev_register+0xe8/0x220 [drm]
[   68.275371]  radeon_pci_probe+0x118/0x188 [radeon]
[   68.275373]  pci_device_probe+0xc0/0x190
[   68.275374]  really_probe+0xf0/0x4d0
[   68.275375]  driver_probe_device+0xfc/0x168
[   68.275376]  device_driver_attach+0x7c/0x88
[   68.275378]  __driver_attach+0xac/0x178
[   68.275379]  bus_for_each_dev+0x78/0xd0
[   68.275380]  driver_attach+0x2c/0x38
[   68.275381]  bus_add_driver+0x14c/0x230
[   68.275383]  driver_register+0x6c/0x128
[   68.275384]  __pci_register_driver+0x4c/0x58
[   68.275385]  radeon_init+0x84/0x1000 [radeon]
[   68.275387]  do_one_initcall+0x4c/0x2d0
[   68.275388]  do_init_module+0x60/0x248
[   68.275389]  load_module+0x2204/0x2970
[   68.275391]  __do_sys_finit_module+0xbc/0x128
[   68.275392]  __arm64_sys_finit_module+0x28/0x38
[   68.275393]  el0_svc_common.constprop.0+0x84/0x1e8
[   68.275395]  do_el0_svc+0x2c/0x98
[   68.275396]  el0_svc+0x20/0x30
[   68.275397]  el0_sync_handler+0xb0/0xb8
[   68.275398]  el0_sync+0x174/0x180
[   68.275400] ---[ end trace 240e9a4e757dda1f ]---
[   68.275471] first 64 bytes after BIOS header start: 24 00 01 01 41 54 4f 4d 00 c0 c3 03 75 01 29 02 cf 00 22 04 00 00 00 00 4b 17 04 a0 04 02 aa a7 4e a8 a0 00 50 43 49 52 02 10 79 67 00 00 18 00 00 00 00 03 80 00 0c 0d 00 00 00 00 41 4d 44 20
[   68.275476] ATOM magic true
[   68.275482] testing if BIOS is AtomBIOS
[   68.275487] Made it past AtomBIOS check
[   68.275551] [drm:radeon_atombios_init [radeon]] *ERROR* Unable to find PCI I/O BAR; using MMIO for ATOM IIO
[   68.275566] ATOM BIOS: R5
[   68.275649] [drm] GPU not posted. posting now...
[   68.278927] radeon 0000:01:00.0: VRAM: 2048M 0x0000000000000000 - 0x000000007FFFFFFF (2048M used)
[   68.278936] radeon 0000:01:00.0: GTT: 1024M 0x0000000080000000 - 0x00000000BFFFFFFF
[   68.278942] [drm] Detected VRAM RAM=2048M, BAR=256M
[   68.278946] [drm] RAM width 64bits DDR
[   68.279144] [TTM] Zone  kernel: Available graphics memory: 947156 KiB
[   68.279150] [TTM] Initializing pool allocator
[   68.279170] [TTM] Initializing DMA pool allocator
[   68.279238] [drm] radeon: 2048M of VRAM memory ready
[   68.279248] [drm] radeon: 1024M of GTT memory ready.
[   68.279286] [drm] Loading CAICOS Microcode
[   68.284583] [drm] Internal thermal controller without fan control
[   68.289796] [drm] radeon: dpm initialized
[   68.295016] [drm] GART: num cpu pages 262144, num gpu pages 262144
[   68.297634] [drm] enabling PCIE gen 2 link speeds, disable with radeon.pcie_gen2=0

```

The crash probably happens somewhere in my code that replaces the bios in memory with the rom.

```
	kfree(rdev->bios);
	printk("freed old BIOS buffer");
	rdev->bios = kzalloc(__222524_rom_len, GFP_KERNEL);
	printk("allocated new BIOS buffer for image");
	memcpy(rdev->bios,&__222524_rom,__222524_rom_len);
	printk("copied BIOS image into buffer");
```

6by9, 2020-12-27:
You had noticed that ```%*ph``` says up to 64 bytes long
https://elixir.bootlin.com/linux/latest/source/Documentation/core-api/printk-formats.rst#L257

```field width 1152254592 too large``` is just a warning. The kernel clamps the maximum size - https://elixir.bootlin.com/linux/latest/source/lib/vsprintf.c#L2507

I'd suspect that you're running quite a long way past that WARN, and it's initialised a load more stuff before blowing up.
Where exactly have you replaced the BIOS?

Coreforge, 2020-12-27:
The printk was the issue. Removed that and now that warning is gone. 
I'm loading the rom directly after this line: https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/radeon_bios.c#L689 which should be directly after it was read (I could probably take the reading part out as it just gets replaced anyways). It still locks up though.

Coreforge, 2020-12-28:
The same thing happens with the bios I dumped with my pc, so that's not the problem anymore. There's nothing happening at the HDMI output if the card, so the issue probably isn't that I don't have a monitor connected. Enabling DRM debug messages would probably be a good way to get an idea where it hangs, but I haven't been able to do that. If I could get the debugger working that could help too, but so far, I haven't gotten it to work.

6by9, 2020-12-28:
Drm debug is easiest to do by adding ```drm.debug=0x3f``` to /boot/cmdline.txt (don't add an carriage returns).
I suspect it'll be blowing up within the probe/bind though, so that may jot be so useful as it is framework level logging.

Coreforge, 2020-12-28:
Drm debug didn't have any useful information. I've tracked down where it hangs using more printk, so it might not be totally accurate, but I've placed msleep(100) after the printks so that there's a bit more time for the data to get out.
It seems to get stuck on this line https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/evergreen.c#L5057
The output I get from dmesg is:
```
[   57.331338] radeon 0000:01:00.0: WB enabled
[   57.331346] radeon_wb_init done. r=0
[   57.351303] radeon 0000:01:00.0: fence driver on ring 0 use gpu addr 0x0000000080000c00
[   57.351311] radeon_fence_driver_ring GFX done. r=0
[   57.459306] radeon 0000:01:00.0: fence driver on ring 3 use gpu addr 0x0000000080000c0c
```

Interestingly, these only got out after i added more printks (I added the second and fourth one), so it's probably not actually in this line, but somewhere later. My guess is the IRQ stuff gets stuck somewhere.

6by9, 2020-12-28:
Are you remembering to add a ```\n``` to the end of all your printk lines? The console only flushes on a ```\n```

Coreforge, 2020-12-28:
I didn't on all. Some after the IRQ section had it, so it didn't get past that, but I didn't have it on some in that section.

[505 more replies]