geerlingguy/raspberry-pi-pcie-devices #8: Here is something I would like someone to test: a full-on south bridge.

xcvista, 2020-10-29:
Specifically, a A320/B350 south bridge. Those 1st generation AMD Ryzen south bridges are really just PCIe multi-downlink bridges with SATA AHCI, USB XHCI, more PCIe ports etc all packed in one chip. If a full blown south bridge is useable, it would make certain complicated projects a lot more viable as there would be a simple solution to get a lot of varied I/O.

ZILtoid1991, 2020-10-30:
I have a feeling that it would lock you out for not using an AMD processor, or would require a full custom firmware, for which the documentation is basically nonexistent outside of OEMs.

xcvista, 2020-10-31:
@ZILtoid1991 It may not be the case if any indications from previous AMD chipsets are true: their AM3 chipsets are used by Loongson and their MIPS platform with HyperTransport FSB with little ill effects. Now we have effectively a PCIe "FSB" which matches the AM4 chipsets.

geerlingguy, 2020-11-23:
For my own purposes, I'd rather leave this testing up to someone else. A few people are starting to get their own CM4s, and if someone wants to do some testing, open an issue, and post some results in this project (and add a listing on the Pi PCI Express DB page), I'll accept that, but I won't be working on it on my own.

closed 2020-11-23 by geerlingguy