geerlingguy/raspberry-pi-pcie-devices #11: Gpu Suggestion

eliteSchwein, 2020-10-30:
Hey there.

Maybe you can try a matrox card? Or a aspeed one? (If you can buy them as a dedicated gpu)

geerlingguy, 2020-11-23:
For now, I'm going to remain focused on testing with Nvidia/AMD cards, just because it seems there's more opportunity for help around those cards. In the future I may definitely take a look at some other video cards though, but for now I'm going to stay focused on the more mainline GPUs from the two largest manufacturers.

closed 2020-11-23 by geerlingguy

PixlRainbow, 2020-11-24:
matrox cards are pretty common in backend rack servers as basic graphics cards for displaying terminals or for just navigating around the Windows GUI. Some of them are also used because they support remote access video streaming on a hardware level. They are also pretty old -- they've been around for over 20 years afaik. Support for them on Linux isn't very good, but they seem to work just fine on standard VGA drivers -- which unfortunately is an x86 feature.