geerlingguy/raspberry-pi-pcie-devices #9: Open your horizon of hardware by using this PCI-E to PCI bridge

ZILtoid1991, 2020-10-30:
https://www.amazon.com/StarTech-com-PCI-Express-Adapter-Card/dp/B0024CV3SA

This in theory would allow you to use PCI cards, like soundcards, or maybe even some quite antiquated graphics cards. I might can help you with driver hacking (there's an open source driver for the SBLive, maybe it can even access the SPU too), however I currently don't own any Raspberry Pi devices, let alone the CM4.

boogermann, 2020-11-01:
@ZILtoid1991 There's a cheaper option that allows 2 devices.
https://www.amazon.com/gp/product/B00KZHDSLQ/
I've been waiting to try a voodoo card with that to play some Glide games with DOSBox or Box86

UnKnoWn-Consortium, 2020-11-01:
Both options use the Pericom (now Diodes) PI7C9X111SL which runs on PCI-E 1.1. 

https://www.diodes.com/part/view/PI7C9X111SL

ZILtoid1991, 2020-11-01:
> 
> 
> @ZILtoid1991 There's a cheaper option that allows 2 devices.
> https://www.amazon.com/gp/product/B00KZHDSLQ/
> I've been waiting to try a voodoo card with that to play some Glide games with DOSBox or Box86

The Voodoo cards most likely don't use any I/O BARs too, so you don't have to workaround it. You might also can get a YMF7xx soundcard working there for OPL3, however that very likely used the I/O BAR, since it's a legacy device.

mi-hol, 2020-11-02:
this could be a great solution for reviving old DVB-C or DVB-S TV-tuner cards

UnKnoWn-Consortium, 2020-11-08:
@mi-hol Maybe those good old analogue TV and capture cards too. The drivers are big enough headaches though.

ZILtoid1991, 2020-11-08:
There are open-source drivers for them, you just have to edit them.

geerlingguy, 2020-11-23:
@ZILtoid1991 hehe, "you just have to..." is the famous last line that leads to days of debugging something you *know* you can get working :D

However, for now at least, I'm going to focus only on PCIe devices, even though I know there are a ton of old PCI boards sitting around (heck, I think my dad has some old FireWire and SCSI boards that would someday be useful for extracting data off old MiniDV tapes and hard drives). But if someone else wants to do some research, open a PR, and add some data to the project, please be my guest. But in terms of active work in this issue queue, I'm going to limit it to newer PCIe devices—_for now_.

closed 2020-12-11 by geerlingguy