geerlingguy/raspberry-pi-pcie-devices #12: Suggestion -  Test Mellanox Infiniband card

jamesy0ung, 2020-10-31:
They are pretty cheap on eBay where I live: https://www.ebay.com.au/sch/i.html?LH_CAds&_fpos&_fspt=1&_mPrRngCbx=1&_sacat&_sadis&_udhi&_udlo&_fosrp=1&_nkw=infiniband%20pci-e&_sop=15&_clu=2&_fcid=15&_localstpos=2227&_stpos=2227&gbr=1

sinetek, 2020-11-01:
maybe you could also test Firewire,  or Acquisition cards.

https://www.amazon.com/Hauppauge-ImpactVCB-Express-Capture-1381/dp/B0068IMT5M/ref=sr_1_13?dchild=1&keywords=Hauppauge&qid=1604192743&sr=8-13

jamesy0ung, 2020-11-01:
Main thing here would be firewire or infiniband because they use dma

geerlingguy, 2020-11-01:
@sinetek - I do have a couple old FireWire drives laying around, and besides ugly messes of Thunderbolt to FireWire adapters, it would be cool to have a way to interface with them natively. Also old digital camcorders that had 'i.Link'.

I don't have any infiniband gear to test with, so if I had a card I could just verify it loads up, but wouldn't be able to do anything with it :/

jamesy0ung, 2020-11-04:
> I don't have any infiniband gear to test with, so if I had a card I could just verify it loads up, but wouldn't be able to do anything with it :/
You can grab 2 and a cable for under $50
Then connect them together

closed 2020-12-01 by geerlingguy