geerlingguy/raspberry-pi-pcie-devices #15: Test 10 Gbps ASUS PCIe card

geerlingguy, 2020-11-06:
TODO: Fill in the details, add it to the index page.

UnKnoWn-Consortium, 2020-11-08:
So it is an Aquantia AQN107 based card. 

Sad that I have not yet received my preordered CM4 and IO board in my country. Otherwise I can test it with a good old Intel X550 10G NIC. I guess the bottleneck is with the PCI-E bus and SoC MSI support regardless.

geerlingguy, 2020-11-08:
@UnKnoWn-Consortium - Ooh, I'd be interested to see how you get along with the X550; I was glancing through listings for a few Intel cards too, though that one was a little out of my 'test bench' price range, and I didn't need the extra port.

I did end up receiving a few Intel 10G SFP+ cards though, so I'll see how far I can get with them too!

But yeah, in the end, I'm pretty sure I'll max things out around 3.2 Gbps on the interface, since that's around the bus limit for the 1x PCIe 2.0 lane on the Pi. We'll see, though!

misaka10086, 2020-11-09:
I don't think CAT5e in a short distance will be a issue. I personally run 10G using CAT5e cable(20m) and get almost full speed(Intel X520). Someone actually tested it in 50m length and found it still work. I would rather switch to SFP+ instead of CAT6+ expensive cable.

geerlingguy, 2020-11-09:
@misaka10086 - That's good to know. Half my house is wired with Cat5e, and the other half Cat6 (non-a), which I wired up when I moved in a decade ago using remnants from other projects.

I kinda wonder if I could justify going fiber pair for future-proofing... though the basement is still unfinished, meaning ripping out cables and pulling new ones is not too strenuous, just a bit messy. Plus, my security system will never (I hope!) need 10, 20, or 40G networking!).

UnKnoWn-Consortium, 2020-11-09:
@geerlingguy Optical fiber cables also comes with CATs like copper cables. For example, the short-distance multi-mode fiber cables have 5 categories (OM1-5) and they are not exactly backward compatible with each other (OM1<>OM2 and OM3<>OM4<>OM5 only) as the light source evolves and the core size changes. So I guess they may not be as futureproof as one would have hoped for. But, of course, with a top spec speed of 100Gbps, they are unquestionably futureproof speed-wise.

geerlingguy, 2020-12-27:
FYI one of the most annoying things about the MikroTik interface for their RouterOS is that to switch to SwitchOS it requires going to the RouterOS web UI (you can also do it via the console), then going to System > Routerboard, then clicking Settings, then you can finally select Switch OS instead of Router OS, and reboot.

It took me way too long to figure that out. RouterOS has a lot of flexibility, but _toooons_ of complexity too. A lot for a little 4 port switch (that I just want to use as a switch ;).

geerlingguy, 2020-12-29:
All right, finally giving this card its first light. First interesting finding:

  - When used with a simple passive 1x to 16x PCIe adapter, `lspci` gives me nothing.
  - When used with my now-preferred, 'don't burn your card' powered external 16x riser ([this Rosewill version](https://pipci.jeffgeerling.com/cards_adapter/pce164p-no3-ver-006-riser.html)), it does seem to appear. So maybe this card needs the extra power.

```
$ sudo lspci -vvvv
01:00.0 Ethernet controller: Aquantia Corp. AQC107 NBase-T/IEEE 802.3bz Ethernet Controller [AQtion] (rev 02)
	Subsystem: ASUSTeK Computer Inc. AQC107 NBase-T/IEEE 802.3bz Ethernet Controller [AQtion]
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Region 0: Memory at 600440000 (64-bit, non-prefetchable) [disabled] [size=64K]
	Region 2: Memory at 600450000 (64-bit, non-prefetchable) [disabled] [size=4K]
	Region 4: Memory at 600000000 (64-bit, non-prefetchable) [disabled] [size=4M]
	[virtual] Expansion ROM at 600400000 [disabled] [size=256K]
	Capabilities: [40] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM L0s L1, Exit Latency L0s unlimited, L1 unlimited
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis+, LTR-, OBFF Not Supported
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [80] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=375mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [90] MSI-X: Enable- Count=32 Masked-
		Vector table: BAR=2 offset=00000000
		PBA: BAR=2 offset=00000200
	Capabilities: [a0] MSI: Enable- Count=1/32 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [c0] Vital Product Data
		Product Name: Atlantic
		Read-only fields:
			[PN] Part number: 3290495095
			[EC] Engineering changes: 0
			[FG] Unknown: 61 62 63
			[LC] Unknown: 64 65 66
			[MN] Manufacture ID: 41 46 44 53 57 45 57 45 42 53 46 44
			[PG] Unknown: 49 49 49
			[SN] Serial number: CPL5938TLKMY
			[V0] Vendor specific: wfewfe
			[V1] Vendor specific: fwewfe
			[V2] Vendor specific: SDFWI
			[RV] Reserved: checksum good, 0 byte(s) reserved
		Read/write fields:
			[YA] Asset tag: 9495829
			[V0] Vendor specific: f34ge4rsg
			[V1] Vendor specific: ger35g5rthghgsa3
			[Y0] System specific: bsdfvbxcz
			[Y1] System specific: fwefewwfe
			[RW] Read-write area: 11 byte(s) free
		End
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [150 v1] Vendor Specific Information: ID=0001 Rev=1 Len=024 <?>
	Capabilities: [180 v1] #19
```

geerlingguy, 2020-12-29:
```
$ dmesg
...
[    1.217331] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.217372] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.217457] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.217556] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0100000000
[    1.250213] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.250579] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.250611] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.250640] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.250738] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.250985] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.254619] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.254870] pci 0000:01:00.0: [1d6a:07b1] type 00 class 0x020000
[    1.254948] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x0000ffff 64bit]
[    1.255003] pci 0000:01:00.0: reg 0x18: [mem 0x00000000-0x00000fff 64bit]
[    1.255056] pci 0000:01:00.0: reg 0x20: [mem 0x00000000-0x003fffff 64bit]
[    1.255097] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0003ffff pref]
[    1.255140] pci 0000:01:00.0: enabling Extended Tags
[    1.255357] pci 0000:01:00.0: supports D1 D2
[    1.255380] pci 0000:01:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    1.255469] pci 0000:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0000:00:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    1.258937] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.258992] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6005fffff]
[    1.259026] pci 0000:01:00.0: BAR 4: assigned [mem 0x600000000-0x6003fffff 64bit]
[    1.259079] pci 0000:01:00.0: BAR 6: assigned [mem 0x600400000-0x60043ffff pref]
[    1.259110] pci 0000:01:00.0: BAR 0: assigned [mem 0x600440000-0x60044ffff 64bit]
[    1.259161] pci 0000:01:00.0: BAR 2: assigned [mem 0x600450000-0x600450fff 64bit]
[    1.259211] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.259243] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6005fffff]
```

geerlingguy, 2020-12-29:
Sadly, it seems like Aquantia no longer exists: https://www.aquantia.com/driver-download/

It was [purchased by Marvell in 2019](https://www.benzinga.com/news/19/05/13671774/marvell-buys-aquantia-in-450m-deal).

So off to Marvell's site to find drivers!

geerlingguy, 2020-12-29:
Looks like https://www.marvell.com/support/downloads.html has Linux drivers for the AQC107, specifically `Marvell AQtion Linux driver` version 2.4.3.

```
# Install kernel headers so you can build the driver.
wget -O headers.sh https://gist.githubusercontent.com/satmandu/a507c59d84737f6d29ff353395819d51
chmod +x headers.sh
sudo ./headers.sh
sudo dpkg -i /root/workdir/build/linux-image-5.10.3-v8+_arm64.deb

# Copy Marvell_Linux_2.4.3.0_207.zip over to the Pi from my Mac.

# Unzip and enter the directory.
unzip Marvell_Linux_2.4.3.0_207.zip
cd unzip Marvell_Linux_2.4.3.0_207

# Untar the Linux driver tarball and enter the directory.
tar xzf atlantic.tar.gz
cd Linux

# Compile the driver.
make
```

Just noting the directions in the downloaded README seem to be written for an older version of the drivers, before Marvell took over. Ah, acquisitions...

geerlingguy, 2020-12-29:
Well that's annoying:

```
/home/pi/Marvell_Linux_2.4.3.0_207/Linux/aq_sysfs.c: In function ‘memreg_mmap’:
/home/pi/Marvell_Linux_2.4.3.0_207/Linux/aq_sysfs.c:227:7: error: ‘struct vm_area_struct’ has no member named ‘vm_page_pgprot’; did you mean ‘vm_page_prot’?
  vma->vm_page_pgprot = prot_dmacoherent(vma->vm_page_prot);
       ^~~~~~~~~~~~~~
       vm_page_prot
/home/pi/Marvell_Linux_2.4.3.0_207/Linux/aq_sysfs.c:227:24: error: implicit declaration of function ‘prot_dmacoherent’; did you mean ‘pgprot_dmacoherent’? [-Werror=implicit-function-declaration]
  vma->vm_page_pgprot = prot_dmacoherent(vma->vm_page_prot);
                        ^~~~~~~~~~~~~~~~
                        pgprot_dmacoherent
```

So. Like with other cards: kernel recompile time? I think so!

geerlingguy, 2020-12-29:
Menuconfig:

```
Device Drivers
  > Network device support
    > PHY Device support and infrastructure
      > Aquantia PHYs
```

Recompiling now...

geerlingguy, 2020-12-29:
Things... kinda locked up. So adding `/etc/modprobe.d/blacklist-aquantia.conf` with `blacklist aquantia` inside.

geerlingguy, 2020-12-29:
Hmm... running `sudo modprobe aquantia` I get nothing. `lspci` still shows the card though.

geerlingguy, 2020-12-29:
Oops. Missed something. I _also_ enabled:

```
Device Drivers
  > Network device support
    > Ethernet driver support
      > aQuantia devices
        > aQuantia AQtion(tm) Support
```

I've recompiled and am copying things over to the Pi now.

geerlingguy, 2020-12-29:
I got the blinkenlights!!!

```
$ ip address
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether b8:27:eb:5c:89:43 brd ff:ff:ff:ff:ff:ff
    inet 10.0.100.120/24 brd 10.0.100.255 scope global dynamic noprefixroute eth0
       valid_lft 86376sec preferred_lft 75576sec
    inet6 fe80::3d02:e299:86de:81d2/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether b8:27:eb:74:f2:6c brd ff:ff:ff:ff:ff:ff
4: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 24:4b:fe:b1:57:65 brd ff:ff:ff:ff:ff:ff
    inet 10.0.100.251/24 brd 10.0.100.255 scope global dynamic noprefixroute eth1
       valid_lft 86375sec preferred_lft 75575sec
    inet6 fe80::6005:c30b:62d4:1a2d/64 scope link 
       valid_lft forever preferred_lft forever
```

And:

```
$ ethtool eth1
Settings for eth1:
	Supported ports: [ TP ]
	Supported link modes:   100baseT/Full 
	                        1000baseT/Full 
	                        10000baseT/Full 
	                        2500baseT/Full 
	                        5000baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Full 
	                        1000baseT/Full 
	                        10000baseT/Full 
	                        2500baseT/Full 
	                        5000baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: 10000Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	Auto-negotiation: on
	MDI-X: Unknown
Cannot get wake-on-lan settings: Operation not permitted
	Current message level: 0x00000005 (5)
			       drv link
	Link detected: yes
```

A picture of the setup:

[image]

geerlingguy, 2020-12-29:
Good news so far—it seems this driver/hardware does better with whatever optimizations it has than the Rosewill did:

```
$ iperf3 -c 10.0.100.143
Connecting to host 10.0.100.143, port 5201
[  5] local 10.0.100.251 port 44516 connected to 10.0.100.143 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   354 MBytes  2.96 Gbits/sec    0    892 KBytes       
[  5]   1.00-2.00   sec   356 MBytes  2.99 Gbits/sec    1    656 KBytes       
[  5]   2.00-3.00   sec   356 MBytes  2.99 Gbits/sec    0    781 KBytes       
[  5]   3.00-4.00   sec   359 MBytes  3.01 Gbits/sec    0    800 KBytes       
[  5]   4.00-5.00   sec   356 MBytes  2.99 Gbits/sec    0    833 KBytes       
[  5]   5.00-6.00   sec   358 MBytes  3.00 Gbits/sec    0    868 KBytes       
[  5]   6.00-7.00   sec   359 MBytes  3.01 Gbits/sec    0    880 KBytes       
[  5]   7.00-8.00   sec   356 MBytes  2.99 Gbits/sec    0    880 KBytes       
[  5]   8.00-9.00   sec   358 MBytes  3.00 Gbits/sec    0    902 KBytes       
[  5]   9.00-10.00  sec   359 MBytes  3.01 Gbits/sec    0    902 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  3.49 GBytes  2.99 Gbits/sec    1             sender
[  5]   0.00-10.01  sec  3.48 GBytes  2.99 Gbits/sec                  receiver
```

That's with MTU 1500, no other modifications. `atop` shows no IRQ limitations either—2% IRQ load, 30% CPU load in total. Trying MTU 9000 next.

geerlingguy, 2020-12-29:
```
pi@raspberrypi:~ $ sudo ip link set dev eth1 mtu 9000
pi@raspberrypi:~ $ iperf3 -c 10.0.100.143
Connecting to host 10.0.100.143, port 5201
[  5] local 10.0.100.251 port 44536 connected to 10.0.100.143 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   387 MBytes  3.25 Gbits/sec    0    883 KBytes       
[  5]   1.00-2.00   sec   390 MBytes  3.27 Gbits/sec    0    883 KBytes       
[  5]   2.00-3.00   sec   390 MBytes  3.27 Gbits/sec    0    883 KBytes       
[  5]   3.00-4.00   sec   389 MBytes  3.26 Gbits/sec    0    926 KBytes       
[  5]   4.00-5.00   sec   389 MBytes  3.26 Gbits/sec    0    926 KBytes       
[  5]   5.00-6.00   sec   389 MBytes  3.26 Gbits/sec    0    926 KBytes       
[  5]   6.00-7.00   sec   390 MBytes  3.27 Gbits/sec    0    926 KBytes       
[  5]   7.00-8.00   sec   389 MBytes  3.26 Gbits/sec    0   1.33 MBytes       
[  5]   8.00-9.00   sec   389 MBytes  3.26 Gbits/sec    0   1.33 MBytes       
[  5]   9.00-10.00  sec   389 MBytes  3.26 Gbits/sec    0   1.33 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  3.80 GBytes  3.26 Gbits/sec    0             sender
[  5]   0.00-10.01  sec  3.80 GBytes  3.26 Gbits/sec                  receiver
```

And I think we've hit our limit. Also going to try overclocking, but I'm less confident that will make a difference since this card/chipset is doing a lot more offload than the other ones I've tested...

geerlingguy, 2020-12-29:
Testing bidirectional at MTU 9000:

```
# Download iperf3 source.
wget https://github.com/esnet/iperf/archive/master.zip

# Unzip the download and enter source directory.
unzip master.zip && cd iperf-master/

# Run configure and make.
./configure
make

# Use the compiled version.
cd src/
```

And the run, resulting in **4.18 Gbps of traffic** both ways (MUCH better than the 2.5 Gbps card!):

```
$ ./iperf3 -c 10.0.100.143 --bidir
Connecting to host 10.0.100.143, port 5201
[  5] local 10.0.100.251 port 44544 connected to 10.0.100.143 port 5201
[  7] local 10.0.100.251 port 44546 connected to 10.0.100.143 port 5201
[ ID][Role] Interval           Transfer     Bitrate         Retr  Cwnd
[  5][TX-C]   0.00-1.00   sec   371 MBytes  3.11 Gbits/sec    0   1.17 MBytes       
[  7][RX-C]   0.00-1.00   sec   152 MBytes  1.28 Gbits/sec                  
[  5][TX-C]   1.00-2.00   sec   369 MBytes  3.09 Gbits/sec    0   1.29 MBytes       
[  7][RX-C]   1.00-2.00   sec   160 MBytes  1.34 Gbits/sec                  
[  5][TX-C]   2.00-3.00   sec   369 MBytes  3.10 Gbits/sec    0   1.29 MBytes       
[  7][RX-C]   2.00-3.00   sec   147 MBytes  1.24 Gbits/sec                  
[  5][TX-C]   3.00-4.00   sec   375 MBytes  3.15 Gbits/sec    0   1.29 MBytes       
[  7][RX-C]   3.00-4.00   sec   108 MBytes   902 Mbits/sec                  
[  5][TX-C]   4.00-5.00   sec   369 MBytes  3.09 Gbits/sec    0   1.29 MBytes       
[  7][RX-C]   4.00-5.00   sec   149 MBytes  1.25 Gbits/sec                  
[  5][TX-C]   5.00-6.00   sec   376 MBytes  3.16 Gbits/sec    0   1.35 MBytes       
[  7][RX-C]   5.00-6.00   sec   102 MBytes   858 Mbits/sec                  
[  5][TX-C]   6.00-7.00   sec   376 MBytes  3.15 Gbits/sec    0   1.35 MBytes       
[  7][RX-C]   6.00-7.00   sec   109 MBytes   910 Mbits/sec                  
[  5][TX-C]   7.00-8.00   sec   375 MBytes  3.15 Gbits/sec    0   1.35 MBytes       
[  7][RX-C]   7.00-8.00   sec   113 MBytes   947 Mbits/sec                  
[  5][TX-C]   8.00-9.00   sec   376 MBytes  3.16 Gbits/sec    0   1.35 MBytes       
[  7][RX-C]   8.00-9.00   sec   109 MBytes   913 Mbits/sec                  
[  5][TX-C]   9.00-10.00  sec   376 MBytes  3.16 Gbits/sec    0   1.35 MBytes       
[  7][RX-C]   9.00-10.00  sec   106 MBytes   891 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec  3.64 GBytes  3.13 Gbits/sec    0             sender
[  5][TX-C]   0.00-10.01  sec  3.64 GBytes  3.13 Gbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec  1.23 GBytes  1.05 Gbits/sec                  sender
[  7][RX-C]   0.00-10.01  sec  1.23 GBytes  1.05 Gbits/sec                  receiver
```

geerlingguy, 2020-12-29:
I was going to use `iperf3` to test bandwidth to both connections at the same time, and even worked on modifying the Ansible playbook to support that, but it seems there's a bug with the `--bind` option on both server and client—at least on the Pi—where it always binds (in either mode) to the first interface (`eth0`) if multiple are connected.

I worked around that on the i340-T4 by connecting each interface to a unique Pi and setting up custom networks between each of them. It looks like if both interfaces are on the same network, `iperf3` can't be used for simultaneous tests over different interfaces.

See https://github.com/esnet/iperf/issues/865 for details.

geerlingguy, 2020-12-29:
Unplugging and replugging cables doesn't seem to show any strange things in the drivers themselves:

```
$ dmesg --follow
...
[ 2432.978695] atlantic 0000:01:00.0 eth1: atlantic: link change old 10000 new 0
[ 2443.324914] atlantic 0000:01:00.0 eth1: atlantic: link change old 0 new 10000

[ 2465.179003] bcmgenet fd580000.ethernet eth0: Link is Down
[ 2474.395068] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
```

geerlingguy, 2020-12-29:
Exact same command: _FIRST_, with both interfaces plugged in:

```
pi@raspberrypi:~/iperf-master/src $ ./iperf3 --bind 10.0.100.251 -c 10.0.100.143
Connecting to host 10.0.100.143, port 5201
[  5] local 10.0.100.251 port 35045 connected to 10.0.100.143 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   114 MBytes   958 Mbits/sec    0    370 KBytes       
[  5]   1.00-2.00   sec   112 MBytes   944 Mbits/sec    0    370 KBytes       
[  5]   2.00-3.00   sec   111 MBytes   933 Mbits/sec    0    386 KBytes       
[  5]   3.00-4.00   sec   112 MBytes   944 Mbits/sec    0    386 KBytes       
[  5]   4.00-5.00   sec   112 MBytes   944 Mbits/sec    0    386 KBytes       
[  5]   5.00-6.00   sec   112 MBytes   943 Mbits/sec    0    386 KBytes       
[  5]   6.00-7.00   sec   111 MBytes   934 Mbits/sec    0    386 KBytes       
[  5]   7.00-8.00   sec   112 MBytes   944 Mbits/sec    0    386 KBytes       
[  5]   8.00-9.00   sec   112 MBytes   944 Mbits/sec    0    386 KBytes       
[  5]   9.00-10.00  sec   111 MBytes   933 Mbits/sec    0    386 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.10 GBytes   942 Mbits/sec    0             sender
[  5]   0.00-10.01  sec  1.09 GBytes   939 Mbits/sec                  receiver

iperf Done.
```

_SECOND_, with the onboard LAN unplugged:

```
pi@raspberrypi:~/iperf-master/src $ ./iperf3 --bind 10.0.100.251 -c 10.0.100.143
Connecting to host 10.0.100.143, port 5201
[  5] local 10.0.100.251 port 44649 connected to 10.0.100.143 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   389 MBytes  3.26 Gbits/sec    0    856 KBytes       
[  5]   1.00-2.00   sec   390 MBytes  3.27 Gbits/sec    0    856 KBytes       
[  5]   2.00-3.00   sec   390 MBytes  3.27 Gbits/sec    0    856 KBytes       
[  5]   3.00-4.00   sec   389 MBytes  3.26 Gbits/sec    0    856 KBytes       
[  5]   4.00-5.00   sec   390 MBytes  3.27 Gbits/sec    0    856 KBytes       
[  5]   5.00-6.00   sec   390 MBytes  3.27 Gbits/sec    0    900 KBytes       
[  5]   6.00-7.00   sec   390 MBytes  3.27 Gbits/sec    0    900 KBytes       
[  5]   7.00-8.00   sec   389 MBytes  3.26 Gbits/sec    0   1.09 MBytes       
[  5]   8.00-9.00   sec   390 MBytes  3.27 Gbits/sec    0   1.09 MBytes       
[  5]   9.00-10.00  sec   389 MBytes  3.26 Gbits/sec    0   1.09 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  3.80 GBytes  3.27 Gbits/sec    0             sender
[  5]   0.00-10.01  sec  3.80 GBytes  3.26 Gbits/sec                  receiver

iperf Done.
```

6by9, 2020-12-29:
Linux does weird things in choosing which physical interface to use if they're both connected to the same network.

We had some head scratching when trying to debug a wifi issue with QoS. Despite SSH'ing to the IP address associated with the Wifi interface, Linux was choosing to route it over the wired one because it had more bandwidth available or was listed first, or something. 

https://serverfault.com/questions/336021/two-network-interfaces-and-two-ip-addresses-on-the-same-subnet-in-linux

geerlingguy, 2020-12-29:
Hmm... so I guess I'll _have_ to test with two separate networks. Luckily I have a pile of 1 Gbps USB 3.0 dongles I can pop onto my Mac :)

NoisyBloke, 2020-12-30:
Just a couple of comments about the ASUS XG-C100C - I have a couple:
Connected end to end, I got them to work flawlessly over about 20m of cat6. But they got rather hot.
I was delighted to find that they support MTU of 16000 (See https://github.com/Aquantia/AQtion/blob/master/README.txt)
Adding a Mikrotik "CRS309-1G-8S+IN" fitted with two "S+RJ10"  I found:
- the S+J10 has a max MTU of 10K and silently fails with bigger 
- the S+J10 would not work with cat6 over that distance. I bought cat8 and that fixed it.

geerlingguy, 2020-12-30:
@NoisyBloke - Thanks for the updates! I have mostly Cat6 cabling around here, but did buy a few Cat8 patch cables (between 6 and 20') that I'll be using for testing. Later on I may go with fiber for my longer runs, but for now my network switch area is only about 15' from my main desk in my office :)

Good to know the MikroTik can get 'over 9000' but not quite 16000!

geerlingguy, 2021-01-01:
Just referencing the fact that I'm running into similar funny networking benchmark results over in #38 — so I'll have to be a lot more careful with `iperf`/`iperf3` and testing different interfaces.

NickMihailov, 2021-01-02:
You should really try the `pci=pcie_bus_perf` kernel boot parameter. As can be seen from your `lspci` output, the `MaxPayload` size has been set to 128 bytes when the device supports 512 bytes. Luckily the PCIe root complex in the RPi4 supports up to 512 bytes so you should really go for this. I don't have a CM4 module but on a regular RPi4, using that option sets the MaxPayload size for the USB3 controller to 512 bytes and it still works fine. I haven't tested it for actual speed, but it should provide up to around 15% better throughput if there are no other bottlenecks.

geerlingguy, 2021-04-29:
Almost finished getting my new 10G/2.5G hybrid home network set up. Will get back to testing this card a bit more.

sramshaw, 2021-06-02:
If minimalism and optimization is of interest, it seems that going for SFP+ and fiber makes sense, at least from what I read [here](https://www.datacenterknowledge.com/archives/2012/11/27/data-center-infrastructure-benefits-of-deploying-sfp-fiber-vs-10gbase-t). Latency and power consumption are improved, that is what convinced me as I made the choice earlier today for my home lab. I will use DAC cables though, not as good but close.

As far as testing Intel cards, I see reasonably priced SFP+ new ones : clones like 10Gtek 's X520-10G-1S , can be found on Amazon around $115 these days. There are some new one like EB-LINK's  EB-X520-DA1  for fiber at $100 only.

geerlingguy, 2021-06-29:
@NickMihailov - Trying your suggestion, I edited `/boot/cmdline.txt`, and set `pci=pcie_bus_perf` before the `rootwait` option. I rebooted, and after `lspci -vvvv` I now get:

```
$ sudo lspci -vvvv
...
		DevCap:	MaxPayload 512 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 512 bytes, MaxReadReq 512 bytes
```

geerlingguy, 2021-06-29:
Changed it back to not specifying the `pci` option, got a consistent `3.02 Gbps`. Changed it back again to `pcie_bus_perf` and got `3.42 Gbps` again. So it definitely makes a difference!

geerlingguy, 2021-06-29:
Changed the MTU to 9000:

```
pi@raspberrypi:~ $ sudo ifconfig eth1 mtu 9000 up
pi@raspberrypi:~ $ ifconfig | grep mtu
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
eth1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 9000
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
```

And now getting:

```
$ iperf3 -c 10.0.100.250
Connecting to host 10.0.100.250, port 5201
[  5] local 10.0.100.141 port 50840 connected to 10.0.100.250 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   427 MBytes  3.58 Gbits/sec                  
[  5]   1.00-2.00   sec   425 MBytes  3.56 Gbits/sec                  
[  5]   2.00-3.00   sec   428 MBytes  3.59 Gbits/sec                  
[  5]   3.00-4.00   sec   428 MBytes  3.59 Gbits/sec                  
[  5]   4.00-5.00   sec   426 MBytes  3.57 Gbits/sec                  
[  5]   5.00-6.00   sec   428 MBytes  3.59 Gbits/sec                  
[  5]   6.00-7.00   sec   428 MBytes  3.59 Gbits/sec                  
[  5]   7.00-8.00   sec   217 MBytes  1.82 Gbits/sec                  
[  5]   8.00-9.00   sec   206 MBytes  1.73 Gbits/sec                  
[  5]   9.00-10.00  sec   378 MBytes  3.17 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  3.70 GBytes  3.18 Gbits/sec                  sender
[  5]   0.00-10.00  sec  3.70 GBytes  3.18 Gbits/sec                  receiver
```

So it seems we reach a point where maybe packet loss starts hitting — IRQ interrupts also went beyond 50% for a moment, so maybe something is queueing up on the Pi's side at that point.

But interesting to see we can reach a short stable period around 3.6 Gbps!

geerlingguy, 2021-07-07:
All right, to sum things up, here's a video on the card: [10 Gigabit Ethernet on a Raspberry Pi?!](https://www.youtube.com/watch?v=FTP5h9jnVx0).

And here's a blog post with more about increasing the speed of the 10 Gbps connection: [Getting faster 10 Gbps Ethernet on the Raspberry Pi](https://www.jeffgeerling.com/blog/2021/getting-faster-10-gbps-ethernet-on-raspberry-pi).

closed 2021-07-07 by geerlingguy

tu1982, 2022-01-25:
> Oops. Missed something. I _also_ enabled:
> 
> ```
> Device Drivers
>   > Network device support
>     > Ethernet driver support
>       > aQuantia devices
>         > aQuantia AQtion(tm) Support
> ```
> 
> I've recompiled and am copying things over to the Pi now.

Hi, need a little help :) Trying to run this card on CM4 and still no luck. I believe I'm using current kernel using `git clone --depth=1 https://github.com/raspberrypi/linux` and compiling "on-board" (not cross-compiling). The problem seems to be in menuconfig. There is option in "General setup" called "Compile also drivers which will not load". This has to be "checked" unless the option "aquantia AQtion(tm) Support" in "Device Drives > Network device support > Ethernet driver support > aquantia devices" is not listed and therefore cannot be checked. I found out the `make bcm2711_defconfig` completely breaks things down - "Compile also drivers which will not load" is not present in menuconfig, running `make bcm2711_defconfig` and  adding CONFIG_AQTION=y into .config manually does not work...

Any suggestions please?

geerlingguy, 2022-01-26:
@tu1982 - If you're using the raspberrypi/linux repository at the latest commit, it should have aquantia's driver available, are you sure you're on the latest commit on their rpi branch?

I haven't ever gone to 'General setup' nor looked at any options for 'Compile also drivers which will not load'—my testing is usually done with a completely clean checkout of the rpi linux kernel source, and then I run `make bcm2711_defconfig` and then `menuconfig` right after.

tu1982, 2022-02-02:
Hi Jeff,
I really appreciate your help... I have to get it up and running...it's personal now :)

> @tu1982 - If you're using the raspberrypi/linux repository at the latest commit, it should have aquantia's driver available, are you sure you're on the latest commit on their rpi branch?

Yes, I believe so. `uname -a` gives something like 5.10.xy-v7l+. In my last try I choose a different branch (5.7.19-v7l+). I hoped it probably might be closer to you kernel version, when you started this project.

I'm quite new to RPi but I tried to check modules and found out, there is no aquantia folder/file in /lib/modules/'kernel_version'/kernel/drivers/net/ethernet and /phy. Should be there? I found them only in /lib/modules/'kernel_version'/build/...

`sudo lspci -vvvv` returns:
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20) (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 61
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 00000000-00000fff [size=4K]
	Memory behind bridge: f8000000-f85fffff [size=6M]
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity+ SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (ok), Width x1 (ok)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt+
		RootCap: CRSVisible+
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Via WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd-
			 AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled, ARIFwd-
			 AtomicOpsCtl: ReqEn- EgressBlck-
		LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS+
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported, DRS-
			 DownstreamComp: Link Down - Not Determined
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
		RootCmd: CERptEn+ NFERptEn+ FERptEn+
		RootSta: CERcvd- MultCERcvd- UERcvd- MultUERcvd-
			 FirstFatal- NonFatalMsg- FatalMsg- IntMsg 0
		ErrorSrc: ERR_COR: 0000 ERR_FATAL/NONFATAL: 0000
	Capabilities: [180 v1] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
[... 182 lines cut ...]
Jan 27 09:35:07 raspberrypi kernel:  r9:d3f65e48 r8:ef25400c r7:00000000 r6:00000000 r5:00000000 r4:d3f65e38
Jan 27 09:35:07 raspberrypi kernel: [<c0ab0a5c>] (devinet_ioctl) from [<c0ab4d10>] (inet_ioctl+0x204/0x32c)
Jan 27 09:35:07 raspberrypi kernel:  r10:00000004 r9:00000000 r8:d442c080 r7:c12b60c0 r6:c12b60c0 r5:00008914
Jan 27 09:35:07 raspberrypi kernel:  r4:bef0e3e4
Jan 27 09:35:07 raspberrypi kernel: [<c0ab4b0c>] (inet_ioctl) from [<c09cc3d0>] (sock_ioctl+0x490/0x5b8)
Jan 27 09:35:07 raspberrypi kernel:  r7:c12b60c0 r6:bef0e3e4 r5:d3e200c0 r4:00008914
Jan 27 09:35:07 raspberrypi kernel: [<c09cbf40>] (sock_ioctl) from [<c043d828>] (ksys_ioctl+0x114/0xa08)
Jan 27 09:35:07 raspberrypi kernel:  r8:d442c080 r7:d3e200c0 r6:bef0e3e4 r5:d3e200c0 r4:00008914
Jan 27 09:35:07 raspberrypi kernel: [<c043d714>] (ksys_ioctl) from [<c043e134>] (sys_ioctl+0x18/0x1c)
[... cut ...]
Jan 27 09:35:07 raspberrypi kernel: 5fc0: 00000041 bef0e45c 0002e574 00000036 00000000 0001b8fc 0001b8f8 00000000
Jan 27 09:35:07 raspberrypi kernel: 5fe0: 0002e084 bef0e3cc 00013208 b6e8c1ac
Jan 27 09:35:07 raspberrypi kernel: Code: e3a00003 ebfa2c16 e1a00004 e89da8f0 (e7f001f2) 
Jan 27 09:35:07 raspberrypi kernel: ---[ end trace 9d93d6d5c04b03bd ]---
Jan 27 09:35:12 raspberrypi kernel: atlantic 0000:01:00.0 eth0: atlantic: link change old 0 new 1000

Can you please see something suspicious?

Thank you very much!

Tomas

geerlingguy, 2022-02-02:
@tu1982 - Ah... it looks like you're running 32-bit Pi OS. I have only tested in 64-bit Pi OS, so maybe that's the issue? I don't know how well the driver will work on a 32-bit system.

tu1982, 2022-02-04:
> @tu1982 - Ah... it looks like you're running 32-bit Pi OS. I have only tested in 64-bit Pi OS, so maybe that's the issue? I don't know how well the driver will work on a 32-bit system.

Yes, you are right. Shame on me... It seems it does not work on 32-bit system.

Thanks a lot.

grantterra, 2024-01-26:
> Well that's annoying:
>
> ```
> /home/pi/Marvell_Linux_2.4.3.0_207/Linux/aq_sysfs.c: In function ‘memreg_mmap’:
> /home/pi/Marvell_Linux_2.4.3.0_207/Linux/aq_sysfs.c:227:7: error: ‘struct vm_area_struct’ [...]

May I ask you how you fixed that?