geerlingguy/raspberry-pi-pcie-devices #19: Idea: crowdfund a microATX motherboard that a Compute Module 4 plugs into

volkertb, 2020-11-08:
Hi Jeff,

I'm probably not the first person to have thought about this, but as far as I understand, it should be perfectly possible to develop a motherboard with a microATX form factor that offers all the typical I/O that an x86 motherboard usually provides, which you can then plug a Raspberry Pi Compute Module 4 into.

It could have an integrated PCIe multiplier, internally wired to a couple of PCIe 1x slots, as well as an M.2 slot for NVMe SSDs and an integrated USB 3.0 hub. Additionally, all the I/O that comes standard on the Compute Module (2x HDMI, ethernet, USB 2.0, audio in/out, etc) could be exposed through ports on the back of the board, with a standard ATX backplate for it included of course.

And lastly, it would of course have a standard 24-pin ATX power connector on the board, so you could use it with any regular ATX PSU.

It would enable people to finally build an affordable ARM64-based upgradable desktop PC, with the possibility to upgrade not only expansion cards, but also the Compute Module itself, once future versions of it are released.

I can't think of anything that would make something like this unfeasible, but perhaps there are some technical aspects that I haven't taken into account here?

Anyway, assuming this would indeed be a feasible product to develop, it would require people with high level electronics and board design skills to design, prototype, test and produce such a board.

Since you are a prominent vlogger about all things Raspberry Pi related, and you've been exploring with the PCIe capabilities of the Compute Module 4 in particular, perhaps you could dedicate a video to this idea and invite people with the required knowledge to start on a project like that, or at least carry out a little feasibility study on it.

I unfortunately lack the skills to work on the hardware side of such a board, but I'd definitely fund such a project! :smiley:

Thanks for considering this whacky idea.

volkertb, 2020-11-08:
Alternatively, perhaps the Raspberry Pi Foundation could be persuaded to develop such a product. I expect there would definitely be a sizable demand for it. It would be an affordable way for people to explore the ARM64 architecture as a basis for desktop computing.

UnKnoWn-Consortium, 2020-11-08:
Just curious why MicroATX (244x244) specifically? 

Typical MicroATX has 4 expansion slots. I doubt if CM4 has sufficient I/O bus to fill them all and, even we can (with a PCI-E switch), feed them enough bandwidth. Also the rear side is a bit too large for just one CM4. 

I wonder if Mini-ITX (170x170) or mini-DTX (203x170) would be more viable form factors.

volkertb, 2020-11-08:
With a multiplier, multiple expansion slots should be perfectly possible, although whatever cards you'd end up installing would be sharing the bandwidth of the single PCIe lane. But there are lots of PCIe cards that don't need the capacity of the entire lane. Just put a couple of PCIe slots on such a board and the user decide on the trade-off between having multiple cards sharing the bandwidth or a single card having access to all the bandwidth. The point of this suggested product is to provide flexibility and some level of upgradability.

But I would also be happy to see a board with any standard form factor like those you mentioned instead. As long as it fits in a regular case, can be hooked up to a regular PSU, and allows the installation PCIe expansion cards (preferably more than one), in addition to exposing all the I/O ports other than the PCIe interface as well.

geerlingguy, 2020-11-09:
I would indeed like a board like this, maybe even Mini ITX, though having either 2 or 4 PCIe slots would be ideal (I'm thinking NVMe + 'something else').

mo-g, 2020-11-13:
I put together a loose concept for this the other day: https://inthe.study/fixing-the-raspberry-pi-desktop/

Summary is similar to OP's idea here - except I explain why no USB 3.0 controller in the post.

One interesting thing - I found a company selling HDMI-CSI adapters for the Pi 3, and the IC they use also supports the extra bandwidth of the 4 lane interface on the compute module - so rather than leaving CSI wasted inside an ATX case, my concept was to expose it as a pair of HDMI in ports that could potentially be used for a DSLR webcam and the output from another computer.

My concept would have three physical (and power pins) x16, x1 lane slots running off a switch chip, with the first and second separated by an m.2 slot, as many motherboards do for not wasting a slot with a GPU's 2-slot cooler. I also worked out that front panel IO is doable with the CM4.

- Problem is, it's likely to have a cost of at least £200 in bulk - which rather limits the market. Though; that's still (after import costs to the UK) half the price of the Macchiato's latest price jump and a quarter or less of the Honeycomb.

mo-g, 2020-11-13:
Further to my last: I'm willing to give up a lane of a four-lane PCIe switch and let y'all have USB3, if you let me make this a standard ATX board with three (or 2+m.2) PCIe slots, and four ISA slots using some kind of Microcontroller+SRAM combo as a bus translator. (A la http://tinyvga.com/avr-isa-vga)

TheGuyDanish, 2020-11-13:
While I'm not an experienced board designer by any means (I have only designed small boards that were merely passive adapters, more than anything) I am quite happy to look into the possibility of designing something like this. I'd be delighted to work with someone more experienced, if anyone would be interested in doing such a thing.

volkertb, 2020-11-13:
Yes! Let's get something going. :smiley:

@mo-g It's funny you should mention ISA, because a few days ago I was considering suggesting the addition of an ISA slot to such a board, for some fun retro tinkering. I ended up not suggesting it here, since I didn't know of any affordable bridge/adapter solutions. The only existing solution I know of are the adapters from ARS technologies, and those are quite pricey: http://www.arstech.com/install/ecom-prodshow/usb2isar.html

But yeah, the microcontroller solution you proposed indeed seems like a much more affordable (and open) solution that would indeed be awesome to have integrated on a board like this.

So how do we proceed with this? Attract some more savvy people who have the necessary experience with PCB and computer electronics design? @geerlingguy Perhaps you can help out in this department? Perhaps mention this idea in one of your videos?

UnKnoWn-Consortium, 2020-11-13:
@mo-g As a side note, the most commonly seen HDMI-to-CSI converter for Pi in the market is usually powered by TC358743 from Toshiba. 

https://toshiba.semicon-storage.com/ap-en/semiconductor/product/interface-bridge-ics-for-mobile-peripheral-devices/hdmir-interface-bridge-ics/detail.TC358743XBG.html

This chipset also has two siblings that support 4K input. 

https://toshiba.semicon-storage.com/ap-en/semiconductor/product/interface-bridge-ics-for-mobile-peripheral-devices/hdmir-interface-bridge-ics.html

mo-g, 2020-11-13:
> This chipset also has two siblings that support 4K input.

I suggest the TC358743 in my article, because it supports 1080P60 with 4 lanes and 1080p30 with 2 lanes. The 4K chips need two channels of 4 lanes, and the Pi only offers 1x4 + 1x2, so you're not going to get past 1080p60 either way.

mo-g, 2020-11-14:
For the record, I'm really joking with the ISA thing on the ATX board - but I *would* like to see an ISA equipped board - ScummVM for Raspberry P-ISA.

But for the main product? Obscure stuff like that is why projects fail. Keep it to the stuff everyone can use, like PCIe, M.2, HDMI in.

TheGuyDanish, 2020-11-14:
Some thoughts I've had about this include having an onboard PCI-e Switch (Probably a PI7C9X2G404SL chip, like all the other cheap-ish boards are using. These can be purchased loose from Mouser and similar vendors for $8-9). Then potentially equip the first PCI-e slot with a PI3PCIE2215 mux. Again, a cheap IC that would serve as a simple switch. With that, the first slot could easily be toggled between an M.2 slot and the physical PCI-e port.

As for the ports themselves, an mATX would have space for 4 ports, but the PI7C... (You get it, the switch chip) can only accommodate three downstream ports. However, seeing as people are going to want to be playing with all manner of sizes of cards. perhaps it would make sense to put one each x16, x8 and x1 slot on the board? Obviously each wired up to the PI7C as x1 electrically.

That way you wouldn't have to fork out money for dodgy adapters... Jeff. ;)

mo-g, 2020-11-14:
> perhaps it would make sense to put one each x16, x8 and x1

It doesn't really. They should all be x16 for maximum compatibility. A 1x card will fit as happily in a 16x slot. The only reason to have shorter slots is for board space or to differentiate them (so you don't put a 16x card in the 4x slot on a Ryzen board by mistake). In our case, the board won't have much else on it, and ALL the slots are 1x.

Also, or - PI7C9X2G606PRDNJAEX - $13.21 at scale on Mouser, and gives 6 ports - four 1x slots AND an M.2 - Or, three with an interposed m.2 as per most gaming mATX boards, plus USB3 as per the Pi 4. Would prevent a lot of arguments. Downside is - there are no PCIe USB host controllers I can see on Mouser, much less a VL806 or VL805. We could also break out the fifth downstream port as a U.2 connector for an SSD or a second (short) m.2. Or just weld a TPU or SATA or something onto it permanently. 

Or, go completely ham, and combine the last lane with a SN65DSI86 and a JHL7540 and own the "Needs USB 3" crowd by putting a thunderbolt port on instead.

(That last bit was a joke again, for the record.)

But the difference in price between a 4 port and six port switch is almost offset by the cost of the MUX IC alone - so little point in not just using the six port switch and avoiding that added complexity, even if you just left the last port unused entirely.

TheGuyDanish, 2020-11-14:
606PR could be a fun one to put on the board, just to take it to its full potential, but then I'd _definitely_ need someone with a little more schematic/design knowledge to help out. 😂 

Even the 404SL is providing me a bit of a challenge to properly draw in on the schematic I'm starting out on. Thankfully most of the features can be more or less lifted from the CM4IO board. (Which I might have done, except replacing the single HDMI port with a [stacked one](https://www.te.com/global-en/product-1888811-1.html) and replacing some power delivery with the ATX 24-pin connector.)

mo-g, 2020-11-14:
I hope you're putting my TC358743's on there with another stacked HDMI port! You'll also need to route i2s to one of them (and according to the CM4 datasheet, do something with i2c)

;)

Oh, and from what I was reading - there's something weird about GLOBAL_EN and RUN_PG which means you'll need some kind of latch (a single cell flip flop?) to make them work properly as FP I/O and also let you latch the ATX enable pin on when the Pi is running.

TheGuyDanish, 2020-11-14:
> TC358743

While playing with this chip and getting it on board would be interesting, I'm unable to find it anywhere for purchase by a mere mortal, and I (sadly) don't have the financial backing to ask Toshiba for an amount that they would even want to bother supplying.

The entire GLOBAL_EN and RUN_PG debacle is something I need to read a bit further into. From just looking at the CM4 datasheet, I doubt that _not touching_ these would have any adverse effects. Though, as the CM4 datasheet explains:

> 2.13. Global_EN
Pulling this pin low puts the CM4 in the lowest possible power down state. After software shutdown Global_EN needs to
be pulled low for > 1ms to restart the power system on the CM4.

I'm not sure what the benefit to pulling GLOBAL_EN low would be, as opposed to removing power from PS_ON on the 24-pin connector, really. But either could make for a good power button implementation.

>2.14. RUN_PG
This pin when high signals that the CM4 has started. Driving this pin low resets the module, this should be done with
caution as if files on a filesystem are open they will not be closed

Not much to comment on here, really. CM4 drives it high when powered on, I guess it could be a good candidate to handle the reset switch on a front panel.

mo-g, 2020-11-14:
> I'm unable to find it anywhere for purchase by a mere mortal

They're $4 each in a pack of 10 on Alibaba: https://www.aliexpress.com/i/32896175425.html or €5 each singly from whoever this is: https://auvidea.eu/product/26139/

Global_EN pulled low will power-on a shutdown Pi (i.e. - it's a normal ATX power button) - but as I recall when I was reading up on this while spec'ing this design, weird shit happens if you leave it floating, so we should latch it high - possibly by using RUN_PG (buffered) as an output to report "powered on" state to the latch and to the ATX PSU enable pin.

TheGuyDanish, 2020-11-14:
Good find on the TC358743XBG! [Guess I really have no excuses now...](https://www.snapeda.com/parts/TC358743XBG/Toshiba/view-part/) I'll probably have a look into adding these to the design tomorrow and putting the entire thing up on Github for people to follow and submit pull requests to.

>Global_EN pulled low will power-on a shutdown Pi (i.e. - it's a normal ATX power button)

While this is true from the Pi's perspective, it would technically still leave all the PSU's rails still active. Whereas if the signal is removed from PS_ON, we get a "true off" state. (Obviously barring +5VSB from the PSU)

mo-g, 2020-11-14:
Woohoo! Send me a link so I can Star and Fork it when it's up!

> While this is true from the Pi's perspective, it would technically still leave all the PSU's rails still active. Whereas if the signal is removed from PS_ON, we get a "true off" state. (Obviously barring +5VSB from the PSU)

This is confusing. I'm talking about Global_EN turning a computer on, you're talking about turning a computer *off*. We seem to be talking about different things...

TheGuyDanish, 2020-11-14:
Chalk it down to me trying to speak technical at 2 in the morning. 😂 

I see what you mean though. I was considering it as a "If I pull this pin low, the system will shutdown", rather than "If I keep this low, the system will remain off", kind of thing. I get your drift.

mo-g, 2020-11-14:
Think you still have this backwards. (Though, it's 2AM up the A1 as well, so I may also be wrong) My understanding was - If you pull it low, the system will turn *on* from a powered-off state, but I think if you keep holding it low after that, bad things happen - so when the system becomes on, you have to disable the switch for the users' own good - so force the signal high. That way the power switch does nothing after boot (like an ATX button pre-the later invention of the 20s force-off timer) and after that time only the reset switch or the wall socket can be used to rescue a stuck system. I don't think Global_EN has any use for us past the initial power-on trigger.

TheGuyDanish, 2020-11-14:
Hmm. I think the best way to test this would be to look into how the CM4IO board handles Global_EN, and when I get my own IO board in (Pimoroni, please.) I can do my own measurements on it to get a more exact understanding. Another one for the TODO tomorrow.

mo-g, 2020-11-14:
Good luck and good night!

TheGuyDanish, 2020-11-14:
It lives!

https://github.com/TheGuyDanish/CM4_MATX

mo-g, 2020-11-14:
And every good Open Source project needs Issues, so I've raised one. ;)

TheGuyDanish, 2020-11-14:
> And every good Open Source project needs Issues, so I've raised one. ;)

I wasn't expecting any less. ;)

volkertb, 2020-11-14:
Impressive start, man! Hereby starred and watched. :slightly_smiling_face:

I lack the hardware engineering expertise, but I'll definitely be following your progress. Let the ideas and the collaboration flow freely!

mo-g, 2020-11-14:
I can't PCB for jack. But, you can still contribute things like use cases, what ports and connections you'd want exposed for your use cases, etc, etc.

geerlingguy, 2020-11-19:
Since there's a lot of activity over on the [CM4_MATX](https://github.com/TheGuyDanish/CM4_MATX) project, I'm going to go ahead and close out this issue and redirect people over there, and also mention I'm following a few other interesting CM4 projects in this issue: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/25

closed 2020-11-19 by geerlingguy

paulwratt, 2020-11-21:
there are already Open Source motherboards (IBM's PPC Power9 desktop motherboard), and a kazillian x86 motherboards

I would rather see CM4 adapter plates for standard x86_64 CPU sockets, and one for the Power9/Power10 sockets.

There would be (easily) 50% less electronics with a socket adapter and no need for physical busses (outside debug & Pi headers) - plus there would be the possibility to use PC cooling systems (ie Gamers Nexus testing max GHz with nitrogen cooling :^)

geerlingguy, 2020-11-21:
@paulwratt - Haha, I can just imagine someone with a 1 HP pump and some massive water cooling rig to get the CM4 from 35°C to like 1°C

mo-g, 2020-11-21:
> there are already Open Source motherboards (IBM's PPC Power9 desktop motherboard), and a kazillian x86 motherboards:
> https://hackaday.com/2018/11/26/a-sub-1000-non-x86-motherboard/

AFAIK, there are two ARM ATX motherboards readily available. One has the same CPU as the CM4 - and it's $450. Has SATA and 23.5GBe ethernet, PCIe x4, but no display output. The other is $750, has more cores and 41GBe, but also no display output. I'm hoping the Pi CM4 ATX board idea could sell enough to target a $200 price point, and provide a functional desktop experience with more expandability at a lower TCO. Yeah, you could get an ITX atom board and some ram for about $150, but there's a lot to be said for having a Pi at the center.

> I would rather see CM4 adapter plates for standard x86_64 CPU sockets, and one for the Power9/Power10 sockets.

This would cost more than an ATX board and offer way less. A 1150 socket has 500 pins of memory interface that the CM4 can't access, and doesn't provide any way to interface with: CSI, DSI, most of the GPIO, ethernet, i2c, i2s... And you'd probably also need to power the adapter board externally or use some very weird boost circuitry to turn 1.8V back into 5V.

If you were lucky, the only connectors you could get to work would be: a single HDMI port, a single DVI port, a single lane in a single PCIe slot. No sata, not front panel header, no ethernet, sound connectors, probably not even USB...

The point of an x86 motherboard is it has to provide a bunch of stuff to the bare CPU that the CM4 is already doing. Memory is integrated to the CM4. Power management is integrated to the CM4. The USB controller is integrated to the CM4. Pretty much every interface that is handled directly on the Pi and available over the SOM connectors is done separately on an ATX board - and the Pi doesn't have the right hardware to interface with it.

> There would be (easily) 50% less electronics with a socket adapter

That's optimistic as well. There isn't actually a lot of electronics needed in building an ATX pi carrier - a lot of things are just straight "connect pin A to socket A". The things we're adding (usb hub, pcie hub, HDMI-CSI) are optional extras we're adding to give fairly standard interfaces on a PC. In a socket converter, there's no way to pass through most of these interfaces - so they just wouldn't work at all. The things available would be the subset of (things provided by a desktop CPU == things provided by a CM4) which is pretty much going to amount to HDMI and a single PCIe lane.

> plus there would be the possibility to use PC cooling systems

That wouldn't work at all. PC cooling systems depend on a bunch of physical parameters, including height - that you could not match with a CM4 adapter board. The CM4 alone would be too tall, before you added the carrier board in between. If a lot of projects take off that use the CM4 (our ATX board, the Turing Pi 2, a RiscPC retrofit, the 486-replacement we've discussed before, etc, etc) there's plenty of chance someone will start building a cooler that mounts to the same mounting holes as the CM4, and is specially milled to contact the different IC's properly. Maybe even a custom waterblock?

I'm sorry - this is a fun idea, but has almost no chance of being doable, and wouldn't give you much of what you're thinking of if you did. A CPU is not an SOC is not an SOM.

WITH THAT IN MIND.

There is nothing (physically at least, patents would probably be a sticker) to stop someone like Broadcom from designing a CPU that could drop in to certain PC sockets - physically identical in things like heat spreader and height. After all, since the memory controllers and PCIe controllers are integrated into modern CPU's - there's not actually all that much that is custom or proprietary - even many Southbridges are interfaced over PCIe these days. So a major company could replicate a CPU (not an SOC or SOM), with ARM cores  that was pin compatible with a desktop motherboard -but the question is - since that would have to be completely custom silicon; how many would they have to sell to get it to the point where they could sell it for $100? I genuinely have no idea. But, the idea of having a 32 core ARM CPU on a RoG STRIX board with 128GB ram and a Quadro... is not unappealing to me either...

volkertb, 2020-11-23:
Linus Torvalds wrote [an often-quoted post](https://www.realworldtech.com/forum/?threadid=183440&curpostid=183486) about how ARM will never truly become a mainstream platform as long as developers don't have access to cheap development hardware. He compared it with how x86 became the dominant platform on the desktop and in the server space, due to everyone developing on that platform at home and at work.

I've been looking for a a cheap and expandable ARM-based motherboard for quite a while now, and it frankly baffles me why there are **still to this day** hardly any good and affordable options available in that space. You'd think some company would have filled in this gap by now.

However, this solution might fit the bill almost perfectly. With the addition of PCIe slots, the one thing I'd still miss would be upgradable RAM. But on the other hand, since the Compute Module 4 can simply be swapped out for a newer version or a variant with more RAM anyway, that's probably not too much of an issue.

A truly affordable and expandable ARM64-based desktop can be built around a microATX solution like this!

jcdutton, 2021-07-17:
I think a ATX board with the CM4 might present some problems.
Please see https://pipci.jeffgeerling.com/
In particular, the difficulty we are having getting any PCIe graphics cards to work with it. As in, at the moment PCIexpress connected GPUs do NOT work with the CM4.
This is most likely due to hardware bugs in the PCIe parts of the BCM2711 chip (on the CM4 card).
The PCIe interface does work for some things though, so might still be useful for some people.
E.g. Make a desktop with a lot of SSD/NVME storage.
One could route the CM4 own graphics to the normal edge connectors of an ATX board.
The GPIO etc, could probably be routed to number of 25 way D-sub connectors.
I know I am considering this, for example, I have a few old PCs where the motherboard is broken, but I still have the case, PSU, fans, disks. So being able to replace it with a ATX board with CM4, or multiple CM4s on it would work well. Essentially something that would let me screw the CM4 board into the same holes that are on a standard case.

volkertb, 2021-07-18:
@jcdutton This topic has been spun off into a project of its own already.

You might want to continue the discussion there: https://github.com/TheGuyDanish/CM4_MATX

TheGuyDanish, 2021-07-18:
> In particular, the difficulty we are having getting any PCIe graphics cards to work with it. As in, at the moment PCIexpress connected GPUs do NOT work with the CM4.
> This is most likely due to [...]

This is more or less exactly what I am imagining with CM4_MATX. I/We know there are plenty of devices that are (yet) incomaptible with the CM4 and it's PCI-e interface. But the goal isn't to make a "fully functional" computer, per se. The goal, at least for the time being, is to make a functional development platform. A board that is easy to acquire and has plenty of options for people who _want_ to try out running a CM4 in something like an old desktop, or want to plug in multiple expansion cards without having to worry about the mess of powered risers and dodgy switch integrations.

This is still in an early state though. I'm no electrical engineer at all, so I'm still looking for those with a skill in things like verifying the design and routing the PCB to give their thoughts, as well as doing a few other CM4 projects in the back that I've not publicized on Github yet.