geerlingguy/raspberry-pi-pcie-devices #22: Test EDUP PCIe Intel AX200 WiFi 6 Card

geerlingguy, 2020-11-09:
I bought an [EDUP PCIe Intel AX200 WiFi 6 Card](https://amzn.to/3pnFF8S), which uses the Intel AX200 chip, which is _supposedly_ friendlier with Linux than the Realtek chip I tried integrating in the ASUS PCE-AC51 over in #20.

TODO: Picture here

Note: It should work with the `iwlwifi` driver—supposedly.

geerlingguy, 2020-11-13:
```
$ sudo lspci -vvv
...
01:00.0 Network controller: Intel Corporation Device 2723 (rev 1a)
	Subsystem: Intel Corporation Device 0084
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Region 0: Memory at 600000000 (64-bit, non-prefetchable) [disabled] [size=16K]
	Capabilities: [c8] Power Management version 3
		Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [d0] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [40] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <512ns, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+ FLReset-
			MaxPayload 128 bytes, MaxReadReq 128 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L1, Exit Latency L0s <4us, L1 <8us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range B, TimeoutDis+, LTR+, OBFF Via WAKE#
		DevCtl2: Completion Timeout: 16ms to 55ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [80] MSI-X: Enable- Count=16 Masked-
		Vector table: BAR=0 offset=00002000
		PBA: BAR=0 offset=00003000
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [14c v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [154 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=30us PortTPowerOnTime=18us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
```

geerlingguy, 2020-11-13:
Rebuilding the kernel, here's where the driver is according to menuconfig:

```
 Symbol: IWLWIFI [=n]                                                                                            │   
  │ Type  : tristate                                                                                                │   
  │ Prompt: Intel Wireless WiFi Next Gen AGN - Wireless-N/Advanced-N/Ultimate-N (iwlwifi)                           │   
  │   Location:                                                                                                     │   
  │     -> Device Drivers                                                                                           │   
  │       -> Network device support (NETDEVICES [=y])                                                               │   
  │         -> Wireless LAN (WLAN [=y])                                                                             │   
  │ (1)       -> Intel devices (WLAN_VENDOR_INTEL [=y])                                                             │   
  │   Defined at drivers/net/wireless/intel/iwlwifi/Kconfig:2                                                       │   
  │   Depends on: NETDEVICES [=y] && WLAN [=y] && WLAN_VENDOR_INTEL [=y] && PCI [=y] && HAS_IOMEM [=y] && CFG80211  │   
  │   Selects: FW_LOADER [=y] 
```

geerlingguy, 2020-11-13:
```
[    3.925098] Intel(R) Wireless WiFi driver for Linux
[    3.925112] Copyright(c) 2003- 2015 Intel Corporation
[    3.925532] pci 0000:00:00.0: enabling device (0000 -> 0002)
[    3.925563] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
[    3.929671] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-50.ucode failed with error -2
[    3.929760] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-49.ucode failed with error -2
[    3.929824] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-48.ucode failed with error -2
[    3.929884] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-47.ucode failed with error -2
[    3.929944] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-46.ucode failed with error -2
[    3.930006] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-45.ucode failed with error -2
[    3.930064] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-44.ucode failed with error -2
[    3.930123] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-43.ucode failed with error -2
[    3.930180] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-42.ucode failed with error -2
[    3.930354] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-41.ucode failed with error -2
[    3.930418] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-40.ucode failed with error -2
[    3.930477] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-39.ucode failed with error -2
[    3.930495] iwlwifi 0000:01:00.0: no suitable firmware found!
[    3.930507] iwlwifi 0000:01:00.0: minimum version required: iwlwifi-cc-a0-39
[    3.930518] iwlwifi 0000:01:00.0: maximum version supported: iwlwifi-cc-a0-50
[    3.930527] iwlwifi 0000:01:00.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
```

Looks like I need to install some firmware...

geerlingguy, 2020-11-13:
Looking at Intel's page https://wireless.wiki.kernel.org/en/users/Drivers/iwlwifi — it looks like it needs MVM (which I have installed, as well as the firmware linked in the table for `Intel® Wi-Fi 6 AX200 160MHz`.

```
$ wget https://wireless.wiki.kernel.org/_media/en/users/drivers/iwlwifi/iwlwifi-cc-46.3cfab8da.0.tgz
$ tar -xvzf iwlwifi-cc-46.3cfab8da.0.tgz
$ cd iwlwifi-cc-46.3cfab8da.0/
$ sudo cp iwlwifi-*.ucode /lib/firmware
$ sudo reboot
```

geerlingguy, 2020-11-13:
Yikes, tons of error output, going to have to blacklist the module and load it manually to see what's going on...

```
$ nano /etc/modprobe.d/blacklist-.conf
(contents of file below:)
blacklist iwlwifi

$ sudo reboot
```

geerlingguy, 2020-11-13:
```
$ sudo modprobe iwlwifi
```

Then concurrently, `dmesg` starts spewing:

```
[   49.784664] Intel(R) Wireless WiFi driver for Linux
[   49.784677] Copyright(c) 2003- 2015 Intel Corporation
[   49.785037] pci 0000:00:00.0: enabling device (0000 -> 0002)
[   49.785064] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
[   49.788534] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-50.ucode failed with error -2
[   49.788599] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-49.ucode failed with error -2
[   49.788654] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-48.ucode failed with error -2
[   49.788704] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-47.ucode failed with error -2
[   49.817473] iwlwifi 0000:01:00.0: loaded firmware version 46.3cfab8da.0 op_mode iwlmvm
[   49.923040] iwlwifi 0000:01:00.0: Detected Intel(R) Wi-Fi 6 AX200 160MHz, REV=0x340
[   50.082225] iwlwifi 0000:01:00.0: base HW address: e0:d4:e8:49:ff:e8
[   50.097951] thermal thermal_zone1: failed to read out thermal zone (-61)
[   50.522448] iwlwifi 0000:01:00.0: Microcode SW error detected.  Restarting 0x2000000.
[   50.522555] iwlwifi 0000:01:00.0: Start IWL Error Log Dump:
[   50.522619] iwlwifi 0000:01:00.0: Status: 0x00000040, count: 6
[   50.522619] iwlwifi 0000:01:00.0: Loaded firmware version: 46.3cfab8da.0
[   50.522629] iwlwifi 0000:01:00.0: 0x00000071 | NMI_INTERRUPT_UMAC_FATAL    
[   50.522633] iwlwifi 0000:01:00.0: 0x00A0BEF5 | trm_hw_status0
[   50.522638] iwlwifi 0000:01:00.0: 0x00000000 | trm_hw_status1
[   50.522639] iwlwifi 0000:01:00.0: 0x004FC308 | branchlink2
[   50.522639] iwlwifi 0000:01:00.0: 0x004E8926 | interruptlink1
[   50.522639] iwlwifi 0000:01:00.0: 0x004E8926 | interruptlink2
[   50.522639] iwlwifi 0000:01:00.0: 0x00016F7E | data1
[   50.522639] iwlwifi 0000:01:00.0: 0x00001000 | data2
[   50.522660] iwlwifi 0000:01:00.0: 0xF0000008 | data3
[   50.522660] iwlwifi 0000:01:00.0: 0x00000000 | beacon time
[   50.522669] iwlwifi 0000:01:00.0: 0x0004DC44 | tsf low
[   50.522674] iwlwifi 0000:01:00.0: 0x00000000 | tsf hi
[   50.522679] iwlwifi 0000:01:00.0: 0x00000000 | time gp1
[   50.522679] iwlwifi 0000:01:00.0: 0x0004DC44 | time gp2
[   50.522679] iwlwifi 0000:01:00.0: 0x00000001 | uCode revision type
[   50.522691] iwlwifi 0000:01:00.0: 0x0000002E | uCode version major
[   50.522691] iwlwifi 0000:01:00.0: 0x3CFAB8DA | uCode version minor
[   50.522691] iwlwifi 0000:01:00.0: 0x00000340 | hw version
[   50.522703] iwlwifi 0000:01:00.0: 0x18C89000 | board version
[   50.522704] iwlwifi 0000:01:00.0: 0x800CF500 | hcmd
[   50.522704] iwlwifi 0000:01:00.0: 0xA0120000 | isr0
[   50.522715] iwlwifi 0000:01:00.0: 0x00000000 | isr1
[   50.522715] iwlwifi 0000:01:00.0: 0x08F00002 | isr2
[   50.522715] iwlwifi 0000:01:00.0: 0x00C0000C | isr3
[   50.522725] iwlwifi 0000:01:00.0: 0x00000000 | isr4
[   50.522728] iwlwifi 0000:01:00.0: 0x000301D1 | last cmd Id
[   50.522728] iwlwifi 0000:01:00.0: 0x00016F7E | wait_event
[   50.522728] iwlwifi 0000:01:00.0: 0x00000000 | l2p_control
[   50.522728] iwlwifi 0000:01:00.0: 0x00002820 | l2p_duration
[   50.522728] iwlwifi 0000:01:00.0: 0x00000000 | l2p_mhvalid
[   50.522728] iwlwifi 0000:01:00.0: 0x00000000 | l2p_addr_match
[   50.522752] iwlwifi 0000:01:00.0: 0x00000009 | lmpm_pmg_sel
[   50.522756] iwlwifi 0000:01:00.0: 0x16042142 | timestamp
[   50.522762] iwlwifi 0000:01:00.0: 0x00000828 | flow_handler
[   50.522856] iwlwifi 0000:01:00.0: Start IWL Error Log Dump:
[   50.522865] iwlwifi 0000:01:00.0: Status: 0x00000040, count: 7
[   50.522871] iwlwifi 0000:01:00.0: 0x20000034 | NMI_INTERRUPT_WDG
[   50.522873] iwlwifi 0000:01:00.0: 0x00000000 | umac branchlink1
[   50.522881] iwlwifi 0000:01:00.0: 0xC008B1C0 | umac branchlink2
[   50.522884] iwlwifi 0000:01:00.0: 0x8048BC5C | umac interruptlink1
[   50.522889] iwlwifi 0000:01:00.0: 0xC0084264 | umac interruptlink2
[   50.522894] iwlwifi 0000:01:00.0: 0x00000004 | umac data1
[   50.522901] iwlwifi 0000:01:00.0: 0xC0084264 | umac data2
[   50.522906] iwlwifi 0000:01:00.0: 0xDEADBEEF | umac data3
[   50.522909] iwlwifi 0000:01:00.0: 0x0000002E | umac major
[   50.522913] iwlwifi 0000:01:00.0: 0x3CFAB8DA | umac minor
[   50.522913] iwlwifi 0000:01:00.0: 0x0004DC2B | frame pointer
[... 136 lines cut ...]
[   50.896551]  ieee80211_open+0x4c/0x68 [mac80211]
[   50.896554]  __dev_open+0xfc/0x1a8
[   50.896557]  __dev_change_flags+0x16c/0x1d0
[   50.896562]  dev_change_flags+0x3c/0x78
[   50.896562]  devinet_ioctl+0x3f0/0x750
[   50.896562]  inet_ioctl+0x208/0x380
[   50.896562]  sock_do_ioctl+0x58/0x2c0
[   50.896562]  sock_ioctl+0x330/0x4d8
[   50.896562]  do_vfs_ioctl+0xc4/0xaa0
[   50.896586]  ksys_ioctl+0x84/0xb8
[   50.896586]  __arm64_sys_ioctl+0x28/0xc8
[   50.896586]  el0_svc_common.constprop.1+0x98/0x1a0
[   50.896586]  el0_svc_handler+0x34/0xa0
[   50.896586]  el0_svc+0x8/0x204
[   50.898495] iwlwifi 0000:01:00.0: Failed to configure RX queues: -5
[   51.002010] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-50.ucode failed with error -2
[   51.002036] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-49.ucode failed with error -2
[   51.002047] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-48.ucode failed with error -2
[   51.002075] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-47.ucode failed with error -2
[   51.006742] iwlwifi 0000:01:00.0: loaded firmware version 46.3cfab8da.0 op_mode iwlmvm
[   51.007659] iwlwifi 0000:01:00.0: Detected Intel(R) Wi-Fi 6 AX200 160MHz, REV=0x340
[   51.166347] iwlwifi 0000:01:00.0: base HW address: e0:d4:e8:49:ff:e8
[   51.180362] thermal thermal_zone1: failed to read out thermal zone (-61)
[   51.598815] iwlwifi 0000:01:00.0: Microcode SW error detected.  Restarting 0x2000000.
```

geerlingguy, 2020-11-13:
First see:

```
[   50.097951] thermal thermal_zone1: failed to read out thermal zone (-61)
```

And just noting:

```
$ grep . /sys/class/thermal/*/*
/sys/class/thermal/thermal_zone0/available_policies:step_wise 
grep: /sys/class/thermal/thermal_zone0/hwmon0: Is a directory
/sys/class/thermal/thermal_zone0/integral_cutoff:0
/sys/class/thermal/thermal_zone0/k_d:0
/sys/class/thermal/thermal_zone0/k_i:0
/sys/class/thermal/thermal_zone0/k_po:0
/sys/class/thermal/thermal_zone0/k_pu:0
/sys/class/thermal/thermal_zone0/mode:enabled
/sys/class/thermal/thermal_zone0/offset:410040
/sys/class/thermal/thermal_zone0/passive:0
/sys/class/thermal/thermal_zone0/policy:step_wise
grep: /sys/class/thermal/thermal_zone0/power: Is a directory
/sys/class/thermal/thermal_zone0/slope:-487
grep: /sys/class/thermal/thermal_zone0/subsystem: Is a directory
/sys/class/thermal/thermal_zone0/sustainable_power:0
/sys/class/thermal/thermal_zone0/temp:46251
/sys/class/thermal/thermal_zone0/type:cpu-thermal
```

lol, there's an open bug from a few years back for this thermal issue: https://bugzilla.kernel.org/show_bug.cgi?id=201761

But moving on, I also see:

```
$ iw list
Wiphy phy288
  max # scan SSIDs: 20
  max scan IEs length: 365 bytes
  max # sched scan SSIDs: 20
  max # match sets: 11
  max # scan plans: 2
  max scan plan interval: 65535
  max scan plan iterations: 254
  Retry short limit: 7
  Retry long limit: 4
  Coverage class: 0 (up to 0m)
  Device supports RSN-IBSS.
  Device supports AP-side u-APSD.
  Device supports T-DLS.
  Supported Ciphers:
    * WEP40 (00-0f-ac:1)
    * WEP104 (00-0f-ac:5)
    * TKIP (00-0f-ac:2)
    * CCMP-128 (00-0f-ac:4)
    * GCMP-128 (00-0f-ac:8)
    * GCMP-256 (00-0f-ac:9)
    * CMAC (00-0f-ac:6)
    * GMAC-128 (00-0f-ac:11)
    * GMAC-256 (00-0f-ac:12)
  Available Antennas: TX 0 RX 0
  Supported interface modes:
     * IBSS
     * managed
     * AP
     * AP/VLAN
     * monitor
     * P2P-client
     * P2P-GO
     * P2P-device
  Band 1:
[... 206 lines cut ...]
  P2P GO supports CT window setting
  P2P GO supports opportunistic powersave setting
  Driver supports full state transitions for AP/GO clients
  Driver supports a userspace MPM
  Driver/device bandwidth changes during BSS lifetime (AP/GO mode)
  Device adds DS IE to probe requests
  Device can update TPC Report IE
  Device supports static SMPS
  Device supports dynamic SMPS
  Device supports WMM-AC admission (TSPECs)
  Device supports configuring vdev MAC-addr on create.
  Device supports randomizing MAC-addr in scans.
  Device supports randomizing MAC-addr in sched scans.
  Device supports randomizing MAC-addr in net-detect scans.
  Supported extended features:
    * [ VHT_IBSS ]: VHT-IBSS
    * [ RRM ]: RRM
    * [ MU_MIMO_AIR_SNIFFER ]: MU-MIMO sniffer
    * [ SCAN_START_TIME ]: scan start timestamp
    * [ BSS_PARENT_TSF ]: BSS last beacon/probe TSF
    * [ SET_SCAN_DWELL ]: scan dwell setting
    * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
    * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
    * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
```

geerlingguy, 2020-11-13:
`ip address` shows:

```
443: wlan1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether e0:d4:e8:49:ff:e8 brd ff:ff:ff:ff:ff:ff
```

And every time the driver resets itself, it increments (now it's up to 464... waiting a few seconds, now it's 482...)

geerlingguy, 2020-11-13:
Related? https://github.com/QubesOS/qubes-issues/issues/5615

geerlingguy, 2020-11-13:
Maybe this patch will help—that's where the error is coming from, anyways... https://lore.kernel.org/linux-wireless/3cab5072-17a2-4d9a-2077-93788971c6c4@invisiblethingslab.com/T/#u

```
diff --git a/drivers/net/wireless/intel/iwlwifi/mvm/fw.c
b/drivers/net/wireless/intel/iwlwifi/mvm/fw.c
index 98263cd37944..943831c9b685 100644
--- a/drivers/net/wireless/intel/iwlwifi/mvm/fw.c
+++ b/drivers/net/wireless/intel/iwlwifi/mvm/fw.c
@@ -1111,15 +1111,6 @@ int iwl_mvm_up(struct iwl_mvm *mvm)
 		goto error;

 	/* Init RSS configuration */
-	if (mvm->trans->trans_cfg->device_family >= IWL_DEVICE_FAMILY_22000) {
-		ret = iwl_configure_rxq(mvm);
-		if (ret) {
-			IWL_ERR(mvm, "Failed to configure RX queues: %d\n",
-				ret);
-			goto error;
-		}
-	}
-
 	if (iwl_mvm_has_new_rx_api(mvm)) {
 		ret = iwl_send_rss_cfg_cmd(mvm);
 		if (ret) {
```

geerlingguy, 2020-11-13:
In my cross-compile environment:

```
$ nano iwlwifi.patch
(paste in contents of the previous comment patch)

$ git apply -v iwlwifi.patch
Checking patch drivers/net/wireless/intel/iwlwifi/mvm/fw.c...
Hunk #1 succeeded at 1367 (offset 256 lines).
Applied patch drivers/net/wireless/intel/iwlwifi/mvm/fw.c cleanly.
```

Recompiling the kernel the 18th time this week :D

geerlingguy, 2020-11-13:
Well heck. What do you know?

```
$ dmesg
[   39.933592] Intel(R) Wireless WiFi driver for Linux
[   39.933606] Copyright(c) 2003- 2015 Intel Corporation
[   39.933984] pci 0000:00:00.0: enabling device (0000 -> 0002)
[   39.934010] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
[   39.937178] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-50.ucode failed with error -2
[   39.937242] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-49.ucode failed with error -2
[   39.937298] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-48.ucode failed with error -2
[   39.937348] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-cc-a0-47.ucode failed with error -2
[   39.966176] iwlwifi 0000:01:00.0: loaded firmware version 46.3cfab8da.0 op_mode iwlmvm
[   40.069057] iwlwifi 0000:01:00.0: Detected Intel(R) Wi-Fi 6 AX200 160MHz, REV=0x340
[   40.234560] iwlwifi 0000:01:00.0: base HW address: e0:d4:e8:49:ff:e8
[   40.259091] thermal thermal_zone1: failed to read out thermal zone (-61)
```

And:

```
$ ip addr
4: wlan1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether e0:d4:e8:49:ff:e8 brd ff:ff:ff:ff:ff:ff
```

The Pi seems much happier. Now, to connect to my WiFi...

geerlingguy, 2020-11-13:
Hmm... this isn't super encouraging though:

```
root@raspberrypi:/home/pi# wpa_cli
wpa_cli v2.8-devel
Copyright (c) 2004-2019, Jouni Malinen <j@w1.fi> and contributors

This software may be distributed under the terms of the BSD license.
See README for more details.

Selected interface 'p2p-dev-wlan1'

Interactive mode

> scan
OK
> scan_results
bssid / frequency / signal level / flags / ssid
> 
```

geerlingguy, 2020-11-14:
But... after setting up config in `wpa_supplicant.conf`, I am seeing:

```
$ ip addr
...
4: wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether e0:d4:e8:49:ff:e8 brd ff:ff:ff:ff:ff:ff
    inet 10.0.100.211/24 brd 10.0.100.255 scope global dynamic noprefixroute wlan1
       valid_lft 86396sec preferred_lft 75596sec
    inet6 fe80::194d:561c:5428:3250/64 scope link 
       valid_lft forever preferred_lft forever
```

So... 🤷‍♂️ 

Let's get some performance testing!

geerlingguy, 2020-11-14:
On the Pi:

```
$ sudo apt install -y iperf
$ iperf -s
```

On my Mac:

```
$ iperf -c 10.0.100.211
------------------------------------------------------------
Client connecting to 10.0.100.211, TCP port 5001
TCP window size:  129 KByte (default)
------------------------------------------------------------
[  4] local 10.0.100.118 port 64560 connected with 10.0.100.211 port 5001
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-10.0 sec   784 MBytes   658 Mbits/sec
```

As I used to say in college: "Noice!"

Compared the the maximum 80 Mbits/sec I can get via the CM4's internal WiFi, even with an external antenna... that's rather speedy!

geerlingguy, 2020-11-14:
Dangit, now that I finally had some success I should probably make a video about it. But who has the time!?

geerlingguy, 2020-11-14:
`atop` output while the Pi is running the iperf server:

[image]

So with all the defaults, ~675 Mbps while Pi is doing the serving, ~930 Mbps when not. It's gonna be interesting to see how real-world copies work on this, but I'll have to wait as dinner calls!

geerlingguy, 2020-11-14:
So... I was thinking, "hey, it would be a great idea to test this WiFi card on a real WiFi 6 network, using a router that works with a 10G network interface, allowing me to test the full throughput against my Mac over 10G wired network..."

Then I was glancing around at the (very few, so far) good WiFi 6 ax routers with 10G network ports, and they all cost $450+ and are out of stock pretty much everywhere. I was looking mostly at the [ASUS RT-AX89X](https://amzn.to/2H2pr3D), mostly because I've stuck with ASUS wireless gear for like 5 years now and never had an issue... but it's out of stock everywhere.

So I'll stick with testing over my [ASUS AC2900](https://amzn.to/3324Bt5)-based 802.11ac network for now.

I mean, getting 1 gig over my current network is nothing shabby, as that's also probably hitting the max throughput I've seen through two of my Macs (a 2016 MBP and a 2019 MBP) over the same network... Heck, maybe my Mac is the slower of the two devices in terms of raw potential throughput!

All that to say, I think the 'year of WiFi 6' in the Geerling household is not in 2020. Unless maybe I can get ASUS to sponsor a video (a video for a router, eh?).

PixlRainbow, 2020-11-15:
How much would it cost to just have a direct ad-hoc connection between two Wi-Fi 6 clients, instead of buying a router?

geerlingguy, 2020-11-15:
@PixlRainbow - I've been looking around unsuccessfully—does anyone make a WiFi 6 / ax adapter for ThunderBolt 3, USB-C/3.2, or any other connection besides internal/PCIe?

I don't have any other devices currently that support PCI express :(

PixlRainbow, 2020-11-15:
Your Mac is a laptop? Or a desktop? I'm guessing it's an older model that doesn't have WiFi 6 built in.
Some laptops and desktops may have a "WiFi card" slot which you may be able to install a WiFi 6 card into. But some newer devices aren't really made to allow end users to open them up and configure their hardware.

geerlingguy, 2020-11-15:
Both my Mac and Dell are laptops, and I just realized the Dell has a replaceable WiFi module—so I just ordered a [AX200NGW Wireless Card in M.2/NGFF](https://amzn.to/2Uv4n9a), and it's coming tomorrow.

Getting ready to do a little surgery on my XPS 13 again (upgraded the SSD a couple years ago), and I'll let you know how it goes. It looks like iperf is also available on Windows, so I'll have to see what happens! (Totally unfamiliar with Windows' networking stack since the last time I did anything networking-related was probably around the time Windows NT was the new hotness).

geerlingguy, 2020-11-17:
The card works great! Working on a short video now just to show the process of upgrading the Dell card.

geerlingguy, 2020-11-18:
Upgraded the Dell, made a video: [How to Upgrade a Dell XPS 13 to WiFi 6 (802.11ax)](https://www.youtube.com/watch?v=C760v9XpkzE).

geerlingguy, 2020-11-19:
Heh. This is probably the most expensive of all the tests so far. I figured...

  - I'm going to be upgrading my home network to 10G soon (at least my office drop).
  - I now have two 802.11ax devices.
  - I might as well upgrade my WiFi router for the time when I inevitably get an M1 Mac that has built-in WiFi 6.

So I ran and grabbed an [ASUS RT-AX86U AX5700 WiFi 6 router](https://amzn.to/3nGBHGw), which is not the top of the heap, and is only dual band... but is better than my older ac router, and has a 2.5 Gbps LAN uplink port (which can be swapped for the WAN port—though here in St. Louis it'll probably be 50 years before we end up getting gigabit ethernet to the home!). I installed it, got it configured, and now I'm planning on doing some more testing from 10G wired to AX wireless on both my Dell laptop and the CM4.

But that has to wait a bit as I'm trying to finish up my NAS explorations with the SATA card first!

mo-g, 2020-11-22:
This is exceedingly interesting. So - similar to Danish Guy's project, I've also been thinking about how the Pi CM4 and recent kernel patches now make a Pi-based Laptop feasible. This suggests that the best way to implement Wi-Fi on such a device would actually not be through the CM4's SDIO chipset, but through a standard m.2 card.

I guess the ideal approach would be to design the laptop to allow an antenna cable to be routed to either a PCIe card or a Wireless CM4; and offer the former through one of those fancy m.2 slots with three screws that could take anything from a wifi card to an NVMe drive?

geerlingguy, 2020-11-22:
@mo-g yes, I think that would be ideal. Basically give the user some flexibility.

Dygear, 2020-12-06:
@geerlingguy We need to stop running into each other like this. People will talk.

Yeah, I'm actually having the same problem as you had here but on a Razer Blade Stealth that I'm using while waiting for my System76 laptop be be delivered. I _really_ don't want to have to recompile the Kernel as you had done here. Surely that shouldn't be the solution given the fact that I'm ...
1. On an x86_64 platform.
2. Should therefore not have a customer kernel from the Pi foundation that's stripped down.
3. Did all of the things that you short of recompiling the kernel and yet here I am and it's still not working.
And yet here I am anyway trying to figure at this problem as well. I don't think the AX200 platform cards are as Linux friendly as they seem to claim.

Will update once I've got this working.

st0nedB, 2020-12-13:
Hey, just stumbled across this after watching your latest video! Regarding the testing of raw 802.11ax performance with `iperf` and ASUS network gear discussed above:
Why not throw Merlin WRT on an ASUS router, enable Entware, then install and run `iperf` server on the router? This way you could exclude the wired connection (which, as you said, might be a bottleneck).
Just did a quick test with my Phone and RT88AX and got around 900 MBit/s without any further optimizations at a reasonable distance (some meters). Steps are quite simple.
1. Download and install Merlin WRT [1], which is a simple as upgrading the default Asus firmware.
2. Plug a USB drive into the router, enable Entware by running the included `entware-setup.sh` or use `amtm` maintained by [2] (depending how much extras you want to throw on your router, see below).
3. Install iperf3 via opkg package manager with `opkg install iperf3`
4. Run `iperf3 -s`, profit!

If you are using Asus network gear, you might be able to make use of some other community-developed tools running directly on the router (adblocking, improved firewalls, ...).

[1] - [MerlinWRT](https://www.asuswrt-merlin.net)
[2] - [amtm](https://diversion.ch/amtm.html)

geerlingguy, 2020-12-17:
Interesting... after upgrading my router, I retested everything today, and I can get 875-900 Mbps both ways, regardless of which host is the iperf3 server now.

Further testing is warranted—I now have a 10 GbE interface hooked up to my Mac, a 2.5 Gbps connection to my WiFi router, and a 10 GbE switch (though I'm not sure if it can drop to 2.5G on the router connection, so that'll have to be something I test later).

I'm guessing I could push more bandwidth through this card with my new router :-O

geerlingguy, 2020-12-19:
Now that `iperf3` has `--bidir` (see https://github.com/esnet/iperf/pull/780), I should also do a test with that mode: `iperf3 -c [ip address] --bidir`

geerlingguy, 2020-12-19:
Testing with the Merlin firmware:

  1. Downloaded beta version for AX86U from here: https://www.asuswrt-merlin.net/download
  2. Follow the [Installation](https://github.com/RMerl/asuswrt-merlin.ng/wiki/Installation) guide (basically update firmware to the downloaded version).
  3. Restart the router, verify things still work fine.
  4. TODO...

geerlingguy, 2020-12-20:
Hmm... the version installed via `opkg` is:

```
# iperf3 --version
iperf 3.1.3
```

It looks like that's a fairly outdated version though, and doesn't support `--bidir`.

geerlingguy, 2020-12-20:
Regular old `iperf` allows `--dualtest`:

```
$ iperf -c 10.0.100.1 --dualtest
...
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.0 sec  78.4 MBytes  65.5 Mbits/sec
[  7]  0.0-10.1 sec   556 MBytes   463 Mbits/sec
```

geerlingguy, 2020-12-20:
On the Pi:

```
iperf -c 10.0.100.1 --bind 10.0.100.211 --dualtest
...
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.0 sec   265 MBytes   222 Mbits/sec
[  4]  0.0-10.0 sec  1.40 GBytes  1.20 Gbits/sec
```

Standard `iperf`: `927 Mbits/sec`
`iperf3`: `928 Mbits/sec`

geerlingguy, 2020-12-22:
Video posted: [WiFi 6 on the Raspberry Pi CM4 makes it Fly! MORE THAN 1 Gbps!](https://www.youtube.com/watch?v=csI19aOJEik)

geerlingguy, 2020-12-22:
And a blog post: [WiFi 6 gets 1.34 Gbps on the Raspberry Pi CM4](https://www.jeffgeerling.com/blog/2020/wifi-6-gets-134-gbps-on-raspberry-pi-cm4).

christofer-f, 2020-12-29:
Hi! I just stumbled over your excellent youtube channel. 
I am thinking that it would be really cool to build an open-source python-based system for wifi 6 access points.

So here are my question...
Is it possible to query the wifi 6 card about:
1) Users connected and there signal strength
2) Disconnect a user
3) Connect a user

If so then one could build a wifi system that automatically passes a connected user around between different access points.

Cisco sells these kind of systems so I thought that it would be fun to build an open source version. 

I was thinking that you could build a rest like interface with fastapi to run on the rasberry and the control all ap:s from a webserver.

geerlingguy, 2021-01-02:
@christofer-f - It seems the AX200 might not be the best chip for the purpose (though it probably can be made to work). It would be cool to be able to do an open source setup like that, though!

geerlingguy, 2021-01-02:
All right, after going through another round of testing for the Intel AX200 Desktop Kit adapted by the MZHOU WiFi/BT adapter in #38, I'm going to re-run the benchmarks today, this time both with the wired network enabled and disabled, and see what kind of results I get:

First testing with `iperf` and `--dualtest`:

```
$ iperf -c 10.0.100.1 --bind 10.0.100.168 --dualtest
```

Then with `iperf3`:

  - eth0 disabled (`sudo ip link set eth0 down`):
    - iperf3 from Pi to router: TODO Mbits/sec
    - iperf3 from router to Pi (`--reverse`): TODO Gbits/sec
  - eth0 enabled (`sudo ip link set eth0 up`):
    - iperf3 from Pi to router: TODO Mbits/sec
    - iperf3 from router to Pi (`--reverse`): TODO Gbits/sec

geerlingguy, 2021-01-08:
Hmm...

```
$ iw dev wlan1 info
Interface wlan1
	ifindex 4
	wdev 0x1
	addr e0:d4:e8:49:ff:e8
	ssid geerling-acn
	type managed
	wiphy 0
	channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
	txpower 22.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
```

It's using 80 MHz, why not 160 MHz?

geerlingguy, 2021-01-09:
Closing issues where testing is at least mostly complete, to keep the issue queue tidy.

I'm going to move the rest of the testing of the AX200 with 80 vs 160 MHz over to #56.

closed 2021-01-09 by geerlingguy

lealog, 2021-05-18:
The URL for amazon is not working.

geerlingguy, 2021-05-18:
@lealog - That's weird. Even if I go into my order history on Amazon, and click through the order link, it gives me a Page Not Found... I guess EDUP dropped the card or something :(

belden, 2024-01-02:
What a fun repo to have found!

The Amazon link does work again now.