geerlingguy/raspberry-pi-pcie-devices #23: Test TV-Cards

JanHBade, 2020-11-11:
https://skinflint.co.uk/?cat=vid&xf=800_PCIe+x1

the cards from digital devices have a nice Linux support

geerlingguy, 2020-11-23:
If I picked up one of these, I would probably want to first see if [tvheadend](https://github.com/tvheadend/tvheadend) will compile on ARM/Pi, and if not, it might not be worth trying to hack together the software side.

It seems like something like the [Hauppauge 1196 HVR-1265](https://amzn.to/397DX5A) would have the best shot at working, as it does have Linux drivers, though it's only 'supported' on Ubuntu. And I'd have to wonder if anyone's ever tested/compiled the drivers for ARM...

TheGuyDanish, 2020-11-23:
Chiming in since I've got a Pi Zero running it beautifully. Tvheadend works great with the Pi and the [TV hat](https://www.raspberrypi.org/products/raspberry-pi-tv-hat). The only challenge is that tvheadend's support has been a bit spotty with Raspbian Buster, and even Stretch is only for armhf. So using it with a newer OS and 64-bit could be a challenge.

Assuming there's a card with a neatly compatible driver, I could see this being a good use case.

~I'll go back to working on my mATX board now, though.~

geerlingguy, 2020-11-23:
Hehe, can't wait to see a copy of that board :)

But extremely good to know—saves me the time I'd have to do testing it out as much as I can without a tuner card. I currently have a HDHomeRun so I might even test it out with that too.

closed 2020-12-01 by geerlingguy