geerlingguy/raspberry-pi-pcie-devices #24: Test SSD

Rucadi, 2020-11-15:
It could be cool to test if a PCI SSD can be installed, and how it affects the responsiveness of the whole system.

geerlingguy, 2020-11-16:
I've been doing a lot of testing with SSDs via SATA adapters (see https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/1) — please keep the discussion in that thread for now. I have a couple other SATA cards I'll be testing soon as well (along with hopefully more NVMe drives too!).

closed 2020-11-16 by geerlingguy

PixlRainbow, 2020-11-16:
I think they are referring to SSDs that are directly integrated into a PCIE card -- not to PCIE adapters for SSDs e.g. looks something like this: [image]

geerlingguy, 2020-11-16:
@PixlRainbow - I see; I would rather continue testing SSDs and NVMe drives and/or adapters directly—there are a ton of cheap (< $20) NVMe PCI express 1x adapters that work with M.2 form-factor drives that are more appealing in terms of convenience for me. I honestly didn't know there were cards that were literal SSD-cards, though I'm wondering if you popped off the cover on that Zotac, if it would just be one or more M.2 drives underneath with some heat sinks attached.

geerlingguy, 2020-11-16:
Ah so TIL these things exist, sorry for doubting you! I'm removing the duplicate label, as apparently there are some more specialized (and usually WAY more expensive) SSD PCIe cards that are basically giant SSD cards, and not SSDs strapped to cards, like this WD Black: https://amzn.to/3pF3FVd

I think it would be interesting to see if one works on the Pi, but that's well beyond my own budget, and the other thing I should note is it looks like these want 4x lanes, for better performance, and that performance would just be wasted on a Pi CM4, with its 1x lane and 5 Gbps of bandwidth (real world < 3.5).