geerlingguy/raspberry-pi-pcie-devices #25: Document some neat CM4 projects/accessories

geerlingguy, 2020-11-16:
For example, today @BG5USN sent me this: https://twitter.com/BG5USN/status/1328331941536477189

[image]

Basically a tiny CM4 IO board. Very neat, great for just working with CM4s (but probably wouldn't be a great replacement for the Pi 4 model B itself), and it has full-size HDMI :)

There's also the Turing Pi 2: https://turingpi.com

[image]

And CM4 MATX, a micro-ATX compliant board design so you can securely install a CM4 inside an ATX case with a PCIe slot in the right place: https://github.com/TheGuyDanish/CM4_MATX

[image]

I don't know where exactly I'll document these things, but I know I'm excited to see some of the different builds people make!

geerlingguy, 2020-11-20:
[image]

The [Gumstix Raspberry Pi CM4 Development Board](https://www.gumstix.com/manufacturer/raspberry-pi/cm4-development-board.html), which has a built-in NVMe slot (I presume it is consuming the PCIe 1x lane).

paulwratt, 2020-11-21:
On release day of the CM4, Gumstix had already produces the following hardware and adaptors for the CM4:
https://www.cnx-software.com/2020/10/19/gumstix-introduces-cm4-to-cm3-adapter-carrier-boards-for-raspberry-pi-compute-module-4/
including a CM3 adaptor that works with the current TuringPi (note the inclusion of a Google Coral accelerator):
[image]

geerlingguy, 2020-11-27:
Another interesting project, the [PiKeeb](https://twitter.com/PiKeeb/status/1332046059695202304):

[image]

[image]

Seems to be a combination mechanical keyboard + CM4 board (similar to Pi 400 in general design), with the addition of a fold-out screen; more details are on the project's subreddit: https://www.reddit.com/r/PiKeeb/ (would be nice to have a page somewhere with consolidated info, because you kinda have to read around in that subreddit to figure out everything the board should be able to do).

geerlingguy, 2020-11-27:
Another interesting project, the [StereoPi V2](https://hackaday.io/project/175788-stereopi-v2-cm4-based-diy-stereoscopic-camera), for stereoscopic photography/video (interesting application for robotics too):

[image]

geerlingguy, 2020-11-29:
User u/mebs_t on Reddit posted this [Compute Module 4 NAS PCB with PCIe](https://www.reddit.com/r/raspberry_pi/comments/jt89dm/compute_module_4_nas_pcb_with_pcie/) board:

[image]

The idea is to have that board with the CM4 connected internally to some drives in a tower enclosure like:

[image]

There are some good links for those interested in putting together a similar kind of project. Also in that thread I found out that [Gumstix is offering free manufacturing for CM4 designs through the end of the year](https://www.gumstix.com/community/special-offer/)—probably some strings attached, but an interesting offer nonetheless.

geerlingguy, 2020-12-04:
Uptime Lab (www.uplab.pro) [posted this blade server concept to Twitter](https://twitter.com/Merocle/status/1334521202329661445):

[image]

It looks like it wouldn't be the most complicated PCB to manufacture, and could hold up to 22 devices (176 GB RAM, 44TB storage with 22 2TB NVMe drives. Gigabit backplane maximum, though, so it wouldn't be able to be fully utilized (more to come on why in my next video!).

geerlingguy, 2020-12-11:
Another really cool idea is the [Piunora from timonsku on Twitter](https://twitter.com/timonsku/status/1337134272625467392):

[image]

[image]

It has a slightly custom layout, but features an M.2 B-key connector on the bottom (something _many_ have desired), and tons of other niceties that make it a well rounded little prototype. Not available for sale, but something like this could be a popular option for those of us who want a 'Pi 4 but with built-in NVMe on the bottom' :)

(@timonsku was also the person behind [this dead-simple carrier board](https://hackaday.com/2020/11/13/easy-carrier-board-for-the-compute-module-4-shows-you-can-do-it-too/) featured on Hackaday.io.)

iandk, 2020-12-11:
Someone should design a nice small board with a few onboard SATA + power ports.

andywarburton, 2020-12-11:
I would love to see a little carrier board tuned for cyberdeck makers. Onboard battery charging plus boost to get us enough amps for all the things, power out for additional devices too full sized hdmi out, 4 USB, ethernet etc (basically a full pi4 with battery management and a few extra terminals for power).

volkertb, 2020-12-12:
It would be cool to have something similar to the Piunora, that would expose the PCIe interface as a Thunderbolt (or USB4?) port. I'm not sure how complicated it would be to implement something like that, though.

Belcarra, 2020-12-13:
Looking for a board that has a SuperSpeed (aka USB 3.x) USB Client port.

GaganCJ, 2020-12-13:
Ethernet port in Piunora would have been nice

geerlingguy, 2020-12-13:
Another project that showed up on the radar today thanks to [@JorisBryssinck](https://twitter.com/JorisBryssinck), the CutiePi tablet:

[image]

Even though the website still mentions the CM3+ Lite, they [recently announced an upgraded design using the CM4](https://twitter.com/cutiepi_io/status/1331871597368467458):

[image]

gabyavra, 2020-12-13:
Anybody found a minimalistic carrier board with only USB C for instance (data + power), so I can use the CM4 like this? https://www.youtube.com/watch?v=IR6sDcKo3V8 (emulated Ethernet via USB c in order to ssh into the device)
Thank you

geerlingguy, 2020-12-16:
Looks like there's also the [MCUzone CM4 4G](https://m.weibo.cn/detail/4577659583463712):

[image]

[image]

Video: https://www.bilibili.com/video/BV1mi4y1V7qw

And gathering what I can from translations, it seems the board is tailored for use with a 4G or 5G modem (through an onboard mini-PCIe slot). The board as pictured has "4K HDMI output, Gigabit wired Ethernet, 4G LTE Cat4 network, dual USB host, and USB-C power supply interface."

arthuraldridge, 2020-12-17:
> Someone should design a nice small board with a few onboard SATA + power ports.

Power would still be an issue but...

https://www.seeedstudio.com/M-2-to-SATA-Converter-5-Ports-p-4726.html

volkertb, 2020-12-18:
I'm frankly surprised that there isn't yet a project or crowdfunding drive to create a smartphone around a CM4. So with a smartphone form factor, including a touch screen, 4G modem, the ability to make voice calls, etc, but having an interface for a CM4 (or compatible device) instead of an integrated SoC. You wouldn't even have to start entirely from scratch for something like this. You could basically take the schematics of the PinePhone and/or the LibRem 5 (I believe they are both open source hardware), and use those as a basis.

The advantage would be a phone that would not only be open with out-of-the-box mainline Linux kernel support, but upgradable as well.

(Let's just hope the Raspberry Pi foundation will not make incompatible changes to the Compute Module form factor so soon again, like they did between the CM3 and the CM4.)

hipboi, 2020-12-18:
Let's do this :-)

hatonthecat, 2020-12-18:
> > Someone should design a nice small board with a few onboard SATA + power ports.
> 
> Power would still be an issue but...
> 
> https://www.seeedstudio.com/M-2-to-SATA-Converter-5-Ports-p-4726.html

mSATA has some of best performance/watt ratio. The Samsung PM851 uses <200mW in active use. 
https://www.amazon.com/Original-512GB-mSATA-Retail-Packaging/dp/B08DLW863Y
There are some USB-mSATA expansion boards for the Pi4 like: https://notenoughtech.com/rpi-hat/x857-msata-for-raspberry-pi-4/
and I've seen some GPIO adapters for mSATA like the PiDrive (2015 status unknown). I think a little boost for write and random reads would be great, Otherwise there is a handful of Class 10 cards that are 1GB transfers in less than a minute for copy/write speeds.

stevefan1999-personal, 2020-12-23:
Test Google Coral via a PCIE to Mini PCIE adaptor (or inverse converter)

https://coral.ai/products/pcie-accelerator/

stevefan1999-personal, 2020-12-23:
https://developers.googleblog.com/2020/09/doubling-down-on-edge-with-corals-new.html?m=1

there is also the dual core variant, though it requires M.2 and i doubt RPi itself would have to capacity to run it at all

TheGuyDanish, 2020-12-23:
There was some talk of this already as I recall, but the Coral PCIe adapters require MSI-X, which the CM4 does not support.

https://developers.googleblog.com/2020/09/doubling-down-on-edge-with-corals-new.html?m=1

stuart-c, 2020-12-23:
The CM4 does support MSI-X and there has been some work to support the Coral PCIe adapters. Google are apparently adjusting their software to try to overcome some of the remaining issues:

https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=294924&p=1780727&hilit=coral#p1780681

I believe the dual PCIe adapter requires two Gen2 x1 lanes, so wouldn't work with the CM4 (as there is only a single lane).

geerlingguy, 2020-12-23:
I'm actually digging into MSI-X support a bit more lately for one [secret] project and for Intel networking cards to be able to run more efficiently.

It seems that there was a recent commit in the RPi Linux kernel (5.10.y branch) that could make some MSI-X requiring cards work: https://github.com/raspberrypi/linux/commit/4a93ff6eb717a8c45532159124efcd5abfb02bac (but not everything is supported).

geerlingguy, 2020-12-27:
Another new NAS board, from @olvint this time: https://github.com/olvint/CM4-NAS-MiniPCIE — It's a fork of https://github.com/mebs/CM4-NAS with a mini PCIe slot (instead of full-size). Pictures speak louder than words:

[image]

[image]

That looks like [this little IOCrest mini PCIe card with the ASMedia 1061R chipset](https://amzn.to/3nOiuTT) for $27. Syba also makes a [full-size mini PCIe card with 4 SATA ports](https://amzn.to/2M8iGPP) for $31.

g30ba1, 2020-12-28:
Wow! There are so many cool projects on course!

I hope that EdgeTPU guys finish the work to get the MSI-X support done and ready to deploy 🙏

ansonhe97, 2020-12-28:
Could also check this out: https://twitter.com/Nicho810/status/1337210218980118529

Dual gigabit Ports for CM4, soft router options :D

geerlingguy, 2020-12-28:
@ansonhe97 - Ah, great find! It looks like that board is using two interfaces—one is the internal CM4 gigabit ethernet interface, and the other is a USB 3.0 interface that goes through the PCIe bus. Clever solution and I'm guessing that cuts down on costs slightly—but it does add a little extra overhead. Probably not so bad with < 2.0 full gigabits.

Here's a picture, for those not wishing to jump through to Twitter:

[image]

Looks like a really cool and tidy little router board, perfect for OpenWRT. Pair that up with the Intel AX200 wifi card, and this could be a pretty cool solution (would need a PCIe switch before the USB 3.0 chip though... adds complexity and board size most likely).

[According to memoryleakyu](https://twitter.com/memoryleakyu/status/1337282171732520960), preorders start for the new year on https://www.seeedstudio.com

ansonhe97, 2020-12-28:
Yes, I'm beta testing the hardware and it works great with OpenWrt! Please check here for some stats: https://twitter.com/Ansonnh/status/1342469626555289600
https://twitter.com/Ansonnh/status/1341996436141948928

stevefan1999-personal, 2020-12-29:
> 
> 
> @ansonhe97 - Ah, great find from @nicho810! It looks like that board is using two interfaces—one is the internal CM4 gigabit ethernet interface, and the other is a USB 3.0 interface that goes through the PCIe bus. Clever solution and I'm guessing that cuts down on costs slightly—but it does add a little extra overhead. Probably not so bad with < 2.0 full gigabits.
> 
> Here's a picture, for those not wishing to jump through to Twitter:
> 
> [image]
> 
> Looks like a really cool and tidy little router board, perfect for OpenWRT. Pair that up with the Intel AX200 wifi card, and this could be a pretty cool solution (would need a PCIe switch before the USB 3.0 chip though... adds complexity and board size most likely).
> 
> [According to memoryleakyu](https://twitter.com/memoryleakyu/status/1337282171732520960), preorders start for the new year on https://www.seeedstudio.com

With a combination of not only OpenWRT, but LXC/virtualization (VyOS/pfSense pick one) and the assistance of another gigabit switch, this can also be fundamentally a very powerful home router + VPN gateway + being able to run all kinds of service! We can even opt to remove the USB for a SATA port to make it ascend to an all in one NAS. The problem is, will there be enough CM4 out here to cater all this kinds of insane needs😂

Holys*it, this is going to be the next-gen custo-ZeroTier!

OverDevices, 2021-01-06:
Hi, I've just launched an Indiegogo campaign to make my Mini-ITX carrier board design a reality. So here is another project to add to your list. Keep up the great work!

[https://www.indiegogo.com/campaigns/over-board-raspberry-pi-4-mini-itx-motherboard](url)

BMicraft, 2021-01-06:
> Hi, I've just launched an Indiegogo campaign to make my Mini-ITX carrier board design a reality. So here is another project to add to your list. Keep up the great work!
> 
> [https://www.indiegogo.com/campaigns/over-board-raspberry-pi-4-mini-itx-motherboard](url)

Unfortunately link seems to be broken in more ways than one and I am unable to correct it. Mind linking your project again @OverDevices?

geerlingguy, 2021-01-06:
@BMicraft - I fixed the link — @OverDevices was using BBCode or something not-markdown, and that mangled the generated link.

geerlingguy, 2021-01-06:
@OverDevices - A couple quick questions in case you might want to answer them here or elsewhere (neat project btw, very compact with a lot of functionality!):

First, in your description about the audio and SATA connector:

> USB 2.0 4-port hub with two physical ports, USB Audio IC with Line-out and Mic-in, and a USB SATA controller for HDD/SSD storage. I've kept the PCI-Express lane clear to the slot for full performance expansion cards and offer the additional functionality using the USB 2.0 bus which gives minimal functionality without compromising expandability.

It seems the way it's implemented the SATA connection will be not only shared with the external USB 2.0 ports (and audio), but will be limited to a maximum of 480 Mbps (~60 MB/sec), which seems pretty slow.

Is there any possibility this could be changed up to use a PCIe switch instead, with a faster PCIe SATA adapter chip that would allow for speeds of up to 400 MB/sec (obviously shared with any other PCIe devices a user chooses to install)?

That would make the board alone a lot more functional, and solve one of the primary use cases—a Pi with a built-in fast hard drive solution without making the user buy an addon PCIe card.

> Raspberry Pi 4 Compute Module not included with Over:Board as you may already have one and they are readily available from online retailers

Heh... "readily available" probably won't be true for many of the models of CM4 for a few months yet!

OverDevices, 2021-01-06:
@BMicraft Apologies, the link I pasted must have been just for me when logged in (my first Indiegogo!), thanks for pointing it out and getting it fixed

@geerlingguy Thanks for the support! It's been a labour of love for a few months now. To answer your questions...

USB SATA - there are a few reasons for this decision, not the least of which is cost...I want to keep the cost of the board to a reasonable minimum to make it available for all. Secondly, the Raspberry Pi 4 supports booting from USB and as you have experienced yourself it is a bit hit and miss which PCI-E controllers/switches are compatible. I wouldn't want to make a board that appears to work ok initially but it turns out has compatibility issues further down the line once we know more about what the Pi 4 CM can and can't actually do. Lastly, as mentioned I wanted to keep the fairly limited PCI-E bandwidth untouched for maximum performance and compatibility for add-ons, whereas the USB 2.0 is generally only used for low-speed devices and compatibility is high.

Readily available - in the UK it is possible to get CM4s from online retailers, the 4Gb RAM models are currently out of stock but we are promised availability within a few weeks from now. So hopefully by the time I can get boards out to people they will all be fully stocked...globally!

Open Hardware License - I am a massive fan, user and contributor to the Open Source community and I would love to make this openly available at some point in the future. I've put a lot of time and effort into it so far and it will take quite a lot more work to make it fully commercially available, so once I've recouped the cost of that this is definitely something I would look to do. Hopefully by then I can have Version 2.0 on the bench!

Generally speaking I've tried to keep things a simple as possible whilst offering as much functionality as is feasible, I'd love this to be a viable product spawning multiple generations of Over:Boards with ever increasing capabilities. I need to get the baseline product right and then we can move on from there with a whole host of onboard controllers, slots and connectors with the community-led driver and software support to go with them. The work you guys are doing here is definitely going to help with that journey!

Thanks guys

geerlingguy, 2021-01-06:
Another project I noticed last night a on a browse of Reddit: [CM4 Handheld - My First Prototype](https://www.reddit.com/r/raspberry_pi/comments/kr112s/cm4_handheld_my_first_prototype/), by u/Temporary-Ability-46 (a.k.a. "Inquisitive Engineer"):

[image]

[image]

[image]

And he posted a brief YouTube video of the thing in action: [Raspberry Pi CM4 Handheld Prototype Test](https://www.youtube.com/watch?v=XAPKUmwa3sU)

harlab, 2021-01-06:
Hi Jeff, we at Harlab are in a good shape to start CM4Ext Nano production soon
[image]
Project page: [https://github.com/harlab/CM4Ext_Nano](url)
[twitter.com/harlab_com](url)

geerlingguy, 2021-01-06:
@harlab - Wow, that's tiny! Nice work, and I'm definitely interested in seeing what kind of things people can do with it.

stephenvalente, 2021-01-07:
@harlab. Looking forward to that board too. Have just managed to source an
official I/O board and a 2Gb Compute module to test things out (4Gb
non-existant in the UK ironically, and 8Gb only sold to bulk customers for
some reason)

harlab, 2021-01-07:
@geerlingguy you’re in a first row to get it for review
@stephenvalente @gabyavra board should be available for purchase in February.
To keep this thread clean for other amazing projects feel free to share your thoughts at CM4Ext Nano Issues [https://github.com/harlab/CM4Ext_Nano/issues/1](url)

mi-hol, 2021-01-08:
@OverDevices appreciate the goal

> **_USB SATA_** - there are a few reasons for this decision, not the least of which is cost...I want to keep the cost of the board to a reasonable minimum to make it available for all. 

On Indiegogo you mentioned a final price of ~ 120 Euro. Which honestly I find a bit concerning because the board plus CM4 (45 -100 Euro) will compete with Celeron J1900 to J4125 based ITX boards in the 60-100 Euro range plus RAM (15-30 Euro) that offer 4 or more SATA ports

stuart-c, 2021-01-09:
> @OverDevices appreciate the goal
>
>
> On Indiegogo you mentioned a final price of ~ 120 Euro. Which honestly I find a bit concerning because the board plus CM4 (45 -100 Euro) will compete with Celeron [...]

In general it is always going to be very difficult/impossible to compete on price with mass produced products. The economies of scale when you are producing millions of a product are significant, so a small run item (like pretty much any of the available add-ons for the Raspberry Pi) has very little chance of being cheaper at comparable specifications.

If you just want the cheapest option something like the Celeron or even just a normal Pi with USB adapters (or a CM4/CMIO with an off the shelf SATA PCIe card) are probably the best option.

However the big advantage you get with CM4 is the flexibility. While a standard Pi or alternative such as an ITX board may be cheaper you have to live with whatever they give you - no way to change the size/shape, add or remove features, etc. You do pay for that. For example a standard Pi 4 is $35 and there are only a couple of lower spec (only 1GB RAM and/or no WiFi) that are slightly cheaper. Adding on the cost of the board you mount the CM4 on (which for all but one of the cheaper options would need a SD slot or equivalent) is pretty much definitely going to come out at more than the cost of a standard Pi.

mi-hol, 2021-01-09:
@stuart-c the board is targeting a specific use case (i.e. NAS). 
First the required features must exist (4 SATA ports), only second flexibility comes into play

stuart-c, 2021-01-09:
> @stuart-c the board is targeting a specific use case (i.e. NAS).
> First the required features must exist (4 SATA ports), only second flexibility comes into play, third decision point is the price difference of alternatives.

I don't see anything on the Indiegogo page saying it is specifically for a NAS use case. It seems to be targeting more general desktop usage (fitting in with a standard desktop sizing). While it can be used as a NAS it does have several features you probably don't need (dual HDMI, audio, RS232), and a single SATA port is sufficient for a basic NAS. You could add additional storage via the PCIe slot (SATA adapters, RAID cards, NVMe adapters).

One thing to be cautions of with regards to the Pi/CM4 for heavy NAS usage is the limited interface speed for storage. The "best" interface would be the PCIe slot connected to an NVMe drive or SATA card, but even with just that single drive the speed is going to be lower than you could get from a mass produced ITX motherboard (as you only have a single x1 lane). Trying to have multiple drives is just going to divide that speed further (unless the only reason for more drives is to allow higher storage density and only one drive is likely to be used at a time).

I think the Pi/CM4 for NAS realistically is limited to a single drive mass storage but lower speed use case - so backups, audio storage, maybe some video (but probably not lots of simultaneous high quality playout streams from disk). Personally for dedicated NAS usage I'd want a board which just has a single NVMe slot (mainly because that would be cheaper than a SATA port), network (maybe with POE) and power input. I probably wouldn't want HDMI ports and maybe no USB either (or maybe a single USB2 port). That would probably be the lowest cost and smallest NAS board option.

dans98, 2021-01-11:
Here is a pretty good looking one that I found over the weekend. My only wish/complaint was to get all the standard I/0 on one edge, so people can make a nice looking mini desktop or server. 

https://forum.seeedstudio.com/t/design-of-carrier-board-for-raspberry-pi-com4/254797
[image]

aessig, 2021-01-13:
Hi guys, we have been busy building our own CM4 Carrier for industrial applications. We called it the TOFU. Let us know what you think. (https://twitter.com/oratek_ch) @anybuche

[image]
[image]

olvint, 2021-01-13:
Tiny Car CM4. 6 layer board from [Markus Kasten](https://twitter.com/markusinabox) 
[Twitter post link](https://twitter.com/MarkusInABox/status/1347920396998344705)
[image]

matthiasjauch, 2021-01-15:
https://www.waveshare.com/compute-module-4-poe-board.htm

geerlingguy, 2021-01-15:
For reference; two videos I've done relating to projects in this thread:

  - [10 Exciting Raspberry Pi Compute Module 4 Boards](https://www.youtube.com/watch?v=4vSGl5kpf-k)
  - [7 New Raspberry Pi CM4 Boards for 2021!](https://www.youtube.com/watch?v=w9giSg6ORMk)

elaverick, 2021-01-15:
Hi Jeff,
  Nothing to do with me, but this CM4 carrier is one I'm interested in at the moment (haven't seen it mentioned here yet).
  
  https://twitter.com/ryanteck/status/1343879441248825345?s=20

dronecz, 2021-01-15:
Hi, I made this simple board (design is [here](https://github.com/dronecz/Minimal_carrier_board_for_CM4)) and I have some more designs in PCB fab 🙂  
[image]

dronecz, 2021-01-15:
> Anybody found a minimalistic carrier board with only USB C for instance (data + power), so I can use the CM4 like this? https://www.youtube.com/watch?v=IR6sDcKo3V8 (emulated Ethernet via USB c in order to ssh into the device)
> Thank you

If you need just USB-C, I can make it ;)

Belcarra, 2021-01-15:
What would be required to get Raspberry Pi 4 to support USB 3 speeds with
Gadget? Is that possible with existing chip?

stuart-c, 2021-01-15:
You would need to connect a USB 3 chip to the PCIe port of the CM4 that supports device mode. Unfortunately any form of USB3 chips seem to be quite hard to get hold of (both supply and documentation) unless you are a major manufacturer. Also most such chips are generally focussed on implementing host mode.

If you are asking about the Pi4 rather than the CM4 then it isn't possible. The USB3 ports are host mode only. Only the USB2 port supports device mode.

rdpeake, 2021-01-15:
That MCU zone board looks like it fills a need I have for simple power via USB C (hopefully pd, or at least high current capable) and the minipcie slot - wish I was more capable of following translation or working out if there were similar boards for sale somewhere.

Thanks for this list!

ivdimitro, 2021-01-15:
I see a lot of talented people so here is a free idea - [Argon One V2 ](https://thepihut.com/products/argon-one-raspberry-pi-4-case) compatible board for CM4. The base requirements can be:
- compatible with the cooling "pillars" in the case
- Include M.2 SATA drive. 80 mm will not fit but 20 or 40 mm should be ok as the board will not lose space for the bulki connecters in the middle. 
- POE compatibility
- the GPIO ports are optional ;)

gabyavra, 2021-01-16:
> Hi, I made this simple board (design is [here](https://github.com/dronecz/Minimal_carrier_board_for_CM4)) and I have some more designs in PCB fab 🙂
> [image]

it might be more minimalistic to make it like the CM4lite above, with the CM4 on one side and the ports on the other side. I designed something but it’s not complete, the size is exactly like CM4 so you can overlap them completely.

@dronecz 

[image]
[image]

dronecz, 2021-01-16:
@gabyavra something like this? https://twitter.com/dronecz/status/1333930283192770560?s=20

gabyavra, 2021-01-16:
> @gabyavra something like this? https://twitter.com/dronecz/status/1333930283192770560?s=20

Oh yes!  @dronecz

dans98, 2021-01-16:
> I see a lot of talented people so here is a free idea - [Argon One V2 ](https://thepihut.com/products/argon-one-raspberry-pi-4-case) compatible board for CM4. The base requirements can be:

I think the Argon one case is a bad fit, because it was designed around the need for the breakout board to get all the I/O ports on the back. I think a properly designed carrier board could be made smaller.

 I've been messing with cad for over 20 years, and could do any of mechanical design needed for a carrier board or a case. what I don't have the skill set to do, is the electrical/pcb design. I know just enough to know that i don't know enough to do it right. For example, getting the traces all the right length with the right spacing etc.
 
 
Here is a case I designed to mount a Zero W to an HQ camera mount. It could be made even smaller/thinner if it was printed out of PC instead of PLA, as PC is a more robust material.
[image]

I wrote A custom python app for it as well. It can stream the maximum resolution h264 the encoder supports, to multiple clients simultaneously (even on a Pi Zero). It's only limited by the I/0 ability of the Pi, and your network connection.  Just for fun I added focus peaking into the interface using webgl.

https://github.com/dans98/pi-h264-to-browser

[41 more replies]

closed 2021-02-20 by geerlingguy