geerlingguy/raspberry-pi-pcie-devices #26: Test GPU (EVGA GeForce GTX 750 Ti)

geerlingguy, 2020-11-23:
I just received a used [EVGA GeForce GTX 750 Ti](https://amzn.to/3l2rrXs), which was generously sent to me by [TODO], and I'd like to put it through the gauntlet—plug it in, see if it powers up, see if it is visible with `lspci -vvvv`, compile `nouveau` and see if it `modprobe`s, see if the official Nvidia ARM driver works... etc.

[image]

Highly related: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/2

vegedb, 2020-11-24:
This is awesome, however its more interesting to have Quadro P400 for plex/emby servers. It's the cheapest Quadro for transcoding costing around 50-100 on ebay. I feel that if this works, it will be the ultimate low power plex server.

Benefits,

**Power**
P400 = 30W
GTX 750 TI = 60-75W

**Codecs**
P400 = 6th Gen NVENC
GTX 750 TI = 4th Gen NVENC (Supports X265)

**Screenshots**
_P400_
[image]

_GTX 750 TI_
[image]

_4k Streams_
[image]

**Additional info**
https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new
https://www.elpamsoft.com/?p=Plex-Hardware-Transcoding

geerlingguy, 2020-11-24:
@vegedb - Right now my goal is to get _any_ card working, and some seem to have a better chance than others :D

Once we can prove that some card actually works, then at that point, I'll start thinking about use cases like media transcoding, AI/ML, gaming, etc.

geerlingguy, 2020-12-23:
```
01:00.0 VGA compatible controller: NVIDIA Corporation GM107 [GeForce GTX 750 Ti] (rev a2) (prog-if 00 [VGA controller])
	Subsystem: eVga.com. Corp. GM107 [GeForce GTX 750 Ti]
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Region 0: Memory at 618000000 (32-bit, non-prefetchable) [disabled] [size=16M]
	Region 1: Memory at 600000000 (64-bit, prefetchable) [disabled] [size=256M]
	Region 3: Memory at 610000000 (64-bit, prefetchable) [disabled] [size=32M]
	Region 5: I/O ports at <unassigned> [disabled]
	[virtual] Expansion ROM at 619000000 [disabled] [size=512K]
	Capabilities: [60] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [68] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [78] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 <64us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <512ns, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range AB, TimeoutDis+, LTR+, OBFF Via message
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [250 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [258 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=255us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=262144ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [128 v1] Power Budgeting <?>
	Capabilities: [600 v1] Vendor Specific Information: ID=0001 Rev=1 Len=024 <?>
	Capabilities: [900 v1] #19

01:00.1 Audio device: NVIDIA Corporation Device 0fbc (rev a1)
	Subsystem: eVga.com. Corp. Device 3751
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin B routed to IRQ 0
	Region 0: Memory at 619080000 (32-bit, non-prefetchable) [disabled] [size=16K]
	Capabilities: [60] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [68] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [78] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 <64us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <512ns, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range AB, TimeoutDis+, LTR+, OBFF Via message
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
```

geerlingguy, 2020-12-23:
```
[    1.257396] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.260080] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.262848] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.265605] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0200000000
[    1.314966] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.317945] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.320502] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.323035] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.325717] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.328558] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.334679] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.337626] pci 0000:01:00.0: [10de:1380] type 00 class 0x030000
[    1.340201] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00ffffff]
[    1.342879] pci 0000:01:00.0: reg 0x14: [mem 0x00000000-0x0fffffff 64bit pref]
[    1.345534] pci 0000:01:00.0: reg 0x1c: [mem 0x00000000-0x01ffffff 64bit pref]
[    1.348120] pci 0000:01:00.0: reg 0x24: [io  0x0000-0x007f]
[    1.350701] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0007ffff pref]
[    1.353566] pci 0000:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0000:00:00.0 (capable of 126.016 Gb/s with 8.0 GT/s PCIe x16 link)
[    1.356323] pci 0000:01:00.0: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[    1.359054] pci 0000:01:00.1: [10de:0fbc] type 00 class 0x040300
[    1.361747] pci 0000:01:00.1: reg 0x10: [mem 0x00000000-0x00003fff]
[    1.368030] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.370696] pci 0000:00:00.0: BAR 9: assigned [mem 0x600000000-0x617ffffff 64bit pref]
[    1.373239] pci 0000:00:00.0: BAR 8: assigned [mem 0x618000000-0x6197fffff]
[    1.375813] pci 0000:01:00.0: BAR 1: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    1.378416] pci 0000:01:00.0: BAR 3: assigned [mem 0x610000000-0x611ffffff 64bit pref]
[    1.380948] pci 0000:01:00.0: BAR 0: assigned [mem 0x618000000-0x618ffffff]
[    1.383428] pci 0000:01:00.0: BAR 6: assigned [mem 0x619000000-0x61907ffff pref]
[    1.385888] pci 0000:01:00.1: BAR 0: assigned [mem 0x619080000-0x619083fff]
[    1.388276] pci 0000:01:00.0: BAR 5: no space for [io  size 0x0080]
[    1.390562] pci 0000:01:00.0: BAR 5: failed to assign [io  size 0x0080]
[    1.392998] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.395355] pci 0000:00:00.0:   bridge window [mem 0x618000000-0x6197fffff]
[    1.397746] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x617ffffff 64bit pref]
[    1.400377] pci 0000:01:00.1: D0 power state depends on 0000:01:00.0
```

That new BAR space increase must've landed in the kernel I built, because I didn't have to manually tweak the BAR space anymore!

geerlingguy, 2020-12-23:
Downloading the proprietary Nvidia AARCH64 Driver first: https://www.nvidia.com/en-us/drivers/unix/linux-aarch64-archive/

geerlingguy, 2020-12-23:
Full log with the errors in the build: https://gist.github.com/geerlingguy/33539fd16a1b2ec7cabc6d86d0e75cd9

geerlingguy, 2020-12-23:
Trying a [cross-compile](https://github.com/geerlingguy/raspberry-pi-pcie-devices/tree/master/extras/cross-compile) with the Nouveau driver (inside menuconfig, under Device Drivers > Graphics support).

geerlingguy, 2020-12-23:
```
$ dmesg --follow
[   57.389199] pci 0000:00:00.0: enabling device (0000 -> 0002)
[   57.389235] nouveau 0000:01:00.0: enabling device (0000 -> 0002)
[   57.389369] nouveau 0000:01:00.0: NVIDIA GM107 (117000a2)
[   57.651947] nouveau 0000:01:00.0: bios: version 82.07.55.00.29
[   59.761566] nouveau 0000:01:00.0: fb: 2048 MiB GDDR5
[   59.761591] nouveau 0000:01:00.0: bus: MMIO read of 00000000 FAULT at 3e6684 [ IBUS ]
...
```

geerlingguy, 2020-12-23:
Someone else with that exact same fault: https://bugzilla.kernel.org/show_bug.cgi?id=202731 — but I'm guessing it could be like with the Radeon 5350, where it's actually failing somewhere else.

geerlingguy, 2020-12-23:
Also in https://bugs.launchpad.net/nouveau/+bug/1684123 (migrated to https://gitlab.freedesktop.org/xorg/driver/xf86-video-nouveau/-/issues/335).

geerlingguy, 2020-12-23:
Trying now with a powered external riser. And a note:

  - I tried the PCE164P-NO3 ver 888, and it resulted in a kernel panic (no boot). Tried 5x before giving up.
  - I tried the PCE164P-NO6 ver 008S, and it resulted in kernel panics (no boot). Tried 3x.
  - I tried the PCE164P-NO3 ver 006, and it resulted in kernel panics if I used the USB 3 cable it came with, but if I swapped in the beefier USB 3 cable that came with my ver 888 board, it booted.

Go figure. Cheap junk doesn't work wonderfully. And the 888 board somehow fried my 2.5G network adapter yesterday, wish I had that on video. Lots of smoke!

geerlingguy, 2020-12-23:
With the ver 006 riser, after I run `sudo modprobe nouveau`, I get:

```
[  172.199242] pci 0000:00:00.0: enabling device (0000 -> 0002)
[  172.199279] nouveau 0000:01:00.0: enabling device (0000 -> 0002)
[  172.199411] nouveau 0000:01:00.0: NVIDIA GM107 (117000a2)
[  172.461834] nouveau 0000:01:00.0: bios: version 82.07.55.00.29
```

And then the system freezes completely. Watching with an external display on the console, I do see a full kernel panic. How can I get that output in text form? Here's what it looks like on the diminutive screen I have hooked up:

[image]

6by9, 2020-12-23:
> Watching with an external display on the console, I do see a full kernel panic. How can I get that output in text form?

UART like https://www.amazon.co.uk/Serial-Converter-Adapter-Prolific-Windows-Black/dp/B08DKM6Q63/ref=sr_1_3 on GPIOs 14&15 + GND, and configured as a console (use raspi-config).
It does depend on how quickly the full kernel gets killed as it takes a little while to dribble everything out at 115200baud.

There is also a kernel config option NOUVEAU_DEBUG that can be set with menuconfig or similar. Crank it up before doing your cross-compile to get lots of debug out. That was the next step I was intending to do in my conversation on the nouveau kernel mailing list, but other things became the priority :-/

geerlingguy, 2020-12-23:
@6by9 - Thanks; it seems like that's the long-term route to go, but seeing as I don't have one of those cables handy and have everything plugged in today, I was thinking maybe I could use `netconsole`...

I added the following in `/boot/cmdline.txt`:

```
debug netconsole=6665@10.0.100.119/,6666@10.0.100.142/
```

And I made sure my Mac (142 / target) was running netcat on UDP port 6666:

```
nc -u -l 6666
```

The Pi would boot, but I don't see any output making its way to my Mac. I confirmed if I ran `nc -u 10.0.100.142 6666` on the Pi directly and typed in text, it made it across to my Mac.

Is netconsole not supported on Pi OS?

geerlingguy, 2020-12-23:
Set debug levels to max:

[image]

Pushing the code over to the Pi now...

geerlingguy, 2020-12-23:
Here's the entire dump of NOUVEAU output: https://gist.github.com/geerlingguy/7a021f3fecf198bf7020b85244e772ee

But I suspect it just got cut off at the kernel panic with more output queued up in a buffer like the Radeon did when we were debugging it. It seems like it was in the middle of the devinit process, as the last main section was:

```
[  182.675218] nouveau 0000:01:00.0: sw: preinit running...
[  182.675223] nouveau 0000:01:00.0: sw: preinit completed in 0us
[  182.675233] nouveau 0000:01:00.0: devinit: running init tables
...
```

geerlingguy, 2020-12-24:
Received the [ADAFRUIT Industries 954 USB-to-TTL Serial Cable](https://amzn.to/3rCcCj0) today from Amazon (less than 18 hours after ordering it!).

  1. I installed the [Silicon Labs driver for macOS](https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/software-installation-mac#silabs-cp210x-drivers-2149358-5).
  2. TODO (I'll be updating this with other steps to get it working on the Pi—seems easy enough :)

geerlingguy, 2020-12-25:
Digging around quite a bit, I found little clues in a few places:

  - [aarch64 Kernel Panic Asynchronous SError Interrupt on large file IO](https://lore.kernel.org/linux-arm-kernel/CA+Vb7hqPvDtv0ahjxa_gM68qsws6-dmtiOPmG6-WB+HZEC=4aw@mail.gmail.com/T/#u)
  - [[aarch64] Kernel crash on v5.1-rc5, __arch_copy_from_user+0x1bc/0x240](https://linuxlists.cc/l/5/linux-ext4/t/2983265/(aarch64)_kernel_crash_on_v5.1-rc5,___arch_copy_from_user+0x1bc_0x240)
  - [Kernel crash on large file operations](http://bbs.t-firefly.com/forum.php?mod=viewthread&tid=2718&extra=page%3D1)
  - [Nouveau: kernel hang on Optimus+Intel+NVidia GeForce 1060m](https://lists.freedesktop.org/archives/nouveau/2017-September/028813.html)

Glancing especially at that last result, I tried setting `nouveau.runpm=0` in `/boot/cmdline.txt`, and rebooted, but same result. Seems like `init_rd32` is the final culprit?

geerlingguy, 2021-03-03:
I might also try the proprietary driver once more for fun on the latest precompiled 64-bit kernel, since kernel headers are now available via package and don't have to be built from source.

jamesy0ung, 2021-03-12:
> I might also try the proprietary driver once more for fun on the latest precompiled 64-bit kernel, since kernel headers are now available via package and don't have to be built from source.

Hey how is it going? I have a GT 730 from the same generation and I'm hoping it will work

geerlingguy, 2021-04-21:
So... trying this again today as I _thought_ I'd be wrapping up work on a video for Friday, but as always, things behave differently if you look at them sideways—in this case, I have a camera on it, so it's doing different things than it did late last year.

This time around, since there the BAR space issue was resolved in newer Pi OS kernels, _and_ since the 64-bit Pi OS now has a proper headers package available that can be installed via apt instead of by compiling things by myself, I'm taking another stab at installing Nvidia's proprietary AARCH64 latest driver from https://www.nvidia.com/en-us/drivers/unix/linux-aarch64-archive/

First I flashed my Pi's drive with the 64-bit beta release, then I booted it and ran:

```
sudo apt-get update
sudo apt-get -y dist-upgrade
sudo reboot
```

I copied Nvidia's driver `.run` file to the Pi, then I ran:

```
sudo ./NVIDIA-Linux-aarch64-460.27.04.run
```

TODO... will be posting more details here soon.

jamesy0ung, 2021-04-21:
Nice, thanks Jeff!

geerlingguy, 2021-04-21:
Finding similar reports here: https://github.com/gnab/rtl8812au/issues/92, and here: https://github.com/raspberrypi/linux/issues/3222

Though it could be a variety of things, I'm really thinking the driver is doing something like `memcopy()` and it's breaking similar to how it broke on the RX 550, and on the Broadcom MegaRAID (the latter of which we got fixed/patched to work around the memory addressing limitations on 64-bit Pi OS and a PCIe bug (see: https://github.com/raspberrypi/linux/issues/4158) — though I'm not sure if the PCIe thing is the problem here.

geerlingguy, 2021-04-24:
From [@uruhakomachin on Twitter](https://twitter.com/uruhakomachin/status/1385984062700068868):

> @geerlingguy I had some tries using Nvidia graphics cards with the official arm64 driver. I had the same/quite similar error as you got, RmInitAdapter failed! (0x25:0x54:1262). 
> 
> So I checked the rm_init_adapter function, and it seemed that the driver tried to init the card with some probably x86 code. 0x400000 is the usual base address for 32-bit applications. Given other asm code here, this func seems to save some regs and then load/exec some init code from the graphics card.
>
> Also, this kind of error has been existed for a long time. Probably have to wait for NVIDIA to fix it...
> 
> For now, I think open source drivers have more chances to succeed

[image]

geerlingguy, 2023-11-20:
Going to see what happens with `nouveau` on Pi 5.

The proprietary drivers weren't built for arm64 (only amd64) back when this card was supported...

geerlingguy, 2023-11-20:
```
pi@pi5:~ $ dmesg | grep nouveau
[    3.121503] nouveau 0000:01:00.0: enabling device (0000 -> 0002)
[    3.121599] nouveau 0000:01:00.0: unknown chipset (ffffffff)
```

geerlingguy, 2024-10-30:
Nouveau on Pi 5 today:

```
[    8.449195] nouveau 0000:01:00.0: DRM: VRAM: 2048 MiB
[    8.449204] nouveau 0000:01:00.0: DRM: GART: 1048576 MiB
[    8.449207] nouveau 0000:01:00.0: DRM: TMDS table version 2.0
[    8.449210] nouveau 0000:01:00.0: DRM: DCB version 4.0
[    8.449212] nouveau 0000:01:00.0: DRM: DCB outp 00: 01000f02 00020030
[    8.449215] nouveau 0000:01:00.0: DRM: DCB outp 01: 02000f00 00000000
[    8.449218] nouveau 0000:01:00.0: DRM: DCB outp 02: 08011f82 00020010
[    8.449220] nouveau 0000:01:00.0: DRM: DCB outp 03: 02822fa6 04420010
[    8.449222] nouveau 0000:01:00.0: DRM: DCB outp 04: 02022f62 00020010
[    8.449224] nouveau 0000:01:00.0: DRM: DCB conn 00: 00001030
[    8.449226] nouveau 0000:01:00.0: DRM: DCB conn 01: 00010161
[    8.449227] nouveau 0000:01:00.0: DRM: DCB conn 02: 00002246
[    8.456760] nouveau 0000:01:00.0: DRM: failed to initialise sync subsystem, -22
[    8.457497] Unable to handle kernel NULL pointer dereference at virtual address 0000000000003fc0
[    8.466327] Mem abort info:
[    8.469126]   ESR = 0x0000000096000145
[    8.472883]   EC = 0x25: DABT (current EL), IL = 32 bits
[    8.478221]   SET = 0, FnV = 0
[    8.481281]   EA = 0, S1PTW = 0
[    8.484426]   FSC = 0x05: level 1 translation fault
[    8.489318] Data abort info:
[    8.492201]   ISV = 0, ISS = 0x00000145, ISS2 = 0x00000000
[    8.497703]   CM = 1, WnR = 1, TnD = 0, TagAccess = 0
[    8.502769]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    8.508097] user pgtable: 16k pages, 47-bit VAs, pgdp=0000000107a40000
[    8.514647] [0000000000003fc0] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[    8.523383] Internal error: Oops: 0000000096000145 [#1] PREEMPT SMP
[    8.529671] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device algif_hash algif_skcipher af_alg bnep spidev aes_ce_blk aes_ce_cipher ghash_ce gf128mul sha2_ce sha256_arm64 sha1_ce hci_uart btbcm raspberrypi_hwmon bluetooth hid_apple ecdh_generic ecc libaes brcmfmac_wcc vc4 i2c_brcmstb brcmfmac spi_bcm2835 snd_soc_hdmi_codec joydev brcmutil nouveau(+) cec cfg80211 drm_dma_helper snd_soc_core hid_multitouch gpio_keys drm_exec rpivid_hevc(C) rfkill i2c_algo_bit v4l2_mem2mem pisp_be drm_display_helper drm_ttm_helper ttm snd_compress snd_pcm_dmaengine videobuf2_dma_contig pwm_fan binfmt_misc snd_pcm videobuf2_memops videobuf2_v4l2 videodev drm_kms_helper snd_timer v3d snd videobuf2_common gpu_sched drm_shmem_helper mc raspberrypi_gpiomem nvmem_rmem rp1_adc uio_pdrv_genirq uio drm i2c_dev fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[    8.608028] CPU: 1 PID: 323 Comm: (udev-worker) Tainted: G        WC         6.6.58-v8-16k+ #12
[    8.616761] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    8.622611] pstate: 00400009 (nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    8.629597] pc : dcache_inval_poc+0x28/0x58
[    8.633795] lr : arch_sync_dma_for_cpu+0x34/0x50
[    8.638425] sp : ffffc00080b1b720
[    8.641744] x29: ffffc00080b1b720 x28: ffffd00083c15a78 x27: 0000000000000001
[    8.648906] x26: ffffc00080b1bce0 x25: ffffc00080b1bc90 x24: ffffffffffffffff
[    8.656066] x23: 0000000000000000 x22: ffff80010154a0c0 x21: 0000000000004000
[    8.663228] x20: 0000000000000000 x19: ffffffffffffffff x18: ffffffffffffffff
[    8.670388] x17: 202c6d6574737973 x16: 62757320636e7973 x15: 0000000000000100
[    8.677549] x14: ffff800105ba2800 x13: 022cabcf3a1654ae x12: 0000000000000001
[    8.684710] x11: ffffd00083c80a28 x10: 0000000000000001 x9 : ffffd0008413e04c
[    8.691871] x8 : 0000000000000000 x7 : ffff8001052108e0 x6 : ffffffefffffffff
[    8.699032] x5 : ffff8001052108e0 x4 : 0000000000000000 x3 : 000000000000003f
[    8.706192] x2 : 0000000000000040 x1 : 0000000000003fc0 x0 : ffffffffffffffff
[    8.713354] Call trace:
[    8.715800]  dcache_inval_poc+0x28/0x58
[    8.719645]  dma_unmap_page_attrs+0x1b4/0x1d0
[    8.724014]  nvkm_fb_dtor+0xc0/0x108 [nouveau]
[    8.728597]  nvkm_subdev_del+0x78/0x128 [nouveau]
[    8.733423]  nvkm_device_del+0x80/0x130 [nouveau]
[    8.738249]  nouveau_drm_probe+0x154/0x230 [nouveau]
[    8.743336]  local_pci_probe+0x48/0xb8
[    8.747094]  pci_device_probe+0xac/0x1c8
[    8.751025]  really_probe+0x150/0x2c0
[    8.754695]  __driver_probe_device+0x80/0x140
[    8.759062]  driver_probe_device+0x44/0x170
[    8.763255]  __driver_attach+0x9c/0x1b0
[    8.767098]  bus_for_each_dev+0x80/0xe8
[    8.770944]  driver_attach+0x2c/0x40
[    8.774525]  bus_add_driver+0xec/0x218
[    8.778280]  driver_register+0x68/0x138
[    8.782124]  __pci_register_driver+0x54/0x68
[    8.786404]  nouveau_drm_init+0x214/0x3ff8 [nouveau]
[    8.791491]  do_one_initcall+0x60/0x2c0
[    8.795335]  do_init_module+0x60/0x218
[    8.799094]  load_module+0x1de0/0x2090
[    8.802852]  __do_sys_init_module+0x19c/0x1e0
[... 128 lines cut ...]
[   10.334681] x11: 0000000472885165 x10: 0000000000000002 x9 : ffffd000840faa48
[   10.341842] x8 : ffff8001031aedc0 x7 : d0d0d0d0a2b5a3a5 x6 : 0000000000000004
[   10.349003] x5 : ffff8001031ae7c0 x4 : 0000000000000064 x3 : 0000000000000004
[   10.356163] x2 : 0000000000000001 x1 : 0000000000000000 x0 : ffffc00083853c20
[   10.363327] Call trace:
[   10.365773]  _raw_spin_lock_irqsave+0x3c/0xa0
[   10.370145]  add_wait_queue+0x30/0xb0
[   10.373816]  d_alloc_parallel+0x414/0x468
[   10.377837]  path_openat+0x6bc/0xf18
[   10.381421]  do_filp_open+0xa4/0x160
[   10.385004]  do_sys_openat2+0xc8/0xf8
[... cut ...]
[   10.407561]  el0t_64_sync_handler+0x100/0x130
[   10.411930]  el0t_64_sync+0x190/0x198
[   10.415605] Code: b9000841 d503201f 52800001 52800022 (88e17c02) 
[   10.421721] ---[ end trace 0000000000000000 ]---
[   10.426353] note: wf-panel-pi[1257] exited with irqs disabled
[   10.432166] note: wf-panel-pi[1257] exited with preempt_count 2
```

Rebooting with `/etc/modprobe.d/blacklist-nouveau.conf` in place, and I will try the latest Nvidia driver: https://www.nvidia.com/en-us/drivers/details/233007/

geerlingguy, 2024-10-30:
```
sudo ./NVIDIA-Linux-aarch64-550.127.05.run
```

After booting up:

```
[    6.566235] NVRM: Chipset not recognized (vendor ID 0x14e4, device ID 0x2712)
[    6.566241] The NVIDIA GPU driver for AArch64 has not been qualified on this platform
               and therefore it is not recommended or intended for use in any production
               environment.
[   37.673624] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x24:0x72:1568)
[   37.673683] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[   37.701373] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x24:0x72:1568)
[   37.701435] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[   37.729248] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x24:0x72:1568)
[   37.729307] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[   37.757167] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x24:0x72:1568)
[   37.757209] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
```

It took a while this time to hit the `RmInitAdapter failed!` warning, at least.

geerlingguy, 2024-10-30:
Installing with the open kernel module:

```
sudo ./NVIDIA-Linux-aarch64-550.127.05.run -m=kernel-open
```

Aww...

```
[    6.097203] nvidia-nvlink: Nvlink Core is being initialized, major device number 508

[    6.099277] nvidia 0000:01:00.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[    6.099290] NVRM: The NVIDIA GPU 0000:01:00.0 (PCI ID: 10de:1380)
               NVRM: installed in this system is not supported by open
               NVRM: nvidia.ko because it does not include the required GPU
               NVRM: System Processor (GSP).
               NVRM: Please see the 'Open Linux Kernel Modules' and 'GSP
               NVRM: Firmware' sections in the driver README, available on
               NVRM: the Linux graphics driver download page at
               NVRM: www.nvidia.com.
[    6.099301] nvidia: probe of 0000:01:00.0 failed with error -1
[    6.099319] NVRM: The NVIDIA probe routine failed for 1 device(s).
[    6.099321] NVRM: None of the NVIDIA devices were initialized.
[    6.147912] nvidia-nvlink: Unregistered Nvlink Core, major device number 508
```

qwertychouskie, 2024-11-07:
Did you mean to close this?

geerlingguy, 2024-11-07:
@qwertychouskie - Yes, I don't think we'll get older generation Nvidia cards working on the Pi. Some of the newer generations have a shot, but these older cards are a lot harder to even debug in nouveau.

geerlingguy, 2025-11-27:
Just confirming that with the open GPU kernel driver fix (see https://github.com/NVIDIA/open-gpu-kernel-modules/pull/972), this card is not supported as it doesn't have the GSP firmware in newer cards:

```
[    8.443644] nvidia-nvlink: Nvlink Core is being initialized, major device number 507
[    8.443655] NVRM: The NVIDIA GPU 0001:01:00.0 (PCI ID: 10de:1380)
               NVRM: installed in this system is not supported by open
               NVRM: nvidia.ko because it does not include the required GPU
               NVRM: System Processor (GSP).
               NVRM: Please see the 'Open Linux Kernel Modules' and 'GSP
               NVRM: Firmware' sections in the driver README, available on
               NVRM: the Linux graphics driver download page at
               NVRM: www.nvidia.com.
[    8.449870] nvidia 0001:01:00.0: probe with driver nvidia failed with error -1
[    8.449973] NVRM: The NVIDIA probe routine failed for 1 device(s).
[    8.449974] NVRM: None of the NVIDIA devices were initialized.
[    8.450477] nvidia-nvlink: Unregistered Nvlink Core, major device number 507
```

closed 2025-11-27 by geerlingguy