geerlingguy/raspberry-pi-pcie-devices #28: Coral edgetpu Accelerators with CM4 IO Board

g30ba1, 2020-11-28:
Hi Jeff, I know that only a reduced group of people is using the Raspberry Pi hardware (Specifically the CM4 and the CM4 IO Board) for Machine Learning projects, but it would be interesting if you can verify if the Coral edgetpu Accelerators (mini PCIe and M.2 B+M Key versions) can be used using the PCIe 2 Gen x1 Socket.

I´ve been looking for information about this, but, well, it is not easy to find info about this.

mini PCIe edgetpu accelerator
[https://coral.ai/products/pcie-accelerator/](url)

M.2 accelerator B+M
[https://coral.ai/products/m2-accelerator-bm/](url)

geerlingguy, 2020-11-28:
@g30ba1 - I imagine I could use a [standard PCIe to Mini PCIe card adapter](https://www.amazon.com/Zerone-Express-Wireless-Ethernet-Converter/dp/B07N2X62LQ/ref=sr_1_9?dchild=1&keywords=mini+pcie+to+pcie+adapter&qid=1606604103&sr=8-9) with that first mini PCIe accelerator? To be honest I haven't used any products from Coral before so it would be an overall learning experience, but might be fun.

g30ba1, 2020-11-29:
Checking the Coral edgetpu datasheet:
[https://coral.ai/docs/mini-pcie/datasheet/](url)

I can see the information about the interface:
"Hardware interface: Half-Mini PCIe card
Serial interface: PCIe Gen2 x1"

Being honest, I don´t know what kind of adapters/cables are the best suited for this, but I'm pretty sure that you can figure out the best way to make it.

If you need assistance to get the code necessary to use de Coral edgetpu accelerator (Models, API and the script that runs the inference) do not hesitate to ask for it.

g30ba1, 2020-11-29:
I have found the google-coral/edgetpu issues thread about mini PCIe, maybe you can find useful information on it:

[https://github.com/google-coral/edgetpu/issues/232](url)

Looks like other people has been using the adapter you mentioned earlier:

"@Namburger I got it working with the adapter you just posted, the Ableconn PEX-MP117 Mini PCI-E to PCI-E Adapter Card. Boot, sleep, restart etc work fine. I tested the edgetpu with the get started example and it worked. After following the setup instructions, I was able to get the reductions in inference time on the TPU for the test model. I am now convinced it was the adapter."

markus-k, 2020-11-29:
I have two EdgeTPUs with m.2 (A+E and B+M) on order. I will report back once they and the CM4 are here. Could be interesting to use multiple accelerators with a PCIe switch.

g30ba1, 2020-11-29:
> I have two EdgeTPUs with m.2 (A+E and B+M) on order. I will report back once they and the CM4 are here. Could be interesting to use multiple accelerators with a PCIe switch.

I can't wait for that report 😅!

I'm planning using several EdgeTPUs at the same time too.

I've been reading about the PyCoral API, and looks like you will be able to use those EdgeTPUs for re-training models too!

Namburger, 2020-11-30:
Hey guys, sorry for delivering the bad new, but the pi4's cpu doesn't support MSI-X as [required by the google apex driver](https://coral.ai/docs/m2/get-started/#requirements) :/

markus-k, 2020-11-30:
Yes I saw that before too. I was hoping may be there is some way around that. Performance over USB2 is quite bad sadly, and there aren't any modules supporting USB3 besides the Coral Stick, which is not really suited for my application.

elFarto, 2020-11-30:
There is a comment in the PCIe driver source code about this (I think):
```
/* Multi MSI is supported by the controller, but not by this driver */
```
It could just be the case the didn't implement it since the USB3 device didn't need it. However getting that implemented, or the documentation needed to implement it is likely impossible.

closed 2020-12-01 by geerlingguy