geerlingguy/raspberry-pi-pcie-devices #30: Test SATA adapter (4-port 2-active ASM1061 chipset)

geerlingguy, 2020-12-07:
This card is a little bit funky...

[image]

It has two internal SATA ports on top, and two external eSATA connectors on back. It seems like, according to the jumper readout, you can switch two interfaces from the ASM1061 chipset (which only supports 2 SATA-III devices over PCIe 1x 2.0) between either the internal or external ports.

Here's a YouTube video on this adapter (or at least one of the generic revisions of it): [Detailed Look At A $6 ASM1061 SATA HBA From eBay](https://www.youtube.com/watch?v=yqQPh8KZhp4). It's basically a super-cheap 2-drive SATA-III controller.

Special thanks to Jacob Hiltz for sending this card to me! See #16 for details about that.

dlednik, 2020-12-31:
@geerlingguy  Any plans for ASM1061R which supports RAID ?

geerlingguy, 2021-01-01:
@dlednik if someone can send me one, maybe. But I'm a bit backlogged testing all these storage adapters as it is!

MrThanlon, 2021-01-30:
@geerlingguy I tested this card and it worked fine. The chip is ASM1062. But we need to recompile kernel with SATA driver.
```
 pi@raspberrypi  ~  sudo lspci   
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 01)
 pi@raspberrypi  ~  sudo fdisk -l /dev/sda
Disk /dev/sda: 55.9 GiB, 60022480896 bytes, 117231408 sectors
Disk model: GLOWAY FER60GS3-
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
 pi@raspberrypi  ~  sudo smartctl -a /dev/sda
smartctl 6.6 2017-11-05 r4594 [armv7l-linux-5.4.83-v7l+] (local build)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Device Model:     GLOWAY FER60GS3-S7
Serial Number:    DA3D05C020ACC0010047
LU WWN Device Id: 5 000000 000000000
Firmware Version: F1AC20
User Capacity:    60,022,480,896 bytes [60.0 GB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Device is:        Not in smartctl database [for details use: -P showall]
ATA Version is:   ACS-2 (minor revision not indicated)
SATA Version is:  SATA 3.1, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Sat Jan 30 12:39:45 2021 GMT
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x00)	Offline data collection activity
					was never started.
					Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0)	The previous self-test routine completed
					without error or no self-test has ever 
					been run.
Total time to complete Offline 
data collection: 		(    0) seconds.
Offline data collection
capabilities: 			 (0x71) SMART execute Offline immediate.
					No Auto Offline data collection support.
					Suspend Offline collection upon new
					command.
					No Offline surface scan supported.
					Self-test supported.
					Conveyance Self-test supported.
					Selective Self-test supported.
SMART capabilities:            (0x0002)	Does not save SMART data before
					entering power-saving mode.
					Supports SMART auto save timer.
Error logging capability:        (0x01)	Error logging supported.
					General Purpose Logging supported.
Short self-test routine 
recommended polling time: 	 (   1) minutes.
Extended self-test routine
recommended polling time: 	 (   1) minutes.
Conveyance self-test routine
recommended polling time: 	 (   1) minutes.
SCT capabilities: 	       (0x003d)	SCT Status supported.
					SCT Error Recovery Control supported.
					SCT Feature Control supported.
					SCT Data Table supported.

SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
[... 25 lines cut ...]
199 UDMA_CRC_Error_Count    0x0000   100   100   050    Old_age   Offline      -       0
232 Available_Reservd_Space 0x0000   100   100   000    Old_age   Offline      -       98
241 Total_LBAs_Written      0x0000   100   100   000    Old_age   Offline      -       52988
242 Total_LBAs_Read         0x0000   100   100   000    Old_age   Offline      -       70158
245 Unknown_Attribute       0x0000   100   100   000    Old_age   Offline      -       114652

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
No self-tests have been logged.  [To run self-tests, use: smartctl -t]

SMART Selective self-test log data structure revision number 1
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
    6        0    65535  Read_scanning was never started
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.

```

mi-hol, 2021-01-31:
@MrThanlon as several different chipsets are mention in this issue, could you please clarify what card you are referencing with "I tested this card" ?

MrThanlon, 2021-02-07:
> @MrThanlon as several cards with different chipsets are mention in this issue, could you please clarify what card you are referencing with "I tested this card" ?

The one on the picture.

Geofferic, 2021-02-07:
> @MrThanlon as several cards with different chipsets are mention in this issue, could you please clarify what card you are referencing with "I tested this card" ?

I assume he means the Diyeeni PCI-E Express to 4-Port Sata/ESATA Riser Adapter Card.

ThomasKaiser, 2021-02-17:
> The chip is ASM1062 ... The one on the picture

On the picture there's an ASM1061 but it doesn't really matter whether it's ASM1061 or ASM1062 since those are software-wise the same, the latter only could establish an x2 link but that doesn't matter anyway with the single lane CM4.

Linux will report an ASM1061 sometimes as ASM1062 and vice versa (depending on version of `pciutils`), see this [nice example](https://forum.frank-mankel.org/topic/418/rockpro64-4gb-probleme-mit-der-pcie-sata-karte) where booting an RockPro64 with two different OS images results in `lspci -vvv` outputting with Ubuntu this

    01:00.0 IDE interface: ASMedia Technology Inc. ASM1061 SATA IDE Controller (rev 02) (prog-if 85 [Master SecO PriO])
	Subsystem: ASMedia Technology Inc. ASM1061 SATA IDE Controller

...and directly afterwards booting Debian Stretch that:

    01:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 02) (prog-if 01 [AHCI 1.0])
	Subsystem: ASMedia Technology Inc. ASM1062 Serial ATA Controller

The card tested there is the same as the one shown in first post of this issue.

Anyway: both ASM1061 and ASM1062 work fine with Linux on ARM but users should be aware that adding SATA port multipliers to the SATA ports [is not the best idea](https://forum.odroid.com/viewtopic.php?p=309103#p309103).

mynameisflorian, 2021-04-17:
I got a similar card working. I haven't done any detailed testing, but my card seems to work with the kernel recompiled as specified below. I will update here if I run into any problems.

Instructions Used:  https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/1#issuecomment-717578358
Model:  MHQJRH PCI Express SATA 3.0 Controller Card
Chipset:  ASM1061 
Amazon Link:  https://www.amazon.com/dp/B07M942LN9

gev, 2021-06-11:
Hi, guys!

I wana to develop own pci/sata bridge on this chip (asm 1061/2). Have anybody an expirience?

clarkalastair, 2024-04-28:
Adding a comment here for documentation's sake. 

Just plugged in a couple disks into my turing pi 2 board populated with CM4s. Looks like the version of the board I have is using ASM1061, and the bridge doesn't seem happy: 

```
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 IDE interface: ASMedia Technology Inc. ASM1061 SATA IDE Controller (rev 02)
```

```
Apr 28 22:45:23 <redacted> kernel: pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00001000/00002000
Apr 28 22:45:23 <redacted> systemd-journald[11767]: Missed 100 kernel messages
```

```
Apr 28 22:45:23 <redacted> kernel: pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
Apr 28 22:45:23 <redacted> kernel: pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
Apr 28 22:45:23 <redacted> kernel: pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00001000/00002000
Apr 28 22:45:23 <redacted> kernel: pcieport 0000:00:00.0:    [12] Timeout
```

Planning to try to understand this a bit more, but throught I'd share, and that I should be able to do some testing with this chipset if desired.