geerlingguy/raspberry-pi-pcie-devices #31: Intel SASUC8I / LSI SAS3081E-R LSI 1068e RAID controller

geerlingguy, 2020-12-07:
I received an [Intel SASUC8I and LSI SAS3081E-R RAID controller](https://amzn.to/3qydoNk) from @kientran — it is similar to the SAS3082E-R I've been testing (see https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/18), but there are some differences. Here's a pretty picture:

[image]

And see this article from servethehome: [Intel SASUC8I and LSI SAS3081E-R LSI 1068e Based RAID Controller Review](https://www.servethehome.com/intel-sasuc8i-lsi-lsi-sas3081er-lsi-1068e-based-raid-controller-review/).

geerlingguy, 2020-12-07:
Just noting I'm seeing in some reviews this card only supports disks up to 2.2 TB ('in the latest firmware') FWIW. How could people in pre-2010 not design cards with 10PB drives in mind!?

kientran, 2020-12-08:
This card had a previous life in a FreeBSD ZFS system and was flashed to the IT firmware for JBOD disk access.  I could have sworn I used it with my 4TB drives, but that was at least 2 file servers ago so who knows  ¯\_(ツ)_/¯

geerlingguy, 2020-12-08:
@kientran - Thanks for the extra info. If it's already flashed to IT, I am hopeful I might be able to have more luck getting it to start up in the Pi than the other two. I now have a separate Intel PC I can use to re-flash cards but I am lacking the time to get that done as I'm trying to focus on a couple other things before end-of-year.

But I do want to get at least one HBA working before 2021! This could be the one...

geerlingguy, 2020-12-30:
Heh... looks like I won't get it tested before 2021, but soon after the year turns over ;)

BryanKadzban, 2024-04-19:
I have one of these on the way here, along with an rpi 5, so I should be able to test things out once both of those arrive. I plan to use it in a NAS with an Intel RES2SV240 SAS expander, and about ten or so SAS drives (300G each, 15k RPM). Assuming it works of course.

Shipping is supposed to take like a week. But I wanted to write down what I was planning.

BryanKadzban, 2024-04-27:
So, it registers in the PCIe subsystem, but in a standard Raspbian environment (as flashed onto an SD card by pishop.us) has no driver.  :-/

```
$ sudo lspci -vvv
0000:01:00.0 SCSI storage controller: Broadcom / LSI SAS1068E PCI-Express Fusion-MPT SAS (rev 08)
        Subsystem: Broadcom / LSI SAS3081E-R 8-Port SAS/SATA Host Bus Adapter
        Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 0
        Region 1: Memory at 1b00210000 (64-bit, non-prefetchable) [disabled] [size=16K]
        Region 3: Memory at 1b00200000 (64-bit, non-prefetchable) [disabled] [size=64K]
        Expansion ROM at 1b00000000 [virtual] [disabled] [size=2M]
        Capabilities: [50] Power Management version 2
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [68] Express (v1) Endpoint, MSI 00
                DevCap: MaxPayload 4096 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
                        ExtTag+ AttnBtn- AttnInd- PwrInd- RBE- FLReset- SlotPowerLimit 0W
                DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 512 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 2.5GT/s, Width x8, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1 (downgraded)
                        TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
        Capabilities: [98] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [b0] MSI-X: Enable- Count=1 Masked-
                Vector table: BAR=1 offset=00002000
                PBA: BAR=1 offset=00003000
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000

$ cat /sys/devices/platform/axi/*.pcie/pci*:*/*/0000:01:00.0/{vendor,device,subsystem_vendor,subsystem_device}
0x1000
0x0058
0x1000
0x3140
```

0x1000 is Broadcom, and 0058 is the SAS1068e, so this is the right sysfs directory ... unfortunately:

```
$ ls -l /sys/devices/platform/axi/*.pcie/pci*:*/*/0000:01:00.0/driver
ls: cannot access '/sys/devices/platform/axi/*.pcie/pci*:*/*/0000:01:00.0/driver': No such file or directory
```

so there's no driver bound, and I therefore assume there's nothing compiled (...because modprobe didn't find it), that matches this vid/pid pair.

I see several references to an mptsas driver (and in fact, I found a driver download for this device, from supermicro of all places, which generates an mptsas module with a vendor/device alias that matches here).  I also think I've seen mptsas related drivers in the kernel configuration tree many years ago.  Maybe I just have to turn one on.  I think I've successfully cloned the raspbian kernel repository; looking at the best way to rebuild this.  I've built a ton of kernels for x86, but none on a Debian system, and none on an aarch64 system.  This is going to be fun!  :)

BryanKadzban, 2024-04-27:
OK, yeah.  I found the linux-config-6.6 package, with a contained -2712 config file; decompressing it and doing a "make oldconfig", then trawling through menuconfig, finds that the Device Drivers -> Fusion MPT section is turned off.  Turning it on provides a driver for SPI devices and one for SAS; the latter matches this card (according to the driver's pci_device table).

I also see a separate ioctl driver, which the help says can be used to flash firmware, so I'm enabling it as well.  (I doubt this device is in IT mode.  We'll see.)

I'll find out how long a pi5 takes to build a kernel now, with -j4.  :)

BryanKadzban, 2024-04-27:
Takes at least an hour.  :)

At boot the driver loaded automatically, and mptsas got bound to the device in lspci.  The connected drives did not show up ... but that turned out to be because I didn't have the SAS cable (the SFF-8087 end, at the LSI card) plugged in all the way.  It hotplugged, spun up, and got recognized as /dev/sdb as soon as I clicked it all the way in.

With the mptctl module loaded, `mpt-status -p` doesn't find anything, which is odd, but maybe that's not a huge deal considering the drive does seem to work.  Just have to figure out how to monitor for failed drives now; almost all the way there.

I did not flash into IT mode, but I don't know how to tell for sure if the card is already in that mode either (I *think* this was new, but not 100% sure).

wcarterjr68, 2024-08-02:
0000:04:00.0 Serial Attached SCSI controller: Broadcom / LSI SAS2008 PCI-Express Fusion-MPT SAS-2 [Falcon] (rev 03)
	Subsystem: IBM SAS2008 PCI-Express Fusion-MPT SAS-2 [Falcon]
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Region 1: Memory at 1b001c0000 (64-bit, non-prefetchable) [disabled] [size=16K]
	Region 3: Memory at 1b00180000 (64-bit, non-prefetchable) [disabled] [size=256K]
	Expansion ROM at 1b00100000 [virtual] [disabled] [size=512K]
	Capabilities: [50] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [68] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 4096 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x8, ASPM L0s, Exit Latency L0s <64ns
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range BC, TimeoutDis+ NROPrPrP- LTR-
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkR

wcarterjr68@vault:~ $ ls -l /sys/devices/platform/axi/*.pcie/pci*:*/*/0000:01:00.0/driver
lrwxrwxrwx 1 root root 0 Aug  1 21:44 /sys/devices/platform/axi/1000110000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/driver -> ../../../../../../../bus/pci/drivers/pcieport

This looks like the driver is loaded, I am not seeing the drives.  I have 4 of the 8 drives connected.  The card I bought is supposed to be in IT mode from the seller.

What do I need to try now.  Ron Carter

wcarterjr68, 2024-08-02:
After working looking at this it looks like I need to recompile the kernal.  What is the best process for completing this process.?

mpt-statusd.service - LSB: Check mpt-status values in the background.
     Loaded: loaded (/etc/init.d/mpt-statusd; generated)
     Active: failed (Result: exit-code) since Thu 2024-08-01 23:13:28 EDT; 2min 18s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 723 ExecStart=/etc/init.d/mpt-statusd start (code=exited, status=1/FAILURE)
        CPU: 19ms

Aug 01 23:13:28 vault systemd[1]: Starting mpt-statusd.service - LSB: Check mpt-status values in the background....
Aug 01 23:13:28 vault mpt-statusd[723]: Starting mpt-status monitor: failed to load mptctl. ... failed!
Aug 01 23:13:28 vault systemd[1]: mpt-statusd.service: Control process exited, code=exited, status=1/FAILURE
Aug 01 23:13:28 vault systemd[1]: mpt-statusd.service: Failed with result 'exit-code'.
Aug 01 23:13:28 vault systemd[1]: Failed to start mpt-statusd.service - LSB: Check mpt-status values in the background..
~

BryanKadzban, 2024-08-14:
The process I followed was here:

https://www.raspberrypi.com/documentation/computers/linux_kernel.html

but it of course doesn't say anything about which driver to enable for this hardware.  To get that, after generating the proper _defconfig for your RPi hardware (see the link), rerun the kernel configuration and enable the needed drivers.  In menuconfig, for example, under "Device Drivers" there's a "Fusion MPT" section that needs to be enabled, then the SAS driver under there needs to be turned on.  (I set it as a module.)