geerlingguy/raspberry-pi-pcie-devices #32: Test NVMe Drives and PCIe Adapters (Samsung 970 EVO Plus, NGFF, etc.)

geerlingguy, 2020-12-12:
I realized I tested some of these devices for my original review video, but I really didn't document things well in an issue on this project (which I started _after_ I finished up that video):

### NVMe Drives

  - [Samsung 970 EVO Plus M.2 NVMe](https://amzn.to/348ArEN)
  - [XPG SX6000 Pro PCIe Gen3x4 M.2 2280 NVMe](https://amzn.to/3oLUkcS)
  - [Kingston A400 SSD M.2 2280 SA400M8 B&M Key](https://amzn.to/37azZIc)

### PCIe NVMe M.2 adapters

  - [NGFF M.2 NVMe M Key SSD to PCIe 1x adapter](https://amzn.to/2Lx69VV)
  - [Xiwai NGFF M Key NVMe AHCI SSD to PCIe 1x adapter](https://amzn.to/2JPArCY)
  - [MZHOU NVME M.2 SSD M Key to PCIe x1 adapter](https://amzn.to/3mdiuv5)

I'll try to document my experiences and some benchmarks with all the drives here. I think I left some notes on the Samsung 970 EVO Plus, at least, elsewhere.

The other thing to test is how each drive (if it works as direct NVMe storage at least) compares in performance via a USB 3.0 adapter (like the [KBR M.2 NVMe USB 3.1 SSD Adapter](https://amzn.to/3oN4uty) compared to directly attached. I know the 970 was a lot faster for random IO.

ScottHolden, 2020-12-14:
As a note for people just receiving their CM4's, I had no luck getting the Ubuntu img (`ubuntu-20.10-preinstalled-server-arm64+raspi`) to boot with a Samsung 970 EVO attached via a generic vertical M.2 to PCIe x1 adapter, but when I switched to RasPiOS (`2020-12-02-raspios-buster-armhf-lite`) it booted without issue!

I also found it was handy to add `nvme-core' to `/etc/modules` to make sure it is loaded at boot time.

(Let me know if this is the wrong place for community submitted notes, happy to place elsewhere 😃 )

geerlingguy, 2020-12-14:
@ScottHolden - Were you able to boot directly from the 970 EVO? Or did you still do the initial boot from the microSD card?

jblanz, 2021-02-13:
@ScottHolden - Did you find out what is blocking ubuntu from booting when the Samsung 970 EVO is attached? Did you manage to boot ubuntu while some other SSD model is attached? I assume you are booting off of an SD-Card or eMMC, correct? Would be very interested to see which SSD can be used in combination with ubuntu.

THX
JOsef

starchivore, 2021-03-05:
CM4 and PCIe with Ubuntu
[https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=301016](https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=301016)
> That specific issue was resolved the downstream 5.4 Raspberry Pi OS Kernel
> [https://github.com/raspberrypi/linux/commit/20f5f742eea1057b8cc3d368658bcd6fb54b90db](https://github.com/raspberrypi/linux/commit/20f5f742eea1057b8cc3d368658bcd6fb54b90db)

kylefmohr, 2022-12-15:
Just wanna add that I purchased the Xiwai M.2 adapter via your Amazon link and it works perfectly with Kali for Linux, no setup necessary. Benchmark screenshot attached. Thanks!
[image]