geerlingguy/raspberry-pi-pcie-devices #33: Gpu on pi

groot4272, 2020-12-15:
Please once try gpu once again on windows on cm4
As drivers are available and recently microsoft also announced x64 emulation on arm

PixlRainbow, 2020-12-15:
He never stopped testing GPUs. It's a work in progress; see [these issues](/geerlingguy/raspberry-pi-pcie-devices/issues?q=is%3Aissue+is%3Aopen+label%3A%22currently+testing%22+GPU). He just hasn't gotten around to uploading any GPU videos yet since he hasn't got enough content to fill a video.

There are also other reasons he isn't testing with Windows on ARM at the moment, see #7

geerlingguy, 2020-12-15:
Yes; thanks for opening the issue, and please follow those other issues for updates. Just installing Windows on the Pi (which isn't officially supported by Microsoft) is a project unto itself, and I don't think it will be the magic bullet here, as there are many other layers to unpack first.

closed 2020-12-15 by geerlingguy