geerlingguy/raspberry-pi-pcie-devices #36: [Presentation] How to make benchmark data better visible & accessible?

mi-hol, 2020-12-20:
Hi Jeff,
you're doing great things here.
The [OpenMediaVault NAS builder community](https://forum.openmediavault.org/index.php?thread/19871-which-energy-efficient-arm-platform-to-choose/&postID=259048#post259048) would greatly benefit from having better access/visibility on SATA/NVMe benchmark results. Do you have plans to publish somehow already?
A [benchmark site for web site dev tools I use](https://www.techempower.com/benchmarks/#section=data-r19&hw=ph&test=fortune) shows what I'm looking for

geerlingguy, 2020-12-24:
@mi-hol - Thanks! I would like to help add some benchmark results—right now I've just had them kinda jumbled around in Numbers files on my local computer (and sometimes in Google Sheets), and then I produce graphs for my videos / blog posts.

At some point I would like to be a little more formal about storing the results somewhere more permanent though...

mi-hol, 2021-01-03:
I
I like the visualization of your benchmarks any pointer show how you create them maybe in an automated approach?

Comparing performance results of individually build systems is attractive to builders (i.e. Gamers) haven't found such a site for NAS builders yet.
As a starting point I found a pwsh script automating data collection providing results as .csv and .xml file via [this blog post](https://blog.yavilevich.com/2015/10/diskspdauto-automating-io-performance-tests-and-result-collection-using-diskspd-and-powershell/)
I've forked it and am working to iron out some glitches
What is you view on this opportunity?

mi-hol, 2023-08-19:
closing due to lack of interest

closed 2023-08-19 by mi-hol