geerlingguy/raspberry-pi-pcie-devices #37: Have you tried to use a PCIe adaptors for M2 wifi boards?

Shokman, 2020-12-20:
One question, have you tried to use [these adaptors](https://www.amazon.co.uk/gp/product/B088NBPWGP/ref=ox_sc_act_title_1?smid=A21CTMDLFXECIV&psc=1) with a M2 2230 wifi device? It might be helpful to understand any drivers issues that you are having with the other boards.

Maybe it is a straight forward "this won't work". The problem is that I do not know why it should not work. If that is the case please feel free to close the issue.

Coreforge, 2020-12-21:
I don't see why they shouldn't work as long as the drivers work. Depending on the card, it might need PCIe or USB, which the official IO board has a header for. Wifi drivers can just generally be fun on linux in my experiance.

geerlingguy, 2020-12-21:
Heh... speaking of which, have you seen https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/22 — that's using the AX200 chip directly, but I also tried with a Realtek 802.11ac chip and was having a rough time getting it working.

I'm actually working (right now, literally!) on a video documenting my long process of getting the AX200 working with the Pi.

@Coreforge is correct in mentioning WiFi drivers being 'fun' in Linux :P

Any M2 WiFi card should work with the appropriate PCIe to M2 slot adapter, and I may try another AX200 card I just got recently. This all assumes you have chosen a wireless chipset with good (or great) wireless support in the kernel. Check https://wireless.wiki.kernel.org for more details.

closed 2020-12-21 by geerlingguy

Shokman, 2020-12-21:
@geerlingguy Oh yes, I saw them both, that is why I thought we might share interest on working with these adaptors, where you can work with any wifi module used in laptops. The boards you choose before have PCIe connectors directly, but I thought you might find better supported boards on linux using the M2 connector. To be honest I do not know if they add something compared with the wifi module on the CM4, but at least you can verify that it is possible so use a wifi board through PCIe on the CM4 and which drivers works best.

In my case I am looking for a sbc where I can use a particular wifi module from intel, which uses M2. And I was wondering if the raspberry pi 4 compute module with the IoT board is an option for me or not. I already have to deal with a patched module so I wanted to be sure that everything works on the connector side...

Anyway, I will be looking forward for the video and congratulations for the work you are doing, best of lucks!

geerlingguy, 2020-12-21:
@Shokman - Thanks! I hope the video will be helpful.

geerlingguy, 2020-12-21:
Also, going to test this: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/38