geerlingguy/raspberry-pi-pcie-devices #39: Optimize kernel compile time in cross-compile environment

geerlingguy, 2020-12-22:
I think there should be a new maxim in computing:

> If you want to figure out a better way to do something, share what you're currently doing in a YouTube video, and at least ten comments will provide suggestions for improvement.

A number of people have mentioned the `make` command to compile the kernel could be a lot faster with a couple tweaks—most notably, using `-j` instead of `-j4` to allow the compiler to use as many jobs as it wants.

I would like to try a few different tweaks to see if I can cut the build times down on my MacBook Pro.

geerlingguy, 2020-12-22:
First attempt: I dropped the job count (using `-j`), and increased the VirtualBox `vb.cpus` to `6`, and let it rip.

Things got... shaky.

A lot of things would compile, and the VM was using ~570% CPU the whole time, but I noticed a lot of compile warnings and errors, like:

```
  CC [M]  drivers/media/common/siano/smscoreapi.o
  CC      drivers/mfd/arizona-core.o
compilation terminated.
compilation terminated.
compilation terminated.
  CC [M]  net/ipv6/netfilter/nf_conntrack_reasm.o
  AR      drivers/iio/resolver/built-in.a
aarch64-linux-gnu-gcc: fatal error: Killed signal terminated program cc1
compilation terminated.
aarch64-linux-gnu-gcc: fatal error: Killed signal terminated program cc1
aarch64-linux-gnu-gcc: fatal error: Killed signal terminated program cc1
aarch64-linux-gnu-gcc: fatal error: Killed signal terminated program cc1
aarch64-linux-gnu-gcc: fatal error: Killed signal terminated program cc1
compilation terminated.
compilation terminated.
compilation terminated.
```

geerlingguy, 2020-12-22:
From [this comment by norse-dougbert](https://www.youtube.com/watch?v=csI19aOJEik&lc=Ugw-FFlUwLZUPmnH-LR4AaABAg.9HY6ELgVFWF9HYc9jLV42d):

> Don't know how many cpus on your build machine, but after years of kernel building my `make -j $PARALLEL`  parameter became 1.6 times actual cpus, hence  8 cores became  `-j 12` or `13`.  Of course drives are much faster now then back then

So going to try `-j10` on another build after resetting the environment.

geerlingguy, 2020-12-22:
Results:

```
$ time make -j10 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs
...
real	10m12.872s
user	55m31.871s
sys	5m29.739s
```

CPU load was around 580% from VBoxHeadless, laptop got... very nice and toasty. It was on my lap, probably should not do that next time!

But 10 minutes! My Mac is running faster than the Focus was now—though I didn't optimize the runs on the Focus either. I'm going to crank it up, giving the VM 8 cores and going to 13 jobs!

geerlingguy, 2020-12-22:
Whee! That got the laptop cranking... 94°C and ~750% avg CPU usage. I'm also going to add an optimization to the vagrantfile to automatically set half the logical cores on the machine.

I might go a little insane and try consuming more. Bad idea? Maybe. For science? It must be done.

geerlingguy, 2020-12-22:
Now trying with 14 logical cores and:

```
make -j22 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs
```

Heh... CPU up to ~1080%, temp went straight up to 94°C again within 30s. Let's see how fast this thing will run, though!

geerlingguy, 2020-12-22:
Looks like diminishing returns (or even slower process) once we pass up 20 concurrent jobs. Trying again with 16 jobs.

geerlingguy, 2020-12-22:
Back to 8 CPU cores (`logical CPU count / 2`), and `-j12`:

```
$ time make -j12 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs
...
TODO
```

I think I'll push a commit with these changes and move on from there!

closed 2020-12-22 by geerlingguy

PixlRainbow, 2020-12-22:
Wonder if it'll be any faster in native AArch64 compile on the new Apple Silicon Macs, not that I'd recommend buying them considering a 10 minute compile time on your current one is already pretty good.

6by9, 2020-12-22:
As a follow-on, rather than having to mess with ejecting SD cards, mounting them, copying, unmounting, and putting the SD card back in.
 
- do the modules_install to a local directory
- enable SSH for root on the Pi (```sudo nano /etc/ssh/sshd_config``` and change ```PermitRootLogin yes```)
- reboot or restart sshd
- copy your SSH public key to /root/.ssh/authorized_keys so it doesn't prompt for a password
- use ```scp -r``` or rsync to copy all the files across your network 

It's also far easier to do it that way on a CM with onboard EMMC than messing with rpiboot to mount the EMMC as a USB storage device.

geerlingguy, 2020-12-23:
@6by9 - That's almost exactly what I'm doing now: https://github.com/geerlingguy/raspberry-pi-pcie-devices/tree/master/extras/cross-compile#copying-built-kernel-via-remote-sshfs-filesystem

It was especially important as I rearranged my desk and now it's almost impossible to get to my card reader (I had to order a new one with a long cable so I can put it back on top of my desk)... much easier to not have to swap cards!

benleb, 2021-01-07:
Did you think about putting the whole builddir/container/vm in a tmpfs? 🚀😄