geerlingguy/raspberry-pi-pcie-devices #41: Make website way more awesome-r

geerlingguy, 2020-12-22:
It's not much of a database with me having to throw a ton of data into one big web page.

I would like to have every card be a 'post' (or custom page) in Jekyll, and then aggregate those posts on the home page, using frontmatter as 'database storage fields'.

Yeah, it'd be better to use a real database and stuff, but for now I'm gonna hack it to make it work. I've been playing around with layouts and I'm happy with what I've come up with, so the first stage is to get all the boards broken out into their own pages.

DiHydro, 2020-12-24:
Can you host it as a wiki? Something like the OpenWRT or https://wikidevi.wi-cat.ru/Main_Page. Basically what most people are about are what device exactly is it, and how can they get it, and then how can you get it to work. I know GitHub even has wiki pages for some projects.

geerlingguy, 2020-12-24:
@DiHydro - A Wiki is (IMO) a bit harder to maintain in a more database-like fashion.

Jekyll is definitely far from a database-style site too, but at least I can use the frontmatter fields to define a standard set of data that I template onto the main page.

If things grow beyond a few dozen cards, I'd likely move to another structure entirely (maybe SQLite or something... still love having a static site as the end result though!).

geerlingguy, 2020-12-25:
One last thing—I think I might try to figure out a way to preprocess all the card pictures to make thumbnail versions during the `jekyll build` process... ideally I don't want to have to re-process every image for the home page, but I also don't want to make users download the full size image for every card on the home page (currently that's a 3.5 MB load!).

geerlingguy, 2020-12-25:
Mmm, [smartly cropped thumbnails](https://kalifi.org/2015/08/smart_post_thumbnails.html) with a custom plugin. Will require me to move the Jekyll build process out to GitHub Actions, most likely though.

Everyone else seems to pre-generate thumbnail images (e.g. by hand, or using Python scripts).

geerlingguy, 2020-12-26:
That was... a lot more complicated than I was hoping, but I think I have things working now.

commit d69e2e12 (2020-12-26): Fixes #41: Fix cache directory for bundle deps.
closed 2020-12-26 by geerlingguy