geerlingguy/raspberry-pi-pcie-devices #44: Test Coral M.2 Accelerator A+E key

geerlingguy, 2020-12-24:
I just bought a [Coral M.2 Accelerator A+E key](https://coral.ai/products/m2-accelerator-ae/) after seeing a lot of buzz about this little 'IoT' TensorFlow-compatible AI accelerator.

TODO: Picture goes here.

I also just received an M.2 A key to PCIe 1x slot adapter card (see https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/38), and I will pop the Coral board into there.

I haven't done much with AI/ML, but apparently one of the big holdups for using this board with the Pi may be overcome soon—see: https://github.com/google-coral/edgetpu/issues/280

(And I don't want to step on @timonsku's toes here either—he was the original inspiration for me getting this particular card, after seeing his [Piunora](https://twitter.com/timonsku/status/1337134272625467392); just figured now is as good a time as ever to take a quick stab at TensorFlow.)

timonsku, 2020-12-24:
No worries,  glad to see more people having an interest in seeing this working :)

Valdiolus, 2020-12-29:
Great, looking forward to solve this issue in my project!

geerlingguy, 2021-01-24:
It has arrived! (Picture uploaded).

I also got a couple other goodies today, though, so I'm going to have to wait to start testing it at least a couple days :( (ah, if only time were infinite!).

Geofferic, 2021-01-24:
Surely time is infinite in Catholic canon. ;)

geerlingguy, 2021-01-24:
@Geofferic - Haha, well yes... but I'm imagining if I can make it to the point where it is indeed infinite—I don't think testing a Coral TPU is going to be my highest priority 😆

geerlingguy, 2021-02-17:
```
$ sudo lspci -vvvv -d 1ac1:089a
05:00.0 System peripheral: Device 1ac1:089a (prog-if ff)
	Subsystem: Device 1ac1:089a
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Region 0: Memory at 600900000 (64-bit, prefetchable) [disabled] [size=16K]
	Region 2: Memory at 600800000 (64-bit, prefetchable) [disabled] [size=1M]
	Capabilities: [80] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 25.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #1, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Not Supported
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [d0] MSI-X: Enable- Count=128 Masked-
		Vector table: BAR=2 offset=00046800
		PBA: BAR=2 offset=00046068
	Capabilities: [e0] MSI: Enable- Count=1/32 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [f8] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [100 v1] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [108 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [110 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [200 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr+ BadTLP- BadDLLP+ Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
```

geerlingguy, 2021-02-17:
Following the [default instructions for setting up the Coral M.2 PCIe card](https://coral.ai/docs/m2/get-started#2a-on-linux), I get the following kernel panic when the device manager starts:

(No problem prior to installing the Coral packages `gasket-dkms` and `libedgetpu1-std`.)

geerlingguy, 2021-02-17:
(Also posted this over to https://github.com/google-coral/edgetpu/issues/280#issuecomment-780913752).

geerlingguy, 2021-04-13:
Latest update from someone over in the Coral issue queue:

> For now, the plan is to wait until the office is open so we can use a PCIe analyzer and confirm this hypothesis. But there doesn't appear to be any additional changes that we can do in SW - the device expecting a host to be able to perform 64-bit read/write is built into the hardware.
> 
> USB is still the recommendation for the CM4. USB2.0 is possible out of box, and USB3.0 may be possible although extra design considerations are required (more info here: https://coral.ai/products/accelerator-module/).

It looks doubtful (though still in the realm of possibility) the PCI Express version of the Coral TPU will work on the current generation of Raspberry Pi. Though I still wonder if it's a similar issue to the 64/32-bit discrepancy that Broadcom had to work around for the MegaRAID card.

If so, there's a possibility the driver could add a one-off to work around the PCIe limitation on the Pi 4, but it would be much nicer for Pi OS or the firmware to somehow make it work more as expected :P

StuartIanNaylor, 2021-07-09:
@geerlingguy the wavshare carriers have a M.2 which should fit a Coral B+M

I am confused at the '64/32-bit discrepancy' as why use 32bit raspiOS but what is the PCIe limitation of the Pi4?

geerlingguy, 2021-07-09:
@StuartIanNaylor - Right now the Coral drivers don't seem to work on either 32-bit or 64-bit Pi OS. But note that the OS type is not related to the weird issues you get on the PCI Express bus (though for some things, the 32-bit Pi OS behaves more consistently).

darkbasic, 2021-10-11:
AFAIK it's an hardware issue which cannot be solved.

darkbasic, 2021-10-11:
That would be awesome, I'd love to be able to use an M.2 Coral TPU.

Valdiolus, 2021-10-30:
> Which is a reccomanded SoC or mobo compatible with Google Coral TPU M.2 Accelerator m.2 E-key ?

Try to use nxp - like one from Coral dev board. But the cost is quite high.

vebmaster, 2021-12-02:
Hello. Do you use power IC (PMIC) and Edge TPU heatsinks and fans for cooling?
Since the datasheet says that it is necessary to use cooling:
https://coral.ai/docs/m2/datasheet/
https://coral.ai/docs/m2-dual-edgetpu/datasheet/

aiden1989acw, 2021-12-08:
I'd maybe recommend following as per description from your first linked datasheet, in section 5.3, possibly a dear to both lower and upper cooling of the TPU to provide maximum performance, especially considering you're wanting to run 24/7, also take special note to possible transient spikes in power up to 3A

geerlingguy, 2022-03-11:
This is still hope, however small: https://github.com/google-coral/edgetpu/issues/280#issuecomment-1065115347

jveitchmichaelis, 2022-03-21:
> Which is a reccomanded SoC or mobo compatible with Google Coral TPU M.2 Accelerator m.2 E-key ?

You can use the Jetson Nano with the M.2 module, see setup notes here:  https://github.com/jveitchmichaelis/edgetpu-yolo/blob/main/hardware.md It works pretty well!

UcefMountacer, 2022-09-16:
what about this ?
https://pipci.jeffgeerling.com/boards_cm/tinycar-cm4-markus-kasten.html

grigio, 2022-09-19:
> 

@jveitchmichaelis it's out of stock in Europe

jveitchmichaelis, 2022-09-19:
Yes unfortunately almost all the popular dev boards are out of stock due to the chip shortage/supply chain issues. If you just want to play with the module then the Coral Dev Board Minis are more available.

geerlingguy, 2023-10-11:
Just wanted to update here that I have been able to get much further on a Raspberry Pi 5; see https://www.jeffgeerling.com/blog/2023/testing-pcie-on-raspberry-pi-5

Right now it seems like a clock sync/retiming issue with the PCIe bus. One Pi engineer said a prototype Pi 5 board with a different PCIe connection worked with his Coral TPU, but he hasn't gotten it working with the standard FPC connection (there are timeouts when you try using the hardware).

Perhaps a better FFC will fix the issue, we'll see!

StuartIanNaylor, 2023-10-11:
@geerlingguy pcie3.0 or 2.0 as do you not neet to set back to 2.0?

geerlingguy, 2023-10-11:
@StuartIanNaylor - I have tried at PCIe Gen 3.0, 2.0, and 1.0, and have encountered the exact same issue on all three speeds.

geerlingguy, 2023-10-26:
Just as a note, I have heard the final version of the FFC (the little flat cable that goes from Pi 5 to a HAT or expansion board) will be [impedance-controlled](https://digipas.com/product/flexible-flat-cable/impedance-control.php)—the one I'm using in my testing is _not_.

It seems like that may solve a lot of the little issues I'm seeing, especially with the Coral and SATA storage controllers.

StuartIanNaylor, 2023-10-26:
Likely Radxa did the same but swapped out, similar as faults occur under load but seem ok at start up.

geerlingguy, 2023-11-17:
From jdb on the Pi forums:

> There are two possible mappings for routing of message-signalled interrupts - the BCM root complex MSI target, and a MSI-x target external to the RC that is completely transparent to software. External MSI-x interrupts are translated into GIC SPIs of which there are limited number on the chip.
> 
> What happens if you swap the msi-parent target from &mip1 to &pcie1 in the devicetree?
>
> https://github.com/raspberrypi/linux/blob/6137fb168c08bd8c41c8421bf26f09ed29479f08/arch/arm/boot/dts/bcm2712.dtsi#L1015C19-L1015C23

I haven't tested that yet, but I also pinged the folks over at Pineberry Pi (they sent over a few impedance-controlled FPC cables I can test with, they seem more reliable than the flat/straight cable I was using originally...), and @mikegapinski mentioned:

```
diff --git a/arch/arm64/configs/bcm2712_defconfig b/arch/arm64/configs/bcm2712_defconfig
index 8ad2775f5..ff2c619c7 100644
--- a/arch/arm64/configs/bcm2712_defconfig
+++ b/arch/arm64/configs/bcm2712_defconfig
@@ -452,9 +452,10 @@ CONFIG_RFKILL_INPUT=y
 CONFIG_NET_9P=m
 CONFIG_NFC=m
 CONFIG_PCI=y
+CONFIG_PCI_MSI=y
 CONFIG_PCIEPORTBUS=y
 CONFIG_PCIEAER=y
-CONFIG_PCIEASPM_POWERSAVE=y
+CONFIG_PCIEASPM_PERFORMANCE=y
 CONFIG_PCIE_DPC=y
 CONFIG_UEVENT_HELPER=y
 CONFIG_DEVTMPFS=y
```

> PCIEASPM / CONFIG_PCIEASPM_POWERSAVE needs to be off. This will get you past all the pcie related errors in DMESG. Only errors I am getting now are from the driver itself after I run the classifier.

There could still be other bugs lurking, but that does make sense. There are some other PCIe devices that seem to barf when you run any kind of powersaving options, even on CM4.

I will test out these theories tomorrow (fingers crossed!).

geerlingguy, 2023-11-17:
If I just add `pcie_aspm=off` to `/boot/cmdline.txt` and reboot, it gets rid of the annoying PCIe Bus Error messages. This is after a few minutes booted, where normally there's a message every 5 seconds:

```
pi@pi5:~ $ dmesg | grep apex
[    2.800975] apex 0000:01:00.0: enabling device (0000 -> 0002)
[    2.801855] apex 0000:01:00.0: Couldn't initialize interrupts: -28
[    7.901248] apex 0000:01:00.0: Apex performance not throttled due to temperature
```

Can you just force `performance` with `pcie_aspm.policy=powersave`?

The Coral still gives me:

```
[  307.543226] apex 0000:01:00.0: Couldn't reinit interrupts: -28
[  307.543245] apex 0000:01:00.0: Permission checking failed.
```

geerlingguy, 2023-11-17:
I'm now attempting to swap the msi-parent target from `&mip1` to `&pcie1` in the devicetree:

```
# Back up the current dtb
sudo cp /boot/firmware/bcm2712-rpi-5-b.dtb /boot/firmware/bcm2712-rpi-5-b.dtb.bak

# Decompile the current dtb (ignore warnings)
dtc -I dtb -O dts /boot/firmware/bcm2712-rpi-5-b.dtb -o ~/test.dts

# Edit the file
nano ~/test.dts

# Change the line: msi-parent = <0x2f>; (under `pcie@110000`)
# To: msi-parent = <0x66>;
# Then save the file.

# Recompile the dtb and move it back to the firmware directory
dtc -I dts -O dtb ~/test.dts -o ~/test.dtb
sudo mv ~/test.dtb /boot/firmware/bcm2712-rpi-5-b.dtb
```

I rebooted, and...

```
pi@pi5:~ $ dmesg | grep apex
[    2.867959] apex 0000:01:00.0: enabling device (0000 -> 0002)
[    7.901233] apex 0000:01:00.0: Apex performance not throttled due to temperature
```

That interrupt message is gone!

Trying out some inference...

```
# dmesg
[   67.872746] apex 0000:01:00.0: mapping size 0x1000 must be page aligned
[   67.872757] apex 0000:01:00.0: mapping size 0x1000 must be page aligned

# Python output
root@c10211e66ef2:~# python3 /usr/share/edgetpu/examples/classify_image.py --model /usr/share/edgetpu/examples/models/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --label /usr/share/edgetpu/examples/models/inat_bird_labels.txt --image /usr/share/edgetpu/examples/images/bird.bmp
F :39] Attempting to fetch value instead of handling error Failed precondition: Could not map pages : 6 (Invalid argument)
Aborted (core dumped)
```

mikegapinski, 2023-11-17:
@geerlingguy can you add the verbose flag to the example? Mirek has our Coral so I won't be able to test until next week.

```
from pycoral.pybind._pywrap_coral import SetVerbosity as set_verbosity
set_verbosity(10)
```

If it is anything mmap related just use sudo. I ended up not using Docker since I wanted to avoid any potential issues from it. You'd need to purge python that comes with raspbian and manually install version 3.9. It works normally

geerlingguy, 2023-11-17:
@mikegapinski - Regarding purging Python, how did you do that specifically? Every time I've tried messing with changing system Python versions on Debian I end up with a tangled mess and regret my decisions.

geerlingguy, 2023-11-17:
Tried 4k page sizes by switching from my custom kernel to `kernel8`. Edited `/boot/config.txt` and added `kernel=kernel8.img`. Rebooted, verified I was on the 4k kernel:

```
pi@pi5:~ $ uname -a
Linux pi5 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
```

Then I rebuilt the DKMS using the oneliner: `ls /var/lib/initramfs-tools | sudo xargs -n1 /usr/lib/dkms/dkms_autoinstaller start` (without this, I don't have the kernel module for `apex` available).

Rebooted, and verified `/dev/apex_0` was present (and dmesg logs are same, no warnings about interrupts).

Ran docker container and...

```
pi@pi5:~ $ sudo docker run -it --device /dev/apex_0:/dev/apex_0 coral /bin/bash
root@20dff9d9ed96:~# python3 /usr/share/edgetpu/examples/classify_image.py --model /usr/share/edgetpu/examples/models/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --label /usr/share/edgetpu/examples/models/inat_bird_labels.txt --image /usr/share/edgetpu/examples/images/bird.bmp
---------------------------
Poecile atricapillus (Black-capped Chickadee)
Score :  0.44140625
---------------------------
Poecile carolinensis (Carolina Chickadee)
Score :  0.29296875
```

IT WORKS!!!

mikegapinski, 2023-11-17:
KUDOS!!!!!!!!!

I knew it was something with the newer kernel, the old one was a 4K one too

geerlingguy, 2023-11-17:
That test was with

```
dtparam=pciex1_gen=1
```

I am now testing on:

```
dtparam=pciex1_gen=3
```

mikegapinski, 2023-11-17:
I had issues with it on gen3, but it is not a gen3 device so it does not really matter. I have a feeling that Gen3 will be tricky on the pi5 on something other than the NVMe drive from the software side

geerlingguy, 2023-11-17:
Blog post with full summary: [A PCIe Coral TPU FINALLY works on Raspberry Pi 5](https://www.jeffgeerling.com/blog/2023/pcie-coral-tpu-finally-works-on-raspberry-pi-5).

mhaligowski, 2024-01-02:
@geerlingguy congrats on running the Coral! I'm more on the ML side and not embedded systems, so I'm not sure if I understand correctly what's going on here, there's no way of connecting an M.2 device to RPi 5 without the kinda ridiculous chain of hardware? I'm looking into https://coral.ai/products/m2-accelerator-dual-edgetpu/ for now, do I understand correctly that I need a dedicated HAT?

geerlingguy, 2024-01-02:
@mhaligowski - You would need to either use an M.2 A+E key HAT (like the AiHat from Pineberry), or adapters with an M-key HAT, or a USB Coral.

mhaligowski, 2024-01-02:
Just saw the AiHat, it's exactly what I need! Even more excited since I'm Polish living in the US, so proud of being able to support business from PL!

IuliuNovac, 2024-01-07:
> Just saw the AiHat, it's exactly what I need! Even more excited since I'm Polish living in the US, so proud of being able to support business from PL!

I have the Ai Hat and can confirm it works, you just need to follow this guide  [https://www.jeffgeerling.com/blog/2023/pcie-coral-tpu-finally-works-on-raspberry-pi-5 .](https://www.jeffgeerling.com/blog/2023/pcie-coral-tpu-finally-works-on-raspberry-pi-5)

The part that caught me off guard was the change in dtb, since without it you get the access denied.

Seems to be working fine with c++, haven't tried python tho.

Skillnoob, 2024-01-21:
Has anyone gotten the coral usb accelerator working yet?
I've run into massive amounts of issues trying to get it working and had no success so far.
I've posted about the issue on the tensorflow issue page but the issue is quite stale:
https://github.com/tensorflow/tensorflow/issues/62371

geerlingguy, 2024-01-21:
@Skillnoob that would be a different issue (this issue is just for the A+E key PCIe accelerator). I have successfully used the USB accelerator with Frigate on two different Pi 4s and two different Pi 5s now, so at a hardware level it should be fine. Note that Python libraries are all over the place, and you have to use older Python versions to get it working with Python.

Skillnoob, 2024-01-21:
@geerlingguy sorry about that.
I just haven't found any resource to get it working yet and I've tried the recommended python 3.9 without it working.

skol101, 2024-03-04:
How can CM4 work with Dual Edge TPU, if the latter supports only 1 PciE lane?

mikegapinski, 2024-03-04:
If you do not have a PCIe switch and a special board that connects the second lane it'll work as a standard single TPU

skol101, 2024-03-04:
> If you do not have a PCIe switch and a special board that connects the second lane it'll work as a standard single TPU

Wouldn't Radxa CM3I with 1 x PCIe 2‑lane(1x2, 1x1+1x1) Host, Gen 3 (16Gbps) be able to use both TPUs?

r3po-1s-Tr3e, 2024-04-05:
> Blog post with full summary: [A PCIe Coral TPU FINALLY works on Raspberry Pi 5](https://www.jeffgeerling.com/blog/2023/pcie-coral-tpu-finally-works-on-raspberry-pi-5).

@geerlingguy , amazing guide. But we have been facing a problem, we are not able to install pcie gasket drivers after changing the device tree settings. Rpi is not able to update its kernel headers after changing the drive tree settings. 
I have tried downgrading rpi os from 6.6 to 6.1, but still it dosent work.

any insight will be appreciated, thanks!

mikegapinski, 2024-04-05:
@r3po-1s-Tr3e follow this thread, it has all the info for Coral. I have not tested the B+M recently but A+E works ok so this one should too

https://gist.github.com/dataslayermedia/714ec5a9601249d9ee754919dea49c7e?permalink_comment_id=4989560#gistcomment-4989560

r3po-1s-Tr3e, 2024-04-05:
@mikegapinski It worked! I changed a few things, if anyone is intrested, here are the details:

Hardware: 
Raspberry pi 5 4gb
Coral TPU M.2 Accelerator B+M key
NVMe Base PCIe extension HAT for Raspberry pi 5
OS: 6.1.0-rpi4-rpi

Procedure which i followed (I ommited a few steps from Jeff Geerling's guide and changed sequence of action):

echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list
curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
sudo apt-get update

sudo apt-get install gasket-dkms libedgetpu1-std

sudo sh -c "echo 'SUBSYSTEM==\"apex\", MODE=\"0660\", GROUP=\"apex\"' >> /etc/udev/rules.d/65-apex.rules"
sudo groupadd apex
sudo adduser $USER apex

Now REBOOT, after reboot check if TPU is detected by:
lspci -nn | grep 089a                 #Output: 03:00.0 System peripheral: Device 1ac1:089a
Check if pcie driver is loaded:
ls /dev/apex_0                           #Output: /dev/apex_0

If i tried to follow these steps after changing the device tree, I would have not been able to install the gasket and pcie drivers because of not being able to install headers.
Further steps:

echo "kernel=kernel8.img" | sudo tee -a /boot/config.txt
Omitted other steps regarding config changes which were mentioned in the guide
Then changed the Device tree settings: https://www.jeffgeerling.com/blog/2023/how-customize-dtb-device-tree-binary-on-raspberry-pi 
(NOTE: If you are on kernel version 6.6, which is currently the latest, change the msi-parent settings to 0x67 instead of 0x66. Source: https://gist.github.com/dataslayermedia/714ec5a9601249d9ee754919dea49c7e?permalink_comment_id=4989560#gistcomment-4989560)

Installed docker:
curl -sSL https://get.docker.com | sh

Then followed the steps from : https://www.jeffgeerling.com/blog/2023/testing-coral-tpu-accelerator-m2-or-pcie-docker to create and run docker image

tested by running the sample image classifier script of tpu

zoldaten, 2024-08-05:
in my case changed the msi-parent settings to 0x67 instead of 0x6e. 
`Linux raspberrypi 6.6.31+rpt-rpi-v8 `

https://gist.github.com/dataslayermedia/714ec5a9601249d9ee754919dea49c7e?permalink_comment_id=4989560

JeffreyPeacock, 2024-09-09:
Please forgive my shorthand.  Just the facts ma'am:

Raspberry Pi 5 8GB
Coral Dual-Edge TPU
[52PI EP-0223 HAT](https://wiki.52pi.com/index.php?title=EP-0223)
`Linux pi5 6.6.47+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux`
```
root@pi5:~# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```
```
root@pi5:~# tail /boot/firmware/config.txt 
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
dtparam=pciex1
dtparam=pciex1_gen=2

# kernel=kernel8.img
```
I can see the device:
```
root@pi5:~# uname -a
Linux pi5 6.6.47+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux

root@pi5:~# lspci -v

0000:01:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU (prog-if ff)
	Subsystem: Global Unichip Corp. Coral Edge TPU
	Flags: fast devsel
	Memory at 1800100000 (64-bit, prefetchable) [disabled] [size=16K]
	Memory at 1800000000 (64-bit, prefetchable) [disabled] [size=1M]
	Capabilities: [80] Express Endpoint, MSI 00
	Capabilities: [d0] MSI-X: Enable- Count=128 Masked-
	Capabilities: [e0] MSI: Enable- Count=1/32 Maskable- 64bit+
	Capabilities: [f8] Power Management version 3
	Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [108] Latency Tolerance Reporting
	Capabilities: [110] L1 PM Substates
	Capabilities: [200] Advanced Error Reporting

root@pi5:~# ll /dev/apex_0 
crw-rw---- 1 root apex 120, 0 Sep  8 19:40 /dev/apex_0

root@pi5:~# ll /usr/lib/aarch64-linux-gnu/libedgetpu*
lrwxrwxrwx 1 root root      17 Jul  9  2021 /usr/lib/aarch64-linux-gnu/libedgetpu.so.1 -> libedgetpu.so.1.0
-rwxr-xr-x 1 root root 1135952 Jul  9  2021 /usr/lib/aarch64-linux-gnu/libedgetpu.so.1.0

```

However, I normally get this:
```
jeffp@pi5:~/Workspace/AI/Coral/pycoral $ python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
I tflite/edgetpu_manager_direct.cc:453] No matching device is already opened for shared ownership.
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x1a6e, product:0x89a
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[4] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[2]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x18d1, product:0x9302
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[4] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[2]
[... 86 lines cut ...]
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jeffp/Workspace/AI/Coral/pycoral/examples/classify_image.py", line 124, in <module>
    main()
  File "/home/jeffp/Workspace/AI/Coral/pycoral/examples/classify_image.py", line 73, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "/home/jeffp/.pyenv/versions/3.9.18/lib/python3.9/site-packages/pycoral/utils/edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/home/jeffp/.pyenv/versions/3.9.18/lib/python3.9/site-packages/pycoral/utils/edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/home/jeffp/.pyenv/versions/3.9.18/lib/python3.9/site-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1
```
An in `syslog`:
```
2024-09-08T19:42:45.548846-07:00 pi5 kernel: [  112.693475] apex 0000:01:00.0: Couldn't reinit interrupts: -28
2024-09-08T19:42:45.548869-07:00 pi5 kernel: [  112.693497] apex 0000:01:00.0: Permission checking failed.
```
But all the permissions are the same and the user id is in the apex group.

So, did I go forward or backwoards?

Any help is appreciated.  Thanks in advance.

JeffreyPeacock, 2024-09-09:
No, I fixed the kernel8.img rebuild errors your going to find above -- used `dpkg-reconfigure linux-image-6.6.47+rpt-rpi-v8`.  Also updated the tree as per posts.  Then I could see the device but I am still stuck with/at:

```
jeffp@pi5:~/Workspace/AI/Coral/pycoral $ python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
I tflite/edgetpu_manager_direct.cc:453] No matching device is already opened for shared ownership.
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x1a6e, product:0x89a
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[4] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[2]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x18d1, product:0x9302
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[4] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[2]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I tflite/edgetpu_context_direct.cc:106] USB always DFU: False (default)
I tflite/edgetpu_context_direct.cc:128] USB bulk-in queue capacity: default
I tflite/edgetpu_context_direct.cc:67] Performance expectation: Max (default)
I ./driver/mmio/host_queue.h:266] Starting in normal mode
I driver/kernel/kernel_registers.cc:83] Opening /dev/apex_0. read_only=0
I tflite/edgetpu_context_direct.cc:401] Failed to open device [Apex (PCIe)] at [/dev/apex_0]: Failed precondition: Device open failed : -1 (Connection timed out)
Traceback (most recent call last):
  File "/home/jeffp/.pyenv/versions/3.9.18/lib/python3.9/site-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/home/jeffp/.pyenv/versions/3.9.18/lib/python3.9/site-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jeffp/Workspace/AI/Coral/pycoral/examples/classify_image.py", line 124, in <module>
    main()
  File "/home/jeffp/Workspace/AI/Coral/pycoral/examples/classify_image.py", line 73, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "/home/jeffp/.pyenv/versions/3.9.18/lib/python3.9/site-packages/pycoral/utils/edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/home/jeffp/.pyenv/versions/3.9.18/lib/python3.9/site-packages/pycoral/utils/edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/home/jeffp/.pyenv/versions/3.9.18/lib/python3.9/site-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1
```

And `syslog` says: 
```

2024-09-08T20:38:43.965306-07:00 pi5 kernel: [   81.724011] apex 0000:01:00.0: RAM did not enable within timeout (12000 ms)
2024-09-08T20:38:43.965331-07:00 pi5 kernel: [   81.724020] apex 0000:01:00.0: Error in device open cb: -110
```