geerlingguy/raspberry-pi-pcie-devices #45: Test USB 3.0 Cards (A ADWITS, Inateck, and Syba)

geerlingguy, 2020-12-24:
I'm adding this issue mostly so I can reference it from elsewhere.

For my initial Pi CM4 review video (https://www.youtube.com/watch?v=HUamq0ey8_M) and blog post (https://www.jeffgeerling.com/blog/2020/raspberry-pi-compute-module-4-review), I reviewed these three USB 3.0 controller cards, and found:

  - The Syba worked great, always giving full 5000 mbps superspeed for USB 3 devices, and didn't require any external power
  - The Inateck worked sometimes, but not always. It would always mount devices, but sometimes a USB 3 device would mount at 480 mbps speeds.
  - The cheap A ADWITS would not work unless you provided it external power, and it would sometimes mount USB 3 devices at 480 mbps just like the Inateck

6by9, 2020-12-24:
I think I made the comment elsewhere, but anyway.
With the USB3 VL805 card I have (Tamkyo - https://www.amazon.co.uk/gp/product/B08HLKCLX4/), if the plug was pushed in too slowly then it negotiated at 480Mbit/s presumably as the high speed USB3 pairs hadn't connected, however it didn't then renegotiate when pushed fully in. I don't see this as a big failing as it is more down to the connector (and probably protocol), but it is worth bearing in mind.

geerlingguy, 2020-12-24:
@6by9 - Woah; that's an interesting find! I'll have to do a little more testing, and see if it's the case with these cards. Luckily they don't require a recompile of the kernel to use, so it's quicker to test with them :D

Zbrooklyn, 2021-03-12:
So this Syba SD-PEX20199 PCIe x1 USB 3.1 & 3.0 adapter

Does it support display on the USB type c or only file transfer

6by9, 2021-03-12:
DisplayPort Alternate Mode on USB-C requires a GPU that supports a DisplayPort connection first. 
https://www.bigmessowires.com/2019/05/19/explaining-4k-60hz-video-through-usb-c-hub/

jamesy0ung, 2021-04-14:
I got my CM4Lite Wireless 4GB today and my card works. It is not the 'A ADAWITS' one and everything I have tested mounts at USB3 speeds if it is plugged in quickly enough.
[image]

DiomedesDominguez, 2021-10-16:
> > Hey, @geerlingguy , have you tested a 7x usb one? https://www.amazon.com/dp/B07Z34BJYH
> 
> It might work. According the the amazon page, it has a VL805 and a VL813 controller on board. According to this message, it might not work. https://www.spinics.net/lists/linux-usb/msg159005.html

So, it have 50% of working?

drlohwk, 2021-11-17:
yes.. I need to know if the 7x USB with VL805 and a VL813 controllers works with CM4 IO Board. Can wait for the findings..