geerlingguy/raspberry-pi-pcie-devices #47: Test Dell Intel 942V6 X520-DA2 Dual-Port 10Gb SFP+ PCIe NIC

geerlingguy, 2020-12-29:
This is a fun one, sent to me by Jacob Hiltz:

TODO: Picture here.

It's a [Dell Intel 942V6 X520-DA2 Dual-Port 10Gb SFP+ PCIe NIC](https://amzn.to/37XWikN), and it's a little over a hundred bucks new. This was pulled from a Dell server and I have a couple SFP+ 10 Gbps cables (see related: #21).

From my experience with the ASUS 10G adapter (see #15), I know I'll probably not get more than about 3.2 Gbps through the card... but it would be very interesting to see if both 10G interfaces show up and could be used simultaneously.

Having better latency and a solid connection like SFP+ opens up some interesting possibilities. And it's just darn fun.

PBXForums, 2021-01-08:
This should work just fine once the intel driver is compiled in:

root@raspberrypi:~# lspci
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
01:00.1 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)

iperf -c 192.168.35.11
------------------------------------------------------------
Client connecting to 192.168.35.11, TCP port 5001
TCP window size:  325 KByte (default)
------------------------------------------------------------
[  3] local 192.168.35.10 port 43468 connected with 192.168.35.11 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  3.69 GBytes  3.17 Gbits/sec

PBXForums, 2021-01-08:
A little more with mtu 9000:

iperf -c 192.168.35.11
------------------------------------------------------------
Client connecting to 192.168.35.11, TCP port 5001
TCP window size:  325 KByte (default)
------------------------------------------------------------
[  3] local 192.168.35.10 port 43576 connected with 192.168.35.11 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  3.90 GBytes  3.35 Gbits/sec

geerlingguy, 2021-01-08:
@PBXForums - Nice! It seems like you're right at that bus limit that I hit testing the ASUS card (3.36 Gbps).

neg2led, 2021-06-21:
Can confirm this works just fine with X520-DA1 and X520-DA2 cards. Using 64-bit raspbian on kernel 5.10.17-v8+, tested with reference design boards from Dell, HPE, SuperMicro, and Intel themselves all work just fine, to nobody's real surprise.

If anyone would like to get it working themselves, since the raspbian default kernel doesn't build `ixgbe` and rebuilding the kernel takes _untold centuries_ (or cross-compilation), [here's a copy of the intel v5.11.3 driver source](https://github.com/neg2led/ixgbe-dkms) with a `dkms.conf` added - clone to `/usr/src/ixgbe-5.11.3` and do the usual dkms install commands.

There's also a `.deb` package in the releases on that repo which _should_ handle all the dkms setup for you.

geerlingguy, 2021-07-02:
If recompiling the kernel, though, in `menuconfig` select:

```
Device Drivers
  > Network device support
    > Ethernet driver support
      > Intel(R) 10GbE PCI Express adapters support
```

geerlingguy, 2021-07-02:
I'll be doing some bandwidth tests soon. One thing I might do is see how fast the card can communicate between ports.

[image]

kripton, 2023-06-25:
I'd assume that #330  and this one are actually the same card. Wanna merge those two? They are confirmed working and I wouldn't see a reason to have the Dell-branded card listed separately

geerlingguy, 2023-06-26:
@kripton - Yep!

korhojoa, 2024-08-10:
I tried a 82599 10GBase-T version. Didn't need to rebuild a kernel. Works, but does require a reasonable (2.5A 12v wouldn't do it, but a 12v 5a did) external power supply to do so. The bus would reset at initialization otherwise and the card would not get detected.
`Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`
```
0000:00:00.0/01:00.0 Ethernet controller [0200]: Intel Corporation 82599 10 Gigabit TN Network Connection [8086:151c] (rev 01)
        Subsystem: Intel Corporation Device [8086:a02c]
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin B routed to IRQ 164
        Region 0: Memory at 1b00000000 (64-bit, non-prefetchable) [size=2M]
        Region 4: Memory at 1b00400000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=1 PME-
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
                Address: 0000000000000000  Data: 0000
                Masking: 00000000  Pending: 00000000
        Capabilities: [70] MSI-X: Enable+ Count=64 Masked-
                Vector table: BAR=4 offset=00000000
                PBA: BAR=4 offset=00002000
        Capabilities: [a0] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0, Latency L0s <512ns, L1 <64us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
                DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+ FLReset-
                        MaxPayload 512 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr+ NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 5GT/s, Width x8, ASPM L0s, Exit Latency L0s <1us
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 5GT/s, Width x1 (downgraded)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [e0] Vital Product Data
                Unknown small resource type 06, will not decode more.
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr+ BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [140 v1] Device Serial Number 68-05-ca-ff-ff-04-fb-68
        Capabilities: [150 v1] Alternative Routing-ID Interpretation (ARI)
                ARICap: MFVC- ACS-, Next Function: 1
                ARICtl: MFVC- ACS-, Function Group: 0
        Capabilities: [160 v1] Single Root I/O Virtualization (SR-IOV)
                IOVCap: Migration- 10BitTagReq- Interrupt Message Number: 000
                IOVCtl: Enable- Migration- Interrupt- MSE- ARIHierarchy- 10BitTagReq-
                IOVSta: Migration-
                Initial VFs: 64, Total VFs: 64, Number of VFs: 0, Function Dependency Link: 00
                VF offset: 384, stride: 2, Device ID: 10ed
                Supported Page Size: 00000553, System Page Size: 00000001
                Region 0: Memory at 0000000000000000 (64-bit, prefetchable)
                Region 3: Memory at 0000000000000000 (64-bit, prefetchable)
[... 55 lines cut ...]
[... cut ...]
                Supported Page Size: 00000553, System Page Size: 00000001
                Region 0: Memory at 0000000000000000 (64-bit, prefetchable)
                Region 3: Memory at 0000000000000000 (64-bit, prefetchable)
                VF Migration: offset: 00000000, BIR: 0
        Kernel driver in use: ixgbe
        Kernel modules: ixgbe
00: 86 80 1c 15 06 04 10 00 01 00 00 02 10 00 80 00
10: 04 00 20 00 00 00 00 00 01 00 00 00 00 00 00 00
20: 04 40 40 00 00 00 00 00 00 00 00 00 86 80 2c a0
30: 00 00 00 00 40 00 00 00 00 00 00 00 26 01 00 00
```