geerlingguy/raspberry-pi-pcie-devices #49: PCI-e card Mzhou-k3 M.2 B-KEY INTERFACE AND SATA SSD

acasadoalonso, 2020-12-31:
I just tested on the RPi CM4 this PCI-e card

https://www.amazon.com/MZHOU-Express-Devices-Motherboard-Adapter/dp/B082D5ZRXT/ref=sr_1_12?dchild=1&keywords=Tarjeta+PCI-E+a+USB+3.0+PCI+Express+Incluye+1+Puerto+USB+C+y+2+Puertos+USB+A%2C+para+Agregar+Dispositivos+SSD+SATA+III+M.2+a+la+PC+o+la+Placa+Base+Mediante+la+Tarjeta+adaptadora+SATA+a+PCIe+3.0+M.2&qid=1609407604&sr=8-12

and it seems to work OK the USB hub with the 3 ports (USB-a and 1 USB-C), great performance indeed, however the M.2 B-KEY INTERFACE AND SATA SSD does not seems to work.
Any ideas ??? I guess that the support from RasPi OS is not ready yet ???

Great job indeed Jeff, I love your videos and your blog.

Feliz Año Nuevo.

AC/.

TheGuyDanish, 2020-12-31:
I imagine that for the SATA signal to work, you need to plug the SATA cable from the card into the motherboard, as unless there's a SATA controller on the card, it won't have any way to expose it to the host device.

acasadoalonso, 2020-12-31:
This is a Raspberry Pi Computer Module 4 with the IO board 
The Lspci  cmd sees the m2 sata but do not recognize it
I think that is an issue of handling the PCI and the b-key M2 SATA interface
I will keep digging and report it

AC/.

acasadoalonso, 2020-12-31:
Thanks from Madrid, Spain.
AC/.

acasadoalonso, 2020-12-31:
Here it is the output of the lspci command:

pi@CM4:~ $ lspci -vvv -m -D -nn -q

Device: 0000:00:00.0
Class:  PCI bridge [0604]
Vendor: Broadcom Limited [14e4]
Device: BCM2711 PCIe Bridge [2711]
Rev:    20

Device: 0000:01:00.0
Class:  USB controller [0c03]
Vendor: Renesas Technology Corp. [1912]
Device: uPD720201 USB 3.0 Host Controller [0014]
Rev:    03
ProgIf: 30

geerlingguy, 2021-01-01:
It looks like it's only seeing a USB 3.0 controller.

It's an interesting little card, but it seems like the M.2 slot is directly attached through the 'SATA input', which would also require separate SATA power, to power an SSD in the M.2 slot.

See the pictures in the Amazon listing—it mentions "Note: This adapter does not support PCI-e NVMe SSD".

So you would need a separate PCIe switch that allows connecting at least two cards, and then you'd need a SATA controller card in the other slot, and then you could connect a SATA cable between the SATA controller card and this card's input, and a power supply to the SATA power input on this card, and then an M.2 SSD could be used with it.

acasadoalonso, 2021-01-01:
Gracias amigo
Great job indeed.
AC/.

acasadoalonso, 2021-01-06:
You can report that the USB hub (in the 3 ports) work perfect.
I would like to see how the SATA M2 works as well.
I see that a card brother https://pipci.jeffgeerling.com/cards_m2/mzhou-nvme-m2-ssd-m-key-adapter.html you reported as working with no issues, I wonder if the combination of USB hub and M.2 and NVMe adapters works as well.
Thanks for the great job that you are doing.

acasadoalonso, 2021-01-06:
[image]

acasadoalonso, 2021-01-06:
[image]

geerlingguy, 2021-07-02:
Just added to the site. Thanks for reporting that the USB side works—to get the M.2 SATA slot working, you'd still need a separate SATA card and a PCI Express switch.

closed 2021-07-02 by geerlingguy