geerlingguy/raspberry-pi-pcie-devices #56: Re-test Intel AX200 WiFi 6 cards - again

geerlingguy, 2021-01-09:
So... in #38 and #22 I tested the Intel AX200 Desktop Kit (via M.2 to PCIe adapter) and the EDUP Intel AX200 PCIe card, respectively. The most recent video I posted was [I lied... WiFi 6 on the Raspberry Pi is NOT faster than Ethernet](https://www.youtube.com/watch?v=1kWAdtoq8TQ)—but _that itself might be another unintentional lie_!

My testing has been _all over the place_, and every day I seem to be learning new (and sometimes seemingly contradictory) facts about WiFi, the AX200 chipset itself, and networking in general.

Today I've learned that it _may_, in fact, be possible to get more than the ~800-1100 Mbps I was getting in my most recent tests with the AX200 to my ASUS RT-AX86U router:

From [Dmitri Arekhta's comment](https://www.youtube.com/watch?v=1kWAdtoq8TQ&lc=Ugz229A4972O2-iS0JF4AaABAg):

> Jeff, Marvell Marvell is wrong. Intel AX200 max PHY rate is 2400Mbps because it's the first two chains mass-market chipset that could do 2x2 MIMO on 160MHz channel. Before AX200 2 chain chipsets were capable of either 2x2 80MHz or 1x1 160MHz, i.e 1200Mbps max PHY rate. I think you have to set 160MHz channel on your router in order to push AX200 to its limits. In ideal RF conditions you might get ~70% UDP throughput from max PHY rate. However, with MCS11 it's a bit harder, but I guess you should do 1.3-1.4Mbps w/o problems. However, the bottleneck might be elsewhere of course.

And follow-up [from Tobias Mädel](https://www.youtube.com/watch?v=1kWAdtoq8TQ&lc=Ugz229A4972O2-iS0JF4AaABAg.9IG8h1nAR1l9IGDe4BYnBC):

>  That might be a regdomain or channel issue. 
> Check "iw reg get" if the channel you're using is allowed 160 MHz bandwidth in your region.
> 
> I totally agree with Dmitri, I've seen 1470 MBit/s of real-world TCP performance via AX200 chips. Your first results were most likely correct.

Then [ThePunkBuster chimes in](https://www.youtube.com/watch?v=1kWAdtoq8TQ&lc=Ugz229A4972O2-iS0JF4AaABAg.9IG8h1nAR1l9IGhL64Z-sw):

> 160 MHz May be enabled, but if the channel width is not forced and selection is left on Auto, the router will set the connection by itself according to RF environment conditions, device support, or signal strength. Also, as Dimitri said, MU-MIMO should be enabled with OFDMA support for full Wi-Fi AX advantage. Marvell Marvell comment was about a single spatial stream and not about multiple ones, which is what all current innovations on Wi-Fi are about.

And after confirming I have the ASUS RT-AX86U:

> So... reviewed the specs, and you should be able to achieve 2,402 Mbit/s on a 2x2 stream using a 160 MHz channel width at 1024-QAM and a 800 ns Guard Interval with your current hardware setup! Both, the intel card and the Asus router should handle the connection. Try again forcing the router and the card parameters :)

So yeah... as I think is patently obvious, I am a n00b when it comes to networking in general (but learning as much as I can as I go!), and there is more testing to be done!

vegedb, 2021-01-14:
@geerlingguy 
why not test AX210 instead?

geerlingguy, 2021-01-14:
@vegedb - I don't have one :)

SuperKali, 2021-02-03:
@geerlingguy hi, I very appreciate your video, but I have got a problem, I don't know how compile this for my, all tutorial that you've they're for mac, I don't have it, can you provide me your build of the system with those fix, because for me doesn't work.

Thanks in advance.

geerlingguy, 2021-02-03:
@SuperKali - This is my exact cross-compile environment. Technically you can run an identical environment on any Windows, Mac, or Linux PC as long as you have VirtualBox and Vagrant installed: https://github.com/geerlingguy/raspberry-pi-pcie-devices/tree/master/extras/cross-compile

SuperKali, 2021-02-03:
@geerlingguy thank you for your answer, the problem is one, I never compiled a kernel and I don't know as to do it, I'm beginner, I hope that you can explain or can help me.

Thanks.

PixlRainbow, 2021-02-04:
@SuperKali  the guide mentions *if you are not using Linux PC, create a virtual machine (VM) and then **run Linux in that VM**, then continue from the "Compiling the kernel" step onwards, adjusting folder paths as needed*

The specific steps for creating a VM and installing Linux within it will vary between different VM vendors, so the details are not included in the guide -- Assuming you are a Windows user you may use either Microsoft Hyper-V, Oracle VirtualBox, VMWare, QEMU or others.

PixlRainbow, 2021-02-04:
which step are you having trouble with? The guide essentially tells you to **perform all steps in a Linux environment**, using a virtual machine to emulate a Linux PC as needed.
[After that, the steps amount to](https://github.com/geerlingguy/raspberry-pi-pcie-devices/tree/master/extras/cross-compile#compiling-the-kernel)
1. download kernel source code
2. create configuration file with linux command
3. compile kernel with linux command)

SuperKali, 2021-02-04:
@PixlRainbow i got this error:

[image]

PixlRainbow, 2021-02-04:
then use a different VM, e.g. use the virtualbox GUI directly, or go with another vendor such as VMWare

Geofferic, 2021-02-04:
Can you upload the file itself, or at least copy/paste line 12?

SuperKali, 2021-02-04:
The line 12 is on the config of your Vagrant

geerlingguy, 2021-02-04:
@SuperKali - Could you move the discussion around kernel compilation over to a Discussion in that tab? I don't want to derail this issue too much before I've even had a chance to start working on it.

SuperKali, 2021-02-05:
@geerlingguy do you think it is possible to make Intel AX200 as AP?

lealog, 2021-05-18:
I'm also doing some tests on this Wi-Fi chipset, and I'm having connectivity issues when connecting on 160MHz. A lot of times the STA is disconnecting and the UDP throughput with only one stream when using iperf is maximum ~450Mbps.

> lshw -c network
>   *-network
>        description: Wireless interface
>        product: Wi-Fi 6 AX200
>        vendor: Intel Corporation
>        physical id: 0
>        bus info: pci@0000:03:00.0
>        logical name: wlan0
>        version: 1a
>        serial: xx:xx:xx:xx:xx
>        width: 64 bits
>        clock: 33MHz
>        capabilities: pm msi pciexpress msix bus_master cap_list ethernet physical wireless
>        configuration: broadcast=yes driver=iwlwifi driverversion=5.8.0-48-generic firmware=48.4fa0041f.0 cc-a0-48.ucode latency=0 link=no multicast=yes wireless=IEEE 802.11

Did you find this issue?

Hiruks, 2021-10-09:
> @geerlingguy do you think it is possible to make Intel AX200 as AP?

Hiruks, 2021-10-09:
Yes it is possible to use the wifi card as an access point if it supports master mode, I am not sure of the exact model of this card, but usually it should work fine with services like Hostapd, I suggest you check out this video if you want some guidance on creating an AP with the Hostapd configurations : https://www.youtube.com/watch?v=EIdSlGwQGyU

geerlingguy, 2021-10-10:
@Hiruks - So far I've been unsuccessful in doing so, though it's been a few months since my last attempts. Was trying to get OpenWRT working with an AX200 for a base station, but it seems like Intel's driver just doesn't allow it.