geerlingguy/raspberry-pi-pcie-devices #57: Raspberry PI Working as a TV Reciver (TBS6522)

yehia2amer, 2021-01-11:
I always had this idea of having a Raspberry PI Working as a Satellite/Terrestrial Reciver and Media Player.
This could finally be achieved if PCIe Tuners works.

**Example of a Dual Tuner PCIe Card:**

TBS6522 Dual Tuner PCIe Card DVB-S2X S2 S T2 T C2 C ISDB-T Multi Standard Digital TV Card Live TV/Window/Linux/HTPC/IPTV Server https://www.amazon.com/dp/B01MR28YAA/

**Also This could be tested on a more specialized linux distro Like:**

[OSMC](https://osmc.tv)

[LibreELEC](https://libreelec.tv/raspberry-pi-4/)

[OpenPCTV](https://sourceforge.net/projects/openpctv/)

stephenvalente, 2021-01-11:
There is already an official Pi DVB hat in the UK. Cannot recall exact compatibility elsewhere though.

https://thepihut.com/blogs/raspberry-pi-tutorials/how-to-stream-digital-tv-with-the-raspberry-pi-tv-hat

6by9, 2021-01-11:
A DVBSky S952 v2 is certainly detected fine.
It needs the relevant kernel module building (option VIDEO_CX23885 / module cx23885). I didn't actually hoook it up to a dish as it had been playing up in my x86 tvheadend machine. (I might try if I get a moment).

TBS cards are pretty grim in that they have never contributed drivers to mainline, so you have to manually compile the modules. Hauppauge are generally better in that regard.

See https://www.linuxtv.org/wiki/index.php/Category:DVB-S2_PCIe_Cards for general guidance on Linux support for DVB PCIe cards.
Don't forget to find the relevant firmware files for the tuner. https://github.com/LibreELEC/dvb-firmware/ is generally a good place to start.

6by9, 2021-01-11:
The Pi TV HAT is DVB-T/T2 only and connects over SPI. There's no easy way to hook multiple of them up to the same Pi though.
https://www.raspberrypi.org/products/raspberry-pi-tv-hat/

stephenvalente, 2021-01-11:
The Pi foundation talked about seeking the relevant licenses for other formats 3 years ago when this was launched. Shame they didn't manage for whatever reason.

6by9, 2021-01-11:
> The Pi foundation talked about seeking the relevant licenses for other formats 3 years ago when this was launched. Shame they didn't manage for whatever reason.

It's not licences, it's finding a chip that does the relevant decode. I have never seen an ATSC SPI DVB demod chip.
DVB-S2 isn't really an option as it requires higher voltages to drive the LNB (the DVBSky S952 takes an additional 6pin supply)

> Would a USB solution work under Linux?

Yes, works fine.
I have 2 Hauppauge WinTV-dualHD DVB-T2 tuners, a DVBSky S960 DVB-S2, and a Geniatech HDStar DVB-S2 tuner hanging off a Pi4 via a hub. tvheadend is quite happy handling the lot. USB3 HDD for storage.

6by9, 2021-01-11:
DVBSky S952 is being awkward.
Kernel splat when first told TVHeadend to start tuning and no services found.  
```
[  150.337678] ------------[ cut here ]------------
[  150.337715] WARNING: CPU: 3 PID: 656 at /home/dave/Pi/linux/kernel/dma/swiotlb.c:683 swiotlb_map+0x404/0x40c
[  150.337735] cx23885 0000:01:00.0: swiotlb addr 0x0000000423cc0000+16384 overflow (mask ffffffff, bus limit 4ffffffff).
[  150.337749] Modules linked in: bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc ts2020 ov2311 cx25840 brcmfmac cx23885 tveeprom brcmutil cx2341x tda18271 videobuf2_dvb videobuf2_dma_sg bcm2835_isp(C) m88ds3103 bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_unicam bcm2835_mmal_vchiq(C) dvb_core v4l2_dv_timings videobuf2_dma_contig v4l2_fwnode videobuf2_vmalloc videobuf2_memops sha256_generic videobuf2_v4l2 videobuf2_common cfg80211 rfkill i2c_mux_pinctrl raspberrypi_hwmon snd_pcm i2c_mux dwc2 snd_timer i2c_bcm2835 snd roles regmap_i2c videodev vc_sm_cma(C) mc rpivid_mem uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[  150.338514] CPU: 3 PID: 656 Comm: tvh:lnxdvb-fron Tainted: G         C        5.10.6-v7l+ #5
[  150.338524] Hardware name: BCM2711
[  150.338533] Backtrace: 
[  150.338561] [<c020c3e8>] (dump_backtrace) from [<c020c718>] (show_stack+0x20/0x24)
[  150.338576]  r7:00000000 r6:60000013 r5:00000000 r4:c12e6abc
[  150.338594] [<c020c6f8>] (show_stack) from [<c0b71708>] (dump_stack+0xd4/0x100)
[  150.338613] [<c0b71634>] (dump_stack) from [<c02216fc>] (__warn+0xd0/0x114)
[  150.338628]  r8:000002ab r7:00000009 r6:00000000 r5:c0e259f8 r4:c4c6ba84 r3:00000000
[  150.338645] [<c022162c>] (__warn) from [<c02217c4>] (warn_slowpath_fmt+0x84/0xc0)
[  150.338659]  r9:c0e259f8 r8:000002ab r7:c02a6bd8 r6:00000009 r5:c0e25a24 r4:c1205048
[  150.338676] [<c0221744>] (warn_slowpath_fmt) from [<c02a6bd8>] (swiotlb_map+0x404/0x40c)
[  150.338689]  r9:c1205048 r8:c1bbb070 r7:00000000 r6:ffffffff r5:00000004 r4:ffffffff
[  150.338709] [<c02a67d4>] (swiotlb_map) from [<c02a41c0>] (dma_direct_map_sg+0x2b0/0x39c)
[  150.338723]  r10:c1bbb070 r9:00000000 r8:04c14000 r7:ffffffff r6:c1bbb1e0 r5:00000004
[  150.338733]  r4:04c14000
[  150.338754] [<c02a3f10>] (dma_direct_map_sg) from [<c02a1720>] (dma_map_sg_attrs+0x44/0xac)
[  150.338768]  r10:00000001 r9:00006000 r8:00001000 r7:00000000 r6:00002dc4 r5:00000006
[  150.338777]  r4:c4020200
[  150.338808] [<c02a16dc>] (dma_map_sg_attrs) from [<bf0bce1c>] (vb2_dma_sg_alloc+0x250/0x1434 [videobuf2_dma_sg])
[  150.338853] [<bf0bcbcc>] (vb2_dma_sg_alloc [videobuf2_dma_sg]) from [<bf513ac4>] (__vb2_queue_alloc+0x1c4/0x460 [videobuf2_common])
[  150.338867]  r10:c3d08858 r9:00000000 r8:c3d08818 r7:c33dbc00 r6:00000000 r5:c3d08818
[  150.338877]  r4:c33dbc00
[  150.338933] [<bf513900>] (__vb2_queue_alloc [videobuf2_common]) from [<bf514390>] (vb2_core_reqbufs+0x2e0/0x44c [videobuf2_common])
[  150.338947]  r10:c4c6bca4 r9:c4c6bca0 r8:00000001 r7:c33db000 r6:c3d08858 r5:c1205048
[  150.338957]  r4:c3d08818
[  150.339010] [<bf5140b0>] (vb2_core_reqbufs [videobuf2_common]) from [<bf514720>] (__vb2_init_fileio+0xd0/0x2f8 [videobuf2_common])
[  150.339024]  r10:00000001 r9:00000000 r8:00000001 r7:00000000 r6:c43b0680 r5:c33db000
[  150.339033]  r4:c3d08818
[  150.339086] [<bf514650>] (__vb2_init_fileio [videobuf2_common]) from [<bf514bac>] (vb2_thread_start+0x88/0x164 [videobuf2_common])
[  150.339100]  r10:00000001 r9:00000000 r8:bf0c2200 r7:c34c8ec0 r6:c43b0680 r5:c3d08810
[  150.339109]  r4:c3d08818
[  150.339149] [<bf514b24>] (vb2_thread_start [videobuf2_common]) from [<bf0c21e0>] (vb2_dvb_start_feed+0x7c/0x9c [videobuf2_dvb])
[  150.339162]  r8:00000000 r7:c3d08b50 r6:c3d0896c r5:c3d08810 r4:c3d08988
[  150.339228] [<bf0c2164>] (vb2_dvb_start_feed [videobuf2_dvb]) from [<bf5f232c>] (dmx_ts_feed_start_filtering+0x60/0xf4 [dvb_core])
[  150.339241]  r6:f0a21000 r5:f0a22000 r4:c3d08988 r3:bf0c2164
[  150.339345] [<bf5f22cc>] (dmx_ts_feed_start_filtering [dvb_core]) from [<bf5ee784>] (dvb_dmxdev_start_feed+0xc0/0xf0 [dvb_core])
[  150.339359]  r8:00000000 r7:c3d08b88 r6:f0b36000 r5:f0a21000 r4:00000000 r3:bf5f22cc
[  150.339458] [<bf5ee6c4>] (dvb_dmxdev_start_feed [dvb_core]) from [<bf5f0034>] (dvb_dmxdev_filter_start+0x90/0x3fc [dvb_core])
[  150.339472]  r9:f0b36228 r8:f0b36004 r7:c3d08b88 r6:f0b36000 r5:c3d08b88 r4:c43b0a80
[  150.339571] [<bf5effa4>] (dvb_dmxdev_filter_start [dvb_core]) from [<bf5f08f4>] (dvb_demux_do_ioctl+0x554/0x5f8 [dvb_core])
[  150.339584]  r8:f0b36000 r7:c3d08b88 r6:c4c6be4c r5:c3d08d7c r4:00000004 r3:c3d08d90
[  150.339682] [<bf5f03a0>] (dvb_demux_do_ioctl [dvb_core]) from [<bf5ee5a8>] (dvb_usercopy+0xd0/0x1a8 [dvb_core])
[  150.339696]  r10:00000001 r9:00000000 r8:c4c6be4c r7:a92f5abc r6:c1205048 r5:00000014
[  150.339706]  r4:40146f2c
[  150.339804] [<bf5ee4d8>] (dvb_usercopy [dvb_core]) from [<bf5eea84>] (dvb_demux_ioctl+0x20/0x24 [dvb_core])
[  150.339818]  r10:00000029 r9:00000000 r8:a92f5abc r7:c3265d80 r6:40146f2c r5:c3265d81
[  150.339828]  r4:c1205048
[  150.339893] [<bf5eea64>] (dvb_demux_ioctl [dvb_core]) from [<c045184c>] (sys_ioctl+0x168/0x8fc)
[  150.339911] [<c04516e4>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x28)
[  150.339922] Exception stack(0xc4c6bfa8 to 0xc4c6bff0)
[  150.339936] bfa0:                   a92f5cd8 00000012 00000029 40146f2c a92f5abc 00000004
[  150.339951] bfc0: a92f5cd8 00000012 00750404 00000036 a92f5bd8 00000020 01af2f9c 01af2f80
[... cut ...]
[    7.201295] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    7.201319] cx23885: cx23885[0]/0: found at 0000:01:00.0, rev: 4, irq: 65, latency: 0, mmio: 0x600000000
```
That's repeatable and needs a full powerdown to identify the card again properly, but then it dies with the ```swiotlb``` splat.

It may be that it's just a PCIe setup issue, but I'm not going to investigate further. I had issues with this card disappearing on my x86 box so it may be that the card has partially died.

yehia2amer, 2021-01-11:
@6by9 , Thanks for sharing all these info.
I would really love to see something like that working. Please let me know if you got time to hook it up and try it. A video would be great !
Also have you tried OSMC or LibreELEC ?

stinger151, 2021-01-14:
Hi all,

I am trying the same with a tbs6984!
https://www.linuxtv.org/wiki/index.php/TBS6984

Unfortunatelly after days of trying various range settings and up to rpi-5.10-y i cannot get it to show in lspci.

I tried the cm4+io board hardware with a intel NIC (NC110T) and the device is listed in lspci right away alongs side the bc pcie bridge.

i even tried increasing the brcm_pcie_link_up(pcie) timeout in pcie-brcmstb.c but it keeps starting with "no link" in dmesg and 0 output in lspci.

am i right that this is before any dvb-specific kernel modules are needed? in my test i did not compile the tbs driver each time but only check in dmesg/lspci for a life sign.... but nothing :(

here the lspci -vv from a x64 ubuntu, can anybody tell me if the card is maybe using some mode the pi does not support and i should give up?

`02:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 02)
        Subsystem: Device 6984:0013
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 16
        Region 0: Memory at f7600000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [40] MSI: Enable- Count=1/32 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [50] Express (v1) Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <256ns, L1 <1us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset- SlotPowerLimit 25.000W
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
                LnkCap: Port #1, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 128 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
        Capabilities: [74] Power Management version 2
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [80] Vendor Specific Information: Len=50 <?>
        Capabilities: [100 v1] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
        Kernel driver in use: SAA716x Budget
        Kernel modules: saa716x_tbs_dvb`

6by9, 2021-01-22:
@stinger151 lspci should work even without the drivers.
Just plugging my Asus GT710-4H graphics card in, I get
```
pi@raspberrypi:~ $ lspci -k
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
	Kernel driver in use: pcieport
01:00.0 VGA compatible controller: NVIDIA Corporation GK208 [GeForce GT 710B] (rev a1)
	Subsystem: ASUSTeK Computer Inc. GK208B [GeForce GT 710]
01:00.1 Audio device: NVIDIA Corporation GK208 HDMI/DP Audio Controller (rev a1)
	Subsystem: ASUSTeK Computer Inc. GK208 HDMI/DP Audio Controller
```

DVB-S2 is going to need additional power connecting through the 6pin power connector on the TBS6984 for the LNB power.
I just tried without that step with my DVBSky S952 and it created a big splat on trying lspci, presumably as parts of the board weren't powered up. That board has been playing up though, so it could be just dead.

I'm keeping my eyes open on Ebay for TBS cards at silly prices, and will test if I find one.

FrancescoRestelli, 2021-01-26:
i could get the card to work by patching the dtsi file and include the  max-link-speed = <1>

for details see here:
https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=300061&p=1805284#p1805284

6by9, 2021-01-28:
Ebay win for a TBS6981 dual DVB-S2 tuner.
Connected and detected straight off
```
pi@raspberrypi:~ $ lspci -k
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
	Kernel driver in use: pcieport
01:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
	Subsystem: Device 6981:8888
```
Looking at the wiki it's a simple CX23885 / CX24132 / CX24117 device, all of which are supported in the mainline kernel. Enable ```VIDEO_CX23885```, rebuild the kernel, install the dvb-firmware, and it shows up.
Then again tvheadend appears to be failing to find services, and is registering 0% SNR. The kernel log has the same swiotlb_map splat as the DVBSky was showing
```
[  584.259298] ------------[ cut here ]------------
[  584.259333] WARNING: CPU: 2 PID: 1118 at /kernel/dma/swiotlb.c:683 swiotlb_map+0x404/0x40c
[  584.259353] cx23885 0000:01:00.0: swiotlb addr 0x0000000417cc0000+16384 overflow (mask ffffffff, bus limit 4ffffffff).
[  584.259365] Modules linked in: bnep cx24117 hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc cx25840 imx219 brcmfmac brcmutil vc4 cec v3d gpu_sched sha256_generic cx23885 cfg80211 tveeprom drm_kms_helper bcm2835_codec(C) rfkill cx2341x raspberrypi_hwmon tda18271 i2c_mux_pinctrl dwc2 videobuf2_dvb videobuf2_dma_sg bcm2835_unicam i2c_bcm2835 v4l2_dv_timings roles m88ds3103 v4l2_fwnode drm bcm2835_v4l2(C) v4l2_mem2mem dvb_core bcm2835_isp(C) videobuf2_vmalloc bcm2835_mmal_vchiq(C) videobuf2_dma_contig i2c_mux videobuf2_memops drm_panel_orientation_quirks videobuf2_v4l2 videobuf2_common vc_sm_cma(C) videodev snd_soc_core mc regmap_i2c snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd rpivid_mem syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[  584.260206] CPU: 2 PID: 1118 Comm: tvh:lnxdvb-fron Tainted: G         C        5.10.10-v7l+ #2
[  584.260215] Hardware name: BCM2711
[  584.260223] Backtrace: 
[  584.260250] [<c020c3e8>] (dump_backtrace) from [<c020c718>] (show_stack+0x20/0x24)
[  584.260262]  r7:00000000 r6:60000013 r5:00000000 r4:c12e6abc
[  584.260281] [<c020c6f8>] (show_stack) from [<c0b6c6cc>] (dump_stack+0xd4/0x100)
[  584.260296] [<c0b6c5f8>] (dump_stack) from [<c02216fc>] (__warn+0xd0/0x114)
[  584.260309]  r8:000002ab r7:00000009 r6:00000000 r5:c0e23804 r4:c6267a84 r3:00000000
[  584.260324] [<c022162c>] (__warn) from [<c02217c4>] (warn_slowpath_fmt+0x84/0xc0)
[  584.260336]  r9:c0e23804 r8:000002ab r7:c02a6be4 r6:00000009 r5:c0e23834 r4:c1205048
[  584.260350] [<c0221744>] (warn_slowpath_fmt) from [<c02a6be4>] (swiotlb_map+0x404/0x40c)
[  584.260362]  r9:c1205048 r8:c1bcb070 r7:00000000 r6:ffffffff r5:00000004 r4:ffffffff
[  584.260378] [<c02a67e0>] (swiotlb_map) from [<c02a41cc>] (dma_direct_map_sg+0x2b0/0x39c)
[  584.260390]  r10:c1bcb070 r9:00000000 r8:062b0000 r7:ffffffff r6:c1bcb1e0 r5:00000004
[  584.260398]  r4:062b0000
[  584.260414] [<c02a3f1c>] (dma_direct_map_sg) from [<c02a172c>] (dma_map_sg_attrs+0x44/0xac)
[  584.260426]  r10:00000001 r9:00006000 r8:00001000 r7:00000000 r6:00002dc4 r5:00000006
[  584.260434]  r4:c415c780
[  584.260460] [<c02a16e8>] (dma_map_sg_attrs) from [<bf245e1c>] (vb2_dma_sg_alloc+0x250/0x1434 [videobuf2_dma_sg])
[  584.260499] [<bf245bcc>] (vb2_dma_sg_alloc [videobuf2_dma_sg]) from [<bf2b5ac4>] (__vb2_queue_alloc+0x1c4/0x460 [videobuf2_common])
[  584.260511]  r10:c45d6058 r9:00000000 r8:c45d6018 r7:c4286400 r6:00000000 r5:c45d6018
[  584.260519]  r4:c4286400
[  584.260566] [<bf2b5900>] (__vb2_queue_alloc [videobuf2_common]) from [<bf2b6390>] (vb2_core_reqbufs+0x2e0/0x44c [videobuf2_common])
[  584.260578]  r10:c6267ca4 r9:c6267ca0 r8:00000001 r7:c4285800 r6:c45d6058 r5:c1205048
[  584.260587]  r4:c45d6018
[  584.260633] [<bf2b60b0>] (vb2_core_reqbufs [videobuf2_common]) from [<bf2b6720>] (__vb2_init_fileio+0xd0/0x2f8 [videobuf2_common])
[  584.260645]  r10:00000001 r9:00000000 r8:00000001 r7:00000000 r6:c3b79140 r5:c4285800
[  584.260653]  r4:c45d6018
[  584.260697] [<bf2b6650>] (__vb2_init_fileio [videobuf2_common]) from [<bf2b6bac>] (vb2_thread_start+0x88/0x164 [videobuf2_common])
[  584.260708]  r10:00000001 r9:00000000 r8:bf24b200 r7:c18d4ec0 r6:c3b79140 r5:c45d6010
[  584.260716]  r4:c45d6018
[  584.260750] [<bf2b6b24>] (vb2_thread_start [videobuf2_common]) from [<bf24b1e0>] (vb2_dvb_start_feed+0x7c/0x9c [videobuf2_dvb])
[  584.260762]  r8:00000000 r7:c45d6350 r6:c45d616c r5:c45d6010 r4:c45d6188
[  584.260819] [<bf24b164>] (vb2_dvb_start_feed [videobuf2_dvb]) from [<bf1fb32c>] (dmx_ts_feed_start_filtering+0x60/0xf4 [dvb_core])
[  584.260829]  r6:f0a4b000 r5:f0a4c000 r4:c45d6188 r3:bf24b164
[  584.260918] [<bf1fb2cc>] (dmx_ts_feed_start_filtering [dvb_core]) from [<bf1f7784>] (dvb_dmxdev_start_feed+0xc0/0xf0 [dvb_core])
[  584.260930]  r8:00000000 r7:c45d6388 r6:f0901000 r5:f0a4b000 r4:00000000 r3:bf1fb2cc
[  584.261015] [<bf1f76c4>] (dvb_dmxdev_start_feed [dvb_core]) from [<bf1f9034>] (dvb_dmxdev_filter_start+0x90/0x3fc [dvb_core])
[  584.261027]  r9:f0901228 r8:f0901004 r7:c45d6388 r6:f0901000 r5:c45d6388 r4:c3b79d40
[  584.261111] [<bf1f8fa4>] (dvb_dmxdev_filter_start [dvb_core]) from [<bf1f98f4>] (dvb_demux_do_ioctl+0x554/0x5f8 [dvb_core])
[  584.261122]  r8:f0901000 r7:c45d6388 r6:c6267e4c r5:c45d657c r4:00000004 r3:c45d6590
[  584.261206] [<bf1f93a0>] (dvb_demux_do_ioctl [dvb_core]) from [<bf1f75a8>] (dvb_usercopy+0xd0/0x1a8 [dvb_core])
[... cut ...]
[  584.261434] 7fc0: a96f7cd8 00000000 007d0404 00000036 a96f7bd8 00000020 014f3794 014f3778
[  584.261444] 7fe0: 007d0af4 a96f7a9c 00659648 b508851c
[  584.261456]  r10:00000036 r9:c6266000 r8:c0200204 r7:00000036 r6:007d0404 r5:00000000
[  584.261464]  r4:a96f7cd8
[  584.261475] ---[ end trace bcba4eab9d84ab56 ]---
```
Ah, looking at the code, it's DMA woes. That would explain it.
https://github.com/raspberrypi/linux/blob/rpi-5.10.y/kernel/dma/swiotlb.c#L678
```/* Ensure that the address returned is DMA'ble */```
Perhaps my DVBSky S952 isn't dead after all then.

pelwell, 2021-01-28:
That address (0x0000000417cc0000) looks like a RAM address _in PCIe space_. Can you find out what phys and dma_addr are in dma_direct_map_page when it fails?

pelwell, 2021-01-28:
Also dev->dma_mask and dev->bus_dma_limit.

6by9, 2021-01-28:
[  584.259353] cx23885 0000:01:00.0: swiotlb addr 0x0000000417cc0000+16384 overflow **(mask ffffffff, bus limit 4ffffffff).**

With extra logging in from

```
[   40.026230] dma_direct_map_page: !dma_capable: phys 0x0000000004170000 dma_addr 0x0000000404170000, swiotlb_force 0
[   40.026253] swiotlb_map: dev->dma_mask ffffffff, dec->bus_dma_limit 4ffffffff
[   40.026270] ------------[ cut here ]------------
[   40.026300] WARNING: CPU: 2 PID: 668 at /home/dave/Pi/linux2711/kernel/dma/swiotlb.c:685 swiotlb_map+0x454/0x484
[   40.026319] cx23885 0000:01:00.0: swiotlb addr 0x0000000417cc0000+16384 overflow (mask ffffffff, bus limit 4ffffffff).
[   40.026332] Modules linked in: bnep cx24117 hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc cx25840 imx219 brcmfmac brcmutil vc4 sha256_generic cx23885 tveeprom cx2341x cec v3d tda18271 cfg80211 videobuf2_dvb gpu_sched videobuf2_dma_sg m88ds3103 dvb_core drm_kms_helper rfkill i2c_mux_pinctrl raspberrypi_hwmon bcm2835_isp(C) i2c_mux bcm2835_v4l2(C) regmap_i2c bcm2835_codec(C) bcm2835_unicam bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig v4l2_dv_timings videobuf2_vmalloc drm v4l2_fwnode dwc2 videobuf2_memops videobuf2_v4l2 i2c_bcm2835 roles videobuf2_common drm_panel_orientation_quirks videodev mc snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd vc_sm_cma(C) syscopyarea sysfillrect sysimgblt fb_sys_fops backlight rpivid_mem uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[   40.027306] CPU: 2 PID: 668 Comm: tvh:lnxdvb-fron Tainted: G         C        5.10.10-v7l+ #5
[   40.027316] Hardware name: BCM2711
[   40.027326] Backtrace: 
[   40.027354] [<c020c3e8>] (dump_backtrace) from [<c020c718>] (show_stack+0x20/0x24)
[   40.027369]  r7:00000000 r6:60000013 r5:00000000 r4:c12e6abc
[   40.027389] [<c020c6f8>] (show_stack) from [<c0b6c84c>] (dump_stack+0xd4/0x100)
[   40.027407] [<c0b6c778>] (dump_stack) from [<c02216fc>] (__warn+0xd0/0x114)
[   40.027422]  r8:000002ad r7:00000009 r6:00000000 r5:c0e23928 r4:c4ea5a74 r3:00000000
[   40.027439] [<c022162c>] (__warn) from [<c02217c4>] (warn_slowpath_fmt+0x84/0xc0)
[   40.027453]  r9:c0e23928 r8:000002ad r7:c02a6d30 r6:00000009 r5:c0e23958 r4:c1205048
[   40.027469] [<c0221744>] (warn_slowpath_fmt) from [<c02a6d30>] (swiotlb_map+0x454/0x484)
[   40.027483]  r9:00000004 r8:17cc3fff r7:00004000 r6:c1205048 r5:00000004 r4:ffffffff
[   40.027502] [<c02a68dc>] (swiotlb_map) from [<c02a4320>] (dma_direct_map_sg+0x3d4/0x468)
[   40.027516]  r10:c4120500 r9:00000000 r8:04170000 r7:ffffffff r6:c1bbb1e0 r5:00000004
[   40.027526]  r4:04170000
[   40.027545] [<c02a3f4c>] (dma_direct_map_sg) from [<c02a175c>] (dma_map_sg_attrs+0x44/0xac)
[   40.027559]  r10:00000001 r9:00006000 r8:00001000 r7:00000000 r6:00002dc4 r5:00000006
[   40.027569]  r4:c4d4f480
[   40.027599] [<c02a1718>] (dma_map_sg_attrs) from [<bf0a5e1c>] (vb2_dma_sg_alloc+0x250/0x1434 [videobuf2_dma_sg])
[   40.027644] [<bf0a5bcc>] (vb2_dma_sg_alloc [videobuf2_dma_sg]) from [<bf1a9ac4>] (__vb2_queue_alloc+0x1c4/0x460 [videobuf2_common])
[   40.027658]  r10:c22e6058 r9:00000000 r8:c22e6018 r7:c3aa0400 r6:00000000 r5:c22e6018
[   40.027668]  r4:c3aa0400
[   40.027725] [<bf1a9900>] (__vb2_queue_alloc [videobuf2_common]) from [<bf1aa390>] (vb2_core_reqbufs+0x2e0/0x44c [videobuf2_common])
[   40.027739]  r10:c4ea5ca4 r9:c4ea5ca0 r8:00000001 r7:c3aa0000 r6:c22e6058 r5:c1205048
[   40.027748]  r4:c22e6018
[   40.027802] [<bf1aa0b0>] (vb2_core_reqbufs [videobuf2_common]) from [<bf1aa720>] (__vb2_init_fileio+0xd0/0x2f8 [videobuf2_common])
[   40.027815]  r10:00000001 r9:00000000 r8:00000001 r7:00000000 r6:c4120c40 r5:c3aa0000
[   40.027825]  r4:c22e6018
[   40.027877] [<bf1aa650>] (__vb2_init_fileio [videobuf2_common]) from [<bf1aabac>] (vb2_thread_start+0x88/0x164 [videobuf2_common])
[   40.027891]  r10:00000001 r9:00000000 r8:bf110200 r7:c18d6ec0 r6:c4120c40 r5:c22e6010
[   40.027900]  r4:c22e6018
[   40.027939] [<bf1aab24>] (vb2_thread_start [videobuf2_common]) from [<bf1101e0>] (vb2_dvb_start_feed+0x7c/0x9c [videobuf2_dvb])
[   40.027953]  r8:00000000 r7:c22e6350 r6:c22e616c r5:c22e6010 r4:c22e6188
[   40.028019] [<bf110164>] (vb2_dvb_start_feed [videobuf2_dvb]) from [<bf20c32c>] (dmx_ts_feed_start_filtering+0x60/0xf4 [dvb_core])
[   40.028032]  r6:f08cb000 r5:f08cc000 r4:c22e6188 r3:bf110164
[   40.028136] [<bf20c2cc>] (dmx_ts_feed_start_filtering [dvb_core]) from [<bf208784>] (dvb_dmxdev_start_feed+0xc0/0xf0 [dvb_core])
[   40.028151]  r8:00000000 r7:c22e6388 r6:f09e0000 r5:f08cb000 r4:00000000 r3:bf20c2cc
[   40.028250] [<bf2086c4>] (dvb_dmxdev_start_feed [dvb_core]) from [<bf20a034>] (dvb_dmxdev_filter_start+0x90/0x3fc [dvb_core])
[   40.028265]  r9:f09e0228 r8:f09e0004 r7:c22e6388 r6:f09e0000 r5:c22e6388 r4:c4120b00
[   40.028363] [<bf209fa4>] (dvb_dmxdev_filter_start [dvb_core]) from [<bf20a8f4>] (dvb_demux_do_ioctl+0x554/0x5f8 [dvb_core])
[   40.028377]  r8:f09e0000 r7:c22e6388 r6:c4ea5e4c r5:c22e657c r4:00000004 r3:c22e6590
[   40.028475] [<bf20a3a0>] (dvb_demux_do_ioctl [dvb_core]) from [<bf2085a8>] (dvb_usercopy+0xd0/0x1a8 [dvb_core])
[   40.028490]  r10:00000001 r9:00000000 r8:c4ea5e4c r7:a96f7abc r6:c1205048 r5:00000014
[   40.028499]  r4:40146f2c
[   40.028597] [<bf2084d8>] (dvb_usercopy [dvb_core]) from [<bf208a84>] (dvb_demux_ioctl+0x20/0x24 [dvb_core])
[   40.028611]  r10:00000023 r9:00000000 r8:a96f7abc r7:c462acc0 r6:40146f2c r5:c462acc1
[   40.028621]  r4:c1205048
[... cut ...]
[   40.040273] dma_direct_map_page: !dma_capable: phys 0x0000000004e68000 dma_addr 0x0000000404e68000, swiotlb_force 0
[   40.040292] swiotlb_map: dev->dma_mask ffffffff, dec->bus_dma_limit 4ffffffff
[   40.040314] dma_direct_map_sg: dma_direct_map_page failed. sg->dma_address 0xffffffffffffffff, 
[   40.040589] dma_direct_map_page: !dma_capable: phys 0x0000000004e6c000 dma_addr 0x0000000404e6c000, swiotlb_force 0
[   40.040607] swiotlb_map: dev->dma_mask ffffffff, dec->bus_dma_limit 4ffffffff
[   40.040624] dma_direct_map_sg: dma_direct_map_page failed. sg->dma_address 0xffffffffffffffff, 
```

6by9, 2021-01-28:
I've kicked off a 64bit kernel build too to see if that changes things.

pelwell, 2021-01-28:
It might. I think the problem is highlighted in the last line of dma_capable:
```
static inline bool dma_capable(struct device *dev, dma_addr_t addr, size_t size,
		bool is_ram)
{
	dma_addr_t end = addr + size - 1;

	if (addr == DMA_MAPPING_ERROR)
		return false;
	if (is_ram && !IS_ENABLED(CONFIG_ARCH_DMA_ADDR_T_64BIT) &&
	    min(addr, end) < phys_to_dma(dev, PFN_PHYS(min_low_pfn)))
		return false;

	return end <= min_not_zero(*dev->dma_mask, dev->bus_dma_limit);
}
```
`dma_mask` is only `DMA_BIT_MASK(32)` - it needs to be at least `DMA_BIT_MASK(35)`.

6by9, 2021-01-28:
Where does the ```DMA_BIT_MASK``` come from? Any plausible way of hacking it on 32bit, or do we say this is going to be 64bit only? (assuming it works - still building)
Trying to fix up the LPAE modes sounds like it may be more hassle than it's worth.

6by9, 2021-01-28:
64bit build failed the same way :-(
```
[   74.906652] dma_direct_map_page: !dma_capable: phys 0x0000000050dc4000 dma_addr 0x0000000450dc4000, swiotlb_force 0
[   74.906678] swiotlb_map: dev->dma_mask ffffffff, dec->bus_dma_limit 4ffffffff
[   74.906691] ------------[ cut here ]------------
[   74.906709] cx23885 0000:01:00.0: swiotlb addr 0x000000041ad44000+16384 overflow (mask ffffffff, bus limit 4ffffffff).
[   74.906756] WARNING: CPU: 0 PID: 752 at /home/dave/Pi/linux2711/kernel/dma/swiotlb.c:685 swiotlb_map+0x2f8/0x310
[   74.906763] Modules linked in: cx24117 bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc cx25840 imx219 brcmfmac brcmutil cx23885 tveeprom sha256_generic cx2341x raspberrypi_hwmon v3d tda18271 bcm2835_unicam gpu_sched vc4 videobuf2_dvb cfg80211 v4l2_dv_timings videobuf2_dma_sg m88ds3103 rfkill bcm2835_v4l2(C) v4l2_fwnode bcm2835_codec(C) cec bcm2835_isp(C) i2c_mux_pinctrl dwc2 dvb_core v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc i2c_mux videobuf2_dma_contig roles videobuf2_memops drm_kms_helper videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) mc regmap_i2c drm i2c_bcm2835 drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[   74.907148] CPU: 0 PID: 752 Comm: tvh:lnxdvb-fron Tainted: G         C        5.10.10-v8+ #1
[   74.907156] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   74.907166] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[   74.907178] pc : swiotlb_map+0x2f8/0x310
[   74.907189] lr : swiotlb_map+0x2f8/0x310
[   74.907195] sp : ffffffc012693810
[   74.907202] x29: ffffffc012693810 x28: ffffff80410b20b0 
[   74.907219] x27: ffffff804a049a80 x26: 0000000450dc4000 
[   74.907235] x25: 0000000000000002 x24: 0000000000000020 
[   74.907250] x23: 0000000000000002 x22: ffffffc011238000 
[   74.907265] x21: 0000000000004000 x20: ffffff80410b20b0 
[   74.907280] x19: 000000001ad44000 x18: ffffffc011238948 
[   74.907295] x17: 0000000000000000 x16: 0000000000000000 
[   74.907310] x15: ffffffc01143ac50 x14: 616d2820776f6c66 
[   74.907325] x13: 7265766f20343833 x12: 36312b3030303434 
[   74.907339] x11: 6461313430303030 x10: 3030307830207264 
[   74.907354] x9 : 646120626c746f69 x8 : 6666666666666666 
[   74.907370] x7 : 0000000000000000 x6 : ffffffc011239000 
[   74.907385] x5 : ffffff80fb7dd8e0 x4 : 0000000000000000 
[   74.907399] x3 : 0000000000000027 x2 : 0000000000000000 
[   74.907414] x1 : e9725a3433a08200 x0 : 0000000000000000 
[   74.907430] Call trace:
[   74.907441]  swiotlb_map+0x2f8/0x310
[   74.907452]  dma_direct_map_sg+0x288/0x2e8
[   74.907461]  dma_map_sg_attrs+0x58/0x88
[   74.907480]  vb2_dma_sg_alloc+0x25c/0x2d0 [videobuf2_dma_sg]
[   74.907504]  __vb2_queue_alloc+0x1b0/0x470 [videobuf2_common]
[   74.907523]  vb2_core_reqbufs+0x2b4/0x410 [videobuf2_common]
[   74.907541]  __vb2_init_fileio+0xbc/0x2d0 [videobuf2_common]
[   74.907559]  vb2_thread_start+0x78/0x1c8 [videobuf2_common]
[   74.907572]  vb2_dvb_start_feed+0x88/0xb0 [videobuf2_dvb]
[   74.907599]  dmx_ts_feed_start_filtering+0x54/0xe8 [dvb_core]
[   74.907625]  dvb_dmxdev_start_feed.isra.12+0xac/0x128 [dvb_core]
[   74.907652]  dvb_dmxdev_filter_start+0x98/0x3c0 [dvb_core]
[   74.907677]  dvb_demux_do_ioctl+0x2e8/0x538 [dvb_core]
[   74.907703]  dvb_usercopy+0x70/0x248 [dvb_core]
[   74.907728]  dvb_demux_ioctl+0x3c/0x50 [dvb_core]
[   74.907741]  __arm64_compat_sys_ioctl+0xd8/0x1a8
[   74.907754]  el0_svc_common.constprop.2+0x7c/0x1b0
[   74.907764]  do_el0_svc_compat+0x2c/0x58
[   74.907775]  el0_svc_compat+0x1c/0x28
[   74.907784]  el0_sync_compat_handler+0x9c/0xe8
[   74.907793]  el0_sync_compat+0x144/0x180
[   74.907801] ---[ end trace 2c99c832ff8ff2e9 ]---
[   74.907843] dma_direct_map_sg: dma_direct_map_page failed. sg->dma_address 0xffffffffffffffff, 
[   74.908121] dma_direct_map_page: !dma_capable: phys 0x0000000050dc8000 dma_addr 0x0000000450dc8000, swiotlb_force 0
[   74.908138] swiotlb_map: dev->dma_mask ffffffff, dec->bus_dma_limit 4ffffffff
```
So the question is where that mask gets set.

6by9, 2021-01-28:
I think I can answer my own question - https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/media/pci/cx23885/cx23885-core.c#L2150
```
err = pci_set_dma_mask(pci_dev, 0xffffffff);
```

6by9, 2021-01-28:
Default value for the dma_mask is also 0xFFFFFFFF.

Trying to increase that to 0xFFFFFFFFF or 0x7FFFFFFFF blows up
```
[   46.058939] ------------[ cut here ]------------
[   46.058962] Failed to get suitable pool for 0000:01:00.0
[   46.059019] WARNING: CPU: 2 PID: 676 at /home/dave/Pi/linux2711/kernel/dma/pool.c:282 dma_alloc_from_pool+0x144/0x1d0
[   46.059026] Modules linked in: bnep cx24117 hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc cx25840 imx219 v3d gpu_sched brcmfmac brcmutil sha256_generic cfg80211 rfkill i2c_mux_pinctrl cx23885 tveeprom cx2341x tda18271 raspberrypi_hwmon videobuf2_dvb videobuf2_dma_sg m88ds3103 dwc2 bcm2835_unicam dvb_core v4l2_dv_timings roles bcm2835_codec(C) bcm2835_v4l2(C) v4l2_fwnode vc4 i2c_mux bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig cec videobuf2_memops videobuf2_v4l2 drm_kms_helper videobuf2_common vc_sm_cma(C) videodev i2c_bcm2835 drm mc drm_panel_orientation_quirks regmap_i2c snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer rpivid_mem snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[   46.059413] CPU: 2 PID: 676 Comm: tvh:lnxdvb-fron Tainted: G         C        5.10.10-v8+ #1
[   46.059420] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   46.059431] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[   46.059443] pc : dma_alloc_from_pool+0x144/0x1d0
[   46.059453] lr : dma_alloc_from_pool+0x144/0x1d0
[   46.059459] sp : ffffffc012193890
[   46.059466] x29: ffffffc012193890 x28: ffffff80403e2b00 
[   46.059483] x27: ffffffc01143b6d0 x26: ffffff80410ba0b0 
[   46.059498] x25: ffffffc012193970 x24: ffffffc01143b6b0 
[   46.059513] x23: ffffffc010110770 x22: 0000000000001000 
[   46.059528] x21: ffffffc010de39b0 x20: 000000001ef00000 
[   46.059544] x19: ffffffc0119a1000 x18: 0000000000000000 
[   46.059558] x17: 0000000000000000 x16: 0000000000000000 
[   46.059574] x15: 0000000000000000 x14: 0000000000000000 
[   46.059588] x13: ffffffc01157d000 x12: 0000000000000000 
[   46.059603] x11: 0000000000000001 x10: 0000000000000040 
[   46.059619] x9 : ffffffc012193890 x8 : 3a31303a30303030 
[   46.059634] x7 : 0000000000000000 x6 : ffffffc011239000 
[   46.059648] x5 : ffffff80fb81d8e0 x4 : 0000000000000000 
[   46.059663] x3 : 0000000000000027 x2 : 0000000000000000 
[   46.059677] x1 : 9d326b27a411a500 x0 : 0000000000000000 
[   46.059694] Call trace:
[   46.059705]  dma_alloc_from_pool+0x144/0x1d0
[   46.059716]  dma_direct_alloc+0x284/0x310
[   46.059725]  dma_alloc_attrs+0xa8/0xc0
[   46.059762]  cx23885_risc_databuffer+0x68/0xf0 [cx23885]
[   46.059790]  cx23885_buf_prepare+0x84/0xd0 [cx23885]
[   46.059817]  buffer_prepare+0x28/0x38 [cx23885]
[   46.059838]  __buf_prepare+0x1dc/0x248 [videobuf2_common]
[   46.059857]  vb2_core_qbuf+0x518/0x5c8 [videobuf2_common]
[   46.059874]  __vb2_init_fileio+0x208/0x2d0 [videobuf2_common]
[   46.059892]  vb2_thread_start+0x78/0x1c8 [videobuf2_common]
[   46.059904]  vb2_dvb_start_feed+0x88/0xb0 [videobuf2_dvb]
[   46.059932]  dmx_ts_feed_start_filtering+0x54/0xe8 [dvb_core]
[   46.059958]  dvb_dmxdev_start_feed.isra.12+0xac/0x128 [dvb_core]
[   46.059983]  dvb_dmxdev_filter_start+0x98/0x3c0 [dvb_core]
[   46.060007]  dvb_demux_do_ioctl+0x2e8/0x538 [dvb_core]
[   46.060032]  dvb_usercopy+0x70/0x248 [dvb_core]
[   46.060056]  dvb_demux_ioctl+0x3c/0x50 [dvb_core]
```
and ending with
```
[   46.128699] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000004
[   46.139670] Mem abort info:
[   46.144468]   ESR = 0x96000046
[   46.149568]   EC = 0x25: DABT (current EL), IL = 32 bits
[   46.156692]   SET = 0, FnV = 0
[   46.160514]   EA = 0, S1PTW = 0
[   46.164431] Data abort info:
[   46.168068]   ISV = 0, ISS = 0x00000046
[   46.172712]   CM = 0, WnR = 1
[   46.176486] user pgtable: 4k pages, 39-bit VAs, pgdp=00000000470c0000
[   46.183757] [0000000000000004] pgd=0000000046a66003, p4d=0000000046a66003, pud=0000000046a66003, pmd=0000000000000000
[   46.195316] Internal error: Oops: 96000046 [#1] PREEMPT SMP
```
from cx23885_buf_queue+0x3c/0x150, presumably as it is trying to queue a buffer that it failed to allocate.

It needs someone who understands PCIe and DMA better than me to work this out.

pelwell, 2021-01-29:
Is this device fully 64-bit capable? I see 64-bit registers and descriptors, but the upper 32-bits are written as zeroes.

6by9, 2021-01-29:
> Is this device fully 64-bit capable? I see 64-bit registers and descriptors, but the upper 32-bits are written as zeroes.

I would expect so as whilst the cx23885 has been around a while, the boards are relatively recent.
```
01:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
	Subsystem: Device 6981:8888
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 72
	Region 0: Memory at 600000000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: [40] Express (v1) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset- SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <4us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
	Capabilities: [80] Power Management version 2
		Flags: PMEClk- DSI+ D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [90] Vital Product Data
		Product Name: "
		End
	Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [200 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed+ WRR32+ WRR64+ WRR128-
		Ctrl:	ArbSelect=WRR64
		Status:	InProgress-
		Port Arbitration Table [240] <?>
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Kernel driver in use: cx23885
	Kernel modules: cx23885
```

Trying my DVBsky S952 (same chipset):
```
01:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
	Subsystem: DVBSky S952
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 72
	Region 0: Memory at 600000000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: [40] Express (v1) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset- SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <4us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
	Capabilities: [80] Power Management version 2
		Flags: PMEClk- DSI+ D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [90] Vital Product Data
		Product Name: "
		End
	Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [200 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed+ WRR32+ WRR64+ WRR128-
		Ctrl:	ArbSelect=WRR64
		Status:	InProgress-
		Port Arbitration Table [240] <?>
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Kernel driver in use: cx23885
	Kernel modules: cx23885
```

I had hoped to win a TBS 6209 last night, but it went higher than I was prepared to pay.
(Octa DVB-T2 tuner - you could receive ALL of FreeView simultaneously with that, and have a tuner left over!)

pelwell, 2021-01-29:
I'm going to try inverting the PCIe address map, putting RAM at zero and peripherals high. I've not found anything that explicitly disallows placing the outbound window above the 4GB mark, except that it would prevent 32-bit devices from working. Of course, putting RAM above 4GB would also do that, so for devices with 4GB+ of RAM it may be no worse...

6by9, 2021-01-29:
If you want to play with actual hardware then I can drop one of these into the office for/on Monday. I think they'll work well enough without an actual satellite signal to provoke this.

6by9, 2021-02-05:
Next attempt failed too. Hauppauge WinTV QuadHD.

It shows up as a 1:2 PCI-e switch, and then two cx23885 decoders (https://www.linuxtv.org/wiki/index.php/Hauppauge_WinTV-quadHD_(DVB-T/T2/C) says they're actually CX23888).
```
pi@raspberrypi:~ $ lspci
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
02:01.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
02:02.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
03:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
04:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
```
Different failure - cx23885 fails to get an IRQ on trying to register the second half of the first adapter.
```
[    6.608792] cx23885: cx23885 driver version 0.0.4 loaded
[    6.609087] pcieport 0000:00:00.0: of_irq_parse_pci: failed with rc=-22
[    6.609171] cx23885 0000:03:00.0: enabling device (0140 -> 0142)
[    6.609267] cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    6.994790] tveeprom: Hauppauge model 166200, rev B4I6, serial# 4036013435
[    6.994830] tveeprom: MAC address is 00:0d:fe:90:ad:7b
[    6.994868] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    6.994906] tveeprom: TV standards ATSC/DVB Digital (eeprom 0x80)
[    6.994921] tveeprom: audio processor is CX23885 (idx 39)
[    6.994956] tveeprom: decoder processor is CX23885 (idx 33)
[    6.994991] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    6.995007] cx23885: cx23885[0]: hauppauge eeprom: model=166200
[    6.995046] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.043498] cx23885: cx23885[0]: cx23885 based dvb card
[    7.043529] cx23885: dvb_register(): board=60 port=1
[    7.087016] i2c i2c-13: Added multiplexed i2c bus 16
[    7.087035] si2168 13-0064: Silicon Labs Si2168-B40 successfully identified
[    7.087049] si2168 13-0064: firmware version: B 4.0.2
[    7.106372] si2157 14-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    7.106482] dvbdev: DVB: registering new adapter (cx23885[0])
[    7.106502] cx23885 0000:03:00.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    7.114044] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.114085] cx23885: cx23885[0]: cx23885 based dvb card
[    7.114133] cx23885: dvb_register(): board=60 port=2
[    7.141326] i2c i2c-13: Added multiplexed i2c bus 17
[    7.141345] si2168 13-0066: Silicon Labs Si2168-B40 successfully identified
[    7.141359] si2168 13-0066: firmware version: B 4.0.2
[    7.157731] si2157 14-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    7.159951] dvbdev: DVB: registering new adapter (cx23885[0])
[    7.159972] cx23885 0000:03:00.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[    7.164802] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    7.164825] cx23885: cx23885[0]/0: found at 0000:03:00.0, rev: 4, irq: 0, latency: 0, mmio: 0x600000000
[    7.164849] cx23885: cx23885[0]: can't get IRQ 0
[    7.178919] cx23885: probe of 0000:03:00.0 failed with error -22
[    7.179215] pcieport 0000:00:00.0: of_irq_parse_pci: failed with rc=-22
```
the second adapter then fails too
```
[    7.179296] cx23885 0000:04:00.0: enabling device (0140 -> 0142)
[    7.179384] cx23885: CORE cx23885[1]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    7.562109] tveeprom: Hauppauge model 166201, rev B4I6, serial# 4036013435
[    7.562128] tveeprom: MAC address is 00:0d:fe:90:ad:7b
[    7.562144] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    7.562161] tveeprom: TV standards ATSC/DVB Digital (eeprom 0x80)
[    7.562176] tveeprom: audio processor is CX23885 (idx 39)
[    7.562190] tveeprom: decoder processor is CX23885 (idx 33)
[    7.562211] tveeprom: has no radio
[    7.562227] cx23885: cx23885[1]: hauppauge eeprom: model=166201
[    7.562245] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.562644] cx23885: cx23885[1]: cx23885 based dvb card
[    7.562666] cx23885: dvb_register(): board=60 port=1
[    7.581945] i2c i2c-13: Added multiplexed i2c bus 16
[    7.581976] si2168 13-0064: Silicon Labs Si2168-B40 successfully identified
[    7.581993] si2168 13-0064: firmware version: B 4.0.2
[    7.596417] si2157 14-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    7.596537] dvbdev: DVB: registering new adapter (cx23885[1])
[    7.596561] cx23885 0000:04:00.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    7.598737] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.598755] cx23885: cx23885[1]: cx23885 based dvb card
[    7.598775] cx23885: dvb_register(): board=60 port=2
[    7.617178] i2c i2c-13: Added multiplexed i2c bus 17
[    7.617208] si2168 13-0066: Silicon Labs Si2168-B40 successfully identified
[    7.617225] si2168 13-0066: firmware version: B 4.0.2
[    7.631269] si2157 14-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    7.631386] dvbdev: DVB: registering new adapter (cx23885[1])
[    7.631409] cx23885 0000:04:00.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[... cut ...] ision = 0xa5
[    7.633754] cx23885: cx23885[1]/0: found at 0000:04:00.0, rev: 4, irq: 0, latency: 0, mmio: 0x600200000
[    7.633780] cx23885: cx23885[1]: can't get IRQ 0
[    7.639626] cx23885: probe of 0000:04:00.0 failed with error -22
```
So it looks like it is the PCIe switch that has the issue, and not cx23885 specifically.

(TBH The Pi works so nicely as a TV Receiver using USB tuners (I have 6 on my Pi4), so switching to these PCIe devices doesn't really add that much, but it has uses).

6by9, 2021-02-05:
I have just updated to 5.10.13, so it's possible something has changed since the 5.10.7 I was testing the other card with.

6by9, 2021-02-05:
Ah, it is switches that are the issue.

Just to make sure I wasn't going crazy I tried my DVBSky S952. Initially connected via an ASMedia ASM1184e switch and it logged the same error as the QuadHD.
Connected direct to the CMIO and it initialises fine (won't scan as it crashes in the same way as the TBS6981).

geerlingguy, 2021-02-28:
Just linking this issue to this discussion: [Test an HDTV tuner card](https://github.com/geerlingguy/raspberry-pi-pcie-devices/discussions/23)...

PixlRainbow, 2021-03-01:
> Is this device fully 64-bit capable? I see 64-bit registers and descriptors, but the upper 32-bits are written as zeroes.

Wonder if this is related to the fact that [the PCIE controller cannot perform 64 bit accesses](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/62#issuecomment-787440564)

6by9, 2021-03-09:
Next attempt - DVBSky S952 v3 dual DVB-S2 tuner.
**SUCCESS!**

It needs ```CONFIG_DVB_SMIPCIE=m``` in the kernel config so that the smipcie module gets built, and the dvb-demod-m88rs6000.fw firmware file.

```
pi@raspberrypi:~ $ sudo lspci -k -v
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 65
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 00000000-00000fff
	Memory behind bridge: c0000000-c00fffff
	Capabilities: [48] Power Management version 3
	Capabilities: [ac] Express Root Port (Slot-), MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
	Capabilities: [240] L1 PM Substates
	Kernel driver in use: pcieport

01:00.0 Multimedia video controller: Spin Master Ltd. PCIe Video Bridge (rev 01)
	Subsystem: DVBSky S952 v3
	Flags: bus master, fast devsel, latency 0, IRQ 65
	Memory at 600000000 (32-bit, non-prefetchable) [size=4K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=1/16 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Kernel driver in use: SMI PCIe driver
	Kernel modules: smipcie
```

geerlingguy, 2021-03-09:
Nice! What software do you use with it? I honestly haven't done much with tuners on Linux before; I used to use EyeTV from Elgato back in the day, but that was never ported to Linux AFAICT.

6by9, 2021-03-09:
One small caveat for S952v3. You MUST connect power to the 6 pin ATX power connector on the card. Just having the 12V via the PCIe connector is not sufficient.

It's probably a case of extra regulators on the card as connecting from the floppy power connector on the CM4IO to the ATX connector is sufficient (at least in my case), so it's not power consumption per se. I do have other satellite receivers connected to the octo LNB on my dish, so it may be that without those then you need more juice to be able to feed the LNB.

I've used tvheadend as packaged in Raspberry Pi OS. It needs a little fiddling to enable the Freesat EPG grabber by default, but with that done it found all the expected services. Streaming to VLC or other players (eg Kodi) works quite happily, or it'll record any programmes you like as either raw or MKV files. It'll subscribe to the entire mux too, so can record (or stream) multiple programmes from the same mux with a single tuner.

I use tvheadend on my normal TV rig too - a Pi4 with a pair of Hauppauge WinTV DualHD USB DVB-T2 tuners and 2 USB DVB-S2 tuners. It's a bit of a fiddle combining the two sources, but once done works fine.

6by9, 2021-03-14:
More play time - TBS 6284 quad DVB-T2 tuner this time.

It uses out of tree drivers, but I've pulled the required bits across from https://github.com/tbsdtv/linux_media (I'll push a tree probably tomorrow, and may discuss if we're prepared to support this in our downstream kernel. TBS seem not to like upstreaming drivers).

Initially it reported pcie "link down", but that rang bells as having been seen before. A forum search pulled up https://www.raspberrypi.org/forums/viewtopic.php?p=1803474#p1803474 with a DT tweak for the TBS6984 (quad DVB-S2) tuner. That works perfectly with the TBS6284 as well - we have a card detected, and have a driver. I probably ought to query that DT change with colleagues who understand PCIe better.

Tuners A&B vs C&D appear to come up swapped around in their order (ie tuner 0 is not A), so initially I'd connected my aerial into the wrong tuner. Swapping to tuner 2 and it's found 7 working Freeview muxes with a total of 162 services, and I've just watched a bit of BBC TWO via it. I haven't tried enabling all 4 tuners at this point.

So another working PCIe card, and I'm now wondering just how many tuners I have that I can run simultaneously from a Pi :-)

TBS6284 uses https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/pci/saa716x/saa716x_budget.c#L2877 which supports a load of other cards. Where the tuner isn't in mainline I've just hacked them out, but there are still a number of other cards that should all work with the patch. That should be clear once I've pushed my tree.

6by9, 2021-03-15:
Further update on DVBSky S952 (not v3).
With https://github.com/raspberrypi/linux/pull/4216/ and ```dtoverlay=pcie-32bit-dma``` I don't get the splat from the DMA subsystem, however I suspect that card is dead as it still won't actually tune. It may well mean that the TBS 6981 I have in the office will work though (same hardware).

6by9, 2021-03-15:
TBS6284 drivers are at https://github.com/6by9/linux/tree/rpi-5.10.y-tbs

Look at https://github.com/6by9/linux/blob/rpi-5.10.y-tbs/drivers/media/pci/saa716x/saa716x_budget.c#L1473 for the list of cards that should in theory be supported.

SonnyWalkman, 2021-06-06:
I have the tbs6902 dual dvb-s/s2 PCIe card m. Works fine in x86 Linux box however, I was hoping there were tbs drivers compiled drives for ARM?
I have a project I'm developing on Nvidia Xavier NX which is 6 core ARM. I like to hook up pcie via M2 socket where the tbs6902 is plugged in. I have seperate 5v & 12v supplies powering the tbs6992 and the m2 to PCIe extender hardware.
Can you confirm if your changers get the TBS sat card works with and on the ARM architecture?

6by9, 2021-06-06:
Very few of the TBS cards have mainline drivers for them - they are largely relying on the out of tree drivers from https://github.com/tbsdtv/linux_media which can give issues in their own right (eg they recompile all the V4L2 framework, so any V4L2 drivers that are in your kernel config but not theirs will be discarded). You'll need to compile them yourself.

You quote both TBS6902 and TBS6992. Is one a typo? The first is a current tbsecp3 based card, whilst the second is a much older Philips SAA7160 based card.
I've not looked at any of the tbsecp3 based cards, largely as I don't have one. My branch was largely looking at the SAA716x based cards, and even then I removed all the weirder hacks from their branch.

PCIe support is also different between x86 and ARM. It very much depends on the hardware implementation on your particular SoC as to whether it works or not. DVB cards are generally easier to get running than graphics cards though.

For Linux driver support I'd generally advise to buy a card from Hauppauge instead of TBS. One of their employees is actively involved in getting driver support into the mainline Linux kernel.

SonnyWalkman, 2021-06-07:
Thanks for your response.. 6by9!
Correct, typo with the 6992. It’s the 6902 dual sat card.
The card works in x86 no worries however as indicated, not sure the drive
can be Criss compiled for ARM devices. I’m sure the pressure overtime
change manufactures view on supporting ARM since there are far more
Computers and small boards which use ARM.

I’ll investigate using the USB type Sat receiver over the PCIe? The PCIe
would have far less overhead than having the USB stack in between hardware
and vl2 driver?

I’ll keep researching and hopefully find a suitable arrangement.

Regards

Rob

6by9, 2021-06-07:
It's not so much a matter of ARM vs x86, but differences in PCIe implementation. PCIe is relatively new to be exposed on ARM platforms. It's issues around cache coherency, PCIe window ranges, and types of access that vary based on implementation.

Upstreaming Linux drivers so that users (whether on x86 or ARM) don't have to jump through hoops to get their devices working is the bigger issue. Most of the major manufacturers do so, but TBS refuse to. 

I'm only concerned with Raspberry Pi as I'm employed by them. Nvidia Xavier NX may just work if you compile the tbsdtv drivers for it, but I can't say. Compile natively rather than trying to cross compile to save yourself a load of hassle.

USB does tend to just work, and the bitrates for DVB tend not to stress USB too highly (a complete mux tends to be <40 Mbit/s). Again look for devices with mainline Linux support to save yourself some grief. USB DVB-S2 devices do seem to be becoming scarcer though.

LubosD, 2021-08-07:
My 2 cents on DVB cards on CM4:

I've tried TBS 6985 with official drivers from TBS. It works fine, but I'm still very disappointed. It DOES NOT work if I place the card behind a [PCE3PCE PCIe switch](https://www.aliexpress.com/item/32829751253.html?spm=a2g0s.9042311.0.0.27424c4dSl9pHa) from AliExpress. The card is simply not shown in `lspci`. There are some I/O BAR related errors in `dmesg`, but that's to be expected.

I tried another DVB-S card (DVBsky something with CX23885 and a Montage demod). Again the same situation - the card is not detected if placed into the PCIe switch, but I don't see any obvious reason why in `dmesg`.

That being said, anything else placed in the PCIe switch works: I've tried a SATA controller, an RS232 controller, an Ethernet NIC etc., all was detected properly. So I thought maybe it's a general problem of DVB cards in this PCIe switch, so I tried using the switch + a DVB card in my desktop PC. There it worked OK.

So only the combination of RPi CM4 + PCIe switch + DVB card doesn't work for me. Change anything in the equation and it does.

FrancescoRestelli, 2021-08-07:
@LubosD try this setting the max-link-speed = <1>; as described here:
 https://www.raspberrypi.org/forums/viewtopic.php?t=300061

seems to work for most people

LubosD, 2021-08-07:
> @LubosD try this setting the max-link-speed = <1>; as described here:
> https://www.raspberrypi.org/forums/viewtopic.php?t=300061
> 
> seems to work for most people

Yes, that made my card work when plugged directly into CM4 I/O board, but not when behind the PCIe switch. When the card is in the switch, it doesn't even appear in this listing:

```
[    1.207043] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.211242] pci 0000:01:00.0: [12d8:2404] type 01 class 0x060400
[    1.215765] pci 0000:02:01.0: [12d8:2404] type 01 class 0x060400
[    1.216672] pci 0000:02:02.0: [12d8:2404] type 01 class 0x060400
[    1.217570] pci 0000:02:03.0: [12d8:2404] type 01 class 0x060400
```

There should be a 1131:7160 here, but it's not. Same problem with my DVBsky card :-(

FrancescoRestelli, 2021-08-07:
sounds more like an issue with the raiser card then the dvb card then.

i used that one and it worked at the time
https://www.amazon.de/gp/product/B07VNGB9KY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

from the picture i see some logic ic on your card, maybe in the driver one can force the link speed to 1 or similar 

sorry i cannot help more then

6by9, 2021-08-08:
As [above](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/57#issuecomment-773996715), I also had an issue with DVB cards behind PCIe switches, and it appeared to be related to the interrupt mapping rather than anything else.
This affected the Hauppauge WinTV QuadHD as it is 2 tuners behind a switch on one card, and a DVBSky S952 behind an ASMedia ASM1184e switch. In both cases though they enumerated, but wouldn't probe the driver as it couldn't get the necessary resources.

PCIe supports a couple of mechanisms for routing interrupts, so it's probably just one combination that doesn't work. I haven't had the time to investigate, and other things are significantly higher priorities at present.
All the kernel code is open source, so if someone else has the desire to work out where and why the error is being raised, then that would save a fair amount of triage time.

LubosD, 2021-08-12:
@6by9 My case is a little different, it's even more low level I'd say. On my CM4, my DVB cards behind the PCIe switch aren't even enumerated. (But the switch+DVB card work in my PC, so it's not a general problem of the switch.)

I dug deeper and found out that the PCIe switch returns vendor ID `0xffffffff` for the PCIe slot with the DVB card just like for empty slots (in `pci_bus_read_dev_vendor_id()`).

It seems the PCIe switch is getting initialized differently on the CM4 (compared to an x86 desktop) or something, which makes the PCIe switch completely shun my DVB cards. But I have no idea what the difference could be or where in the kernel source look next.

LubosD, 2021-08-13:
> It seems the PCIe switch is getting initialized differently on the CM4 (compared to an x86 desktop) or something, which makes the PCIe switch completely shun my DVB cards. But I have no idea what the difference could be or where in the kernel source look next.

Heureka, the device(s) get discovered after I run:

```
echo 1 > /sys/bus/pci/rescan
```

So maybe the switch is just a little slow, which is not a problem on much slowly booting x86 systems?

LubosD, 2021-08-22:
> 
> Heureka, the device(s) get discovered after I run:
> 
> ```
> echo 1 > /sys/bus/pci/rescan
> ```
> 
> So maybe the switch is just a little slow, which is not a problem on much more slowly booting x86 systems?

Nope, that's not the (only) problem. After rescanning, I get 

```
Unhandled fault: asynchronous external abort (0x1211) at 0x00000000
```

whenever the driver tries to access the register space of the DVB card behind the PCIe switch. This space is provided by something like `ioremap(pci_resource_start(...), pci_resource_len(...))`.

I find it suspicious that while the space assigned to the DVB card is marked as 64bit:
```
pci 0000:03:00.0: BAR 0: assigned [mem 0x600000000-0x6001fffff 64bit]
```

the assignment to the "pcieport", where the card is plugged into, doesn't carry this flag:

```
pcieport 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6001fffff]
pcieport 0000:01:00.0: BAR 8: assigned [mem 0x600000000-0x6001fffff]
pcieport 0000:02:01.0: BAR 8: assigned [mem 0x600000000-0x6001fffff]
```

Also, when I plug in a working non-DVB device into the PCIe switch, the BAR assignments are printed as "pci" instead of "pcieport":

```
pci 0000:02:01.0: BAR 9: assigned [mem 0x600000000-0x6000fffff 64bit pref]
```
No idea if my findings are of any significance.

6by9, 2021-11-18:
TBS6909 turned out to be dead :-( At least I got a refund.

Tried a combined rig with DVBSky S952 v3, TBS6985, and TBS6284 via a 4 way PCIe switch, alongside my 4 port SATA card (Marvell 9215).
DVB cards need the pcie-32bit-dma tweak. The SATA card refuses to work with that in place. Joys! I might be able to get storage via a USB3 card instead, but that card was in the office. Not a priority though.

geerlingguy, 2021-11-18:
> DVB cards need the pcie-32bit-dma tweak. The SATA card refuses to work with that in place.

It seems a lot of older cards (especially PCIe gen 1) require the tweak to work. Kind of annoying, but I think the solution is to either build specifically for the older card and sacrifice some newer gear (though there might be some SATA II card that works with it enabled), or give up the functionality from that older card.

pcmds, 2022-01-13:
IZI-FINGERPRINT IS A SOFTWARE DESIGNED FOR RECORDING VIDEO AND AUDIO FROM VARIOUS SOURCES SUCH AS EXTERNAL DEVICES, ONLINE MEDIAS, WEBCAM, YOUTUBE, LIVE STREAMS, ETC. 

IT RECORDS MULTIPLE CHANNELS AND DEVICES SIMULTANEOUSLY.

DEMO VIDEO ON YOUTUBE: https://www.youtube.com/watch?v=_5wVGrauRPo

6by9, 2022-01-14:
Whoops.
```
pi@raspberrypi:~ $ lspci -nn
00:00.0 PCI bridge [0604]: Broadcom Limited Device [14e4:2711] (rev 20)
01:00.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1184e PCIe Switch Port [1b21:1184]
02:01.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1184e PCIe Switch Port [1b21:1184]
02:03.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1184e PCIe Switch Port [1b21:1184]
02:05.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1184e PCIe Switch Port [1b21:1184]
02:07.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1184e PCIe Switch Port [1b21:1184]
03:00.0 Multimedia controller [0480]: TBS Technologies DVB-S2 4 Tuner PCIe Card [544d:6178]
pi@raspberrypi:~ $ sudo lspci -v
...
03:00.0 Multimedia controller: TBS Technologies DVB-S2 4 Tuner PCIe Card
	Subsystem: Device 6909:0001
	Flags: fast devsel
	Memory at 600000000 (32-bit, non-prefetchable) [disabled] [size=256K]
	Capabilities: [50] Power Management version 3
	Capabilities: [70] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [90] Express Endpoint, MSI 00
	Capabilities: [100] Device Serial Number 00-00-00-00-00-00-00-00
```
Not sure why it reports as a 4 tuner card as it's a TBS6909 octa DVB-S2 tuner.
Now to try getting the drivers to compile....

6by9, 2022-01-16:
I've created a monster!
[image]
Sadly it doesn't want to work through my ASM1184e PCIe switch, so I can't combine it with any other cards :-(
It required one little tweak to the driver to use ```dma_alloc_coherent(..., GFP_KERNEL)``` instead of ```pci_alloc_consistent()``` as the later uses the restricted GFP_ATOMIC heap.

(It looks to be an interrupt issue - complains of failing to enable MSI interrupt and falling back to polling, but then I2C transactions time out which would imply no interrupts happening)

6by9, 2022-01-16:
Er, it helps if you don't leave ```pci=nosmi``` lying around on your kernel command line if you want to enable MSI.

The monster just grew.
[image]
- Hauppauge winTV quadHD for 4 DVB-T2 tuners
- TBS 6909 for 8 DVB-S2 tuners
- DVBSky S952 v3 for 2 more DVB-S2 tuners.

The cx23885 in the Hauppauge card needs ```dtoverlay=pcie-32bit-dma```, and also seem to have an interrupt issue (nothing handles the triggered interrupt, so the kernel disables it). They do appear to route the interrupts through the 2 PCIe switches now though.
The other cards don't appear to require it, but the TBS6909 doesn't seem to actually work without it. 

My TBS6284 (quad DVB-T2) and TBS6985 aren't being recognised in the PCIe switch. I'll have to see if they work direct, otherwise it may be that they're just dead.

Now to sort out some form of storage around this device and see just how many channels I can record at once :-)

6by9, 2022-01-16:
A VL805 works fine in 32bit DMA mode, so that deals with storage.

Add USB and I can connect all my USB tuners too!
[image]
```
pi@raspberrypi:~ $ ls /dev/dvb*
adapter0  adapter10  adapter12  adapter14  adapter16  adapter18  adapter2   adapter3  adapter5  adapter7  adapter9
adapter1  adapter11  adapter13  adapter15  adapter17  adapter19  adapter20  adapter4  adapter6  adapter8
```
```
pi@raspberrypi:~ $ lspci -tv
-[0000:00]---00.0-[01-09]----00.0-[02-09]--+-01.0-[03-06]----00.0-[04-06]--+-01.0-[05]----00.0  Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder
                                           |                               \-02.0-[06]----00.0  Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder
                                           +-03.0-[07]----00.0  VIA Technologies, Inc. VL805 USB 3.0 Host Controller
                                           +-05.0-[08]----00.0  TBS Technologies DVB-S2 4 Tuner PCIe Card
                                           \-07.0-[09]----00.0  Spin Master Ltd. PCIe Video Bridge
```
```
pi@raspberrypi:~ $ lsusb -tv
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
    |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 4, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 4, If 0, Class=Vendor Specific Class, Driver=dvb_usb_dvbsky, 480M
            |__ Port 2: Dev 5, If 0, Class=Vendor Specific Class, Driver=em28xx, 480M
            |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=dw2102, 480M
            |__ Port 4: Dev 7, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 3: Dev 8, If 0, Class=Vendor Specific Class, Driver=em28xx, 480M
                |__ Port 4: Dev 9, If 0, Class=Vendor Specific Class, Driver=em28xx, 480M
```
I guess 9 DVB-T2 tuners is a touch excessive when there are only 7 active muxes in this area.

6by9, 2022-01-16:
Last update for now. 
TBS6284 quad DVB-T2 and TBS6985 quad DVB-S2 work fine when directly connected to the CM4, so there appears to be something about the PCIe switch that they don't like.
I do have an alternate chipset PCIe switch somewhere, so may try that (if I can find it).

FrancescoRestelli, 2022-01-19:
hi all,

i think i made a step forward in including the tbs 6984 using an asm1184e raiser board

it´s quite late here so i just give the quick version for now, also no stream test has been done just enumeration worked.

first i added 2 scripts  on the pi and installed setpci
https://www.alexforencich.com/wiki/en/pcie/set-speed
https://unix.stackexchange.com/a/474378

then i set all asmedia enum´s to linkspeed 1 
here an example for one enum:  
sudo ./setpci.sh 02:07.0 1
sudo ./pcireset.sh 02:07.0

maybe we can also just set that one port and keep the pi and the other ports on gen2 speed?

and sure enough after that it appears to be alive! :)
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
02:01.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
02:03.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
02:05.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
02:07.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
07:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 02)

hope that helps and let me know how the testing goes

br stinger1/stinger151/francesco

6by9, 2022-01-19:
@FrancescoRestelli Thank you!

I'd started reading through the information about PCI configuration blocks, but had some real work to do and therefore dropped it.
I'll give it a go tonight when at home - I can see cascaded PCIe switches in my future to accommodate 5 PCIe tuner cards. Shame I smashed one of the PCIe connectors off my 3 way switch, so I won't be able to connect the USB3 card as well.
AIUI Yes you should be able to drop the speed of just one port to keep as much bandwidth as possible.

6by9, 2022-01-20:
Hmm, some further tweaking required.
Trying the above with a TBS6909 in slot 3, a TBS6985 in slot 1, and TBS6284 in slot 4, the device reset appears to reset the TBS6909, gives a kernel ```Unhandled fault: asynchronous external abort```, and a second instance shows up on port 1.

Run it with just the 6284 and 6985 connected, only one card is detected at a time, but they don't initialise happily
```
[  106.035017] pci_bus 0000:06: busn_res: [bus 06] is released
[  106.608485] pci 0000:02:07.0: [1b21:1184] type 01 class 0x060400
[  106.608669] pci 0000:02:07.0: enabling Extended Tags
[  106.608896] pci 0000:02:07.0: PME# supported from D0 D3hot D3cold
[  106.611845] pcieport 0000:02:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[  106.618489] pci 0000:02:07.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[  106.618877] pci 0000:06:00.0: [1131:7160] type 00 class 0x048000
[  106.618974] pci 0000:06:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit]
[  106.619353] pci 0000:06:00.0: supports D1 D2
[  106.619371] pci 0000:06:00.0: PME# supported from D0 D1 D2
[  106.623284] PCI: bus6: Fast back to back transfers disabled
[  106.623311] pci_bus 0000:06: busn_res: [bus 06] end is updated to 06
[  106.623634] pci_bus 0000:07: busn_res: can not insert [bus 07-06] under [bus 02-06] (conflicts with (null) [bus 02-06])
[  106.626204] PCI: bus7: Fast back to back transfers enabled
[  106.626225] pci_bus 0000:07: busn_res: [bus 07-06] end is updated to 07
[  106.626243] pci_bus 0000:07: busn_res: can not insert [bus 07] under [bus 02-06] (conflicts with (null) [bus 02-06])
[  106.626266] pci 0000:02:07.0: devices behind bridge are unusable because [bus 07] cannot be assigned for them
[  106.626284] pcieport 0000:01:00.0: bridge has subordinate 06 but max busn 07
[  106.626337] pcieport 0000:01:00.0: BAR 8: no space for [mem size 0x00100000]
[  106.626353] pcieport 0000:01:00.0: BAR 8: failed to assign [mem size 0x00100000]
[  106.626370] pcieport 0000:02:01.0: BAR 8: no space for [mem size 0x00100000]
[  106.626383] pcieport 0000:02:01.0: BAR 8: failed to assign [mem size 0x00100000]
[  106.626398] pci 0000:06:00.0: BAR 0: no space for [mem size 0x00100000 64bit]
[  106.626412] pci 0000:06:00.0: BAR 0: failed to assign [mem size 0x00100000 64bit]
[  106.626427] pci 0000:02:07.0: PCI bridge to [bus 07]
[  106.663543] SAA716x Budget 0000:06:00.0: SAA716x PCI Initialization failed
[  185.206854] pci 0000:02:07.0: [1b21:1184] type 01 class 0x060400
[  185.207037] pci 0000:02:07.0: enabling Extended Tags
[  185.207265] pci 0000:02:07.0: PME# supported from D0 D3hot D3cold
[  185.210077] pcieport 0000:02:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[  185.215664] pci 0000:02:07.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[  185.218720] pci_bus 0000:06: busn_res: [bus 06] end is updated to 06
[  185.219025] pci_bus 0000:07: busn_res: can not insert [bus 07-06] under [bus 02-06] (conflicts with (null) [bus 02-06])
[  185.221840] PCI: bus7: Fast back to back transfers enabled
[  185.221863] pci_bus 0000:07: busn_res: [bus 07-06] end is updated to 07
[  185.221883] pci_bus 0000:07: busn_res: can not insert [bus 07] under [bus 02-06] (conflicts with (null) [bus 02-06])
[  185.221908] pci 0000:02:07.0: devices behind bridge are unusable because [bus 07] cannot be assigned for them
[  185.221929] pcieport 0000:01:00.0: bridge has subordinate 06 but max busn 07
[  185.221986] pci 0000:02:07.0: PCI bridge to [bus 07]
```

Kitoko86, 2022-12-06:
Hi,

I succesfully installed a TBS6902 on Raspberry Pi OS with 5.15.79-v7l+ compiled kernel but I'm facing a strange issue with this device. I can tune a satellite frequency however, no matter the application that I used, the signal doesn't get lock. There is no PCIe switch on my board.

lsmod command output:
```
pi@raspberrypi:~/minisatip $ lsmod
Module                  Size  Used by
8021q                  32768  0
garp                   16384  1 8021q
stp                    16384  1 garp
llc                    16384  2 garp,stp
brcmfmac              335872  0
vc4                   294912  11
brcmutil               24576  1 brcmfmac
snd_soc_hdmi_codec     20480  2
cec                    49152  1 vc4
cfg80211              794624  1 brcmfmac
drm_kms_helper        286720  2 vc4
av201x                 16384  2
rfkill                 32768  4 cfg80211
snd_soc_core          237568  2 vc4,snd_soc_hdmi_codec
v3d                    81920  8
snd_compress           20480  1 snd_soc_core
gpu_sched              45056  1 v3d
snd_bcm2835            28672  0
snd_pcm_dmaengine      16384  1 snd_soc_core
tbsecp3                49152  11
snd_pcm               122880  5 snd_compress,snd_pcm_dmaengine,snd_soc_hdmi_codec,snd_bcm2835,snd_soc_core
i2c_brcmstb            16384  0
tas2101                20480  3 tbsecp3
gx1133                 24576  1 tbsecp3
raspberrypi_hwmon      16384  0
snd_timer              36864  1 snd_pcm
dvb_core              114688  1 tbsecp3
i2c_mux                16384  1 gx1133
vc_sm_cma              32768  0
snd                    86016  6 snd_compress,snd_soc_hdmi_codec,snd_timer,snd_bcm2835,snd_soc_core,snd_pcm
videodev              249856  0
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
mc                     45056  2 videodev,dvb_core
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
nvmem_rmem             16384  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
drm                   548864  14 v3d,vc4,gpu_sched,drm_kms_helper
i2c_dev                20480  0
fuse                  135168  3
drm_panel_orientation_quirks    16384  1 drm
backlight              20480  2 drm_kms_helper,drm
ip_tables              28672  0
x_tables               32768  1 ip_tables
ipv6                  528384  46
```
dmesg command output:
```
pi@raspberrypi:~/minisatip $ dmesg | grep -i tbs
[    5.516420] TBSECP3 driver 0000:01:00.0: enabling device (0140 -> 0142)
[    5.516511] TBSECP3 driver 0000:01:00.0: TurboSight TBS 6902 DVB-S/S2 
[    5.806780] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    5.926694] TBSECP3 driver 0000:01:00.0: MAC address 00:22:ab:f1:3d:68
[    6.078838] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6902 DVB-S/S2 )...
[    6.079159] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    6.231250] TBSECP3 driver 0000:01:00.0: MAC address 00:22:ab:f1:3d:69
[    6.344528] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6902 DVB-S/S2 )...
[    6.344812] TBSECP3 driver 0000:01:00.0: TurboSight TBS 6902 DVB-S/S2 : PCI 0000:01:00.0, IRQ 79, MMIO 0x0
[   10.384758] TBSECP3 driver 0000:01:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (950000..2150000)
[   11.054730] TBSECP3 driver 0000:01:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)

```

FrancescoRestelli, 2022-12-06:
@Kitoko86  make sure you have set "dtoverlay=pcie-32bit-dma" in /boot/config.txt if i remember right this happend when the dma range is wrong and therefore no data can flow

[6 more replies]