geerlingguy/raspberry-pi-pcie-devices #59: Wavshare

matthiasjauch, 2021-01-15:
https://www.waveshare.com/compute-module-4-poe-board.htm

mi-hol, 2021-01-15:
@matthiasjauch maybe you wanted to change the title to give a clear hint on what is is about?
I.e. "Waveshare Raspberry Pi Compute Module 4 IO Board With PoE Feature"

matthiasjauch, 2021-01-15:
hi,

I think the principle of the Raspberry Pi with IO board and CM has more future than the classic Raspi. I think these are the entry-level computers of today - besides that I favor the risc-v architecture.

best regards from switzerland

p.s. my brand new computer is a digiroule 2U

> Am 15.01.2021 um 15:27 schrieb mi-hol <notifications@github.com>:
> 
> maybe you wanted to change the title to give a clear hint on what is is about?
> I.e. "Waveshare Raspberry Pi Compute Module 4 IO Board With PoE Feature"

mi-hol, 2021-01-15:
@matthiasjauch title if issue has still typo "Wavshare" and is unclear

matthiasjauch, 2021-01-15:
Sorry bud i am a old school it man - do what you want...

geerlingguy, 2021-01-15:
Please see https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/25#issuecomment-760921807 — this is in the official 'neat CM4 projects/accessories' thread now :)

closed 2021-01-15 by geerlingguy

matthiasjauch, 2021-01-15:
nice

br oldtimer

thanks to Niklaus Wirth and his project Oberon