geerlingguy/raspberry-pi-pcie-devices #60: No boot when with some PCIe cards

Corsaire01, 2021-01-20:
I found troubles with some PCIe cards, when inserted the Pi boots up to the rainbow screen but stays here. The emmc access LED is on in fixed lit position.

trouble cards :
- Earthsoft PT3 DVB tuner  https://earthsoft.jp/PT3/specification.html
- Intel® 82574L Gigabit Ethernet Controller  https://www.intel.com/content/www/us/en/products/network-io/ethernet/gigabit-adapters/ct-desktop.html

Is there something wrong in the firmware regarding the handling of some PCIe cards ?

PixlRainbow, 2021-01-21:
[Enable verbose boot messages](https://raspberrypi.stackexchange.com/questions/77655/how-to-enable-and-see-the-bootup-messages-of-the-system-console) to see any debug messages, and [turn off the rainbow boot splash](https://scribles.net/silent-boot-up-on-raspbian-stretch/) so that it doesn't obscure the text.

Corsaire01, 2021-01-21:
> Note that certain cards may consume more power than the Pi can supply and may need a powered PCIe extension.

Yeah, I thought about that one, so I put the card on a Pcie extension cord with power (the kind used by miners) but same result.

Corsaire01, 2021-01-21:
Tried to Enable verbose boot messages and turn off the rainbow boot splash, all I get is a blank screen.

Corsaire01, 2021-01-23:
I had some doubts so I pushed my testings and I found an interesting piece... 

My problems happens with Ubuntu for the raspi, raspbian does boot.
Is something wrong in the boot eeprom in Ubuntu ?