geerlingguy/raspberry-pi-pcie-devices #62: Test GPU (ASRock Rack M2_VGA)

geerlingguy, 2021-01-24:
This is a fun one—it's definitely not the kind of GPU where people would ask "will it run Crysis", especially since a headline feature is _2D graphics acceleration_ (not even 3D yet!).

But the [ASRock Rack M2_VGA](https://www.asrockrack.com/general/productdetail.asp?Model=M2_VGA#Specifications) is an M.2 form factor graphics card that sports a lone VGA port and 16MB (yeah, MB, not GB) of DDR graphics memory.

[image]

I doubt it will even be as fast as the built-in graphics on the Pi, but it would be interesting to see if it works. It uses the [SiliconMotion SM750](https://www.versalogic.com/wp-content/themes/vsl-new/assets/resources/support/mpee-v5/SM750_Datasheet_v.1.4.pdf) graphics chip, which actually supports up to two DVI/HDMI/VGA displays, as well as two video inputs which can be overlaid on those outputs.

The chip is mostly known for being helpful in embedded or server graphics situations, and is not a 'powerhouse' by any means. Just a little utilitarian chip that sips less than 2W of power maximum (making it suitable for lower-power scenarios where you still need a display or two, but don't do gaming or ML/AI applications on it).

It seems like there's a mainline driver since a few years ago (SM750), and it would be interesting to see if it 'just works' (compared to the other cards). It seems like the chip itself uses BIOS (and was designed in 2012), so that gives me a little pause.

But the chip is simple enough and documented enough that I wonder if we could bring it up manually if the driver starts barfing on memory allocations like all the other GPUs I've tested have (AMD and Nvidia).

geerlingguy, 2021-01-24:
Other server vendors also build the same chip (SM750) into full PCIe cards, e.g. [Sunix's VGA0419L PCI-E card](https://www.sunix.com/en/product_detail.php?cid=1&kid=2&gid=15&pid=1763).

And in [this video](https://www.youtube.com/watch?v=RK4yQnesWuk), a rep demos a USB version of the video card and a few other form factors.

Finally, some [commentary from Phoronix](https://www.phoronix.com/forums/forum/linux-graphics-x-org-drivers/x-org-drm/31332-silicon-motion-extends-linux-graphics-driver) seems to indicate the early versions of the driver were a bit rough (but hopefully adequate for ARM64... we'll see!).

geerlingguy, 2021-01-24:
I was hoping the card would leech power off the PCI-E bus but it looks like you _have_ to use the included 4-pin molex to 3-pin jumper to power the board, otherwise it doesn't show up on the Pi at all.

I'm using an external / separate PSU and it looks like it puts 5v on the outer pins and GND in the middle.

Anyways, here's the deets:

```
pi@raspberrypi:~ $ sudo lspci -vvvv
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20) (prog-if 00 [Normal decode])
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 00000000-00000fff
	Memory behind bridge: fff00000-000fffff
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna- CRSVisible+
		RootCap: CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via WAKE# ARIFwd-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled ARIFwd-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [180 v1] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
	Capabilities: [240 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=8us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=1us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us

01:00.0 VGA compatible controller: Silicon Motion, Inc. SM750 (rev a1) (prog-if 00 [VGA controller])
	Subsystem: Silicon Motion, Inc. SM750
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 255
	Region 0: Memory at <unassigned> (32-bit, prefetchable) [disabled]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2- AuxCurrent=375mA PME(D0+,D1+,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
[... 14 lines cut ...]
		LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [b0] MSI-X: Enable- Count=1 Masked-
		Vector table: BAR=5 offset=00000000
		PBA: BAR=5 offset=00000000
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [140 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
```

geerlingguy, 2021-01-24:
```
$ dmesg
...
[    1.013251] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.013270] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.013327] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x0603ffffff -> 0x00f8000000
[    1.013383] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0100000000
[    1.047197] brcm-pcie fd500000.pcie: link up, 2.5 GT/s x1 (SSC)
[    1.047488] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.047503] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.047520] pci_bus 0000:00: root bus resource [mem 0x600000000-0x603ffffff] (bus address [0xf8000000-0xfbffffff])
[    1.047572] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.047792] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.051200] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.051391] pci 0000:01:00.0: [126f:0750] type 00 class 0x030000
[    1.051459] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x03ffffff pref]
[    1.051487] pci 0000:01:00.0: reg 0x14: [mem 0x00000000-0x001fffff]
[    1.051576] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
[    1.051768] pci 0000:01:00.0: supports D1
[    1.051779] pci 0000:01:00.0: PME# supported from D0 D1 D3hot
[    1.051973] pci 0000:01:00.0: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[    1.055078] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.055115] pci 0000:00:00.0: BAR 8: no space for [mem size 0x06000000]
[    1.055127] pci 0000:00:00.0: BAR 8: failed to assign [mem size 0x06000000]
[    1.055143] pci 0000:01:00.0: BAR 0: no space for [mem size 0x04000000 pref]
[    1.055155] pci 0000:01:00.0: BAR 0: failed to assign [mem size 0x04000000 pref]
[    1.055166] pci 0000:01:00.0: BAR 1: no space for [mem size 0x00200000]
[    1.055178] pci 0000:01:00.0: BAR 1: failed to assign [mem size 0x00200000]
[    1.055190] pci 0000:01:00.0: BAR 6: no space for [mem size 0x00010000 pref]
[    1.055201] pci 0000:01:00.0: BAR 6: failed to assign [mem size 0x00010000 pref]
[    1.055214] pci 0000:00:00.0: PCI bridge to [bus 01]
```

So might need to expand the BAR. Good thing I don't see any IO BAR in there 😅

Anyways, that's all for now, I'm going to recompile the kernel later (maybe tomorrow, maybe next week) and add in the SM750 driver, and see where we end up from there. All using the 64-bit Pi OS beta for now.

geerlingguy, 2021-01-24:
Enabled:

```
Device Drivers > Staging drivers > Silicon Motion SM750 framebuffer support
```

PixlRainbow, 2021-01-25:
> as well as two video inputs which can be **_overlaid_** on those outputs.

Pi HUD? 🤔

69K-ram, 2021-01-28:
I was just about to suggest this to "GPU" to you after seeing the LTT video about it, but you already have one lol. 
Good luck! Maybe this one is so low spec it will just work?

geerlingguy, 2021-01-28:
Reference for anyone who doesn't subscribe to LTT: [THIS is a Graphics Card..??](https://www.youtube.com/watch?v=DXo2m_XlH4s)

(I wish I had a team of tech nuts who could do all the testing and script writing for me... just jelly :)

geerlingguy, 2021-01-28:
Well that's annoying. After recompiling the kernel and booting with the VGA card powered up and plugged in, the Pi starts booting to the point where the display starts flashing the cursor... then completely freezes.

Here's what that looks like:

[image]

(That signal's coming through the HDMI port on the IO Board, not from the VGA output.)

I've tried rebooting multiple times, but it always gets stuck a few seconds in and then there's no response from anything. If I turn off power to the VGA card, it boots just fine.

geerlingguy, 2021-01-28:
I used `sudo raspi-config` to boot to console (instead of GUI), and also to disable the splash screen, and then rebooted with the VGA card enabled so I could see what was coming out in the console.

Every single time it gets stuck _around_ here:

[image]

```
[   OK   ] Started Forward Password Requests to Plymouth Directory Watch.
```

So maybe something that's trying to initialize some graphics rendering (the Plymouth Boot Screen...). The annoying thing is there doesn't seem to be a separate driver for the SM750 so I can't just disable it using a modprobe blacklist.

This could be the shortest test cycle ever—I can't yet think of anything else to try :/

geerlingguy, 2021-01-28:
Just noting that without the card powered up, when it boots, it hits that 'Show Plymouth Boot Screen' in the log, then the screen flashes to black, then the screen comes back, outputs some more startup data, then ends with:

```
           Starting Terminate Plymouth Boot Screen...
```

So something in the Plymouth Boot Screen process seems to kinda explode when this card is active.

geerlingguy, 2021-01-28:
Uninstalling plymouth in light of the fact that I don't see how to completely disable it otherwise (all the online guides assume you're using GRUB, but Pi OS does not):

```
sudo apt-get purge -y plymouth*
```

geerlingguy, 2021-01-28:
Well, now it locks up completely at:

```
[   OK   ] Started udev Kernel Device Manager
```

So at least it doesn't lock up at the Plymouth stage anymore :D

geerlingguy, 2021-01-28:
I'm recompiling the kernel _without_ `Silicon Motion SM750 framebuffer support` to see if it makes a difference.

geerlingguy, 2021-01-28:
Okay, so without that compiled in, the thing boots with the board powered up... and even better, with the updated kernel (which has the better BAR space mapping committed a few months ago), the registers all look good:

```
[    1.245812] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.247456] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.249244] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.251088] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0100000000
[    1.286525] brcm-pcie fd500000.pcie: link up, 2.5 GT/s PCIe x1 (SSC)
[    1.287844] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.288874] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.289826] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.291763] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.292968] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.297601] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.299700] pci 0000:01:00.0: [126f:0750] type 00 class 0x030000
[    1.300732] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x03ffffff pref]
[    1.302611] pci 0000:01:00.0: reg 0x14: [mem 0x00000000-0x001fffff]
[    1.303687] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
[    1.305835] pci 0000:01:00.0: supports D1
[    1.306794] pci 0000:01:00.0: PME# supported from D0 D1 D3hot
[    1.307945] pci 0000:01:00.0: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[    1.323777] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.324778] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x605ffffff]
[    1.326596] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x603ffffff pref]
[    1.328471] pci 0000:01:00.0: BAR 1: assigned [mem 0x604000000-0x6041fffff]
[    1.330393] pci 0000:01:00.0: BAR 6: assigned [mem 0x604200000-0x60420ffff pref]
[    1.332347] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.333381] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x605ffffff]
```

Going to do some testing and see what happens with X.org.

geerlingguy, 2021-01-28:
Well, nothing. It only outputs over HDMI.

geerlingguy, 2021-01-28:
Hmm... driver being worked on over here: https://gitlab.com/sudipm/sm750/-/tree/sm750/drivers/gpu/drm/sm750 (from 2 years ago, ha!).

See: https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/staging/sm750fb/TODO

I emailed one of the contacts listed in the driver docs... who knows if I'll get a response.

geerlingguy, 2021-01-29:
So... I tried pulling the patch from that drm driver and popping it into the 5.10.y branch—but when I build I get:

```
  CC [M]  drivers/gpu/drm/sm750/smi_drv.o
drivers/gpu/drm/sm750/smi_drv.c:12:10: fatal error: drm/drmP.h: No such file or directory
 #include <drm/drmP.h>
          ^~~~~~~~~~~~
compilation terminated.
make[4]: *** [scripts/Makefile.build:279: drivers/gpu/drm/sm750/smi_drv.o] Error 1
make[3]: *** [scripts/Makefile.build:496: drivers/gpu/drm/sm750] Error 2
make[3]: *** Waiting for unfinished jobs....
```

So it looks like that driver may be built for an older kernel version and isn't compatible with the latest kernel :/

geerlingguy, 2021-01-29:
Looks like that change was in kernel 5.5—so maybe running the current stable kernel version (5.4) would let it work? It's worth a try... See https://github.com/DisplayLink/evdi/issues/185

geerlingguy, 2021-01-29:
Yeah... so doing a few manual patches to that patch to try to make it compile leads to dozens more compile errors, so I'm going to drop trying to get it working on 5.10.y, and jump back to 5.4.y.

geerlingguy, 2021-01-29:
With the patch in place, the kernel build on 5.4.y seemed to work. Copied it over to the Pi, rebooted and...

```
[    1.324708] pci 0000:00:00.0: BAR 8: no space for [mem size 0x06000000]
[    1.325653] pci 0000:00:00.0: BAR 8: failed to assign [mem size 0x06000000]
[    1.327425] pci 0000:01:00.0: BAR 0: no space for [mem size 0x04000000 pref]
[    1.329274] pci 0000:01:00.0: BAR 0: failed to assign [mem size 0x04000000 pref]
[    1.331175] pci 0000:01:00.0: BAR 1: no space for [mem size 0x00200000]
[    1.332164] pci 0000:01:00.0: BAR 1: failed to assign [mem size 0x00200000]
[    1.334106] pci 0000:01:00.0: BAR 6: no space for [mem size 0x00010000 pref]
[    1.336053] pci 0000:01:00.0: BAR 6: failed to assign [mem size 0x00010000 pref]
```

Have to [adjust the BAR space manually](https://gist.github.com/geerlingguy/9d78ea34cab8e18d71ee5954417429df) on this old kernel.

geerlingguy, 2021-01-29:
Oh... well... that's not right:

```
$ uname -r
5.10.9-v8+
```

geerlingguy, 2021-01-29:
Ugh... I can't get these drivers to build for the life of me!

```
In file included from drivers/gpu/drm/sm750/smi_drv.c:15:
drivers/gpu/drm/sm750/smi_drv.h:178:32: error: field ‘mem_global_ref’ has incomplete type
    struct drm_global_reference mem_global_ref;
                                ^~~~~~~~~~~~~~
drivers/gpu/drm/sm750/smi_drv.h:179:29: error: field ‘bo_global_ref’ has incomplete type
    struct ttm_bo_global_ref bo_global_ref;
                             ^~~~~~~~~~~~~
drivers/gpu/drm/sm750/smi_drv.h:296:50: warning: ‘struct reservation_object’ declared inside parameter list will not be visible outside of this definition or declaration
      uint32_t flags, struct sg_table *sg, struct reservation_object *resv, struct smi_bo **psmibo);
                                                  ^~~~~~~~~~~~~~~~~~
drivers/gpu/drm/sm750/smi_drv.c: In function ‘smi_drm_freeze’:
drivers/gpu/drm/sm750/smi_drv.c:107:2: error: implicit declaration of function ‘drm_kms_helper_poll_disable’; did you mean ‘drm_fb_helper_pan_display’? [-Werror=implicit-function-declaration]
  drm_kms_helper_poll_disable(dev);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  drm_fb_helper_pan_display
drivers/gpu/drm/sm750/smi_drv.c: In function ‘smi_drm_resume’:
drivers/gpu/drm/sm750/smi_drv.c:154:2: error: implicit declaration of function ‘drm_kms_helper_poll_enable’; did you mean ‘drm_fb_helper_fill_info’? [-Werror=implicit-function-declaration]
  drm_kms_helper_poll_enable(dev);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~
  drm_fb_helper_fill_info
drivers/gpu/drm/sm750/smi_drv.c:144:21: warning: unused variable ‘sdev’ [-Wunused-variable]
  struct smi_device *sdev = dev->dev_private;
                     ^~~~
drivers/gpu/drm/sm750/smi_drv.c: At top level:
drivers/gpu/drm/sm750/smi_drv.c:276:39: error: ‘DRIVER_IRQ_SHARED’ undeclared here (not in a function); did you mean ‘TIMER_IRQSAFE’?
  .driver_features = DRIVER_HAVE_IRQ | DRIVER_IRQ_SHARED | DRIVER_GEM |
                                       ^~~~~~~~~~~~~~~~~
                                       TIMER_IRQSAFE
drivers/gpu/drm/sm750/smi_drv.c:277:9: error: ‘DRIVER_PRIME’ undeclared here (not in a function); did you mean ‘DRIVER_NAME’?
         DRIVER_PRIME | DRIVER_MODESET,
         ^~~~~~~~~~~~
         DRIVER_NAME
drivers/gpu/drm/sm750/smi_drv.c:309:3: error: ‘struct drm_driver’ has no member named ‘gem_prime_res_obj’; did you mean ‘gem_prime_export’?
  .gem_prime_res_obj = smi_gem_prime_res_obj,
   ^~~~~~~~~~~~~~~~~
   gem_prime_export
drivers/gpu/drm/sm750/smi_drv.c:309:23: error: initialization of ‘struct sg_table * (*)(struct drm_gem_object *)’ from incompatible pointer type ‘struct reservation_object * (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-types]
  .gem_prime_res_obj = smi_gem_prime_res_obj,
                       ^~~~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/sm750/smi_drv.c:309:23: note: (near initialization for ‘driver.gem_prime_get_sg_table’)
In file included from drivers/gpu/drm/sm750/smi_fbdev.c:15:
drivers/gpu/drm/sm750/smi_drv.h:178:32: error: field ‘mem_global_ref’ has incomplete type
    struct drm_global_reference mem_global_ref;
                                ^~~~~~~~~~~~~~
drivers/gpu/drm/sm750/smi_drv.h:179:29: error: field ‘bo_global_ref’ has incomplete type
    struct ttm_bo_global_ref bo_global_ref;
                             ^~~~~~~~~~~~~
drivers/gpu/drm/sm750/smi_drv.h:296:50: warning: ‘struct reservation_object’ declared inside parameter list will not be visible outside of this definition or declaration
      uint32_t flags, struct sg_table *sg, struct reservation_object *resv, struct smi_bo **psmibo);
                                                  ^~~~~~~~~~~~~~~~~~
drivers/gpu/drm/sm750/smi_fbdev.c: In function ‘smifb_create_object’:
drivers/gpu/drm/sm750/smi_fbdev.c:48:21: warning: unused variable ‘cdev’ [-Wunused-variable]
  struct smi_device *cdev = dev->dev_private;
                     ^~~~
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:266: drivers/gpu/drm/sm750/smi_drv.o] Error 1
make[4]: *** Waiting for unfinished jobs....
drivers/gpu/drm/sm750/smi_fbdev.c: In function ‘smifb_create’:
drivers/gpu/drm/sm750/smi_fbdev.c:161:2: error: implicit declaration of function ‘drm_fb_helper_fill_fix’; did you mean ‘drm_fb_helper_fill_info’? [-Werror=implicit-function-declaration]
  drm_fb_helper_fill_fix(info, fb->pitches[0], fb->format->depth);
  ^~~~~~~~~~~~~~~~~~~~~~
  drm_fb_helper_fill_info
  AR      drivers/i3c/built-in.a
drivers/gpu/drm/sm750/smi_fbdev.c:163:2: error: implicit declaration of function ‘drm_fb_helper_fill_var’; did you mean ‘drm_fb_helper_fill_info’? [-Werror=implicit-function-declaration]
  drm_fb_helper_fill_var(info, &gfbdev->helper, sizes->fb_width,
  ^~~~~~~~~~~~~~~~~~~~~~
  drm_fb_helper_fill_info
drivers/gpu/drm/sm750/smi_fbdev.c:175:58: error: ‘struct ttm_buffer_object’ has no member named ‘vma_node’
  drm_vma_offset_remove(&bo->bo.bdev->vma_manager, &bo->bo.vma_node);
                                                          ^
  CC [M]  drivers/gpu/drm/tiny/ili9341.o
drivers/gpu/drm/sm750/smi_fbdev.c: In function ‘smi_fbdev_destroy’:
drivers/gpu/drm/sm750/smi_fbdev.c:205:4: error: implicit declaration of function ‘drm_gem_object_unreference_unlocked’; did you mean ‘drm_gem_object_put_unlocked’? [-Werror=implicit-function-declaration]
    drm_gem_object_unreference_unlocked(gfb->obj);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    drm_gem_object_put_unlocked
  AR      drivers/idle/built-in.a
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:266: drivers/gpu/drm/sm750/smi_fbdev.o] Error 1
make[3]: *** [scripts/Makefile.build:500: drivers/gpu/drm/sm750] Error 2
make[3]: *** Waiting for unfinished jobs....
```

I think I may have to throw in the towel on this GPU as well.

Why are all the [okay but not great] working drivers only written for Windows?

69K-ram, 2021-01-29:
Respect for even getting this far, I wouldn't have slightest idea where to start, and I sure wouldn't have tried that long.

In the LTT video, he mentions it's using a Windows feature called [WARP](https://docs.microsoft.com/en-us/windows/win32/direct3darticles/directx-warp#casual-games). And the [Wikipedia article about WARP](https://en.wikipedia.org/wiki/Windows_Advanced_Rasterization_Platform) says it uses "just-in-time compilation to x86 machine code". Is it possible the driver is dependent on x86 instructions or something like that? I have no idea how the Linux driver for it works though.

PixlRainbow, 2021-01-30:
The thing is though, WARP is a DirectX feature, not a driver feature.

Coreforge, 2021-02-01:
Could you try disabling fbdev emulation? Since it crashes after udev, it could be the console trying to access the fbdev, which could lock up the whole pi if something doesn't work, like using 64bit writes. You'll have to add 
`drm_kms_helper.fbdev_emulation=0`
to cmdline.txt. Most things involving GRUB also just change the kernel command line, so you can do the same by adding the arguments to cmdline.txt

geerlingguy, 2021-02-01:
I added `drm_kms_helper.fbdev_emulation=0` to `/boot/cmdline.txt` on the Pi, and am building the latest 5.10.y kernel with the SM750 staging driver... we'll see how this goes.

geerlingguy, 2021-02-02:
@Coreforge - Sadly, that seems to have not made any difference:

https://user-images.githubusercontent.com/481677/106538867-9586bb00-64c2-11eb-9c1b-e6e1fcbbf034.mov

geerlingguy, 2021-02-19:
Digging through the driver code, I do notice a bunch of `memset_io()` calls in `sm750.c`:

```c
	if (info) {
		par = info->par;
		crtc = &par->crtc;
		cursor = &crtc->cursor;
		memset_io(cursor->vstart, 0x0, cursor->size);
		memset_io(crtc->vScreen, 0x0, crtc->vidmem_size);
		lynxfb_ops_set_par(info);
		fb_set_suspend(info, 0);
	}

	info = sm750_dev->fbinfo[1];

	if (info) {
		par = info->par;
		crtc = &par->crtc;
		cursor = &crtc->cursor;
		memset_io(cursor->vstart, 0x0, cursor->size);
		memset_io(crtc->vScreen, 0x0, crtc->vidmem_size);
		lynxfb_ops_set_par(info);
		fb_set_suspend(info, 0);
	}
```

Also:

```c
	pr_info("crtc->cursor.mmio = %p\n", crtc->cursor.mmio);
	crtc->cursor.maxH = crtc->cursor.maxW = 64;
	crtc->cursor.size = crtc->cursor.maxH * crtc->cursor.maxW * 2 / 8;
	crtc->cursor.vstart = sm750_dev->pvMem + crtc->cursor.offset;

	memset_io(crtc->cursor.vstart, 0, crtc->cursor.size);
	if (!g_hwcursor) {
		lynxfb_ops.fb_cursor = NULL;
		sm750_hw_cursor_disable(&crtc->cursor);
	}
```

And:

```c
	/* call chip specific mmap routine */
	err = hw_sm750_map(sm750_dev, pdev);
	if (err)
		return err;

	if (!sm750_dev->mtrr_off)
		sm750_dev->mtrr.vram = arch_phys_wc_add(sm750_dev->vidmem_start,
							sm750_dev->vidmem_size);

	memset_io(sm750_dev->pvMem, 0, sm750_dev->vidmem_size);

	pci_set_drvdata(pdev, sm750_dev);
```

Judging by experiences in both https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/4#issuecomment-766275193 and https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/6#issuecomment-728751399, perhaps changing some of this could help.

geerlingguy, 2021-02-20:
[paul maydaynight on YouTube](https://www.youtube.com/watch?v=Ozwyh_Pfe6g&lc=Ugw3VB5i-iE0uAfQ1c54AaABAg.9Jy6FAUr4Om9JycPxb2Hk5) asks if I've considered the work from the Linaro team on this driver...

I found [this listing of patches by gary guo](https://patches.linaro.org/patches/724/) and I might try to see if I can reach him.

Hacker1245, 2021-02-22:
> [paul maydaynight on YouTube](https://www.youtube.com/watch?v=Ozwyh_Pfe6g&lc=Ugw3VB5i-iE0uAfQ1c54AaABAg.9Jy6FAUr4Om9JycPxb2Hk5) asks if I've considered the work from the Linaro team on this driver...
> 
> I found [this listing of patches by gary guo](https://patches.linaro.org/patches/724/), like one [mentioning bar space addressing](https://patches.linaro.org/patch/132401/) and I might try to see if I can reach him.
> 
> **Edit**: The emails associated with those patches seem dead.

Not sure if that would help or if it is even the correct guy but maybe you can try contacting him via LinkedIn
https://www.linkedin.com/in/heyi-guo-70129288

geerlingguy, 2021-02-22:
@Hacker1245 Worth a try, I sent an inmail message :)

Also got a response from ASRock, basically "This thing is meant for Servers and they're mostly Intel, soooo..." :D

kwinz, 2021-02-28:
@geerlingguy After reading through your complete journey (and I am really impressed with and deeply thankful for your perseverance) with the compute module 4 I think everything so far has been much, much harder for us because of this:

> I think we may be out of luck, in this case, as @pelwell mentions that the spec sheet for the BCM2711 says:
> Supports accessing external PCIe configuration space and memory space **(no support for I/O space).**
> [...] The PCI spec requires it, but not all platforms (especially non-x86 platforms) support that.

I think it's essential to communicate to the Raspberry foundation that we would really love to see that supported in the Pi 5 / Compute Module 5. They should know about our struggle. I mean maybe we could hack around that limitation with certain GPU model drivers, but the only true fix is to have a Broadcom SOC that properly implements the PCIe specification. Is the Pi foundation aware of this problem yet?

Coreforge, 2021-02-28:
With newer Cards the IO space shouldn't be the issue as it mostly isn't required. The radeon cards for example typically use the IO BAR for atom IO, but can also do the same over regular bars. I think the bigger issue is that the pcie controller doesn't support 64bit accesses, so you have to use the 32bit OS, which doesn't solve everything either, or replace every relevant 64bit access in the driver with two 32bit ones, which is tedious.

geerlingguy, 2021-02-28:
Yeah, I don't think most driver maintainers really care about the IO bar thing—the cards that depend on it are generally 10+ years old, and many (most?) drivers nowadays don't care if it's not available.

The bigger problem I've started hitting (besides little bugs that affect all ARM64 systems like the patch required to get the Intel WiFi driver working) is the memory issues @Coreforge alludes to. I'm pretty sure that's what's hitting us on the GPUs at this point, and it's also what's making it hard to get things like the Google Coral TPU working, and what we had to spend some time back-and-forth with on the Broadcom RAID card I just covered recently.

kwinz, 2021-02-28:
Very interresting! Thanks for the update, I was wondering if you are talking the the Raspberry guys since they are the Broadcom customers and should be aware of the stuggles that we as users deal with. Maybe it could be fixed it the next revision if we talk to each other.

>bigger issue is that the pcie controller doesn't support 64bit accesses

@Coreforge Sorry for the naive question: Is that a hardware or a software (Kernel driver) issue? Do you have a link with more info on the pcie controller limitations?

Coreforge, 2021-03-01:
I think it's a hardware limitation, but I don't know much about that stuff either. As long as a card works on 32bit systems though, it shouldn't matter too much, it just might be a lot of work to replace all 64bit accesses. It doesn't work to just not use 64bit accesses for the whole kernel though, so they need to be individually replaced in the drivers. I don't have any more information on the controllers limitations either.

kwinz, 2021-03-04:
Thanks @Coreforge. Would the maintainers not resist to Raspberry Pi specific changes spread all over the Kernel in individual drivers? It seems to me there should be a better solution. But I don't know enough unfortunately.

PixlRainbow, 2021-03-04:
> It doesn't work to just not use 64bit accesses for the whole kernel though, so they need to be individually replaced in the drivers

Wait, what do you mean? What happens if you just modify the function that performs 64 bit reads such that it internally performs two 32 bit reads, but retains its original interface transparently.

pelwell, 2021-03-04:
@kwinz The raspberrypi/linux repo is open to hosting modest downstream patches to thirdparty drivers if it allows them to run on BCM2711.

Coreforge, 2021-03-04:
> Wait, what do you mean? What happens if you just modify the function that performs 64 bit reads such that it internally performs two 32 bit reads, but retains its original interface and apparent behavior transparently?

I thought this caused it to hang up, but that was only if I removed the writeq-stuff in memset_io (doesn't make too much sense why as the arm code doesn't even have anything bigger than writeb in that function and it works there). It doens't lock up if you do that, but it doesn't really help either. It could also be that I just replaced it with the wrong code, I don't really know any assembly, and I just copied the code from the writel function.

geerlingguy, 2021-04-23:
Weird coincidence: on YouTube, [Marco said he has a Mini PCIe card—Innodisk EMPV-1201](https://www.youtube.com/watch?v=MxcafwjWw24&lc=Ugx6_zqw2eOmKNWxV554AaABAg), that he may be able to send for testing.

I looked on their [Downloads page](https://www.innodisk.com/en/support_and_service/download) (search for 'EMPV-1201') and found a driver that includes Linux binaries for RHEL 6, Ubuntu 16, etc. (but no newer OSes...), but also includes the source—for an SM750!

Might have to see if I can build that source for the M2_VGA and get it working. Probably not, but it looks like the code is at least different than the sm750fb driver in the Linux kernel source...

piexil, 2021-04-26:
lo_hi_readq and lo_hi_writeq exsist as helpers for doing 64bit I/o over 32bit buses, the NVMe driver uses this to do all of it's 64bit mmio.

samveen, 2021-06-29:
> Weird coincidence: on YouTube, [Marco said he has a Mini PCIe card—Innodisk EMPV-1201](https://www.youtube.com/watch?v=MxcafwjWw24&lc=Ugx6_zqw2eOmKNWxV554AaABAg), that he may be able to send for [...]

@geerlingguy In case you haven't already looked at the the drivers in the package you mentioned, they are NOT kernel drivers, but Xorg drivers. These are not going to help with the kernel driver issues.

geerlingguy, 2021-06-29:
@samveen - Well shucks! Thanks for letting me know.

bobafetthotmail, 2021-07-27:
A question that is not related to raspberry Pi: I've been trying to understand if this card has UEFI GOP option ROM. I need a card to use in a system where CSM is disabled and I would like to not waste power and space with a real GPU.

Did you dump the firmware of this card (and can send it to me so I can have a look) or have you tried installing it in a PC and then disabled CSM from UEFI settings? If you see something on screen connected to this card with CSM disabled then it has this option ROM, otherwise it does not.

sfxworks, 2021-08-24:
Where can one buy this?

bobafetthotmail, 2021-08-25:
There are some on Ebay from a german seller

sfxworks, 2021-08-25:
Any other sources? I could have sworn I have seen these before for about 14 USD

TobleMiner, 2022-02-20:
Hey all,

I got hold of a SM750-based mini PCIe graphics card last weekend. Today I had some time to try it out with a CM4 and investigate the issues you are seeing here. I can confirm that the CM4 does indeed hang when trying to insert the sm750fb kernel module.  
However there is a simple (partial) fix. After doing a bit of debugging it seems that any writes more than 256 byte in size to memory BARs of the SM750 cause the BCM2711 to hang. Those kinds of writes happen when memset_io is called on a memory area with write combining enabled. The following patch maps the SM750 framebuffer BAR without enabling write combining:
```
diff --git a/drivers/staging/sm750fb/sm750_hw.c b/drivers/staging/sm750fb/sm750_hw.c
index 7136d751cff5..01e515ed7e9b 100644
--- a/drivers/staging/sm750fb/sm750_hw.c
+++ b/drivers/staging/sm750fb/sm750_hw.c
@@ -78,8 +78,8 @@ int hw_sm750_map(struct sm750_dev *sm750_dev, struct pci_dev *pdev)
 		sm750_dev->vidmem_start, sm750_dev->vidmem_size);
 
 	/* reserve the vidmem space of smi adaptor */
-	sm750_dev->pvMem = ioremap_wc(sm750_dev->vidmem_start,
-				      sm750_dev->vidmem_size);
+	sm750_dev->pvMem = ioremap(sm750_dev->vidmem_start,
+				   sm750_dev->vidmem_size);
 	if (!sm750_dev->pvMem) {
 		pr_err("Map video memory failed\n");
 		ret = -EFAULT;
```

Note however that this patch does not prevent large writes to the frambuffer BAR in general. An external application can still issue large writes that might cause a hang. It is however good enough to get a framebuffer console:
[image]
You might have to remap the framebuffer console to see anything at all if you have another display attached via HDMI: `sudo con2fbmap 1 1`

I'm not sure why the CM4 hangs when issuing large writes to the framebuffer BAR. I've not looked into it any further yet but am thinking about attaching a JTAG debugger to my CM4 to see what exactly the CPU is doing when it hangs. I might also do more tests with other random PCIe cards and see how they behave when issuing large writes to them.

In general I think this is a good point to continue debugging from. Maybe someone at Broadcom will find this infomation useful and can give some insight into what is happening.

geerlingguy, 2022-02-20:
@TobleMiner - Wow, thanks for taking up the mantle on this one—the write-combining fix is something I didn't even think to look for (most of the things I've been testing relate to switching to safer memcpy routines (e.g. what @Coreforge is doing with the Radeon driver in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/4). Interesting... I need to pull out this card again now!

Coreforge, 2022-02-20:
If you can find out something via JTAG, I'd be interested in that too. From what I've seen when debugging over JTAG with a second pi as a JTAG adapter, the cpu somehow goes into 32bit mode and I can't connect to it anymore. I haven't tried seeing what exactly happens when it crashes though. Maybe there are other parts than the cpu in the SoC that can be accessed over JTAG too?

TobleMiner, 2022-02-20:
I've just done some more tests with an Intel e1000e network card and a simple test driver that does nothing but map a single memory bar from that card. It then tries to perform writes of various sizes to that bar.
Behaviour is much the same as with the SM750. Everyting up to and including 256 byte works fine. Above that e.g. at 512 bytes the Pi freezes.  
Think this might be either an issue with (configuration of) the bus interface of the Broadcom PCIe host controller or the PCIe host controller itself?  
Repeated those tests on the latest rpi-firmware release with Linux 5.15.24 for completeness sake, same result.

paulwratt, 2022-02-21:
Wasn't one of the work-arounds in previous #4 experiments a function that broke data down to 256 max data transactions?

With the CPU moving to 32bit, is that a result of the Driver, or a compiler issue using the wrong op-code (from when ARM was still only 32bit)?

Coreforge, 2022-02-21:
The 256 byte limit might be because the pi might just not write combine under 256 bytes, but it could also be something else overflowing. Any transactions over 4 bytes don't work reliably though, but don't necessairly result in the pi locking up (reading 8 bytes just repeats the first 4 bytes for example).
I don't know why the cpu does what it does, and it also might not have been a switch to 32 bits, I'm not sure anymore (been a while since I've used JTAG on the pi). I couldn't connect to it anymore though through OpenOCD, and I think it said something about Armv7. I didn't try to single step to a lockup though (might be easier with the test driver).

paulwratt, 2022-02-21:
(I wish I was in a position to help out with some of this diagnosis)

TobleMiner, 2022-03-05:
I opened an issue detailing this behaviour on raspberrypi/linux and got a response from a former Broadcom, now Raspberry Pi foundation engineer: https://github.com/raspberrypi/linux/issues/4928#issuecomment-1059839682

It seems to boil down to the fact that the PCIe root complex on the BCM2711 can only support aligned 32bit access. Thus remapping PCIe BARs as normal memory will never work properly. No one expects normal memory to have any alignment requirements. Thus a whole bunch of PCI drivers and also userspace software using those drivers will just not work on the Pi 4 without rewriting them to do only aligned accesses. That would be a huge amount of work though and not useful beyond just "making it work" on the Pi.

pelwell, 2022-03-06:
> the BCM2711 can only support aligned 32bit access.

It supports aligned accesses _up to_ 32 bits. The linked comment has been amended.

TobleMiner, 2022-03-07:
Ah, sorry for the confusion. I've now edited the comment to reflect that, thanks for bringing it up!

geerlingguy, 2022-03-16:
All right, so I managed to apply @TobleMiner's patch in [this comment](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/62#issuecomment-1046319009) to the latest kernel source, and compile the kernel with `CONFIG_FB_SM750=m` (under `Device Drivers -> Staging drivers -> Silicon Motion SM750 framebuffer support` in `menuconfig`).

I tried booting an image with the full Pi OS and window manager, but when it initialized (with one HDMI display in HDMI0 and VGA connected to the M2_VGA card), it seemed to lock up. Got further than usual though. I'm going to try just a console version.

Coreforge, 2022-03-16:
I don't think a window manager is fully working on any gpu yet. Instead of reflashing, just booting it up without the gpu and disabling graphical boot in raspi-config should be faster and has the same effect.

[12 more replies]

closed 2022-04-19 by geerlingguy