geerlingguy/raspberry-pi-pcie-devices #63: Iris Xe (DG1) discrete graphics cards

volkertb, 2021-01-26:
So Intel is finally entering the discrete graphics business, hopefully breathing some new life in a market that has been dominated by the AMD/NVIDIA duopoly for the last two decades.

According to [this article on Phoronix](https://www.phoronix.com/scan.php?page=news_item&px=Intel-Iris-Xe-DG1-Desktop), the Linux drivers aren't quite ready yet, but it might be something to eventually test.

Of course, we shouldn't be surprised if the Intel graphics driver developers would show at least some reluctance towards adding support for non-x86 architectures, let alone prioritizing it. :grin:

volkertb, 2021-01-26:
Aww, according to this article, these cards will only be compatible with certain Intel-based motherboards and will not be sold separately, at least not at first. :confused:

Hopefully, the cards will eventually be sold separately as well, and the BIOS-dependency that they have on those select motherboards will be worked around somehow. The fact that Intel graphics drivers for Linux are open source may prove helpeful in that regard.

Perhaps they're only selling the card bundled with a limited selection of Intel CPUs and motherborads for now, just to test the waters and to let the graphics architecture mature a bit, before making it more widely available.

geerlingguy, 2021-01-28:
More info from Ars Technica: [Intel’s first discrete GPU in 20+ years is only for OEMs, integrators](https://arstechnica.com/gadgets/2021/01/intel-releases-dg1-discrete-graphics-cards-to-oems-and-integrators/).

Sadly, unless the 'only works with select motherboards' thing changes, I don't think I'm going to attempt to acquire one of these things :(

closed 2021-01-28 by geerlingguy

pgwipeout, 2021-05-03:
So these will only work with select motherboards because they will require the motherboard to store and load the firmware image, as the cards have no onboard firmware.
It may be feasible to load the firmware after the fact in Linux, and I wouldn't be surprised if someone figured out how to do this in short order after their general release.