geerlingguy/raspberry-pi-pcie-devices #64: JMB582 / JMB585 SATA Controller Chipset

PastuDan, 2021-01-31:
I'm currently testing an [IOCrest JMB582 storage card](https://amzn.to/3aiQ2Ux).

Testing ongoing. I'm getting a kernel panic during boot whenever the card is plugged in. I've tried increasing the BAR space, and compiling the kernel with AHCI SATA modules.

https://user-images.githubusercontent.com/1296162/106399917-58420080-63d0-11eb-936f-a33bb9d0c72b.mov

geerlingguy, 2021-01-31:
Are you using an external PCIe riser or switch, or just plugging into the official IO Board directly?

PastuDan, 2021-02-01:
Plugging in directly

geerlingguy, 2021-02-02:
I split out the two cards since even though they use the same family chipset, there could be differences. I'm also going to add my pictures to the first comment on this issue :)

PastuDan, 2021-02-02:
Awesome, thanks Jeff!!

ThomasKaiser, 2021-02-23:
@PastuDan according to #88 JMB585 is confirmed to work. Maybe that helps with your JMB582 as well?

geerlingguy, 2021-03-05:
Just dropping some notes in here:

```
make menuconfig

>Device Drivers:
  > Serial ATA and Parallel ATA drivers (libata) (enable this - M)
    > AHCI SATA support (enable this - M)
    > Marvell SATA support (enable this - M)
```

geerlingguy, 2021-03-05:
Just wiring it up with the new kernel in place:

```
01:00.0 SATA controller: JMicron Technology Corp. Device 0585 (prog-if 01 [AHCI 1.0])
	Subsystem: JMicron Technology Corp. Device 0000
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 48
	Region 0: I/O ports at <unassigned> [disabled]
	Region 1: I/O ports at <unassigned> [disabled]
	Region 2: I/O ports at <unassigned> [disabled]
	Region 3: I/O ports at <unassigned> [disabled]
	Region 4: I/O ports at <unassigned> [disabled]
	Region 5: Memory at 600010000 (32-bit, non-prefetchable) [size=8K]
	[virtual] Expansion ROM at 600000000 [disabled] [size=64K]
	Capabilities: [80] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [90] MSI: Enable+ Count=1/8 Maskable- 64bit+
		Address: 00000000fffffffc  Data: 6540
	Capabilities: [c0] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0, Latency L0s <1us, L1 <1us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ UncorrErr- FatalErr- UnsuppReq+ AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x2, ASPM not supported, Exit Latency L0s <256ns, L1 <8us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range B, TimeoutDis+, LTR-, OBFF Via message
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [150 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [160 v1] Power Budgeting <?>
	Capabilities: [1b8 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [300 v1] #19
	Capabilities: [900 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1- L1_PM_Substates-
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-

		L1SubCtl2:
	Kernel driver in use: ahci
	Kernel modules: ahci
```

And in dmesg:

```
[    1.208280] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.208321] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.208406] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.208497] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.243054] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[... 23 lines cut ...]
[    1.252210] pci 0000:01:00.0: BAR 1: failed to assign [io  size 0x0080]
[    1.252234] pci 0000:01:00.0: BAR 2: no space for [io  size 0x0080]
[    1.252257] pci 0000:01:00.0: BAR 2: failed to assign [io  size 0x0080]
[    1.252281] pci 0000:01:00.0: BAR 3: no space for [io  size 0x0080]
[    1.252304] pci 0000:01:00.0: BAR 3: failed to assign [io  size 0x0080]
[    1.252328] pci 0000:01:00.0: BAR 4: no space for [io  size 0x0080]
[    1.252351] pci 0000:01:00.0: BAR 4: failed to assign [io  size 0x0080]
[    1.252377] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.252409] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    4.281691] pci 0000:00:00.0: enabling device (0000 -> 0002)
...
[    4.043411] libata version 3.00 loaded.
[    4.384453] ata1: SATA max UDMA/133 abar m8192@0x600010000 port 0x600010100 irq 48
[    4.384471] ata2: SATA max UDMA/133 abar m8192@0x600010000 port 0x600010180 irq 48
[    4.384485] ata3: SATA max UDMA/133 abar m8192@0x600010000 port 0x600010200 irq 48
[    4.384498] ata4: SATA max UDMA/133 abar m8192@0x600010000 port 0x600010280 irq 48
[    4.384511] ata5: SATA max UDMA/133 abar m8192@0x600010000 port 0x600010300 irq 48
[    4.700649] ata1: SATA link down (SStatus 0 SControl 300)
[    5.013874] ata2: SATA link down (SStatus 0 SControl 300)
[    5.328680] ata3: SATA link down (SStatus 0 SControl 300)
[    5.647804] ata4: SATA link down (SStatus 0 SControl 300)
[    5.963220] ata5: SATA link down (SStatus 0 SControl 300)
```

(I don't have any devices plugged in currently.) No lockup, yay!

geerlingguy, 2021-03-05:
```
$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    0  7.3T  0 disk 
sdb           8:16   0  7.3T  0 disk 
sdc           8:32   0  7.3T  0 disk 
sdd           8:48   0  7.3T  0 disk 
mmcblk0     179:0    0 59.5G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 59.2G  0 part /
```

(Even working through my PCI switch, yay!)

geerlingguy, 2021-03-05:
I'm seeing around 395 MB/sec (3.16 Gbps) through the card during a RAID 5 parity calculation (initializing the disk through OMV, which seems to be using mdadm).

geerlingguy, 2021-03-07:
| Benchmark | Result 1 | Result 2 | Result 3 | Average |
| --- | --- | --- | --- | --- |
| hdparm | TODO | TODO | TODO | TODO |
| dd write | TODO | TODO | TODO | TODO |
| 4k read | TODO | TODO | TODO | TODO |
| 4k write | TODO | TODO | TODO | TODO |

ThomasKaiser, 2021-03-07:
What is this for a setup? RAID-5 set up by `mdadm`? If so which settings and which devices?

And I'm still curious which purpose sequential read and write tests should have that are done with 128K or even 8K block size...

geerlingguy, 2021-03-07:
@ThomasKaiser - It's mostly a way for me to have a line of comparison back to my original tests starting in around 2014, ranging from slow microSD cards up to SSDs. Note that there is no single benchmark I've found that can really be well-rounded that can target every kind of storage device... I just like the consistency.

I will likely also be doing some `fio` comparisons, and in these GitHub issues I mostly document things for my own reference later.

The specific benchmark script I'm using _here_ is https://github.com/geerlingguy/raspberry-pi-dramble/blob/master/setup/benchmarks/disk-benchmark.sh

geerlingguy, 2021-03-08:
Some NFS benchmarks on the same RAID 5 setup (mdadm via OMV), over the 2.5 Gbps network (I am able to get 2.2 Gbps from Pi to Mac, ~0.8 Gbps from Mac to Pi. Not sure why it's not very asymmetric.).

```
$ time dd if=/dev/zero of=/Volumes/NFS/test bs=1024k count=8k
8192+0 records in
8192+0 records out
8589934592 bytes transferred in 376.259394 secs (22829821 bytes/sec)

real	6:17.02
user	0.023
sys	52.107
```

(22.83 MB/sec... not super amazing :/)

And `atop` is showing all three disks hitting 85%+ utilization, with `md0` hitting 100% quite frequently.

geerlingguy, 2021-03-08:
And using `pv`:

```
# From Mac to Pi, 8 GiB file.
$ pv ~/Downloads/test > /Volumes/NFS/test
8.00GiB 0:07:04 [19.3MiB/s] [===============================================================================>] 100%  

# From Pi to Mac, same file:
$ pv /Volumes/NFS/test > ~/Downloads/test
8.00GiB 0:01:08 [ 119MiB/s] [===============================================================================>] 100%
```

geerlingguy, 2021-03-08:
And `pv` over the 1 GbE built-in interface:

```
# From Mac to Pi, 8 GiB file.
$ pv /Volumes/NFS/test > ~/Downloads/test
8.00GiB 0:01:31 [89.2MiB/s] [===============================================================================>] 100% 

# From Pi to Mac, same file:
$ pv ~/Downloads/test > /Volumes/NFS/test
8.00GiB 0:24:45 [5.51MiB/s] [===============================================================================>] 100%
```

geerlingguy, 2021-03-08:
Going to test out RAID 0 next, just to see what the max performance is for SATA + mdadm raid. These numbers are... not overwhelming. But hardware RAID may be able to fix it. We'll see.

geerlingguy, 2021-03-08:
Using RAID 0 on the 2.5G interface:

```
# Pi to Mac:
$ pv /Volumes/NFS/test > ~/Downloads/test
8.00GiB 0:01:06 [ 123MiB/s] [===============================================================================>] 100% 

# Mac to Pi:
$ pv ~/Downloads/test > /Volumes/NFS/test
8.00GiB 0:05:38 [24.2MiB/s] [===============================================================================>] 100% 
```

geerlingguy, 2021-03-08:
And just a single drive (sda) on the 2.5G interface:

```
# Pi to Mac:
$ pv /Volumes/NFS/test > ~/Downloads/test
8.00GiB 0:01:03 [ 128MiB/s] [===============================================================================>] 100%

# Mac to Pi:
$ pv ~/Downloads/test > /Volumes/NFS/test
8.00GiB 0:05:36 [24.4MiB/s] [===============================================================================>] 100%            
```

geerlingguy, 2021-03-08:
I'm going to test the same thing, taking the PCIe switch and 2.5GbE card out of the loop, to see if that's causing any kind of bottleneck with the writes.

geerlingguy, 2021-03-08:
And to be clear, testing the drive itself:

```
hdparm - 252.83 MB/sec
dd - 127 MB/s
4k rand read - 30.16 MiB/sec
4k rand write - 4.67 MiB/sec

geerlingguy, 2021-03-08:
I also enabled SMB/CIFS and tried copying through there... same issue. Writes over the network are incredibly slow in comparison to reads (especially on the 1 Gbps interface), and I can't see anything that would seem to be a culprit.

```
$ uname -a
Linux raspberrypi 5.10.20-v8+ #1 SMP PREEMPT Fri Mar 5 15:27:59 UTC 2021 aarch64 GNU/Linux
```

Just for reference.

geerlingguy, 2021-03-08:
Added an overclock to `/boot/config.txt`:

```
over_voltage=6
arm_freq=2147
gpu_freq=750
```

Rebooted, tried again, still just as slow (maybe a _tiny_ bit faster? Not easy to measure though without a very very long burn-in test.)

ThomasKaiser, 2021-03-08:
> It's mostly a way for me to have a line of comparison back to my original tests starting in around 2014

Well, it doesn't make a lot of sense since back then in 2014 all you had to test with were rather crippled Raspberry Pis when speaking of I/O capabilities (both network and disk storage behind a single Hi-Speed USB lane). And most probably your choice of tools back then was influenced by [the RPi community](https://linux-sunxi.org/Sunxi_devices_as_NAS#Use_the_right_tools).

Anyway: this whole test methodology makes no sense at all and the same is true for using anachronistic RAID _today_ in such a setup. You staple layers of complexity on another without caring about the involved drawbacks. For example if you don't aline RAID parameters with disk parameters you end up with horribly low write performance since every block access at the filesystem layer results in accessing 2 blocks at the block device layer below.

You also need at least 3 machines to do network testing (the 3rd always as a reference to identify bottlenecks, in your above testings most probably your Mac being the culprit[1]). If networking is _not_ confirmed to be free of problems (iperf +940 Mbits/sec in both directions without retransmits) then it makes no sense to continue with NAS tests since one known bottleneck is already in the mix that will invalidate all further results.

While I understand that this whole project is about the joy of tinkering and 'just for fun' it's a bit sad to use inappropriate methodology since the time and efforts you and others spend on generating 'performance numbers' are somewhat wasted.

If the goal is to determine realistic real-world maximum sequential throughput rates for a storage device then you clearly can neither use `hdparm` (limited by a horribly low block size of hardcoded 128K and by a test execution of just 3 seconds) nor `dd` with even worse parameters (8K block size and 50 iterations). You need to use at least 1 MB block sizes since this is [what happens in real-world situations as well since many years](https://www.helios.de/viewart-de.html?id=1711) and a test execution large enough to accommodate for ramping up cpufreq or leaving (deep) sleep states inside SSD storage. So we're back at [this](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/90#issuecomment-790378077) again :)

[1] As an example for why 'passive benchmarking' (in contrast to [active benchmarking](http://www.brendangregg.com/activebenchmarking.html)) between a Mac and a Linux system can show horribly low write performance through the network 'nobody can explain' look back almost 1.5 decades ago when we needed to [adjust the `net.inet.tcp.delayed_ack` parameter in MacOS to cope with more modern TCP/IP stack implementations on another OS at the other end of the network cable](https://www.helios.de/web/EN/support/TI/110.html}.

ThomasKaiser, 2021-03-08:
> Does OMV cause some strange overhead on the Pi's networking stack

No (I should know since having provided dedicated OMV images for many SBC RPi included for some time and most of the installation routine's code by me).

The OMV installation routine does change something wrt networking when executed on Armbian installs but this doesn't apply here (Raspbian or 'Pi OS' as it's called today). Also on Armbian installs there's [a lot of tuning here and there](https://github.com/armbian/build/blob/master/packages/bsp/common/usr/lib/armbian/armbian-hardware-optimization#L97-L361) but this also does not apply here on the RPi (one thing I would check is `/proc/interrupts` prior and after benchmark execution to see whether `cpu0` makes up for a nice bottleneck wrt IRQ processing).

Wrt test methodology: you need to test from bottom to top (storage individually, network individually, only then both combined – and all of this with appropriate parameters). As repeated frequently: testing with inappropriate block sizes will just provide numbers without meaning (which use case is your dd test with `bs=8k count=50k` representing?).

Please keep in mind that hdparm's benchmark modes have been developed last century. Back then DMA was a hot thing compared to PIO and 128K block sizes could be considered huge. Back then HDDs had capacities no average operating system update we now download through the Internet every other day would fit on. Things have changed, use cases have changed, testing methodology needs to be changed accordingly.

geerlingguy, 2021-03-08:
@ThomasKaiser - What do you think of these changes?

https://github.com/geerlingguy/raspberry-pi-dramble/commit/a0b1708a92a9abc9b13343735d3b34beb67af626

It runs a 1M sequential read test using `fio`. And I wanted to use `fio` to test 1M sequential writes... but that destroys a mounted disk (I want to be able to run the benchmarks on an already-existing/in-use disk, it looks like `fio`'s write tests are destructive and meant only for benchmarking / testing on a test environment, not running against production situations).

So instead I also run 1M sequential tests using `iozone`, as well as 4k random tests using `iozone`.

First set of results seems good from this script: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/92#issuecomment-792883731 — however I find it odd that iozone's performance numbers don't quite match up to fio's (though the sequential write speeds are close between the two).

geerlingguy, 2021-03-08:
All right, testing speeds on an Asustor NAS too, I'm seeing the same _really weird_ discrepancy between Mac-to-device vs device-to-Mac.

I swapped out the Cat5e patch cable for Cat8 and a Cat7. No difference (Pi to Mac always 2.2 Gbps, NAS to Mac always 2.35 Gbps... Mac to Pi and Mac to NAS fluctuating wildly around 500-700 Mbps).

I also swapped out Thunderbolt 3 cables (tried three different cables). No difference, very close results.

Tried my gigabit port on my CalDigit Thunderbolt 3 dock, and it's solidly hitting 940 Mbps from Mac to Pi and Mac to NAS. So something is _really_ funky with the OWC Thunderbolt 3 10 Gbps adapter. Used the same port on the Mikrotik switch too, and even tried a different port and copper SFP+ adapter, and nothing made a difference.

I'm going to turn off the OWC adapter and let it cool down and see if that makes any difference. Would stink to have to find another $100+ 10 gig adapter for the Mac. Any good USB 2.5 Gbps adapters out there?

Edit: Just ordered an [Inateck 2.5G Ethernet Adapter ET1001](https://amzn.to/3ccmQiX) USB-C adapter. Better to have more than one option.

PastuDan, 2021-03-08:
Jeff, I can’t stress enough how much I appreciate your thorough testing!! This is immensely helpful while I design around these chips :)

geerlingguy, 2021-03-08:
@PastuDan - Glad to help! At least I can determine now that I don't think there's any issue with the JMB582/585—they have performed flawlessly, even through the PCIe switch. I really need to figure out this networking issue on my Mac though!

geerlingguy, 2021-03-09:
Doing more debugging—I've even changed out the cable to the NAS (which I already did once, but once more with a brand new and known-good-to-10G-cable), and that didn't help. I also confirmed both ends of the connection are showing as 'Full' duplex. The NAS at 2500Mb/s (`ethtool eth0`), and my Mac at 10 Gbps (in Network preference pane).

Also confirmed that all devices are configured with normal MTU (1500).

geerlingguy, 2021-03-11:
All right, so more notes as I get back to testing... soon:

  - My OWC 10G TB3 adapter is definitely on the fritz. Tested it on my Dell XPS 13 laptop, same weird result (really sluggish uplink, fine downlink). Tried three different TB3 cables, four different network cables, plugged into 1G switch and 10G switch using different SFP+ connectors, all these things known good, and nothing made a difference. RMA'ed it and hopefully replacement will work fine.
  - I received the [Inateck 2.5 GbE USB-C adapter](https://amzn.to/2NajrsF). It uses a Realtek chip inside... which apparently doesn't work with 2.5G on Mac out of the box. Downloaded Realtek's drivers (which are very janky for Mac), and they did nothing to help. Stuck at 1G there. At least it's 940 Mbps up and down :P
  - So I ordered a [Plugable 2.5G USB-C adapter](https://amzn.to/2OfCBhp), which _also_ says it's compatible with macOS... but looks like it also uses a Realtek chip and may require a driver install for 2.5G data transfer. Grr. From a reviewer:

> I have a 2018 MacBook Pro running macOS 10.15 Catalina. This device works at 1gbe without any config, but the driver from the Realtek website must be installed for 2.5gbe operation. Installation is very quick and simple and no more config is required after rebooting. One thing of note is that Apple informs you that the Realtek driver will not be compatible with the next version of macOS (10.16), so I'll be curious to see if Realtek releases an update for the next release. I hope so.

We'll see. Worst case, I do all the rest of my testing over on my HP desktop, with the ASUS 10G adapter in it.

mi-hol, 2021-03-15:
> do all the rest of my testing over on my HP desktop, with the ASUS 10G adapter in it.

I'd go that route as I had bad experience to get networking on macOS 10.13 and later to work at higher speeds in a huge DC environment too. macOS is targeted at consumers getting proper technical support for networking turned out to be impossible (at least in Germany).

geerlingguy, 2021-03-15:
The plot thickens. I was using a [MikroTik 10G Switch/Router](https://amzn.to/38Htkpc) (in switch mode) for all this testing, but once I started seeing the HP desktop _also_ have weird issues, using a Mellanox ConnectX2 card, and the ASUS adapter, where it would be 2.3 Gbps down but only 600-800 Mbps up... I'm doubting everything I know.

I have no idea why bandwidth is behaving weird on the MikroTik, but since I now isolated every other potential fault, I decided to bite the bullet an buy a full-fledged QNAP switch (this one: [QNAP QSW-M2108-2C 10-port/8x 2.5 GbE](https://amzn.to/3tmE9F3)), and re-run all the tests on it.

And lo and behold... 2.3 Gbps both directions, all the time. So, it seems like it may _not_ have had anything to do with the Mac. And now I'm scratching my head over how the MikroTik could be having issues. I have it booted into SwOS (so the router overhead is not there), it's not super hot, and it always gives 2.3 Gbps in _one_ direction.

Until I get back my RMA'd OWC adapter, I won't be able to test beyond 2.5 Gbps.

ANYWAYS, that's a HUGE tangent on this card—and an expensive one—so I'll be getting back to my performance testing soon :)

mister2d, 2021-03-21:
Do you have a running list of known good hardware yet. I wouldn't mind testing this out myself.

Thanks.

geerlingguy, 2021-03-26:
@mister2d - My blog and YouTube channel will have a post in just a bit with all the details of my 2.5G Pi NAS build (the hardware side), so check that out when it's released!

Otherwise, all the info in this repository gets aggregated into https://pipci.jeffgeerling.com (I need to run through and update some things though... haven't taken a pass for a couple weeks as I've been heads-down in some projects).

PastuDan, 2021-03-26:
@geerlingguy following your instructions I was able to get the JMB585 chipset card working, but not the JMB582. Still having issues with the kernel panic at boot.

Unfortunately my board is based around the JMB582, and switching to the 585 would be rather costly if you don’t need 5 drives. At volume pricing they run $13 and $25 respectively.

[image]
[image]

geerlingguy, 2021-03-26:
Video is up: https://www.youtube.com/watch?v=vBccak8f-VY

And @PastuDan, I'll try testing out the 582 you sent soon!

geerlingguy, 2021-05-12:
All right, finally had some time to plug the thing in... on a fresh boot with just the plain old Pi OS ARM64 build installed, I also get the kernel panic if I have the card plugged in. I'm going to pull out my USB/UART adapter and get a dump of the actual text for comparison.

geerlingguy, 2021-05-12:
Unfortunately had trouble getting the output (maybe the kernel panic happens too early to start output over uart?), so here's a picture:

[image]

PastuDan, 2021-05-12:
Bummer. Glad to know I'm not crazy though! I didn't think to tap the UART for those kernel logs, that's a good idea.

I'll see if I can give that a try later. There's got to be some helpful output at the top of those logs that my video doesn't capture.

geerlingguy, 2021-05-12:
The panic might occur too early for the console to start up, in which case I wonder if the settings under `Enabling early console (earlycon) for Linux` in [the UART configuration guide](https://www.raspberrypi.org/documentation/configuration/uart.md) might be required. Unfortunately I need to drop back to another project right now or else I'd give it a go.

geerlingguy, 2021-07-02:
All right, I'm going to call this as 'won't work', at least for the JMB582. Weird that other related chips in the same family _do_, but for now I'll update the site with the latest findings so others can just move on to a SATA controller that is known working (since there are a number now that have been tested and verified!).

closed 2021-07-02 by geerlingguy

solstyce, 2021-08-13:
Hi @geerlingguy ,
I'm trying to make JMB582 / JMB585 SATA Controller Chipset working with my set-up but without success. 
I'm using : 
Compute module 4, 1Gb RAM with 16Gb eMMC without WIFI.
This plugged on the "Waveshare Raspberry Pi Compute Module 4 IO Board" 
[image]

I installed latest raspiOS version and upgraded it.

LSPCI give me this answer : 
[image]
and this in verbose mode : 
[image]

Did I missed something ? do you have any idea how I could this this issue? 

Thanks for help :)

solstyce, 2021-08-13:
Resolved by manually updating Kernel version : 
https://www.raspberrypi.org/documentation/computers/linux_kernel.html

[image]

Result : 

[image]

grasmax, 2023-08-29:
Have successfullly testet this configuration (used as pv controller (https://github.com/grasmax/AcOnOff) and local storage):
- Raspberry Pi CM4IO Board
- CM4001032 Raspberry Pi Compute Module 4, 1GB-RAM, 32GB-eMMC, BCM2711, ARM Cortex-A72
- raspberry pi os lite (32bit) v11
- raspberry pi os (32bit) v11
- IO CREST JMB582 2 Port SATA III PCI-e 3.0 x1 Non-RAID Controller Karte Jmicro Chipsatz SI-PEX40148
- 2TB WD20EFZX
Thanks for making pi pcieable...
Rainer

geerlingguy, 2023-10-13:
Testing this on the Raspberry Pi 5:

```
pi@pi5:~ $ lspci
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0000:01:00.0 SATA controller: JMicron Technology Corp. JMB58x AHCI SATA controller
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0001:01:00.0 Ethernet controller: Device 1de4:0001
```

And more info:

```
sudo lspci -vvvv
0000:01:00.0 SATA controller: JMicron Technology Corp. JMB58x AHCI SATA controller (prog-if 01 [AHCI 1.0])
	Subsystem: JMicron Technology Corp. JMB58x AHCI SATA controller
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 169
	Region 5: Memory at 1b00010000 (32-bit, non-prefetchable) [size=8K]
	Expansion ROM at 1b00000000 [virtual] [disabled] [size=64K]
	Capabilities: [80] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [90] MSI: Enable+ Count=1/8 Maskable+ 64bit+
		Address: 000000ffffffe000  Data: 0008
		Masking: 000000fe  Pending: 00000000
	Capabilities: [c0] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0, Latency L0s <1us, L1 <1us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 512 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x1, ASPM not supported
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x1
			TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range B, TimeoutDis+ NROPrPrP- LTR-
			 10BitTagComp- 10BitTagReq- OBFF Via message, ExtFmt+ EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS-
			 AtomicOpsCap: 32bit+ 64bit+ 128bitCAS+
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
			 EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr+ BadTLP- BadDLLP+ Rollover- Timeout+ AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [150 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [160 v1] Power Budgeting <?>
	Capabilities: [1b8 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [300 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: LaneErr at lane: 0
	Capabilities: [900 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=255us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Kernel driver in use: ahci
	Kernel modules: ahci
```

geerlingguy, 2023-10-13:
At PCIe Gen 3.0 speeds, I get a fail on the Pi 5:

```
[    8.062038] ata1: found unknown device (class 0)
[   12.722035] ata1: softreset failed (device not ready)
[   12.865621] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[   12.865634] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[   12.865639] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00000080/00002000
[   12.865643] pcieport 0000:00:00.0:    [ 7] BadDLLP               
[   18.074038] ata1: found unknown device (class 0)
[   22.734035] ata1: softreset failed (device not ready)
[   26.992262] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[   26.992271] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[   26.992275] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00000080/00002000
[   26.992279] pcieport 0000:00:00.0:    [ 7] BadDLLP               
[   28.086038] ata1: found unknown device (class 0)
[   33.286035] ata1: link is slow to respond, please be patient (ready=0)
[   35.687176] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[   35.687189] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[   35.687193] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00000080/00002000
[   35.687198] pcieport 0000:00:00.0:    [ 7] BadDLLP               
[   36.098686] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[   36.098695] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[   36.098699] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00000080/00002000
[   36.098703] pcieport 0000:00:00.0:    [ 7] BadDLLP               
[   50.874258] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[   50.874268] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[   50.874272] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00000080/00002000
[   50.874276] pcieport 0000:00:00.0:    [ 7] BadDLLP               
[   57.766035] ata1: softreset failed (device not ready)
[   57.766042] ata1: limiting SATA link speed to 3.0 Gbps
[   62.802035] ata1: softreset failed (device not ready)
[   62.802047] ata1: reset failed, giving up
[   63.116217] ata2: SATA link down (SStatus 0 SControl 300)
```

PeterSuh-Q3, 2023-10-17:
I use Silverstone's SS-ECS07 using JMB585.
https://www.silverstonetek.com/en/product/info/expansion-cards/ECS07/

The strange thing is that there is no problem during normal booting.
Upon reboot, the following log (dmesg) occurs.
It seems almost similar to the log posted above.

The OS is Synology DSM 7 with XPENOLOGY REDPILL.

Will this problem be solved with the kernel module patch below?
https://lore.kernel.org/lkml/20220907105130.13797-1-mdlin@jmicron.com/T/

`[    2.812512] redpill: loading out-of-tree module taints kernel.
[    2.819555] redpill: module verification failed: signature and/or required key missing - tainting kernel
[    2.846488] PCI host bridge to bus 0001:01
[    2.851421] pci_bus 0001:01: root bus resource [io  0x0000-0xffff]
[    2.858882] pci_bus 0001:01: root bus resource [mem 0x00000000-0x7fffffffff]
[    2.867376] pci_bus 0001:01: root bus resource [bus 00-ff]
[    2.873980] pci 0001:01:00.0: [1b4b:9215] type 00 class 0x010601
[    2.881226] pci 0001:01:00.0: Can't map mv9235 registers
[    2.887654] PCI host bridge to bus 0001:02
[    2.892604] pci_bus 0001:02: root bus resource [io  0x0000-0xffff]
[    2.900063] pci_bus 0001:02: root bus resource [mem 0x00000000-0x7fffffffff]
[    2.908547] pci_bus 0001:02: busn_res: can not insert [bus 02-ff] under domain [bus 00-ff] (conflicts with (null) [bus 
01-ff])
[    2.922288] pci_bus 0001:02: root bus resource [bus 00-ff]
[    2.928892] pci 0001:02:00.0: [8086:1539] type 00 class 0x020000
[    2.936168] PCI host bridge to bus 0001:03
[    2.941108] pci_bus 0001:03: root bus resource [io  0x0000-0xffff]
[    2.948566] pci_bus 0001:03: root bus resource [mem 0x00000000-0x7fffffffff]
[    2.957068] pci_bus 0001:03: busn_res: can not insert [bus 03-ff] under domain [bus 00-ff] (conflicts with (null) [bus 
01-ff])
[    2.970798] pci_bus 0001:03: root bus resource [bus 00-ff]
[    2.977414] pci 0001:03:00.0: [8086:1539] type 00 class 0x020000
[    2.984693] PCI host bridge to bus 0001:00
[    2.989638] pci_bus 0001:00: root bus resource [io  0x0000-0xffff]
[    2.997106] pci_bus 0001:00: root bus resource [mem 0x00000000-0x7fffffffff]
[    3.005609] pci_bus 0001:00: busn_res: can not insert [bus 00-ff] under domain [bus 00-ff] (conflicts with (null) [bus 
01-ff])
[    3.019349] pci_bus 0001:00: root bus resource [bus 00-ff]
[    3.025998] pci 0001:00:12.0: [8086:5ae3] type 00 class 0x010601
[    3.033275] pci 0001:00:13.0: [8086:5ad8] type 00 class 0x060400
[    3.040525] pci 0001:00:13.0: ignoring class 0x060400 (doesn't match header type 00)
[    3.049871] pci 0001:00:14.0: [8086:5ad6] type 00 class 0x060400
[    3.057121] pci 0001:00:14.0: ignoring class 0x060400 (doesn't match header type 00)
[    3.066476] pci 0001:00:15.0: [8086:5aa8] type 00 class 0x0c0330
[    3.073805] pci 0001:00:16.0: [8086:5aac] type 00 class 0x118000
[    3.081086] pci 0001:00:18.0: [8086:5abc] type 00 class 0x118000
[    3.088372] pci 0001:00:19.0: [8086:5ac6] type 00 class 0x118000
[    3.095628] pci 0001:00:19.2: [8086:5ac6] type 00 class 0x118000
[    3.102907] pci 0001:00:1f.0: [8086:5ad4] type 00 class 0x0c0500
[    3.110179] pci 0001:00:1f.1: [8086:5ad4] type 00 class 0x0c0500
[    3.123116] DMAR: Host address width 39
[    3.127756] DMAR: DRHD base: 0x000000fed90000 flags: 0x0
[    3.134189] DMAR: dmar0: reg_base_addr fed90000 ver 1:0 cap 1c0000c40660462 ecap 19e2ff0505e
[    3.144367] DMAR: DRHD base: 0x000000fed91000 flags: 0x1
[    3.150779] DMAR: dmar1: reg_base_addr fed91000 ver 1:0 cap d2008c40660462 ecap f050da
[    3.160321] DMAR: RMRR base: 0x000000c2b84000 end: 0x000000c2dcdfff
[    3.167877] DMAR: RMRR base: 0x000000c3800000 end: 0x000000c7ffffff
[    3.175450] DMAR: No ATSR found
[    3.179287] DMAR: dmar0: Using Queued invalidation
[    3.185057] DMAR: dmar1: Using Queued invalidation
[    3.191041] DMAR: Setting RMRR:
[    3.194839] DMAR: Setting identity map for device 0000:00:02.0 [0xc3800000 - 0xc7ffffff]
[    3.204578] DMAR: Setting identity map for device 0000:00:14.0 [0xc2b84000 - 0xc2dcdfff]
[    3.214348] DMAR: Prepare 0-16MiB unity mapping for LPC
[    3.220649] DMAR: Setting identity map for device 0000:00:1f.0 [0x0 - 0xffffff]
[    3.229759] DMAR: Intel(R) Virtualization Technology for Directed I/O
[    3.237604] iommu: Adding device 0000:00:00.0 to group 0
[    3.244034] iommu: Adding device 0000:00:01.0 to group 1
[... 98 lines cut ...]
[    3.499183] pcieport 0000:00:1d.0:   device [8086:a29a] error status/mask=00000001/00002000
[    3.499184] pcieport 0000:00:1d.0:    [ 0] Receiver Error        
[    3.499207] pcieport 0000:00:1d.0: AER: Multiple Corrected error received: id=00e8
[    3.499241] pcieport 0000:00:1d.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e8(Receiver ID)
[    3.499242] pcieport 0000:00:1d.0:   device [8086:a29a] error status/mask=00000001/00002000
[    3.499242] pcieport 0000:00:1d.0:    [ 0] Receiver Error        
[    3.499269] pcieport 0000:00:1d.0: AER: Multiple Corrected error received: id=00e8
[    3.499304] pcieport 0000:00:1d.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e8(Receiver ID)
[    3.499305] pcieport 0000:00:1d.0:   device [8086:a29a] error status/mask=00000001/00002000
[    3.499305] pcieport 0000:00:1d.0:    [ 0] Receiver Error        
[    3.499328] pcieport 0000:00:1d.0: AER: Multiple Corrected error received: id=00e8
[    3.499363] pcieport 0000:00:1d.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e8(Receiver ID)
[... cut ...]
[    3.499480] pcieport 0000:00:1d.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e8(Receiver ID)
[    3.499481] pcieport 0000:00:1d.0:   device [8086:a29a] error status/mask=00000001/00002000
[    3.499481] pcieport 0000:00:1d.0:    [ 0] Receiver Error        
[    3.499504] pcieport 0000:00:1d.0: AER: Multiple Corrected error received: id=00e8
[    3.499539] pcieport 0000:00:1d.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e8(Receiver ID)
[    3.499540] pcieport 0000:00:1d.0:   device [8086:a29a] error status/mask=00000001/00002000`

commit be354a65 (2021-07-02): Fixes #64: Close the loop on the JMB582 controller kernel panics.