geerlingguy/raspberry-pi-pcie-devices #71: XTRX PCIe Software Defined Radio

albydnc, 2021-02-09:
It would be nice to see this working, since lots of people are into SDR on the pi. Unfortunately I don't have one not a cm4, so I'm just posting here for reference.

geerlingguy, 2021-02-19:
It would be nice to test, but I know I won't be able to justify the cost of one just for the testing (I have a USB SDR from RTL-SDR.com and it's adequate for my rather humble needs (tuning into some OTA radio stations so I can listen to them on my phone since dumb sports leagues black out all home team games even if you pay them lots of money).

trabucayre, 2021-05-06:
I have succesfully used an XTRX board with the CM4 but this required:

`dtoverlay=pcie-32bit-dma` must be added in `config.txt`

[VA_DMA_ADDR_FIXUP](https://github.com/xtrx-sdr/xtrx_linux_pcie_drv/blob/master/xtrx.c#L181) must be defined in the driver

geerlingguy, 2021-05-06:
@trabucayre - Excellent, thanks for posting the update! Is there any chance you could take a picture of the card by itself (preferrably on a plain white background) so I could use that image on the site?

trabucayre, 2021-06-17:
Sorry for delay. This [file](http://kmf2.trabucayre.com/ggm/xtrx_pcieAdapter.jpg) is okay for you?

geerlingguy, 2021-09-09:
It's on the site, thanks!

closed 2021-09-09 by geerlingguy

vjarao, 2021-11-30:
Good evening ... thanks for sharing this SDR option. Do you know what would it take to connect 2 for these boards and sync them? I anticipate I might not be trivial

trabucayre, 2021-11-30:
this [board](https://www.crowdsupply.com/fairwaves/xync) has more than one XTRX. But don't know if it's possible to use it with raspberry

vjarao, 2022-01-30:
Good evening community. My XTRX arrived and now I need little help. As I am installing this in the CM4: 1) I addeddtoverlay=pcie-32bit-dma to /boot/config.txt, and 2) should update the driver ... and here is where I need a hint. In order to do this, which of the pre-build packages should I use from the following link? or shall I manually compile and after make the driver changes?
https://github.com/xtrx-sdr/images

vjarao, 2022-01-31:
Good evening ... following up my previous post / question. I decided to do a manual compilation of host libraries from the sources directory as described here:

https://github.com/xtrx-sdr/images#xtrx-installed-into-minipcie-slot-prevents-the-system-from-booting-dont-get-even-to-bios

I could not install "libqcustomplot1.3" because it was not found. Therefore, I installed everything else without any errors.

I am also able to reproduce / obtain the output of sudo "lspci -v -d 10ee:"
However, when I run "sudo ./test_xtrx -t -l2" I get the following error: " xtrx_open: -19". This somehow makes sense since I "/dev/xtrx0" is not on the list of my devices ...

Therefore, at this point I am considering to conclude that I have an incorrect PICe to miniPCI adapter ... No LED ever lit on the xtrx board either

@trabucayre do you mind to share which PCIe to miniPCI adapter you used for your CM4? I used this one: 
https://www.amazon.com/WSDMAVIS-Adapter-Wireless-Network-Board/dp/B08R5D4MSB/ref=sr_1_28?crid=1P6P515VOV7WK&keywords=pcie+to+mini+pcie&qid=1643589673&sprefix=pcie+to+mini+pci%2Caps%2C96&sr=8-28

vjarao, 2022-01-31:
Getting close !!!
I found out from https://github.com/xtrx-sdr/images/issues/37 that adding "coherent_pool=32M" on /boot/cmdline.txt allows me now to see xtrx0 on dev/ and also get an output when requesting  ./test_xtrx -t -l2
Master: 32.000 MHz; RX rate: 4.000 MHz; TX rate: 0.000 MHz
RX tunned: 900000000.000000
RX bandwidth: 2000000.000000
RX LNA gain: 15.000000
RX PGA gain: 0.000000
RX TIA gain: 9.000000
RX CYCLES=1 SAMPLES=16384 SLICE=16384 (PARTS=1)
PROCESSED RX SLICE 0 /0: res 0 TS:    8192       5376 us DELTA     19 us LATE   3347 us 16384 samples
RX STAT Overruns:0
Success!
Processed 1 devs, each: RX 2 x 1.508 = 3.017 MSPS (WIRE: 12.066365)    TX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000 MB/s)

now ... to gnuradio. So far I am still getting the following error:
[image]

trabucayre, 2022-02-01:
You have to rebuild gr-osmosdr with `-DENABLE_XTRX=ON` to have support for XTRX.
For connector, I have used a dremel to cut connector. Now I'm able to plug directly x2 or x4 PCIe board on cm4-io

vjarao, 2022-02-01:
@trabucayre thank you for the hint. I think I also need to install gnuradio 3.9. I had installed versions 3.8.2.0
[image]

trabucayre, 2022-02-01:
I use gr-osmosdr 0.2.3 (latest release) with 3.8.2.0

vjarao, 2022-02-01:
@trabucayre thanks ... I spent the last 2 hours trying to re-build with and without the aforementioned option. No matter what I do I always get the same error.   I am not a programmer nor software eng.  so I guess I have lot more to read and learn.

trabucayre, 2022-02-02:
Could you check if gr-osmosdr is not installed by your package manager? Maybe there is a conflict.
distro's one may take precedence.
If so please remove distro's package.

vjarao, 2022-02-02:
@trabucayre Thanks for the tip.  I did thought about it and removed everything and build gnuradio from scratch. Unfortunately did not work. I followed this https://wiki.gnuradio.org/index.php/InstallingGR#From_Source

vjarao, 2022-02-03:
@trabucayre @geerlingguy Question to both of you ... isn't bullseye a 64bit OS?

pi@CM41:/usr/local/lib/xtrx $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Will this a possible cause of why I cannot build and install gr-osmosdr with XTRX support?

trabucayre, 2022-02-04:
Please check wich options enabled/disabled at the end of gnuradio's cmake maybe runtime is missing (but it's a bit weird).
As far I know about rpi distribs raspbian is 32bits but theoritically this as no impact on using osmosdr (You have just lost quite all SIMD features)

vjarao, 2022-02-04:
@trabucayre Thank you very much. Indeed that is the last wall I am hitting - gr-osmosdr cannot find gnu runtime.

I worked the past 6 days on this and I am bit exhausted. I found out that there are many dependencies that are not yet linked / compatible.

I can build from source: 1) Volk, which is necessary, 2) gnuradio 3.8.5.0-5, including enable and disable options (flags), but when I try 3) build from source gr-osmosdr it complains about not finding gnuradio runtime. I check and update accordingly my $PATH, my .bashrc and my .profile. All are good and as per the available documentation, examples and community advises. Every other gnu package works but gr-osmosdr.

Now, the main page for gr-osmosdr states that Gnuradio v3.9 is required. See attached screenshot ... with this a chain reaction starts. This is because Gnuradio 3.9 requires the newest Volk version. This new Volk version is not yet compatible with CM4, then cannot install gnuradio v3.9 and cannot install gr-osmosdr with XTRX support.

I hired a SW Eng. and we will work over the weekend on this. I will let you know how it goes.

Monday I will have to decide to keep or return the board.

My lack of progress should not be interpreted as a complain, excuse or any detriment.

I thank you and everyone in the community for such amazing job and I am hoping my contribution could help others. 

[image]

vjarao, 2022-02-04:
@trabucayre I honestly do not know what I did but now it seems to be working ! I started from fresh and perhaps some links where restored to originals ??? This was my 3rd time starting from fresh.

Thank you all for your support.

vjarao, 2022-07-19:
Good morning ... several months ran with our major issues. However, after few recent updates and many projects on my RPi CM4 , I decided to clean up the Pi OS by doing a 1) new clean install following the procedure we found here before. XTRX correctly installs and is recognized by the Pi OS when lspcie. 2) gnu radio installs without any problems.

However, now I am facing a new issue. When running any flow graph with xtrx (osmocom source) it halts with the following message:

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107100; UHD_3.15.0.0-4
06:57:43.288234 INFO:   api_drivers_init:109 [ DEF] Probing 'libxtrxll_pcie.so.0' low-level library
06:57:43.294410 INFO:   api_drivers_init:109 [ DEF] Probing 'libxtrxll_libusb3380.so.0' low-level library
06:57:43.399633 DEBUG:  xtrxllpciev0_discovery:263 [PCIE] pcie: Found `pcie:///dev/xtrx0`
[INFO] Using format CF32.

>>> Done (return code -11)
[image]

On device arguments I tried "xtrx", "xtrx=0" and "xtrx = pcie:///dev/xtrx0" without any luck.

Any suggestions ?

trabucayre, 2022-07-19:
I use
`"xtrx,dev=/dev/xtrx0"`
(with double quote)

vjarao, 2022-07-19:
Thanks. Didn't work. 

I ran update and upgrade and found out there is an issue with my dkms.... I think I need to work that first ...

vjarao@raspberrypi:~ $ sudo apt dist-upgrade -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up xtrx-dkms (0.0.1+git20190320.5ae3a3e-2) ...
Removing old xtrx-0.0.1+git20190320.5ae3a3e-2 DKMS files...

------------------------------
Deleting module version: 0.0.1+git20190320.5ae3a3e-2
completely from the DKMS tree.
------------------------------
Done.
Loading new xtrx-0.0.1+git20190320.5ae3a3e-2 DKMS files...
It is likely that 5.15.32-v7l+ belongs to a chroot's host
Building for 5.15.32+, 5.15.32-v7+, 5.15.32-v7l+ and 5.15.32-v8+
Building initial module for 5.15.32+
Error! Bad return status for module build on kernel: 5.15.32+ (armv7l)
Consult /var/lib/dkms/xtrx/0.0.1+git20190320.5ae3a3e-2/build/make.log for more information.
dpkg: error processing package xtrx-dkms (--configure):
 installed xtrx-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 xtrx-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

trabucayre, 2022-07-19:
Driver is broken with recent kernel release (have to create a PR)
you have to replace:
```
pbufs[i].virt = pci_alloc_consistent(d->pdev, buflen, &pbufs[i].phys);
```
by
```
pbufs[i].virt = dma_alloc_coherent(&d->pdev->dev, buflen, &pbufs[i].phys, GFP_KERNEL);
```

```
pci_free_consistent(d->pdev
```
by
```
dma_free_coherent(&d->pdev->dev
```

And finally
```
if (pci_set_consistent_dma_mask(pdev, DMA_BIT_MASK(32))) {
```
by
```
if (dma_set_mask_and_coherent(&pdev->dev, DMA_BIT_MASK(32))) {
```

Warning: kernel build fine but not tested if it's works

vjarao, 2022-07-19:
Ohhh ... thanks. I will wait for the update because if I start digging on the driver I am most likely to break more stuff.

Thank you very much for your help & support

trabucayre, 2022-07-19:
I'm not sure to have time to test before create a PR and fairwaves people seems slow to merge patch...
But maybe I could create a patch -> you have just to apply this one without worrying to modify code by yourself.

vjarao, 2022-07-19:
That will be fantastic.

vjarao, 2022-07-21:
@trabucayre  Did you attached the patch? I do not see it.

trabucayre, 2022-07-21:
[0001-xtrx-fix-PCI-DMA-allow-free-with-kernel-5.18.txt](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/9156257/0001-xtrx-fix-PCI-DMA-allow-free-with-kernel-5.18.txt)

trabucayre, 2022-07-21:
You have to move at **xtrx_linux_pcie_drv** directory an to use 
```
patch -p1 < /somewhere/0001-xtrx-fix-PCI-DMA-allow-free-with-kernel-5.18.txt
```
to apply this fix.

trabucayre, 2022-07-29:
@vjarao any news? Patch is working as expected?
Thanks.

vjarao, 2022-07-29:
Good morning. Well the applied path went through without any errors. XTRX is on the list of /dev/ and all provided examples works. I have not yet installed gnuradio ... I am hoping to do this weekend. I will share my results

trabucayre, 2022-07-29:
Thanks!
A colleague has tested too and, consequently I have created this [PR](https://github.com/xtrx-sdr/xtrx_linux_pcie_drv/pull/14)

vjarao, 2022-07-30:
good morning, Volk installed without any problems. Then I followed with gnuradio 3.10 without any problems again. My current issue is that I cannot find the excitable to start gnuradio nor to add it to my path ... work in progress ...

vjarao, 2022-07-31:
Good evening. I could not make work GNURADIo 3.9 nor 3.10. However, 3.8 installs without any problems.

[image]

Next wall to overcome is building gr-osmosdr (https://gitea.osmocom.org/argilo/gr-osmosdr). I had 2 failures on make ../ where the errors contradicts minimal gnuradio version dependency ... bellow is the error.

-- The CXX compiler identification is GNU 10.2.1
-- The C compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
CMake Error at CMakeLists.txt:45 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.9".

  The following configuration files were considered but not accepted:

 /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.5.git

-- Configuring incomplete, errors occurred!

vjarao, 2022-08-05:
> Thanks!
> A colleague has tested too and, consequently I have created this [PR](https://github.com/xtrx-sdr/xtrx_linux_pcie_drv/pull/14)

Did it work for him/her?