geerlingguy/raspberry-pi-pcie-devices #73: Building a custom Kernel to improve compatibility

noah1510, 2021-02-15:
One way to increase the compatibility might be to enable some 'unusual' kernel options for a custom build.
This issue is a collection of kernel options that might increase the compatibility, if you leave one in a comments I will add it to the list.

- FB_FOREIGN_ENDIAN (under `device drivers/graphics support` since basically forever)

geerlingguy, 2021-02-19:
@noah1510 - Thanks for opening the issue; I'm trying to figure out, though, what in particular this would enable? Is there a specific card or class of cards you think would be better able to work with this flag on?

It seems like there are a lot of different options that would be required by different cards (e.g. `CONFIG_IRQ_POLL` for the Broadcom card in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/72, or `CONFIG_SATA_AHCI` for SATA cards, e.g. https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/1).

I'm not sure what the value is of having this separate issue open, unless you think we could build a custom kernel that has everything someone would need when messing around with more cards?

Coreforge, 2021-02-20:
I doubt the framebuffer foreign endian support would help much as the pi and x86 based computers are little endian, so if it's not needed normally, it shouldn't be needed here either. It also doesn't look to be implemented in the fbdev driver atleast.

noah1510, 2021-02-20:
> Is there a specific card or class of cards you think would be better able to work with this flag on?

I just wanted to create a place to collect options that might increase compatibility with all kind of PCIe devices.

> I'm not sure what the value is of having this separate issue open, unless you think we could build a custom kernel that has everything someone would need when messing around with more cards?

Exactly that. This issue can also be referenced when there are more specific problems and might help solve similar problems.

Since I don't have a raspberry pi 4 yet there is nothing specific I need.
This thread might be more better off being a discussion rather than an issue but they are not enabled on this repository (yet).

geerlingguy, 2021-02-20:
There are discussions ;)

I'll migrate this issue to a Discussion then.

closed 2021-02-20 by geerlingguy