geerlingguy/raspberry-pi-pcie-devices #74: Legacy PCI adaptors on Pi 4 CM

ArtemisMax, 2021-02-15:
This seems to be the most likely place to find out if anyone has tried using one of these PCIe to PCI adaptors yet. If not I will be getting one in a few months to try using legacy SCSI HDDs and Zip Drives with the Pi 4. I will post it here when I try it.
[image]

6by9, 2021-02-15:
Already tried and failed against an M-Audio Audiophile 24/96, RTL8029 10base2/T network card, and WinTV HVR1100.
thagrol reported similar failures.
https://www.raspberrypi.org/forums/viewtopic.php?p=1772997#p1772997

Most PCI cards will be expecting IO regions as they were effectively ISA I/O cards with a PCI front end.

SCSI should be backwards compatible, so a later SCSI card should be able to talk to earlier devices if you can find the right cables.

ArtemisMax, 2021-02-15:
thats disapointing. I the card I was going to use is actually 64 bit pci for a power mac G4 so it may not be using those IO addresses. I was kind of hopeing if that worked I could move on to using PCI sound blaster compatable sound cards to use with DOSBOX on the Pi 4 but thoes are much more likely to use those missing IO BARs.
When I have time next week I'll have a go with my PCIe SCSI card.

geerlingguy, 2021-02-19:
Yeah, unfortunately I'm not thinking we'll see much success there, and I'm not going to go out of my way testing any adapters because I still have enough PCIe cards to test that have a better chance of working :)

closed 2021-02-19 by geerlingguy