geerlingguy/raspberry-pi-pcie-devices #75: Test Killer1535 M.2 WiFi Card

geerlingguy, 2021-02-17:
The [Killer1535](https://amzn.to/3bjSzP3) is a pretty run of the mill WiFi card used in many laptops. Not sure how easy it will be to install the drivers or recompile the kernel with them on the Pi though.

TODO: Image of card here.

The AX200 was easy enough (but required a patch...). Hopefully these cards are easier.

geerlingguy, 2021-02-17:
```
$ sudo lspci -vvvv -d 168c:003e
0b:00.0 Network controller: Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)
	Subsystem: Bigfoot Networks, Inc. QCA6174 802.11ac Wireless Network Adapter
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 255
	Region 0: Memory at 600000000 (64-bit, non-prefetchable) [disabled] [size=2M]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/8 Maskable+ 64bit-
		Address: 00000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 <64us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 25.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis+, LTR+, OBFF Via message
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [148 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [168 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [178 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [180 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=50us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
```

geerlingguy, 2021-02-20:
Driver: https://wireless.wiki.kernel.org/en/users/drivers/ath10k

Firmware: https://wireless.wiki.kernel.org/en/users/drivers/ath10k/firmware?s[]=qca6174

Known bug/limitation from the wifi wiki:

> TX speeds are extremely poor on certain chips (QCA6174 is one). A [patch](https://gist.github.com/harrykipper/d1bedb234c4af0692f7ccd33329a02d7) solves the issue in most cases ([source](https://bbs.archlinux.org/viewtopic.php?pid=1689990#p1689990))

YumingChang02, 2021-09-22:
Currently running Dell DW1820 ( same chipset QCA6174 as killer 1535 ) on PIzza **which is behind a pci plx switch** have problem during initialization...

[   10.782379] pcieport 0000:00:00.0: of_irq_parse_pci: failed with rc=-22
[   10.789027] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   10.797193] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   10.803870] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   10.811220] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   10.818460] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   10.824802] ath10k: failed to probe PCI : -12, retry-count: 0
[   10.840595] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   10.849144] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   10.855804] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   10.863157] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   10.870403] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   10.876746] ath10k: failed to probe PCI : -12, retry-count: 1
[   10.892525] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   10.900647] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   10.907316] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   10.914654] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   10.921894] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   10.928206] ath10k: failed to probe PCI : -12, retry-count: 2
[   10.943995] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   10.952113] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   10.958761] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   10.966096] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   10.973357] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   10.979668] ath10k: failed to probe PCI : -12, retry-count: 3
[   10.995441] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   11.003614] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   11.010255] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   11.017601] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   11.024852] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   11.031182] ath10k: failed to probe PCI : -12, retry-count: 4
[   11.046960] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   11.055075] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   11.061731] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   11.069059] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   11.076301] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   11.082619] ath10k: failed to probe PCI : -12, retry-count: 5
[   11.098398] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   11.106509] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   11.113168] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   11.120493] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   11.127731] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   11.134080] ath10k: failed to probe PCI : -12, retry-count: 6
[   11.149853] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   11.157971] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   11.164611] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   11.171932] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   11.179167] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   11.185503] ath10k: failed to probe PCI : -12, retry-count: 7
[   11.201296] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   11.209409] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   11.216044] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   11.223365] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   11.230598] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   11.237013] ath10k: failed to probe PCI : -12, retry-count: 8
[   11.252809] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   11.260966] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   11.267605] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   11.274929] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   11.282171] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   11.288501] ath10k: failed to probe PCI : -12, retry-count: 9
[   11.304282] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x3e.
[   11.312381] ath10k_pci 0000:05:00.0: failed to alloc CE src ring 0: -12
[   11.319028] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipe 0: -12
[   11.326348] ath10k_pci 0000:05:00.0: failed to allocate copy engine pipes: -12
[   11.333583] ath10k_pci 0000:05:00.0: failed to setup resource: -12
[   11.339890] ath10k: failed to probe PCI : -12, retry-count: 10
[   11.355785] ath10k_pci: probe of 0000:05:00.0 failed with error -12

running on self build openwrt
OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.231.26241-422c175