geerlingguy/raspberry-pi-pcie-devices #85: JMB535 - IOCREST M.2 (PCIe 3.0) to 5 Ports SATA

Mes-ser, 2021-02-22:
Bought this [IOCREST M.2 (PCIe 3.0) to 5 Ports](https://www.aliexpress.com/item/33014907444.html) (You can get them cheaper)
and tried to make it work with cm4 4GB

[image]

Already increased the BAR memory, but I still got this `BAR 0: no space for [io  size 0x0080]`
```
brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.149430] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.149532] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.149641] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.175024] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.175408] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.175445] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.175479] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.175587] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.175848] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.179131] PCI: bus0: Fast back to back transfers disabled
[    1.179173] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.179458] pci 0000:01:00.0: [197b:0585] type 00 class 0x010601
[    1.179534] pci 0000:01:00.0: reg 0x10: [io  0x0000-0x007f]
[    1.179585] pci 0000:01:00.0: reg 0x14: [io  0x0000-0x007f]
[    1.179635] pci 0000:01:00.0: reg 0x18: [io  0x0000-0x007f]
[    1.179684] pci 0000:01:00.0: reg 0x1c: [io  0x0000-0x007f]
[    1.179734] pci 0000:01:00.0: reg 0x20: [io  0x0000-0x007f]
[    1.179784] pci 0000:01:00.0: reg 0x24: [mem 0x00000000-0x00001fff]
[    1.179835] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
[    1.180107] pci 0000:01:00.0: PME# supported from D3hot
[    1.180216] pci 0000:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0000:00:00.0 (capable of 15.752 Gb/s with 8.0 GT/s PCIe x2 link)
[    1.183371] PCI: bus1: Fast back to back transfers disabled
[    1.183408] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.183534] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.183578] pci 0000:01:00.0: BAR 6: assigned [mem 0x600000000-0x60000ffff pref]
[    1.183617] pci 0000:01:00.0: BAR 5: assigned [mem 0x600010000-0x600011fff]
[    1.183658] pci 0000:01:00.0: BAR 0: no space for [io  size 0x0080]
[    1.183690] pci 0000:01:00.0: BAR 0: failed to assign [io  size 0x0080]
[    1.183722] pci 0000:01:00.0: BAR 1: no space for [io  size 0x0080]
[    1.183754] pci 0000:01:00.0: BAR 1: failed to assign [io  size 0x0080]
[    1.183786] pci 0000:01:00.0: BAR 2: no space for [io  size 0x0080]
[    1.183817] pci 0000:01:00.0: BAR 2: failed to assign [io  size 0x0080]
[    1.183849] pci 0000:01:00.0: BAR 3: no space for [io  size 0x0080]
[    1.183880] pci 0000:01:00.0: BAR 3: failed to assign [io  size 0x0080]
[    1.183912] pci 0000:01:00.0: BAR 4: no space for [io  size 0x0080]
[    1.183943] pci 0000:01:00.0: BAR 4: failed to assign [io  size 0x0080]
[    1.183979] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.184018] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    1.184439] pcieport 0000:00:00.0: enabling device (0140 -> 0142)
[    1.184683] pcieport 0000:00:00.0: PME: Signaling with IRQ 62
[    1.192597] iproc-rng200 fe104000.rng: hwrng registered
```

BAR memory
```
pcie@7d500000 {
                        ...
                        ranges = < 0x2000000 0x00 0xc0000000 0x06 0x00 0x00 0x40000000 >;
                        dma-ranges = < 0x2000000 0x00 0x00 0x00 0x00 0x00 0xc0000000 >;
                        ...
                };
```

`sudo lspci -v` shows
```
01:00.0 SATA controller: JMicron Technology Corp. Device 0585 (prog-if 01 [AHCI 1.0])
        Subsystem: JMicron Technology Corp. Device 0000
        Flags: fast devsel, IRQ 255
        I/O ports at <unassigned> [disabled]
        I/O ports at <unassigned> [disabled]
        I/O ports at <unassigned> [disabled]
        I/O ports at <unassigned> [disabled]
        I/O ports at <unassigned> [disabled]
        Memory at 600010000 (32-bit, non-prefetchable) [disabled] [size=8K]
        [virtual] Expansion ROM at 600000000 [disabled] [size=64K]
        Capabilities: [80] Power Management version 3
        Capabilities: [90] MSI: Enable- Count=1/8 Maskable- 64bit+
        Capabilities: [c0] Express Legacy Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [150] Device Serial Number 00-00-00-00-00-00-00-00
        Capabilities: [160] Power Budgeting <?>
        Capabilities: [1b8] Latency Tolerance Reporting
        Capabilities: [300] #19
        Capabilities: [900] L1 PM Substates
```

ENV:
```
Linux raspberrypi 5.10.16-v7l+ #1402 SMP Tue Feb 16 14:12:18 GMT 2021 armv7l GNU/Linux
```
At this point I have no idea how to go further with this.
Confirmed that it works - connected to windows PC and it shows HDD.

6by9, 2021-02-22:
```
pi@raspberrypi:~/tmp $ lspci -k
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 10)
	Kernel driver in use: pcieport
01:00.0 USB controller: VIA Technologies, Inc. VL805 USB 3.0 Host Controller (rev 01)
	Subsystem: VIA Technologies, Inc. VL805 USB 3.0 Host Controller
	Kernel driver in use: xhci_hcd
```
I suspect that you have no driver for the device, so there will be no kernel driver listed as in use.

If you're lucky then adding the configs ```CONFIG_ATA``` and ```CONFIG_SATA_AHCI``` to the kernel may bring it to life.

Mes-ser, 2021-02-22:
> ```
> pi@raspberrypi:~/tmp $ lspci -k
> 00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 10)
> 	Kernel driver in use: pcieport
> 01:00.0 USB controller: VIA Technologies, Inc. VL805 USB 3.0 Host [...]

Ah! I forgot about that. Thanks! Now it's working.

```
pi@raspberrypi:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 238.5G  0 disk
├─sda1        8:1    0   499M  0 part
├─sda2        8:2    0   100M  0 part
├─sda3        8:3    0    16M  0 part
├─sda4        8:4    0 237.3G  0 part
└─sda5        8:5    0   561M  0 part
sdb           8:16   0 232.9G  0 disk
└─sdb1        8:17   0 232.9G  0 part
mmcblk0     179:0    0  59.5G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  59.2G  0 part /
```
I will close this issue after testing it with 5 drives connected at once.

geerlingguy, 2021-02-22:
@Mes-ser - Go ahead and leave open the issue. If you can get this working, I'd love to add it to the database!

geerlingguy, 2021-02-26:
Looks like this card is also [available on Amazon](https://amzn.to/3sz8gbM), slightly cheaper than AliExpress.

ThomasKaiser, 2021-02-28:
IOCREST is just a brand selling stuff. This JMB585 card uses key B-M coding (many other M.2 cards with this chip are key M only) and `IO-M2TO585-V10` is silkscreened on it next to the SATA ports. By simply searching for JMB585 on Aliexpress you get the same card from various sellers, e.g. [here for 20 bucks less](https://aliexpress.com/item/1005002191782456.html).

geerlingguy, 2021-02-28:
@ThomasKaiser - True, but fwiw, some people have an easier time buying from Amazon than AliExpress. Honestly it's always hit or miss no matter what the vendor lately. I'm still waiting on a few things I've ordered from both places over 2 months ago to finish off a couple projects :(

geerlingguy, 2021-03-17:
I just ordered one for some of my own testing. I'll be adding this to the site soon!

Mes-ser, 2021-04-25:
Few weeks passed, but finally I've got my hand on hdds and enclosure so tested with 500GB drives (raid0 & individual) everything seems to work
Installed OMV and no problems there either.
[image]

htruong, 2021-07-11:
I've bought a [IO CREST branded card](https://www.amazon.com/gp/product/B07ST9CPND/) with this exact chipset and it worked out of the box with no modifications other than sawing the PCI-E x1 slot.

```
pi@raspberrypi:~ $ lspci
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 SATA controller: JMicron Technology Corp. Device 0585

pi@raspberrypi:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
sda           8:0    0 465.8G  0 disk
`-sda1        8:1    0 465.8G  0 part
  `-md0       9:0    0 931.3G  0 raid5
sdb           8:16   0 465.8G  0 disk
`-sdb1        8:17   0 465.8G  0 part
  `-md0       9:0    0 931.3G  0 raid5
sdc           8:32   0 465.8G  0 disk
`-sdc1        8:33   0 465.8G  0 part
  `-md0       9:0    0 931.3G  0 raid5
mmcblk0     179:0    0 119.3G  0 disk
|-mmcblk0p1 179:1    0   256M  0 part  /boot
`-mmcblk0p2 179:2    0   119G  0 part  /
```

dededoubled, 2021-08-05:
sorry for humble questin, because i'am beginner to this
can i use on raspberry pi 4 or only cm4 ?
i want to build nas

Coreforge, 2021-08-06:
Basically only on a cm4. Technically it's possible to desolder the USB3.0 controller on a Pi 4 and get access to the PCIe bus that way, but since the cm4 exists now, there isn't really any point in doing that.

mateuszbrzezinski, 2021-08-10:
Is it possible to connect PCIe switch to cm4 board, then connect 2 or 3 x pcie to m.2 adapters, and then connect multiple this JMB585 adapters? so I could connect 10 or 15 sata hdds to my PI?

BeauSlim, 2021-10-01:
> Is it possible to connect PCIe switch to cm4 board, then connect 2 or 3 x pcie to m.2 adapters, and then connect multiple this JMB585 adapters? so I could connect 10 or 15 sata hdds to my PI?

I just tested this board with a CM4 running Ubuntu Impish and a 4-drive port multiplier box (Mediasonic HF2-SU3S2) connected to a single port over eSATA. Writes to a 4xRAIDZ pool are really slow (40MB/s) but reads at a respectable 200MB/s.  I believe that if combined with JMB575-based port multiplier boards, you should be able to connect 5x5=25 drives to a single JMB585 controller.  

As a total aside, I also put this card into a cheap WavLink Thunderdrive (UTE-02) external nvme enclosure, and it works with my M1 Macbook Air, with no extra drivers needed.   Neat!

bigboipete, 2022-05-07:
> If you're lucky then adding the configs `CONFIG_ATA` and `CONFIG_SATA_AHCI` to the kernel may bring it to life.

@6by9 (or anyone else) could you elaborate on this a bit for beginners? What is cruicial for these configs and where to put them? Or is there at least a good link to a documentation to get information about what to do?

I have to admit, that I'm currently just following the topic and have never touched a kernel yet. But reading this, I'm more and more convinced to give it a try for a Raspi-SATA-Nas...

Or is this obsolete now, since [Raspberry Pi OS received SATA support](https://www.jeffgeerling.com/comment/17464) out of the box?

NetanelHarris, 2022-05-31:
> Or is this obsolete now, since [Raspberry Pi OS received SATA support](https://www.jeffgeerling.com/comment/17464) out of the box?

So it's written [here](https://pipci.jeffgeerling.com/cards_storage/iocrest-jmb585-m2-sata.html#:~:text=Since%20mid%2D2021%2C%20SATA%20support%20is%20built%20into%20the%20Raspberry%20Pi%20kernel%2C%20so%20assuming%20you%20have%20updated%20to%20the%20latest%20version%20(sudo%20apt%20upgrade%20%2Dy)%2C%20this%20card%20should%20work%20out%20of%20the%20box.).

abidsamdani, 2024-01-12:
> I just ordered one for some of my own testing. I'll be adding this to the site soon!

@geerlingguyL any update on using IOCREST M.2 (PCIe 3.0) to 5 Ports with Raspberry Pi 5 ?

flaksirus, 2024-01-20:
I have tried it with random hat from aliexpress - works just fine. 
only did this 
`sudo apt upgrade -y`
and 
added these lines to /boot/config.txt
```
dtparam=pciex1
dtparam=pciex1_gen=3
```

KaKaKaaaaaaaa, 2024-06-16:
i Try connect JMB585 to pi M.2 HAT+ (like this https://www.waveshare.com/wiki/Raspberry_Pi_M.2_HAT%2B) and not working.
`      Memory behind bridge: 00000000-005fffff [size=6M] [32-bit]
    `
ACT LED in HAT+ does not light, only the PWR diode is lit red. I don't see controler SATA. After turning on PI, the LEDs in the SATA controller will flash once. How to check what is not working?

thx for help.

geerlingguy, 2024-11-22:
On a Pi 5, I'm seeing a lot of system slowdown when I have this card connected. I see the following in `dmesg` (when it finally loads in!):

```
[   10.131432] ata1: link is slow to respond, please be patient (ready=0)
[   10.457220] ata1: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   10.714443] ahci 0000:01:00.0: AHCI controller unavailable!
[   11.013931] ahci 0000:01:00.0: AHCI controller unavailable!
[   18.116772] ata2: failed to resume link (SControl FFFFFFFF)
[   22.746423] ata2: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   24.482542] ahci 0000:01:00.0: AHCI controller unavailable!
[   25.061294] ahci 0000:01:00.0: AHCI controller unavailable!
[   33.032770] ata3: failed to resume link (SControl FFFFFFFF)
[   37.662534] ata3: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   39.398752] ahci 0000:01:00.0: AHCI controller unavailable!
[   39.977551] ahci 0000:01:00.0: AHCI controller unavailable!
[   47.948777] ata4: failed to resume link (SControl FFFFFFFF)
[   52.578476] ata4: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   54.314641] ahci 0000:01:00.0: AHCI controller unavailable!
[   54.893632] ahci 0000:01:00.0: AHCI controller unavailable!
[   62.864776] ata5: failed to resume link (SControl FFFFFFFF)
[   67.496990] ata5: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   69.236128] ahci 0000:01:00.0: AHCI controller unavailable!
[   69.826400] ahci 0000:01:00.0: AHCI controller unavailable!
[   77.800772] ata6: failed to resume link (SControl FFFFFFFF)
[   82.430418] ata6: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   84.166542] ahci 0000:01:00.0: AHCI controller unavailable!
[   84.745374] ahci 0000:01:00.0: AHCI controller unavailable!
[   92.716770] ata9: failed to resume link (SControl FFFFFFFF)
[   97.346618] ata9: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   99.083004] ahci 0000:01:00.0: AHCI controller unavailable!
[   99.661781] ahci 0000:01:00.0: AHCI controller unavailable!
[  104.169517] Process accounting resumed
[  104.506400] CUSE: failed to register chrdev region
[  104.508140] CUSE: failed to register chrdev region
[  104.821482] Adding 524284k swap on /var/swap.  Priority:-2 extents:13 across:1294332k 
[  107.653717] ata10: failed to resume link (SControl FFFFFFFF)
[  109.102018] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx
[  112.289491] ata10: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[  114.026768] ahci 0000:01:00.0: AHCI controller unavailable!
[  114.615747] ahci 0000:01:00.0: AHCI controller unavailable!
[  122.596784] ata11: failed to resume link (SControl FFFFFFFF)
[  127.228114] ata11: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[  128.964315] ahci 0000:01:00.0: AHCI controller unavailable!
[  129.544183] ahci 0000:01:00.0: AHCI controller unavailable!
```

geerlingguy, 2024-11-22:
I found [this issue on the Pi forums with the exact same symptoms](https://forums.raspberrypi.com/viewtopic.php?t=363682). The recommendation from jdb was to add the following compatibility options in `/boot/firmware/config.txt`:

```
dtoverlay=pciex1-compat-pi5,no-mip
dtoverlay=pcie-32bit-dma-pi5
```

This issue was also tracked here: https://github.com/raspberrypi/linux/issues/6214

geerlingguy, 2024-11-22:
With those in place, I can see the block device, but I am still getting some link errors and it looks like it's downgrading the speed on me:

```
[    6.431564] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    6.431574] ata1.00: failed to IDENTIFY (I/O error, err_mask=0x40)
[   11.805990] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[   11.806301] ata1.00: supports DRM functions and may not be fully accessible
[   11.806304] ata1.00: ATA-11: Samsung SSD 870 QVO 8TB, SVQ02B6Q, max UDMA/133
[   11.806677] ata1.00: 15628053168 sectors, multi 1: LBA48 NCQ (depth 32), AA
[   11.811255] ata1.00: Features: Trust Dev-Sleep NCQ-sndrcv
[   11.811574] ata1.00: supports DRM functions and may not be fully accessible
[   11.816827] ata1.00: configured for UDMA/133
[   11.817104] ata1.00: supports DRM functions and may not be fully accessible
[   11.822661] ata1.00: supports DRM functions and may not be fully accessible
[   11.831688] scsi 1:0:0:0: Direct-Access     ATA      Samsung SSD 870  2B6Q PQ: 0 ANSI: 5
[   11.832336] ata1.00: Enabling discard_zeroes_data
[   11.832370] sd 1:0:0:0: [sdb] 15628053168 512-byte logical blocks: (8.00 TB/7.28 TiB)
[   11.832392] sd 1:0:0:0: [sdb] Write Protect is off
[   11.832397] sd 1:0:0:0: [sdb] Mode Sense: 00 3a 00 00
[   11.832414] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   11.832453] sd 1:0:0:0: [sdb] Preferred minimum I/O size 512 bytes
[   11.835642] sd 1:0:0:0: Attached scsi generic sg1 type 0
[   11.836507] ata1.00: Enabling discard_zeroes_data
[   11.999536] ata1: limiting SATA link speed to 3.0 Gbps
[   11.999542] ata1.00: exception Emask 0x11 SAct 0x8000 SErr 0x0 action 0x6 frozen
[   11.999546] ata1.00: irq_stat 0x48000008, interface fatal error
[   11.999550] ata1.00: failed command: READ FPDMA QUEUED
[   11.999552] ata1.00: cmd 60/08:78:00:00:00/00:00:00:00:00/40 tag 15 ncq dma 4096 in
                        res 41/84:00:00:00:00/00:00:00:00:00/00 Emask 0x10 (ATA bus error)
[   11.999560] ata1.00: status: { DRDY ERR }
[   11.999563] ata1.00: error: { ICRC ABRT }
[   11.999568] ata1: hard resetting link
[   12.149989] ata2: SATA link down (SStatus 0 SControl 300)
[   12.313991] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 320)
[   12.314285] ata1.00: supports DRM functions and may not be fully accessible
[   12.320472] ata1.00: supports DRM functions and may not be fully accessible
[   12.321188] ata1.00: Read log 0x00 page 0x00 failed, Emask 0x1
[   12.321191] ata1.00: NCQ Send/Recv Log not supported
[   12.321193] ata1.00: Read log 0x00 page 0x00 failed, Emask 0x40
[   12.321195] ata1.00: ATA Identify Device Log not supported
[   12.321197] ata1.00: Security Log not supported
[   12.321199] ata1.00: Read log 0x00 page 0x00 failed, Emask 0x40
[   12.321201] ata1.00: Read log 0x00 page 0x00 failed, Emask 0x40
[   12.321206] ata1.00: configured for UDMA/133
[   12.321230] ata1: limiting SATA link speed to 1.5 Gbps
[   12.461990] ata3: SATA link down (SStatus 0 SControl 300)
[   12.633995] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 310)
[   12.635069] ata1.00: supports DRM functions and may not be fully accessible
[   12.635503] ata1.00: Security Log not supported
[   12.636198] ata1.00: supports DRM functions and may not be fully accessible
[   12.636566] ata1.00: Security Log not supported
[   12.636922] ata1.00: configured for UDMA/133
[   12.636930] ata1: EH complete
[   12.675753]  sdb: sdb1 sdb2
[   12.675932] sd 1:0:0:0: [sdb] Attached SCSI disk
```

geerlingguy, 2024-11-22:
Testing with fio, I'm still getting over 300 MB/sec at least (and these aren't the highest performance SATA SSDs...):

```
$ sudo fio ssd-read.fio
sda1: (g=0): rw=read, bs=(R) 256KiB-256KiB, (W) 256KiB-256KiB, (T) 256KiB-256KiB, ioengine=libaio, iodepth=32
fio-3.33
Starting 1 process
Jobs: 1 (f=1): [R(1)][100.0%][r=354MiB/s][r=1414 IOPS][eta 00m:00s]
sda1: (groupid=0, jobs=1): err= 0: pid=2227: Fri Nov 22 12:16:42 2024
  read: IOPS=1265, BW=317MiB/s (332MB/s)(9503MiB/30006msec)
    slat (usec): min=8, max=407131, avg=788.26, stdev=4497.11
    clat (msec): min=4, max=634, avg=24.48, stdev=27.42
     lat (msec): min=6, max=634, avg=25.27, stdev=27.84
    clat percentiles (msec):
     |  1.00th=[   21],  5.00th=[   21], 10.00th=[   21], 20.00th=[   21],
     | 30.00th=[   21], 40.00th=[   22], 50.00th=[   22], 60.00th=[   22],
     | 70.00th=[   22], 80.00th=[   22], 90.00th=[   23], 95.00th=[   29],
     | 99.00th=[  226], 99.50th=[  230], 99.90th=[  234], 99.95th=[  634],
     | 99.99th=[  634]
   bw (  KiB/s): min=40960, max=374784, per=99.97%, avg=324183.17, stdev=79160.91, samples=60
   iops        : min=  160, max= 1464, avg=1266.22, stdev=309.18, samples=60
  lat (msec)   : 10=0.02%, 20=0.05%, 50=98.87%, 250=1.07%, 500=0.01%
  lat (msec)   : 750=0.08%
  cpu          : usr=0.24%, sys=1.61%, ctx=20874, majf=0, minf=37
  IO depths    : 1=0.0%, 2=0.0%, 4=0.0%, 8=0.0%, 16=0.0%, 32=100.0%, >=64=0.0%
     submit    : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.0%, 64=0.0%, >=64=0.0%
     complete  : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.1%, 64=0.0%, >=64=0.0%
     issued rwts: total=37978,0,0,0 short=0,0,0,0 dropped=0,0,0,0
     latency   : target=0, window=0, percentile=100.00%, depth=32

Run status group 0 (all jobs):
   READ: bw=317MiB/s (332MB/s), 317MiB/s-317MiB/s (332MB/s-332MB/s), io=9503MiB (9964MB), run=30006-30006msec

Disk stats (read/write):
  sda: ios=21681/43, merge=21685/3, ticks=138250/244, in_queue=138493, util=95.10%
```

Andreas-Halse, 2024-12-13:
@geerlingguy thanks so much for your effort. I wanted to share this:

https://youtu.be/AfqOx9NSbe8?si=3XckTp-WoNV0IHMW

You saved me a lot of time.