geerlingguy/raspberry-pi-pcie-devices #86: Add ModBerry, ClusBerry, and AI GATEWAY from Techbase

geerlingguy, 2021-02-23:
The [ModBerry 500 CM4](https://modberry.techbase.eu/compute-module-4/) is billed as an 'Industrial IoT Computer based on the Compute Module 4:

[image]

The [ClusBerry 9500-CM4](https://clusberry.techbase.eu) is billed as an 'Industrial Raspberry Pi Compute Module 4 Cluster'. It has a DIN rail case, a shared heatsink, and the standard set of modules includes an I/O board, a 'Communication Gateway', and an 'AI Gateway', all with different IO connections and purposes:

[image]

The [AI GATEWAY 9500-CM4](https://modberry.techbase.eu/modberry/ai-gateway-series-with-compute-module-4-and-google-coral-tpu/) marries a CM4 with a Google Coral TPU, over PCIe (interestingly... I'm guessing they haven't been able to sell this through yet due to the PCIe driver bugs everyone's been running into—see #44:

[image]

geerlingguy, 2021-09-07:
Modberry, done!

geerlingguy, 2021-09-08:
AI GATEWAY has been added too.

punnypenguins, 2021-09-10:
@geerlingguy All three of these have been added now, right?

geerlingguy, 2021-09-11:
@punnypenguins - Yep!

closed 2021-09-11 by geerlingguy