geerlingguy/raspberry-pi-pcie-devices #88: Raspberry CM4 Carrier Board with galvanically-isolated interfaces

sukhoi12, 2021-02-23:
Raspberry Compute Module 4 Board with ISO Interfaces
Galvanically-Isolated Modbus, 1-Wire, CAN | LAN | HDMI | LCD | USB | RTC | RELAY | QT Application for small Displays | Open Source

This is not just another Raspberry PCB.
Not just IO Board. 
Open Source - Hardware and Software.

This Carrier board was created by engineers for engineers, creators, learners, hobbyist.

CM Hunter Integrates the most common communication protocols:

•	RS485/Modbus isolated
•	CAN 2.0B isolated
•	1-Wire isolated
•	HDMI 2.0
•	ETHERNET Gigabit
•	USB 2.0
•	2.4 GHz, 5.0 GHz IEEE 802.11 b/g/n/ac wireless /some CM4 versions/
•	Bluetooth 5.0, BLE /some CM4 versions/
+ Real Time Clock, Relay, LCD Display, SD/MMC card, Cooling Fan

CM Hunter is coming soon in Kickstarter. It will participate in the crowdfunding activity soon, don't hesitate to give your support!

geerlingguy, 2021-02-23:
@sukhoi12 - Do you have an image you could add here as well (just drag and drop in the original post or in a new comment); it sounds like an interesting board!

sukhoi12, 2021-02-23:
[image]
[image]
[image]
[image]
[image]
[image]

closed 2021-03-17 by geerlingguy

davidindra, 2022-02-23:
Hello,
may I ask what's the state of this board? I didn't found the Kickstarter project and even the mentioned GitHub repository is empty. Please, can the schematics be obtained? @sukhoi12 
Thanks for your response!

commit 6b758b54 (2021-03-17): Fixes #88: Add CM Hunter board.