geerlingguy/raspberry-pi-pcie-devices #91: Test M.2 NGFF NVMe M-key SSD to M.2 Key A/E Adapter

geerlingguy, 2021-03-05:
I, uh, got carried away when I found these things on Amazon, and I bought their entire stock: [Timack M.2 NGFF nvme SSD to M.2 Key A/E Adapter (Vertical Installation) for 2280 Type SSD](https://amzn.to/3ebaIBq).

TODO: Picture here.

They allow you to plug typical NVMe drives (B/M or M-key) into an A/E slot. That's it. They work with any size, but only come with a vertical 'strut' to support a 2280-sized drive. And they don't work with SATA SSDs, only NVMe.

I'm considering testing these with the WD PC SN520s I have over in #90. It'll be fun. I think.

geerlingguy, 2021-03-19:
It's also in a video: https://www.youtube.com/watch?v=AoNxDe1a-X8

geerlingguy, 2021-03-19:
(And it was fun.)

closed 2021-04-30 by geerlingguy

JeWe37, 2022-07-03:
Hey, I've been looking to use these for a project of my own(old laptop as a NAS) but only been able to find very limited information on them. Do you happen to know if these are going to give you PCIe x1 or x2 speeds to the M-key interface(if hooked up to a proper A+E key slot of course)? I'm guessing in the video the limiting factor was the uplink anyway, so that gave me no real indication.

andrebrait, 2023-08-20:
> Hey, I've been looking to use these for a project of my own(old laptop as a NAS) but only been able to find very limited information on them. Do you happen to know if these are going to give you [...]

I also needed this answer.

@geerlingguy any chance you know whether these can give you 2 lanes out of the A+E slot (if the slot on the MB or hub even gives you 2x to start with, of course)

commit 46ba6bbc (2021-04-30): Fixes #91: Add M-key to A/E-key NVMe SSD adapter to M.2 section.