geerlingguy/raspberry-pi-pcie-devices #92: Test KingSpec 128GB M.2 2242 NVME PCIE SSD

geerlingguy, 2021-03-05:
I am picking up a [KingSpec 128GB M.2 2242 NVME PCIE SSD](https://amzn.to/3rnTnJm) since it's one of the few NVMe drives available in B+M and in the 42mm length. I have... special plans.

TODO: Image here.

You'll note one of the [Compute Module boards](https://pipci.jeffgeerling.com/boards_cm) has an M.2 B-key slot on the underside... I may have acquired said board and want to do some testing on it.

starchivore, 2021-03-05:
Could those special plans be related to stuff like this?

[https://www.raspberrypi.org/forums/viewtopic.php?t=305034](https://www.raspberrypi.org/forums/viewtopic.php?t=305034)
> This makes it easier for OS distro maintainers to utilize new features and unblocks the beta releases for larger changes e.g. **native NVMe boot**.

Was that particular feature merely mentioned as an example, or more like a teaser of some sort?

geerlingguy, 2021-03-05:
@starchivore - Eventually yes, but right now I'm kind of building up an arsenal of SSDs in practically every form factor imaginable, because it seems like every new board that comes out has a different slot depth and different key :D

geerlingguy, 2021-03-08:
```
01:00.0 Non-Volatile memory controller: Silicon Motion, Inc. Device 2263 (rev 03) (prog-if 02 [NVM Express])
	Subsystem: Silicon Motion, Inc. Device 2263
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 47
	Region 0: Memory at 600000000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/8 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM not supported, Exit Latency L0s <1us, L1 <8us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Not Supported
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: Unknown
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [b0] MSI-X: Enable+ Count=16 Masked-
		Vector table: BAR=0 offset=00002000
		PBA: BAR=0 offset=00002100
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout+ NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [158 v1] #19
	Capabilities: [178 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [180 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2- ASPM_L1.1- L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us
		L1SubCtl2: T_PwrOn=10us
	Kernel driver in use: nvme
	Kernel modules: nvme
```

geerlingguy, 2021-03-08:
```
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0  14.9G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  14.6G  0 part /
nvme0n1     259:0    0 119.2G  0 disk 
```

geerlingguy, 2021-03-08:
```
$ dmesg
...
[    1.214856] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.214897] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.214979] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.215067] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.261893] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.262277] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.262307] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.262334] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.262429] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.262673] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.266343] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.266601] pci 0000:01:00.0: [126f:2263] type 00 class 0x010802
[    1.266680] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00003fff 64bit]
[    1.267085] pci 0000:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0000:00:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    1.270538] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.270592] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.270624] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600003fff 64bit]
[    1.270676] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.270705] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
```

geerlingguy, 2021-03-08:
| Benchmark | Result |
| --- | --- |
| fio 1M seq read | 395MiB/s (414MB/s) |
| fio 1M seq write | 381MiB/s (400MB/s) |
| iozone 1M seq read | 369.77 MiB/s |
| iozone 1M seq write | 361.34 MiB/s |
| iozone 4k rand read | 39.57 MiB/s |
| iozone 4k rand write | 88.07 MiB/s |

geerlingguy, 2021-03-08:
Comparing the speeds to a similar price-scale Kingston SATA SSD inside the Argon One M.2 case:

[image]

geerlingguy, 2021-03-12:
This drive was featured in the TOFU video: https://www.youtube.com/watch?v=m-QSQ24_8LY

And I just added the info to the site now.

closed 2021-03-12 by geerlingguy

BeauSlim, 2021-10-14:
Wanting a 512GB 2242 drive, I ordered a KingSpec from AliExpress.  
It has memory chips on both sides of the PCB, unlike the smaller capacity drives.  It would fit the Waveshare CM4-IO-BASE-B if waveshare hadn't soldered a giant screw lug at the 2230 position.  It still mostly fits.  
The KingSpec's thermal sensor is confused.  I don't know if it is merely broken or if KingSpec is trying to be sneaky, but as soon as you start writing to the drive, temperature readings drop at least 10C.  I suspect sneakiness because this drive runs HOT, reaching 75C (externally measured) on long writes.

commit 49aca334 (2021-03-12): Fixes #92: Add KingSpec 128GB M.2 2242 NVMe SSD.