geerlingguy/raspberry-pi-pcie-devices #94: Pi2 Design CM4AES

starchivore, 2021-03-06:
While CM4AES might seem to be fairly expensive, let's compare that to another RPi4-based streamer for audiophiles:

[https://recordclean.com.au/4/20770/42895.html](https://recordclean.com.au/4/20770/42895.html)

It's certainly meant for a very different (niche) market:

[https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-3#post-334545](https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-3#post-334545)
> I am hesitant to speak about price right now but it will probably be in the $400 to $500 range for the digital only OLED version and upwards of $800 to $1K for the full analog with display. these prices are certainly inline with what's on the market for network streamers, but a far cry from rolling your own!.

[https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-4#post-339464](https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-4#post-339464)
> Display - No OLED, but CM4 status and 44.1/48 status LEDs
> Size - 124 x 73 x 40 mm, Galaxy GX140 from Modu in Italy
> nVME - 2242 size up to 2TB! www.amazon.com/Sabrent-DRAM-Less-Internal-Performance-SB-1342-2TB
> USB - Front Panel USB that has 3 modes (controlled by slide switch on back)
> A) Normal - host port for external storage or DAC (USB2 only though)
> B) Download - Using Utility Program on PC/Mac to program Compute Module eMMC
> C) Boot - boot from USB drive, same functionality as SD Card

[https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-6#post-339881](https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-6#post-339881)
[image]

[image]

[image]

geerlingguy, 2021-03-06:
That's a very interesting design! One question I had is whether the M.2 slot is intended for an SSD, or if the board pictured in the renders is a custom board?

starchivore, 2021-03-06:
Bingo, that's exactly why I'm SO looking forward to native NVMe boot myself:

[https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-6#post-339881](https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-6#post-339881)
> On the plus side it now allows for a full size NVMe SSD.

And then it's definitely a custom board since they're incorporating so many different components for the sake of sound quality:

[https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-3#post-334454](https://www.superbestaudiofriends.org/index.php?threads/pi2-design-idiot-proof-ready-to-go-rpi-aes-streamer.10471/page-3#post-334454)
> The design of the PI2AES and the upcoming CM4AES (I know terrible name, but I'm an engineer and we suck at marketing) uses two LT3045 LDO's in the critical sections. One powers the NDK NZ2520SD clocks and the other powers the WM8804. The Compute Module itself is powered from 5V switcher that derives it's power from the 24V input, which could be linear or switching. Using a linear will reduce overall system noise, but the LT3045's have <80dB of PSRR (Power Supply Rejection Ratio). With 6mv noise (the spec of the on-board 5V switching regulator) the LT3045's reduce that to <1uV!

BTW, there's also another similar product coming from Allo at some point:

[https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-187.html#post6384400](https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-187.html#post6384400)
> Yes we are already working on RPI4 CM

[https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-189.html#post6391328](https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-189.html#post6391328)
> We are talking completely new design in order to take full advantage of PCIe 2gen connection . Expect lots of changes

[https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-190.html#post6402105](https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-190.html#post6402105)
> I have no timeline. I have started the design and will bring it to the market .
> 
> Lots of new ideas on the power , this time discreet . Not only they show lower noise than LT304x but they are also incredibly low impedance .
> 
> PCIe slot will be provided (one extra)
> 
> We expect to need a beefy PSU , 2.5A for RPI4 and then for PCIe , depending on what you got attached to it , at least 1A (if you have SSD) or even 3A of you have a power hungry device.

[https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-191.html#post6404652](https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-191.html#post6404652)
> We will give an extra pcie connector. You can connect what you want.. Or not

[https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-191.html#post6405268](https://www.diyaudio.com/forums/vendor-s-bazaar/337859-shanti-dual-lps-5v-3a-5v-1-5a-191.html#post6405268)
> Ok to bring clarity... RPI4 has PCIe bus. We will open the bus and give 2 connectors...one for SSD and an extra one.You can connect (for example) a network card , or a WIFI card...(anything that is PCIe and has Linux drivers)

Technically Allo's existing customers could replace CM3+ inside their [USBridge Signature](https://allo.com/sparky/usbridge-signature-pcb.html) with CM4 after adding this adapter from Gumstix, as long as they don't need access to the PCIe bus:

[https://www.gumstix.com/manufacturer/cm4-uprev.html](https://www.gumstix.com/manufacturer/cm4-uprev.html)

Getting a tiny little bit OT here, I just wanna highlight what could be done for both kernel space and user space in order to improve the sound quality:

[https://github.com/klslz/squeezelite](https://github.com/klslz/squeezelite)
[https://github.com/klslz/tuningkitpcp](https://github.com/klslz/tuningkitpcp)
[https://soundcheck-audio.blogspot.com](https://soundcheck-audio.blogspot.com)
> All tools are supporting RPi3, RPi4 and related CM modules. The toolset supports 32-bit and 64-bit pCP versions.

[https://www.facebook.com/permalink.php?story_fbid=965376323936240&id=565378050602738](https://www.facebook.com/permalink.php?story_fbid=965376323936240&id=565378050602738)
[https://www.diyaudio.com/forums/vendor-s-bazaar/355137-symphonic-mpd-4.html#post6222917](https://www.diyaudio.com/forums/vendor-s-bazaar/355137-symphonic-mpd-4.html#post6222917)
> The kernel module built into symphonic-mpd is written from scratch.
> 
> One is the implementation of Real-Time Driver Model (RTDM) for the Xenomai kernel.
> A playback engine that bypasses the ALSA library and ALSA driver to feed PCM frames to the serializer's FIFO, similar to how an FPGA buffer playback is done on a tiny FIFO on a raspberry pi.
> 
> The other kernel module implements a ring buffer to receive PCM data from playback software running on the Linux kernel (mpd, shirtport-sync, spotify connect, etc.) and pass it to the real-time domain of the Xenomai kernel in a lockless.
> 
> Tuning the DMA is a small part of the effort we are doing with symphonic-mpd, which is implemented by modifying the source of the Linux kernel.

[https://github.com/sam0402/pcp-44.1KHz/tree/master/xenomai3](https://github.com/sam0402/pcp-44.1KHz/tree/master/xenomai3)
> Linux Kernel 4.19.122 Xenomai Patch
> CONFIG_HZ 44.1KHz Patch

Now we could understand why there are so many audio HAT options out there:

[https://www.hifiberry.com/products/](https://www.hifiberry.com/products/)
[https://www.audiophonics.fr/en/dac-and-interfaces-for-raspberry-pi-c-6384.html](https://www.audiophonics.fr/en/dac-and-interfaces-for-raspberry-pi-c-6384.html)

geerlingguy, 2021-03-17:
Added!

commit 0461ccd8 (2021-03-17): Fixes #94: Add CM4AES board for pro-quality audio output.
closed 2021-03-17 by geerlingguy