geerlingguy/raspberry-pi-pcie-devices #97: no space for[io size 0x0100]

makermuyi, 2021-03-10:
use for CH368 no space for[io size 0x0100]:
[    0.929678] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.929702] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    0.929774] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x0603ffffff -> 0x00f8000000
[    0.929846] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x007fffffff -> 0x0400000000
[    0.955290] brcm-pcie fd500000.pcie: link up, 2.5 GT/s x1 (SSC)
[    0.955616] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    0.955636] pci_bus 0000:00: root bus resource [bus 00-ff]
[    0.955656] pci_bus 0000:00: root bus resource [mem 0x600000000-0x603ffffff] (bus address [0xf8000000-0xfbffffff])
[    0.955713] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    0.955980] pci 0000:00:00.0: PME# supported from D0 D3hot
[    0.959311] PCI: bus0: Fast back to back transfers disabled
[    0.959335] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    0.959561] pci 0000:01:00.0: [1c00:5834] type 00 class 0x100000
[    0.959643] pci 0000:01:00.0: reg 0x10: [io  0x0000-0x00ff]
[    0.959681] pci 0000:01:00.0: reg 0x14: [mem 0x00000000-0x00007fff pref]
[    0.959718] pci 0000:01:00.0: reg 0x18: [io  0x0000-0x0003]
[    0.959816] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x00007fff pref]
[    0.960015] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    0.963261] PCI: bus1: Fast back to back transfers disabled
[    0.963282] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    0.963351] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    0.963374] pci 0000:01:00.0: BAR 1: assigned [mem 0x600000000-0x600007fff pref]
[    0.963401] pci 0000:01:00.0: BAR 6: assigned [mem 0x600008000-0x60000ffff pref]
[    0.963418] pci 0000:01:00.0: BAR 0: no space for [io  size 0x0100]
[    0.963434] pci 0000:01:00.0: BAR 0: failed to assign [io  size 0x0100]
[    0.963450] pci 0000:01:00.0: BAR 2: no space for [io  size 0x0004]
[    0.963465] pci 0000:01:00.0: BAR 2: failed to assign [io  size 0x0004]
[    0.963482] pci 0000:00:00.0: PCI bridge to [bus 01]
[    0.963506] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    0.963839] pcieport 0000:00:00.0: enabling device (0140 -> 0142)
[    0.964067] pcieport 0000:00:00.0: PME: Signaling with IRQ 60
[    0.964457] pcieport 0000:00:00.0: AER: enabled with IRQ 60

6by9, 2021-03-10:
IO BARs are not supported on ARM.
They're effectively emulating the old ```in``` and ```out``` IO opcodes from x86 which has independent memory and I/O space. ARM has no such concept as all peripherals are memory mapped.

It depends on the driver as to whether the io BARs are required or not. Many graphics cards will advertise IO BARs for the legacy VGA interface for the PC BIOS, but the main drivers won't use them.

geerlingguy, 2021-03-10:
@makermuyi - There are also a number of other devices that have IO bar space but the failed assignment doesn't cause an issue (the drivers just ignore it).

Please consider adding more details about the card you're testing, what you've done to try to get it working, etc. (instead of just pasting the dmesg logs). I'm not quite sure if you're asking for some help here, providing some extra information about a card that's currently being tested...?

geerlingguy, 2021-04-30:
Closing this; please open a new issue if you run into any roadblocks with a particular card, and be sure to offer more detail about what card you're testing, how you're testing it, etc. :)

closed 2021-04-30 by geerlingguy