geerlingguy/raspberry-pi-pcie-devices #100: Sound cards?

SqyD, 2021-03-15:
I've been wondering if/how PCIe sound cards would work. I've been looking for low latency, multi channel sound options for my Pi based piano project.
Considering:
* I2S devices (like the Hifiberry etc) are limited to just 2 channels.
* USB devices have suboptimal latency on the Pi. (well, at least on the P3, which was the last time I checked.)
Something like:
* Cheap gamer cards like the Creative Labs Soundblaster Z https://us.creative.com/p/sound-blaster/sound-blaster-z
* Midrange cards like the ESI Maya44ex: https://www.esi-audio.com/products/maya44ex/
* There are insanely expensive pro options as well like the RME ones (https://www.rme-audio.de/internal-cards.html) Would be interesting to see if it works but it's not a realistic option for a hobby project like mine.

Darry77, 2021-04-19:
I am going to be trying some PCI Express sound cards on a Pi 4 for my S/PDIF mixer project . 
See https://www.raspberrypi.org/forums/viewtopic.php?t=306055

Also, my experience has been that UAC2 USB audio devices that operate in USB 2.0 high speed mode can have much lower latency than the more common UAC USB 1.1 (a.k.a. USB 2.0 "full speed") audio devices .   See this https://www.vogons.org/viewtopic.php?p=951015#p951015 and this https://www.vogons.org/viewtopic.php?p=953521#p953521

geerlingguy, 2021-04-30:
I added a new category today: https://pipci.jeffgeerling.com/#multimedia-av-and-sound-cards

For any individual cards or CM4 projects, please feel free to open separate issues (a few other ones are open too ;).

closed 2021-04-30 by geerlingguy