geerlingguy/raspberry-pi-pcie-devices #101: MirkoPC - Tiny CM4 carrier board with full-size M.2 slot

geerlingguy, 2021-03-15:
Noticed this isn't on the site yet, but looks pretty cool. Very much like the TOFU I just tested, but more targeted towards a desktop use case, and less on the industrial side. It has a 12V input, it also has a built in DAC and RTC battery slot, so it fits the needs of a full-on desktop a little nicer.

[image]

[image]

More pics of the first board here: https://twitter.com/Mirek34416541/status/1368839261886283777

closed 2021-03-17 by geerlingguy

marzy57, 2021-10-07:
Please! swap out the USB 2.0 FOR 3.0

TheGuyDanish, 2021-10-07:
> Please! swap out the USB 2.0 FOR 3.0

One does not simply add USB 3. Looking at the pictures, it seems he's elected to forego USB 3 to use the PCI-e slot for an NVMe SSD. Adding a PCI-e switch to allow both would take a large amount of space which is already at a premium on the board.

geerlingguy, 2021-10-08:
Not only that; if you put a PCIe switch in the loop, the beta NVMe boot option doesn't work, too. So you have to sacrifice a bit to get that.

punnypenguins, 2022-04-11:
Live site page for this board: https://pipci.jeffgeerling.com/boards_cm/mirkopc.html

commit 96afc35b (2021-03-17): Fixes #101: Add MirkoPC to CM4 boards section.