geerlingguy/raspberry-pi-pcie-devices #103: USB 3.0 uPD720201 working

rtreffer, 2021-03-21:
I have been running a uPD720201 based card for a while now.

The card is a [6amLifestyle USB 3.0 PCIe card from amazon](https://www.amazon.de/gp/product/B07S6S9Y96/).
It required a [mad flashing](https://github.com/markusj/upd72020x-load/issues/15) session to working reliably (or use upd72020x-load).

I would assume that any uPD720201 card with decent on-card flash should work out of the box. This chip seems to be a common competitor vs. the VL805.

geerlingguy, 2021-03-22:
@rtreffer - Can you list out a little more of the steps you followed to get the flashing done?

rtreffer, 2021-03-22:
Sure.... There are mixed reports about this card online. When you buy such a card you end up with one of the following

1. A card without flash memory, requires a firmware upload on each boot ([upd72020x-load](https://github.com/markusj/upd72020x-load) or via the [kernel](https://github.com/torvalds/linux/commit/8bd5741e3145e40c1e4f422fa5f1b9d7fe0644b3) - both depend on `K2013080.mem`) - this is the worst as it can't work without a driver that uploads the firmware
1. A card with flash and an old firmware. New firmware can be uploaded (same as 1.) or flashed (turning it into 3.)
1. A card with flash and a recent firmware. Nothing to worry about.

# Prerequisite

1. A [flashrom compatible programmer with SOIC8 clamp](https://www.amazon.de/gp/product/B08J2PPHDY/) - needed for recovery
1. A decent firmware image, e.g. `k2013080.mem`. That's firmware version `2.0.1.3` - I was able to flash `2.0.2.6`, too. Check the [denisandroid/uPD72020x-Firmware repository](https://github.com/denisandroid/uPD72020x-Firmware).
1. [upd72020x-load](https://github.com/markusj/upd72020x-load)
1. The block erase size from the [uPD72020x datasheet](https://cdn.datasheetspdf.com/pdf-down/U/P/D/UPD720201-Renesas.pdf) 

| Vendor | Product | ID | config | Block erase size |
| - | - | - | - | - |
| Macronix | MX25L512E | `0x00C2 0x2010` | `0x0000 0x0700` | 16K |
| Macronix | MX25L1006E | `0x00C2 0x2011` | `0x0000 0x0700` | 16K |
| Macronix | MX25L2006E | `0x00C2 0x2012` | `0x0000 0x0700` | 16K |
| Macronix | MX25L4006E | `0x00C2 0x2013` | `0x0000 0x0700` | 16K |
| Macronix | MX25L5121E | `0x00C2 0x2210` | `0x0000 0x0550` | 16K |
| Macronix | MX25L1021E | `0x00C2 0x2211` | `0x0000 0x0550` | 16K |
| Winbond | W25X10BV | `0x00ef 0x3011` | `0x0000 0x0700` | 16K |
| Winbond | W25X20BV | `0x00ef 0x3012` | `0x0000 0x0700` | 16K |
| Winbond | W25X40BV | `0x00ef 0x3013` | `0x0000 0x0700` | 16K |
| Numonyx | M25P05-A  | `0x0020 0x2010` | `0x0000 0x0750` | 32K |
| Numonyx | M25P10-A  | `0x0020 0x2011` | `0x0000 0x0750` | 32K |
| Numonyx | M25P20  | `0x0020 0x2012` | `0x0000 0x0760` | 64K |
| Numonyx | M25P40  | `0x0020 0x2013` | `0x0000 0x0760` | 64K |
| Chingis | Pm25LD512C  | `0x019d 0x20ff` | `0x0000 0x0700` | 16K |
| Chingis | Pm25LD512C2  | `0x019d 0x207f` | `0x0000 0x0700` | 16K |
| ATMEL | AT25F512B  | `0x001f 0x6500` | `0x0000 0x0700` | 16K |
| EON | EN25F05  | `0x001c 0x3110` | `0x0000 0x0700` | 16K |
| EON | EN25F10  | `0x001c 0x3111` | `0x0000 0x0700` | 16K |
| EON | EN25F20  | `0x001c 0x3112` | `0x0000 0x0700` | 16K |
| EON | EN25F40  | `0x001c 0x3113` | `0x0000 0x0700` | 16K |
| AMIC | A25L512  | `0x0037 0x3010` | `0x0000 0x0700` | 16K |
| AMIC | A25L010  | `0x0037 0x3011` | `0x0000 0x0700` | 16K |
| AMIC | A25L020  | `0x0037 0x3012` | `0x0000 0x0700` | 16K |
| AMIC | A25L040  | `0x0037 0x3013` | `0x0000 0x0700` | 16K |
| SST | SST25VF512A  | `0x00bf 0x0048` | `0x0001 0x0791` | 16K |
| SST | SST25VF010A  | `0x00bf 0x0048` | `0x0001 0x0791` | 16K |

The block size will _usually_ be 16kb.

Get a card, and identify the flash chip
[image]
[image]

So here we have a flash chip labeled T25S40. Which seems to be the [Bright Moon Semiconductor T25S40](https://cdn.datasheetspdf.com/pdf-down/T/2/5/T25S40-BrightMoonSemiconductor.pdf)

## Take a backup of the current firmware

### Online backup of the firmware

My card was at `PCI 04:00.0` which means bus 4, device 0, function 0. Try downloading the firmware from the controller:
```
# ./upd72020x-load -b 4 -d 0 -f 0 -r -o backup
Doing the reading
bus = 4 
dev = 0 
fct = 0 
fname = backup
Found an UPD720201 chipset
got firmware version: 202609
EEPROM installed
got rom_info: 5e2013
got rom_config: 0
setting rom_config: 700
 ======> PASSED
```
The `rom_info` is the chip ID as found in the datasheet (and the same as the actual flash ID you get via SPI). The ROM configuration is chip dependent - I had to add the T25S40 to the list of supported devices, mostly guessing the right setup. [Adding a new chip to upd72020x-load](https://github.com/markusj/upd72020x-load/pull/14/files) is easy - but it should usually not be needed as the list of supported chips is pretty extensive and I have no idea why one would do the extra work of picking a different flash chip :shrug: 

### Offline backup

This is where flashrom + soic8 clamp come into play. The datasheet of your flash lists a pinout
```
       .---------.
    .--+.   °   .+--.
CS# | 1 |       | 8 | VCC
    `--+'       `+--'
    .--+.       .+--.
 SO | 2 |       | 7 | HOLD#
    `--+'       `+--'
    .--+.       .+--.
WP# | 3 |       | 6 | CLK
    `--+'       `+--'
    .--+.       .+--.
GND | 4 |       | 5 | SI
    `--+'       `+--'
       `---------'
```
This is a default BIOS/SPI pinout, something that works out of the box with most programmers. If your mapping is exotic then you'll need a breadboard or a flexible clamp to remap the pins.

[image]
[image]

I was unlucky enough that the chip was not supported by flashrom, but given the datasheet and some hints it was easy to [add T25S40 to flashrom](https://github.com/flashrom/flashrom/pull/200/files). Most people should not need this - as said this is an exotic chip not listed in the datasheet (why? :cry:)

Generating the flash dump:
```
# ./flashrom -p ch341a_spi -r dump
flashrom v1.2-208-g8059157-dirty on Linux 5.8.0-45-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found BrightMoon flash chip "T25S40" (512 kB, SPI) on ch341a_spi.
Reading flash... done.
```

I'll add dedicated comments for the upload part :see_no_evil:

rtreffer, 2021-03-22:
# Flash layout

The `mem` files usually used to make the uPD72020x work are the firmware. But the flash contains two copies of "Vendor Specific Configuration" + "Firmware".

The flash layout is
```
.-----------------------------------.
| Block 0                           |
| .-------------------------------. |
| | vendor specific configuration | |
| | - - - - - - - - - - - - - - - | |
| | firmware                      | |
| | - - - - - - - - - - - - - - - | |
| | 0xff 0xff 0xff 0xff .... 0xff | |
| `-------------------------------' |
| - - - - - - - - - - - - - - - - - |
| Block 1                           |
| .-------------------------------. |
| | vendor specific configuration | |
| | - - - - - - - - - - - - - - - | |
| | firmware                      | |
| | - - - - - - - - - - - - - - - | |
| | 0xff 0xff 0xff 0xff .... 0xff | |
| `-------------------------------' |
| - - - - - - - - - - - - - - - - - |
| Block 2..n  0xff 0xff  ....  0xff |
`-----------------------------------'
```
With blocksize usually being 16KB.
**Updating the flash online with upd72020x-load requires a single block**

# Vendor specific configuration

The vendor specific configuration is explained in the [uPD72020x datasheet](https://cdn.datasheetspdf.com/pdf-down/U/P/D/UPD720201-Renesas.pdf).
I dumped and analyzed my vendor specific configuration:
```
0x28 0x00                               ; length prefix, the next 40 bytes are all vendor specific configuration + CRC
0x00 0xff 0x01 0xff 0x02 0xff 0x03 0xff ; set pci subsystem vendor / subsystem to 0xff ff ff ff
0x0c 0x00 0x0d 0x00 0x0e 0x00 0x0f 0x00 ; set PHY control 0 to 0x00 - undocumented HW init
0x10 0x00 0x11 0x00 0x12 0x00 0x13 0x00 ; set PHY control 1 to 0x00 - undocumented HW init - may break uPD720202
0x14 0x00 0x15 0x00                     ; PHY control 2, battery charging SDP only
0x16 0x55                               ; PHY control 2, Tr/Tf fine control, default value - uPD720202 has a different default
0x17 0x00                               ; PHY control 2, HW init
0x18 0x00 0x19 0x00 0x1a 0x01 0x1b 0x05 ; host configuration register, default values
0xe7 0xef                               ; CRC16
```

- The vendor specific configuration sets all PCIe registers - usually to default values
- The vendor specific configuration sets the magic HWInit registers to 0
- For a "normal" uPD720201 card a working setup should be `\x28\x00\x00\xff\x01\xff\x02\xff\x03\xff\x0c\x00\x0d\x00\x0e\x00\x0f\x00\x10\x00\x11\x00\x12\x00\x13\x00\x14\x00\x15\x00\x16\x55\x17\x00\x18\x00\x19\x00\x1a\x01\x1b\x05\xe7\xef` - the uPD7020202 will likely need a different configuration!

**The firmware starts with `0xAA 0x55` so if possible reuse your own vendor specific configuration, especially if the checksum is not `0xE7 0xEF`**.

**The original flash should contain 2 copies of the firmware. Look for the second firmware to verify that it is inside the second block, e.g. starting at `0x4000` (16kb) plus vendor specific configuration size**

# Generating a flash block

My flash is 512kb, and my block size is 16kb (according to flash layout, datasheet says something different....)
```
# generate an empty block
for i in $(seq 1024) ; do echo -en '\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'; done > empty_block
# write a vendor specific configuration
echo -en '\x28\x00\x00\xff\x01\xff\x02\xff\x03\xff\x0c\x00\x0d\x00\x0e\x00\x0f\x00\x10\x00\x11\x00\x12\x00\x13\x00\x14\x00\x15\x00\x16\x55\x17\x00\x18\x00\x19\x00\x1a\x01\x1b\x05\xe7\xef' > vendor_specific_configuration
# build firmware for flashing
cat vendor_specific_configuration K2013080.mem > fw_flash
# put firmware flash layout into a block
cp empty_block fw_block
dd if=fw_flash of=fw_block  conv=notrunc
```
**fw_block can be uploaded online with upd72020x-load**

# Generating a full flash layout

```
cat fw_block fw_block > fw_flash_full
for i in $(seq 8) ; do cat empty_block empty_block empty_block empty_block; done > empty_flash
cp empty_flash flash_rom
dd if=fw_flash_full of=flash_rom  conv=notrunc
```

# Flashing the new firmware

```
./flashrom -p ch341a_spi -w flash_rom 
flashrom v1.2-208-g8059157-dirty on Linux 5.8.0-45-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found BrightMoon flash chip "T25S40" (512 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.
```

Reading the eeprom from userspace shows that the firmware is at version `2.0.1.3`
```
# ./upd72020x-load -b 4 -d 0 -f 0 -r -o out
Doing the reading
bus = 4 
dev = 0 
fct = 0 
fname = out 
Found an UPD720201 chipset
got firmware version: 201308
EEPROM installed
got rom_info: 5e2013
got rom_config: 0
setting rom_config: 700
 ======> PASSED
```

geerlingguy, 2021-03-22:
Oh wow, above and beyond! Thanks for posting all the detail. I'll be getting the card up on the site soon.

rtreffer, 2021-03-22:
So this was hard work because

- I YOLOd a firmware upload initially
- The flash chip was unique (hopefully only in this design)
- The card arrived with an old/non-operational firmware

But I was lucky / could make it work because

- Datasheets were available
- Chip pinout was common
- There were no components in the way of the clamp

I also flashed an I350 this way and it was way harder. The clamp was hard to get onto the chip and the pinout was unique (mirrored?). But I had success with that, too.

This is also the first time I used flashrom. So I would say it is potentially beginner-friendly.

closed 2021-07-02 by geerlingguy

wuzhongjie1997, 2023-04-11:
> Regarding this card, can you share a benchmark with usage from 1 to all 4 ports been used at the same time? Basically I need to know if this card share the ~5Gbps in all four USB ports or we can go berserk and have ~5Gbps per port?

Probably not gonna work anyways 'cause the PCIe Gen2 x1 interface only provides 5Gbps

retpolanne, 2023-06-25:
Hey! Bringing the discussion here since I've been emailing @rtreffer a lot about this (thank you Rene, btw! ) 

I've been documenting my Renesas journey here: https://blog.retpolanne.com/kernel-dev/renesas/2023/06/20/renesas.html

My board has probably the weirdest and most obscure EEPROMs possibly and it behaves in a very strange way. By that, I mean it is very slow for some reason and I'm troubleshooting that.

commit 2ea045bc (2021-07-02): Fixes #103: USB 3.0 Renesys UPD720201 generic 4-port card.