geerlingguy/raspberry-pi-pcie-devices #104: ASM1184e PCIe switch working (common in 1->4 mining cards)

rtreffer, 2021-03-21:
[ASHATA PCI-E 1 to 4 Adapter Card on amazon](https://www.amazon.de/gp/product/B08FCMFJL1)

The card is a simple ASM1184e based PCIe switch. It works well, my current topology is
```
# lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
02:01.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
02:03.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
02:05.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
02:07.0 PCI bridge: ASMedia Technology Inc. ASM1184e PCIe Switch Port
03:00.0 Non-Volatile memory controller: Toshiba Corporation XG4 NVMe SSD Controller (rev 01)
04:00.0 USB controller: Renesas Technology Corp. uPD720201 USB 3.0 Host Controller (rev 03)
06:00.0 SATA controller: ASMedia Technology Inc. Device 0625 (rev 01)
# lspc -tv
-[0000:00]---00.0-[01-06]----00.0-[02-06]--+-01.0-[03]----00.0  Toshiba Corporation XG4 NVMe SSD Controller
                                           +-03.0-[04]----00.0  Renesas Technology Corp. uPD720201 USB 3.0 Host Controller
                                           +-05.0-[05]--
                                           \-07.0-[06]----00.0  ASMedia Technology Inc. Device 0625
```

This chip is very common on 1->4 mining adapters.

I have not seen any issues, NVMe as root device works. USB 3.0 works. SATA controller seems to work (not tested, but it detects disks)

closed 2021-07-02 by geerlingguy

guibom, 2021-09-03:
Did you have to compile a custom kernel @rtreffer?

I bought another card that uses the same ASM1184e chip, but it doesn't get recognized by lspci.
(It could be that the specific risers I'm using are not supported...)

[image]

nezra, 2021-09-14:
I can confirm, no custom kernel. Worked out of the box with Raspbian and open media vault

nezra, 2021-09-14:
I can also confirm these can be daisy chained for multiple ports.

nezra, 2021-09-17:
Here's an odd issue, wondering if anyone has seen or can explain.

I can boot with 4 nvme drives attached, and 4 nvme drives plus a pcie is hub for 5 devices, but the moment a 5th nvme drive is added, it goes into a kernel panic and won't finish booting. Anyone have thoughts as to why?

It boots fine with two or 3 of these controllers daisy chained no issues. It's just when a 5th or more nvme drive is added, it dies.

geerlingguy, 2021-09-17:
@nezra - NVMe drives are a little weird in how they tie into the PCI Express bus. On something like a Threadripper, there's plenty of lanes and bandwidth for many drives. When you start using PCIe switches, things get dicey on the Pi's x1 Gen 2 bus. I've heard some people testing up to 10+ PCIe devices, and I've personally tested 4x NVMe drives but never more than that, so it could be that 4 is the max, or maybe the model you're using maxes out at 4.

sonic74, 2021-12-11:
VER006 has a jumper _PCI-E_ / _M2/MPCIE_. Does anyone know what it's for?

fryc88, 2022-03-02:
> VER006 has a jumper _PCI-E_ / _M2/MPCIE_. Does anyone know what it's for?

I'm guessing - for exactly what it says.

sonic74, 2022-03-02:
In front of the S-ATA power connector:
https://i.ebayimg.com/images/g/PZIAAOSwV9NiAmBr/s-l1600.jpg

6by9, 2022-03-02:
It looks like there may be a version with breakout from an mPCI-e slot - https://miningshop.pl/en/products/splitter-pci-e-riser-pce4pce-a0-233.html

sonic74, 2022-03-02:
aah, and there the jumper is actually set. thanks for the info!

iccoldbeer, 2022-04-05:
tbs 6925 not seen when in ASM1184e PCIe switch ports, but works fine in cm4 slot same for other tbs dvbs2 card. this switch working for nvme boot drive in all ports. I have ordered usb 3.0

dmeziere, 2022-11-17:
I have the same card as @sonic74 on CM4 + CM4IO, running OMV 6.0.46-5 (Shaitan). It looks like only the 2 middle ports work. The 2 outer ones can't see my ASM1166 Sata controller. Any chance it could be software related ? I also played around with an Intel I225-V ethernet, but without success at all.

dmeziere, 2022-11-23:
Update : My ASM 1184 works on all ports (checked in my PC). The problem of port 1 and 4 not working seems more related to the Raspberry Pi or  the ASM 1166 Sata controller.

commit 4aa3fec1 (2021-07-02): Fixes #104: ASM1184e PCIe Switch from ASHATA.