geerlingguy/raspberry-pi-pcie-devices #110: Waveshare Mini Base Board

julled, 2021-04-10:
There is a new miniature Base Board from Waveshare available: https://www.waveshare.com/cm4-io-base-b.htm

Very nice compact style.

I ordered mine and will test the m.2 connector with a mikrotik wifi card via m.2->minipcie adapter + dual camera on it.

geerlingguy, 2021-04-11:
Thanks for adding this in, I just merged the PR, and I'll touch up the image / link so it appears correctly.

geerlingguy, 2021-04-11:
(I especially like how it finds a way to cram all that extra stuff on the bottom side... it's practically the whole IO Board but in a Pi 4 form factor!)

I'll close this issue once I can confirm the board appears on the site.

msgilligan, 2021-05-16:
I ordered one -- along with the Waveshare 4.3" capacitive touch OLED and case. Their site didn't explicitly say this would work, but it looks like the mounting holes for this are the same as the Pi 4 and the case is open where the connectors go (as opposed to holes cut for specific connectors) so it should work.

julled, 2021-05-16:
I can confirm that the board works as expected. Except the USB ports, which dont work at all.

The mikrotik card works with a m.2 key m -> mini pcie adapter (https://www.delock.de/produkt/64103/merkmale.html?setLanguage=en)  and after adding some buffer capacitors to the 5V rail and connecting the poweramp to the powersupply (https://forum.mikrotik.com/viewtopic.php?t=79292)

geerlingguy, 2021-05-16:
@julled - You probably need to explicitly enable the USB 2.0 ports: https://www.jeffgeerling.com/blog/2020/usb-20-ports-not-working-on-compute-module-4-check-your-overlays

julled, 2021-05-16:
> @julled - You probably need to explicitly enable the USB 2.0 ports: https://www.jeffgeerling.com/blog/2020/usb-20-ports-not-working-on-compute-module-4-check-your-overlays

thank you for your hint, i didnt know this!

just tried, but it didnt help :(

fanoush, 2021-05-27:
The A model is $19.99, not $25, also B model went down in price
https://pipci.jeffgeerling.com/boards_cm/waveshare-cm4-io-base.html
so maybe remove the price or update it
or maybe just keep it outdated :-)

geerlingguy, 2021-05-27:
Heh, time to update the prices! If nobody else submits a PR with the updates I'll get to it at some point.

geerlingguy, 2021-05-28:
Updated!

closed 2021-05-28 by geerlingguy

msgilligan, 2021-05-28:
$20 is pretty amazing price. It brings down the price of a CM4-based system almost to the level of a regular Pi 4.

Johnnie390, 2021-07-02:
Hello all, has anyone actually an NVME SSD using this board?

Regards,

J.

msgilligan, 2021-07-02:
> Hello all, has anyone actually an NVME SSD using this board?

Yes, but I haven't got it to boot yet.

Johnnie390, 2021-07-02:
Morning Sean, what NVME SSD are you using? 

I have some more information regarding flashing the appropriate (NVME-Boot-Ready) EEPROM to the CM4. More to follow.

Johnnie390, 2021-07-02:
@msgilligan in your case - good morning...

msgilligan, 2021-07-02:
I'll have to check when I get home, if I forget ping me...

Danish-Food-Informatics, 2021-07-07:
### USB ports
In order to get the USB port working with CM4 board - it is usually only necessary to include a line
otg_mode=1
in config.txt

Johnnie390, 2021-07-07:
@msgilligan - The information on enabling NVME boot on a CM4 (irrespective of carrier board) is diffuse, sometimes contradictory and not infrequently inaccurate.
Whether you have a CM4 with or without an eMMC (flash) chip or not is irrelevant. You _must_ manipulate the EEPROM on the CM4 to enable NVME booting from another Linux or doze machine using a Beta EEPROM from not earlier than the 29th of May 2021 - see https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/beta. It is not difficult but you need to get your head around the procedure first. Use a search machine of your choice to look for CM4 EEPROM and/or CM4 NVME boot.

mynameisflorian, 2021-07-17:
Tested 2 NVME drives with this board (A model) -- a 128GB Samsung and a 256GB Western Digital (links below). I am not attempting to boot off of the NVME drives yet.

With a lite cm4 (no emmc), i was able to get both to work

With a regular cm4 (w/ emmc), only the 128gb worked. the larger 256gb hung at boot and the whole thing got quite warm. Maybe there's something i need to update with the boot rom?

Product Links:
https://www.amazon.com/gp/product/B08ZBKS4NB
https://www.amazon.com/gp/product/B096CWZ48N

Johnnie390, 2021-07-17:
Salut Florian, I use the "B" version of this board using a CM4 lite. No idea about the "A" version of same. 
I currently use a WDC PC SN520 (256GB - M.2-2230) with the abovementioned board to boot with no problems. No overheating, hiccups etc.
My BIOS/firmware/millicode is as follows -

root@moode:/Transit# CM4_ENABLE_RPI_EEPROM_UPDATE=1 rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Tue  6 Jul 10:44:53 UTC 2021 (1625568293)
    LATEST: Tue  6 Jul 10:44:53 UTC 2021 (1625568293)
   RELEASE: beta (/lib/firmware/raspberrypi/bootloader/beta)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 
    LATEST:

mynameisflorian, 2021-07-17:
I was able to get both drives working with the lite version, it's the cm4 with emmc that was giving me trouble (specifically the 256GB WD drive).

Here's the exact models of ssds...
- Samsung - PM99 - 128GB [MZ-9LQ128A, DPN 0CNYM8, PN MZ9LQ128HBHQ-000D1]
- Western Digital - SN530 - 256GB

technolhodgy, 2021-08-07:
> > Hello all, has anyone actually an NVME SSD using this board?
> 
> Yes, but I haven't got it to boot yet.

Same here
copied CD card to nvme fine but './rpiboot -d nvme' just hangs waiting for brcm2835/6/7/2711

mynameisflorian, 2021-08-16:
I'm not sure of the exact process off of the top of my head, but it shouldn't be terribly hard to setup Raspbian to do the initial boot from the MMC/SD (just enough to be able to do the next step)  and then hand-over to the nvme for the rest of the boot. Maybe a second or two longer to boot up, but should be the same runtime performance vs doing it all via NVME

dans98, 2021-08-20:
@mynameisflorian  based on your comment of getting the samsung 991 to work, I did some digging and found all 4 models available.

- 128GB https://www.amazon.com/Samsung-Internal-MZ-9LQ128A-MZ9LQ128HBHQ-000D1-Package/dp/B08ZBKS4NB/
- 256GB https://www.amazon.com/Samsung-256GB-PM991-MZALQ256HAJD-MZ-ALQ2560/dp/B08K79T9G5/
- 512GB https://www.amazon.com/Samsung-512GB-PM991-MZALQ512HALU-MZ-ALQ5120/dp/B097ZKKLYR/
- 1TB https://www.amazon.com/Samsung-MZALQ1T0HALB-MZ-ALQ1T00-Desktop-Ultrabook/dp/B096RW13N6/

I just ordered the 512GB version since the 1tb version is out of stock. 

I'll report back when it arrives next week and let everyone know if it fits properly or not.

BeauSlim, 2021-09-14:
Do those with the A model find that NVME drives cause power warnings?  Perusing the Waveshare wiki, they mention that the M.2 slot on the A only supports "up to 1.5 A" and if you have problems "buy version B".  Sure enough, in the schematics, the "A" model has a couple regulators, specced for 1A each, and the "B" model has a proper TI TLV62130 that can do 3A.   I'm wondering if I should just return the A that arrived today and get the B.  I'd test a bit myself but my CM4 module probably won't be here for at least another week.  
Jeff may wish to add this difference to the summary for the board?

BeauSlim, 2021-09-25:
In response to my own question, I'm finding that desktop SSDs do seem to have problems on the "A" board.  I'm getting strange issues where the CPU frequency ramps up to max when running benchmarks, but then can't throttle back down, and reams of errors are issued by the kernel.  Some SSDs also seem to cause the PCIe bus to fail to initialize on a reboot.  I think I'll order a "B" version to see if it helps.  I also have a KingSpec SSD on order to test with.

omares, 2021-11-16:
I would like to use the Waveshare mini base board in a [1U rack mount from myelectronics](https://www.myelectronics.nl/us/raspberry-pi-19-inch-1u-rack-mount-for-5x-rbpi-fro.html) and power the PIs using PoE hats. Can someone confirm if the board is PoE hat compatible? Bonus point if you think it should fit into the rack? :D

TMillesich, 2021-11-16:
> I would like to use the Waveshare mini base board in a [1U rack mount from myelectronics](https://www.myelectronics.nl/us/raspberry-pi-19-inch-1u-rack-mount-for-5x-rbpi-fro.html) and power the PIs [...]

The Waveshare mini base board is missing the 4 pin PoE header, which is mandatory for any PoE hat.

omares, 2021-11-16:
Ah, dang it. Thank you @TMillesich

nosferatie, 2022-03-08:
Sorry this may be off topic but has anyone got the HDMI1 FFC to work? I am not having any luck. I'm using Mini Base Board (A)

I've entabled:
 dtoverlay=dwc2,dr_mode=host

I am not using the DSI not the two CSI. There is no /boot/dt-blob.bin file (as per Wiki)
I've tried hdmi_safe=1 and hdmi_force_hotplug=1

HDMI0 works fine. Did I miss anything?

tunip, 2022-03-23:
Are these NVME read/write results ok?
```
read:
pi@raspberrypi:~ $ sudo hdparm -t /dev/nvme0n1p1 
/dev/nvme0n1p1:
 Timing buffered disk reads: 482 MB in  3.01 seconds = 160.34 MB/sec

pi@raspberrypi:~ $ sudo hdparm -t --direct /dev/nvme0n1p1 
/dev/nvme0n1p1:
 Timing O_DIRECT disk reads: 466 MB in  3.01 seconds = 155.06 MB/sec

write:
pi@raspberrypi:~ $ sudo dd if=/dev/zero of=/mnt/testfile_1G bs=1024 count=1024000
1024000+0 records in
1024000+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 13.8456 s, 75.7 MB/s
```

geerlingguy, 2022-03-23:
@tunip - That seems extremely slow; what kind of NVMe SSD are you testing? Most of the ones I tested max out the Pi's bus at 400+ MB/sec for sequential reads/writes.

Some very cheap drives without DRAM/fast controllers are more in the 200+ MB/sec range, but I haven't seen any under 200 MB/sec.

tunip, 2022-03-23:
It is a Kioxia KBG40ZNS256G (used in a Waveshare Mini Base Board, A model)
```
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     19NCP003VQ28         KBG40ZNS256G TOSHIBA MEMORY              1         256.06  GB / 256.06  GB    512   B +  0 B   AEGA0103
```

geerlingguy, 2022-03-23:
@tunip - Ah, the BG4 is a bit slower—here are my speeds: https://pipci.jeffgeerling.com/cards_m2/kioxia-bg4-m2-nvme-ssd.html

Trackhe, 2022-04-09:
did anyone tried a rpi v2 camera on the board? I cant get it working. Only: supported=1 detected=0, i verified that the camera works with a nvidia jetson

Trackhe, 2022-04-09:
I did it with the dt-blob from here: https://www.waveshare.com/wiki/CM4-IO-BASE-A

acocalypso, 2022-09-18:
Hi all I try to boot from nvme with a cm4 8gb emmc and the waveshare board. I used the latest eeprom and set the boot order to use nvme first.

I flashed my nvme 128gb from gigabyte with RPI-Imager and the cm4 is not booting up for me. 

With HDMI connected I see only a grey screen. 
The activity LEDs are all solid (no flashing)

I will try to setup serial connection to get a clue what's happening. Did anyone encounter a similar issue?

BeauSlim, 2022-09-22:
> I flashed my nvme 128gb from gigabyte with RPI-Imager and the cm4 is not booting up for me.

My first guess is power.  Some SSDs pull more than the PS or board (especially the A version) can give.

acocalypso, 2022-10-05:
The one I'm using only needs1A.
I'd this already too much for the A version?

[image]

fanoush, 2022-10-06:
> I'd this already too much for the A version?

No if that sticker is correct. My SN520 has higher value and works (and their docs say 1.5A). Set boot back to sdcard/emmc, keep ssd there and check if if it is is accessible after booting from something else. If yes then it is not a power issue.

fanoush, 2022-12-02:
Hello, if anyone with this A board is interested in low power 2242 version of WD SN520 128GB I saw good offer and  jumped at it, more details here https://forums.raspberrypi.com/viewtopic.php?t=343361 , also tested by aBUGSworstnightmare over here https://forums.raspberrypi.com/viewtopic.php?p=2059063#p2059063

For now I have more than I need and some, but if there was serious interest I could still get more - like 20 pieces of 2242 128GB. Or even more pieces of 2280 variant of the same thing but these are obviously not so well suited for this board. I have actually cut one SN520 2280 to 2242 size and it works just fine as the board is just empty space  but now  when I got  the real 2242 thing  there is no need for that :-)

Sadly this offer is just for 128GB size but for Pi this may still be more than needed.

Logian, 2023-01-13:
I would be very interested to know from someone who own's this board, whether it's capable of 4kp60. Specifications from Waveshare stating only 30Hz at 4k. But if I'm not mistaken, HDMI signals source from the CM4 itself without any other active chip on the circuit board. So I would assume that this may be only a matter of adding "hdmi_enable_4kp60=1" to "config.txt". Am I wrong here?

msgilligan, 2024-01-26:
> @msgilligan - The information on enabling NVME boot on a CM4 (irrespective of carrier board) is diffuse, sometimes contradictory and not infrequently inaccurate. Whether you have a CM4 with or [...]

I finally did this 2 days ago. I had been using the board for about two years booting from SD but running of NVMe.

I flashed the EEPROM *after* my Waveshare Base board failed and I moved the CM4 to the official Raspberry PI I/O Board.

msgilligan, 2024-01-26:
**WARNING:** It looks like my Waveshare Base Board B died after I plugging in a 27W (5V) Raspberry Pi 5 compatible adapter (the one from BOXPUT.) I've contacted Waveshare about this issue, but have not received much of a response yet.